zigpy/zha-device-handlers #1824: [BUG] Siren TS0216 exposed entities doesn't work

Charisketi, 2022-10-13:
I have a siren that looks like this:

https://zigbee.blakadder.com/Tuya_TS0216.html

The siren was recognized by zha. The on / off switch works as well as the tamper switch. In the home assistant, I have some entities with functions such as the alarm type, the volume, the strobe function, but they do not work. The only way to change the volume is from the device itself. How can I activate these features in my device management?

[image]

Device signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=512, server_mask=11264, maximum_outgoing_transfer_size=512, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0403",
      "in_clusters": [
        "0x0000",
        "0x000a",
        "0x0500",
        "0x0502",
        "0xef01"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYZB01_0wcfvptl",
  "model": "TS0216",
  "class": "zigpy.device.Device"
}
```

javicalle, 2022-10-20:
Can you enable the debug logs and attach the relevant info when you interact with the physical device?
* https://www.home-assistant.io/integrations/zha/#debug-logging

It will be hepfull if you can correlate the logs with your actions. For example: "these are the logs when pushing the + button", "these are the logs when volume is max", etc

Logs when you select options in HA are welcomed too.

Charisketi, 2022-10-20:
I enable logger with these settings:

javicalle, 2022-10-22:
That seems the pairing logs for your device.

Please enable at least also the `homeassistant.components.zha: debug`.

And maintain some related logs to events. We need to track something like:
```
2022-10-12 21:05:14.754 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB4E7), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=231, profile_id=260, cluster_id=61184, data=Serialized[b'\t_\x01\x00\t\x01\x04\x00\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=216, rssi=-46)
2022-10-12 21:05:14.755 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Received ZCL frame: b'\t_\x01\x00\t\x01\x04\x00\x01\x00'
2022-10-12 21:05:14.756 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=95, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-12 21:05:14.757 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Decoded ZCL frame: AtlanticSwitchManufCluster:get_data(data=TuyaCommand(status=0, tsn=9, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-10-12 21:05:14.758 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Received command 0x01 (TSN 95): get_data(data=TuyaCommand(status=0, tsn=9, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-10-12 21:05:14.775 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=95, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-12 21:05:14.776 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
2022-10-12 21:05:14.777 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB4E7), dst_ep=1, source_route=None, extended_timeout=False, tsn=95, profile_id=260, cluster_id=61184, data=Serialized[b'\x18_\x0b\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-10-12 21:05:14.779 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xb4e7, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=95), 19, b'\x18_\x0b\x01\x00')

2022-10-12 21:05:14.922 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=232), 216, -46, 0xb4e7, 255, 255, b'\t_\x01\x00\t\x01\x04\x00\x01\x00']
2022-10-12 21:05:14.923 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB4E7), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=232, profile_id=260, cluster_id=61184, data=Serialized[b'\t_\x01\x00\t\x01\x04\x00\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=216, rssi=-46)
2022-10-12 21:05:14.924 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Received ZCL frame: b'\t_\x01\x00\t\x01\x04\x00\x01\x00'
2022-10-12 21:05:14.925 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=95, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-12 21:05:14.927 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Decoded ZCL frame: AtlanticSwitchManufCluster:get_data(data=TuyaCommand(status=0, tsn=9, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))

2022-10-11 00:24:17.293 DEBUG (MainThread) [homeassistant.components.zha.entity] light.tz3210_ngqk6jia_ts110e_light: setting transitioning flag to True
2022-10-11 00:24:17.296 DEBUG (MainThread) [zigpy.zcl] [0xD1C0:1:0x0008] Sending Cluster Command. Cluster Command is 4, Arguments are ()
2022-10-11 00:24:17.297 DEBUG (MainThread) [zigpy.zcl] [0xD1C0:1:0x0008] Setting brightness to 1000
[... cut ...] ssi=None)
2022-10-11 00:24:17.342 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xD1C0), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=8, TSN=7, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x01\x07\xF0\xE8\x03\x00\x00')
2022-10-11 00:24:17.377 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
```

Charisketi, 2022-10-24:
Hello! 
I use conbee 2 and ZHA.
I changed my logger to: 
```
logger:
  default: info
  logs:
    homeassistant.core: debug
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug
```
My log file is masive because i use a lot of zigbee devices. So i try to filter log results using "TS0216" and "_TYZB01_0wcfvptl"
Below are filtered results. 

```
	Line  1339: 2022-10-24 13:29:23.918 DEBUG (MainThread) [zigpy.appdb] [0xf228:1:0x0000] Attribute id: 4 value: _TYZB01_0wcfvptl
	Line  1955: 2022-10-24 13:29:23.950 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TYZB01_0wcfvptl TS0216 (::::::)
	Line  2759: 2022-10-24 13:29:24.331 DEBUG (MainThread) [zigpy.appdb] [0xf228:1:0x0000] Attribute id: 4 value: _TYZB01_0wcfvptl
	Line  2949: 2022-10-24 13:29:24.784 DEBUG (MainThread) [zigpy_deconz.zigbee.application] device: 0xf228 - _TYZB01_0wcfvptl TS0216, FFD=True, Rx_on_when_idle=True
	Line  3545: 2022-10-24 13:29:25.035 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] [0xF228](_TYZB01_0wcfvptl TS0216) restored as 'available', last seen: 0:00:58 ago, consider_unavailable_time: 7200 seconds
	Line  1505: 2022-10-24 13:17:39.949 DEBUG (MainThread) [zigpy.appdb] [0xf228:1:0x0000] Attribute id: 5 value: TS0216
	Line  2125: 2022-10-24 13:17:40.282 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TYZB01_0wcfvptl TS0216 (:::::)
	Line  3065: 2022-10-24 13:17:41.365 DEBUG (MainThread) [zigpy.appdb] [0xf228:1:0x0000] Attribute id: 5 value: TS0216
	Line  3372: 2022-10-24 13:17:42.275 DEBUG (MainThread) [zigpy_deconz.zigbee.application] device: 0xf228 - _TYZB01_0wcfvptl TS0216, FFD=True, Rx_on_when_idle=True
	Line  3954: 2022-10-24 13:17:42.546 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] [0xF228](_TYZB01_0wcfvptl TS0216) restored as 'available', last seen: 0:03:00 ago, consider_unavailable_time: 7200 seconds
	Line  4090: 2022-10-24 13:17:42.559 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF228](TS0216): started initialization
	Line  4091: 2022-10-24 13:17:42.559 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xF228:ZDO](TS0216): 'async_initialize' stage succeeded
	Line  6159: 2022-10-24 13:17:42.647 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF228](TS0216): power source: Mains
	Line  6160: 2022-10-24 13:17:42.647 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF228](TS0216): completed initialization
	Line  6233: 2022-10-24 13:17:42.653 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF228](TS0216): started initialization
	Line  6234: 2022-10-24 13:17:42.653 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xF228:ZDO](TS0216): 'async_initialize' stage succeeded
	Line 15714: 2022-10-24 13:18:32.019 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF228](TS0216): power source: Mains
	Line 15715: 2022-10-24 13:18:32.019 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF228](TS0216): completed initialization
	Line 16495: 2022-10-24 13:19:09.546 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF228](TS0216): Device seen - marking the device available and resetting counter
	Line 16496: 2022-10-24 13:19:09.546 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF228](TS0216): Update device availability -  device available: True - new availability: True - changed: False
	Line 26079: 2022-10-24 13:20:36.546 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF228](TS0216): Device seen - marking the device available and resetting counter
	Line 26080: 2022-10-24 13:20:36.546 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF228](TS0216): Update device availability -  device available: True - new availability: True - changed: False
```
I also download diagnostics from the device:

[zha-26ec3139cb678489c574327ee16f28bc-_TYZB01_0wcfvptl TS0216-7e99fe56646d5cec7de410aa3312bee8.json.txt](https://github.com/zigpy/zha-device-handlers/files/9851112/zha-26ec3139cb678489c574327ee16f28bc-_TYZB01_0wcfvptl.TS0216-7e99fe56646d5cec7de410aa3312bee8.json.txt)

Is there a way to filter the results you need? I also can upload entire log file if you need it.

Charisketi, 2022-11-02:
@javicalle can i offer any other information that might help? I realize I'm not providing enough information. You can guide me if you want to be able to contribute what is needed. I would appreciate any help possible.

javicalle, 2022-11-03:
Ok, let's see if we can move forward on this issue.

In addition to the indicated filters, it would be very useful if you include the Nwk filter of the device (you can see it in the 'Zigbee info' of the device view). It seems to be `0xf228`
We need to have the traces of the Zigpy library to have more information.

I am aware that the debug logs can be very verbose. 
One thing that can be very helpfull is to have a tail of the `homeassistant.log` log activated and interact with the device. That way it is easier to identify which logs generate each event. Correlating the logs with the events is very helpful.

javicalle, 2022-11-03:
Ok, according to Z2M is just some extended `IasWd` cluster.

That is my very early version for that device:
<details open>
<summary>ts0216.py</summary>

```python
"""Map from manufacturer to standard clusters for the NEO Siren device."""
import logging

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Ota, Time
from zigpy.zcl.clusters.security import IasWd, IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

TUYA_VOLUME_ATTR = 0x0002  # [0]/[1]/[2] Volume 0-low, 2-high
TUYA_ALARM_ATTR = 0xF000  # [0]/[1] Alarm!

_LOGGER = logging.getLogger(__name__)

class TuyaIasWd(IasWd):
    """Tuya IasWd cluster with custom attributes."""

    attributes = IasWd.attributes.copy()
    attributes.update(
        {
            TUYA_VOLUME_ATTR: ("volume", t.uint8_t, True),
            TUYA_ALARM_ATTR: ("alarm", t.Bool, True),
        }
    )

class TuyaSiren(CustomDevice):
    """TS0216 Tuya Siren."""

    signature = {
        #  endpoint=1 profile=260 device_type=81 device_version=1 input_clusters=[0, 4, 5, 61184]
        #  output_clusters=[25, 10]>
        MODELS_INFO: [("_TZE200_t1blo2bj", "TS0601")],
        ENDPOINTS: {
            1: {
                # "profile_id": 260,
                # "device_type": "0x0403",
                # "in_clusters": ["0x0000","0x000a","0x0500","0x0502","0xef01"],
                # "out_clusters": ["0x0019"]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_WARNING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Time.cluster_id,
                    IasZone.cluster_id,
                    IasWd.cluster_id,
                    0xEF01,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Time.cluster_id,
                    IasZone.cluster_id,
                    TuyaIasWd,
                    0xEF01,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        }
    }
```

</details>

You will need to enable the local quirk in your instalation. There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Create a `ts0216.py` file in your local quirk folder and put the suggested quirk in the file.

Restart HA, remove and pair you device again. Check that the device is loading the quirk and that the device signature has changed.

For now there will only be 2 new attributes in the `IasWd` cluster, no UI control yet.
You should be able to modify the configuration by changing the values from the 'Manage Zigbee Device' popup in the device view and selecting the `TuyaIasWd` cluster.

javicalle, 2022-11-03:
For future reference:
* https://github.com/Koenkk/zigbee-herdsman-converters/blob/0f9cd2d29cca25ea6318506e10e6693a8ea16e18/devices/tuya.js#L2932-L2947
* https://github.com/Koenkk/zigbee-herdsman-converters/blob/0f9cd2d29cca25ea6318506e10e6693a8ea16e18/converters/fromZigbee.js#L2746-L2760
* https://github.com/Koenkk/zigbee-herdsman-converters/blob/0f9cd2d29cca25ea6318506e10e6693a8ea16e18/converters/toZigbee.js#L6146-L6176

Charisketi, 2022-11-03:
Hello @javicalle ! 
I followed your instructions. Now i'm able to see TuyaIasWd cluster in "Manage Zigbee Device". The problem is that i'm not able to register any attribute. If i register and then read the attribute i always get "None". The device signature is:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=512, server_mask=11264, maximum_outgoing_transfer_size=512, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0403",
      "in_clusters": [
        "0x0000",
        "0x000a",
        "0x0500",
        "0x0502",
        "0xef01"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYZB01_0wcfvptl",
  "model": "TS0216",
  "class": "zigpy.device.Device"
}
```
[image]

[image]

Also on home assistant log i get these errors:

```
Unexpected error handling the frame

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/uart.py", line 84, in data_received
    self._api.data_received(frame)
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/api.py", line 373, in data_received
    getattr(self, f"_handle_{command.name}")(data)
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/api.py", line 505, in _handle_aps_data_indication
    self._app.handle_rx(
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/zigbee/application.py", line 426, in handle_rx
    self.packet_received(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 842, in packet_received
    self.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 348, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 371, in handle_message
    return endpoint.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 224, in handle_message
    handler(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 377, in handle_message
    self.handle_cluster_general_request(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 412, in handle_cluster_general_request
    value = self.attributes[attr.attrid].type(attr.value.value)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 91, in __new__
    n = super().__new__(cls, *args, **kwargs)
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'data8'
```

`[0xF228:1:0x0502] Failed to convert attribute 0x0002 from (<class 'str'>) to type <class 'zigpy.types.basic.uint8_t'>: invalid literal for int() with base 10: ''`

```
Unexpected error handling the frame

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/uart.py", line 84, in data_received
    self._api.data_received(frame)
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/api.py", line 373, in data_received
    getattr(self, f"_handle_{command.name}")(data)
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/api.py", line 505, in _handle_aps_data_indication
    self._app.handle_rx(
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/zigbee/application.py", line 426, in handle_rx
    self.packet_received(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 842, in packet_received
    self.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 348, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 371, in handle_message
    return endpoint.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 224, in handle_message
    handler(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 377, in handle_message
    self.handle_cluster_general_request(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 412, in handle_cluster_general_request
    value = self.attributes[attr.attrid].type(attr.value.value)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 295, in __call__
    value = int(value)
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'data8'
```

javicalle, 2022-11-04:
You must put the value in the 'value' field, and leave the 'manufacturer code override' in blank.

Also check if the value gets updated when you change the volume from the device (select the attribute and push the 'get attribute' after every change).

Any log from your tests are welcomed.

MattWestb, 2022-11-04:
@Charisketi The `"class": "zigpy.device.Device"` is saying that the quirk is not loaded of the system so somthing its not OK i the quirk signature.
Have you adding your device in `MODELS_INFO:` ?

javicalle, 2022-11-04:
🤦🏻‍♂️ Ouuuch!
@MattWestb thanks for chasing the error!

I have updated this part of the code. But in the screenshoot it seems that the `TuyaIasWd` is there, so must have fixed that.

MattWestb, 2022-11-04:
ZHA some time is not loading the quirk also then its OK defined (happens for my then testing and editing quirk intensive in the system).
Can trying doing one reload of ZHA integration and see if its loading the quirk OK.
Also as you was saying the tuya specific cluster is being used so its little strange.

javicalle, 2022-11-04:
> Also as you was saying the tuya specific cluster is being used so its little strange.

Yes, there is a 'new' `0xEF01` cluster but in Z2M seems that there is no use for it (for now).

Charisketi, 2022-11-04:
@MattWestb the signature is not changed with custom quirk and is the same with signature i had before. I follow the instructions with the custom quirk and as i can see in quirk @javicalle posted there is a `MODELS_INFO: [("_TZE200_t1blo2bj", "TS0601")`

In my custom_zha_quirks folder with ts0216.py is created also a folder __pycache__ .

I try to change the value and not "manufacturer code override" as you told me and is working for volume and duration.

I also check if the values updated when i manualy change volume from fysical device and they are updated

One strange behavior with siren ui is that when i click the switch to turn it on, it auto switch to off in 7 sec. This has nothing to do with siren duration, because siren keeps working after ui switch is turned off according to duration i give in the device cluster configuration.

javicalle, 2022-11-04:
Delete the `__pycache__` folder on every restart.
I have changed the `MODELS_INFO` part. Update the code with the current version from my comment, delete the `__pycache__` folder, restart HA and re pair the device.

javicalle, 2022-11-04:
So, more or less, the device is functional now?

Charisketi, 2022-11-04:
I try to try it from the beginning. So i delete my siren device also both files on custom quirks folder, then i put in custom quirks folder the new code file and restart home assistant and now i get this error on my zha intergration.

```
Error setting up entry ConBee II - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2190765-if00, s/n: DE2190765 - dresden elektronik ingenieurtechnik GmbH - 1CF1:0030 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 365, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 101, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0216.py", line 37, in <module>
    class TuyaSiren(CustomDevice):
  File "/config/custom_zha_quirks/ts0216.py", line 41, in TuyaSiren
    MODEL: "TS0216",
NameError: name 'MODEL' is not defined
```

I try also to add device first, then add quirk file in folder and then restart, but this way my device became unavailable and i can't remove it.

javicalle, 2022-11-04:
Ouuuuch^2

Code updated. I forget the new import.
Can you try again?

Charisketi, 2022-11-04:
Ok. :) Now i see that signature is changed. Device is working the same way i told you in my previous comment. Changing clusters i can change device volume and duration.

Charisketi, 2022-11-04:
I try to send squawk using cluster command with no success. I also try to send squawk using home assistant service "Zigbee Home Automation: Warning device squawk" but with no success. Is this posible?

javicalle, 2022-11-04:
What is 'squawk'?

In Z2M the device has a custom command, but I have not reviewed yet.

dmulcahey, 2022-11-04:
> What is 'squawk'?
> 
> In Z2M the device has a custom command, but I have not reviewed yet.

It’s an OOTB ZCL command.

javicalle, 2022-11-04:
Z2M only implements the `start_warning` command in an imaginative way. I have no evidence that the `squawk` can be done.

javicalle, 2022-11-04:
David, the switch that apears in HA is created by the `ZHASiren`, isn't it?

Charisketi, 2022-11-04:
No there are no logs when using squawk service. Only "Warning device warn" is working just to turn on siren.

dmulcahey, 2022-11-04:
> David, the switch that apears in HA is created by the `ZHASiren`, isn't it?

Yes

Charisketi, 2022-11-10:
So I guess the only features available on this device are volume and duration?

Is it necessary to keep the custom quirk or will the volume and duration features be integrated in the next update?

Charisketi, 2022-12-07:
Hello @javicalle! Will there be a quirk for siren support in terms of volume and duration? Do you need anything more from me?

javicalle, 2022-12-07:
> Do you need anything more from me?

No, I just don't have enough spare time.

ZHA already have implemented the siren entity and a few config selects:
* https://github.com/home-assistant/core/blob/41e4f1097793ea33a52e11daf147dd954cfaab99/homeassistant/components/zha/siren.py#L105
* https://github.com/home-assistant/core/blob/41e4f1097793ea33a52e11daf147dd954cfaab99/homeassistant/components/zha/select.py#L130-L156

We need to understand how attributes are handled in ZHA (I think the value of the volume is cached in ZHA, but the duration is not saved in any way) and find a way to translate the values to (and from) the Tuya attributes. 
An implementation specific to Tuya sirens might be the easiest path, but right now that's something I can't afford.
If someone wants to try it I can give support but nothing else.

closed 2023-07-10 by github-actions[bot]

florismetzner, 2024-03-14:
Hi,

I have a comparable device called:
  "manufacturer": "_TYZB01_4obovpbi",
  "model": "TS0216",

Can use the same quirk? I'm able to start the sirene via device -> "warn" and turn it of via sirene: off ...but...I get errors that the device is reachable which makes the situation unworkable. Anny suggestions?  

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4619, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0403",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0x0502",
        "0x0b05"
      ],
      "output_clusters": [
        "0x0004",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TYZB01_4obovpbi",
  "model": "TS0216",
  "class": "zigpy.device.Device"
}
```