zigpy/zha-device-handlers #1826: [Device Support Request] Heiman CO_V15 Monoxide Carbon sensor

cobirnm, 2022-10-14:
**Is your feature request related to a problem? Please describe.**
When Heiman CO_V15 Monoxide Carbon sensor is added to HA it reports as a "Mains" powerd device and after some hours drops out of the network

**Describe the solution you'd like**
If possible a custom quirk to correct it.

<details>

<summary>Device signature</summary>

```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, manufacturer_code=48042, maximum_buffer_size=64, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.ExtendedSimpleDescriptorListAvailable|ExtendedActiveEndpointListAvailable: 3>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0500"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Heiman",
  "model": "CO_V15",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
22-10-14 15:39:50.280 DEBUG (MainThread) [zigpy.zcl] [0xEF61:1:0x000a] Received ZCL frame: b'\x10\x96\x00\x00\x00'
2022-10-14 15:39:50.282 DEBUG (MainThread) [zigpy.zcl] [0xEF61:1:0x000a] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 0>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=150, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-10-14 15:39:50.286 DEBUG (MainThread) [zigpy.zcl] [0xEF61:1:0x000a] Decoded ZCL frame: Time:Read_Attributes(attribute_ids=[0])
2022-10-14 15:39:50.287 DEBUG (MainThread) [zigpy.zcl] [0xEF61:1:0x000a] Received command 0x00 (TSN 150): Read_Attributes(attribute_ids=[0])
2022-10-14 15:39:50.290 DEBUG (MainThread) [zigpy.zcl] [0xEF61:1:0x000a] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=150, command_id=<GeneralCommand.Read_Attributes_rsp: 1>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-14 15:39:50.292 DEBUG (MainThread) [zigpy.zcl] [0xEF61:1:0x000a] Sending reply: Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0000, status=<Status.SUCCESS: 0>, value=TypeValue(type=UTCTime, value=719073590))])
2022-10-14 15:39:50.296 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xEF61), dst_ep=1, source_route=None, extended_timeout=False, tsn=150, profile_id=260, cluster_id=10, data=Serialized[b'\x18\x96\x01\x00\x00\x00\xe261\xdc*'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-10-14 15:39:50.297 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=AddressMode.NWK_AND_IEEE address=0xef61>, ep: 1, profile: 0x0104, cluster_id: 0x000a, data: b'1096000000'
2022-10-14 15:39:50.299 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (26, 111, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0xef61 endpoint=1>, 260, 10, 1, b'\x18\x96\x01\x00\x00\x00\xe261\xdc*', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-10-14 15:39:50.300 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x12d80021001a006f000261ef0104010a00010b00189601000000e23631dc2a0200
2022-10-14 15:39:50.325 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x12d80009000200226f
2022-10-14 15:39:50.326 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_request[2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 111]
2022-10-14 15:39:50.326 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 111]
2022-10-14 15:39:50.326 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0ed9000700a600
2022-10-14 15:39:50.327 DEBUG (MainThread) [zigpy_deconz.api] Received command device_state_changed[<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-10-14 15:39:50.327 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-10-14 15:39:50.328 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-10-14 15:39:50.328 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04d90007000000
2022-10-14 15:39:50.341 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04d90013000c00226f0261ef01010000000000
2022-10-14 15:39:50.342 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_confirm[12, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 111, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0xef61 endpoint=1>, 1, <TXStatus.SUCCESS: 0>, 0, 0, 0, 0]
2022-10-14 15:39:50.343 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 111: 00
2022-10-14 15:39:50.343 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x6f 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0xef61 endpoint=1>, status: 0x00
2022-10-14 15:39:50.421 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1fda000e0007002fdda1230f0000
2022-10-14 15:39:50.421 DEBUG (MainThread) [zigpy_deconz.api] Received command simplified_beacon[7, 56623, 9121, 15, 0, 0]
2022-10-14 15:39:50.421 DEBUG (MainThread) [zigpy_deconz.api] Received simplified beacon frame: source=0xdd2f, pan_id=0x23a1, channel=15, flags=0x00, update_id=0x00
2022-10-14 15:39:50.437 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1fdb0015000e004a6fa1230f00006638a1230f0000
2022-10-14 15:39:50.438 DEBUG (MainThread) [zigpy_deconz.api] Received command simplified_beacon[14, 28490, 9121, 15, 0, 0]
2022-10-14 15:39:50.439 DEBUG (MainThread) [zigpy_deconz.api] Received simplified beacon frame: source=0x6f4a, pan_id=0x23a1, channel=15, flags=0x00, update_id=0x00

```

</details>

**Additional context**
[image]

javicalle, 2022-10-18:
It seems that there is another case like this.

Can you try with that quirks:

```python
"""Smoke Sensor."""

import zigpy.profiles.zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Alarms, Basic, Identify, Ota, PowerConfiguration
from zigpy.zcl.clusters.security import IasWd, IasZone
import zigpy.zdo.types

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    NODE_DESCRIPTOR,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.heiman import HEIMAN

class HeimanSmokYDLV10(CustomDevice):
    """YDLV10 quirk."""

    # NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=132,
    # manufacturer_code=48042, maximum_buffer_size=64, maximum_incoming_transfer_size=0,
    # server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=3)
    # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=1026,
    # device_version=0, input_clusters=[0, 3, 1280, 1, 9, 1282], output_clusters=[25])
    signature = {
        MODELS_INFO: [(HEIMAN, "SMOK_YDLV10")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zigpy.profiles.zha.PROFILE_ID,
                DEVICE_TYPE: zigpy.profiles.zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Alarms.cluster_id,
                    IasZone.cluster_id,
                    IasWd.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }

    replacement = {
        NODE_DESCRIPTOR: zigpy.zdo.types.NodeDescriptor(
            0x02, 0x40, 0x84 & 0b1111_1011, 0xBBAA, 0x40, 0x0000, 0x0000, 0x0000, 0x03
        ),
        ENDPOINTS: {
            1: {
                PROFILE_ID: zigpy.profiles.zha.PROFILE_ID,
                DEVICE_TYPE: zigpy.profiles.zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Alarms.cluster_id,
                    IasZone.cluster_id,
                    IasWd.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }

class HeimanSmokCO_V15(CustomDevice):
    """CO_V15 quirk."""

    # NodeDescriptor(
    #     logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, 
    #     frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, 
    #     manufacturer_code=48042, maximum_buffer_size=64, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, 
    #     descriptor_capability_field=<DescriptorCapability.ExtendedSimpleDescriptorListAvailable|ExtendedActiveEndpointListAvailable: 3>, 
    #     *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, 
    #     *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False
    # )"
    signature = {
        MODELS_INFO: [(HEIMAN, "CO_V15")],
        ENDPOINTS: {
            # "profile_id": 260,"device_type": "0x0402",
            # "in_clusters": ["0x0000","0x0001","0x0003","0x0009","0x0500"],
            # "out_clusters": ["0x0019"]
            1: {
                PROFILE_ID: zigpy.profiles.zha.PROFILE_ID,
                DEVICE_TYPE: zigpy.profiles.zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Alarms.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }

    replacement = {
        NODE_DESCRIPTOR: zigpy.zdo.types.NodeDescriptor(
            0x02, 0x40, 0x84 & 0b1111_1011, 0xBBAA, 0x40, 0x0000, 0x0000, 0x0000, 0x03
        ),
        ENDPOINTS: {
            1: {
                PROFILE_ID: zigpy.profiles.zha.PROFILE_ID,
                DEVICE_TYPE: zigpy.profiles.zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Alarms.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }
```

There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Create a new `smoke.py` file in your local quirks folder.
Put the proposed code in the new file.
Restart HA and pair you device again. Check that the device is loading the quirk and that the device signatura has changed.

javicalle, 2022-10-20:
No, don't close yet.

PR #1847 (2022-10-22): Make `CO_V15` battery powered
Fixes: #1826
commit 566bccd9 (2022-10-22): Make `CO_V15` battery powered (#1847)

* Make `CO_V15` battery powered

Fixes: #1826

* fix checks
closed 2022-10-22 by dmulcahey