zigpy/zha-device-handlers #1828: [Device Support Request] Aqara Smoke Detector JY-GZ-01AQ

wishie, 2022-10-15:
I am looking to help add support for this smoke detector into ZHA.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.sensor_smoke.acn03",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.10.1",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Australia/Brisbane",
    "os_name": "Linux",
    "os_version": "5.19.9-Unraid",
    "run_as_root": true
  },
  "custom_components": {
    "watchman": {
      "version": "0.5.1",
      "requirements": [
        "prettytable==3.0.0"
      ]
    },
    "fullykiosk": {
      "version": "1.1.0",
      "requirements": [
        "python-fullykiosk==0.0.11"
      ]
    },
    "goodwe": {
      "version": "1.0.0",
      "requirements": [
        "goodwe==0.2.20"
      ]
[... 329 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
The device apparently has full support in Z2M, and the details of that can be seen here: https://www.zigbee2mqtt.io/devices/JY-GZ-01AQ.html

javicalle, 2022-10-20:
Can you tell us what is working and what isn't?

I believe that basic detection can be done but make all the Z2M sensors available will require a lot of code:
* https://github.com/Koenkk/zigbee-herdsman-converters/blob/170c63839d924ca85bc59537fc113945a27e8fea/devices/xiaomi.js#L1549-L1588
* https://github.com/Koenkk/zigbee-herdsman-converters/blob/170c63839d924ca85bc59537fc113945a27e8fea/lib/xiaomi.js#L373-L515

wishie, 2022-10-20:
The smoke detector only shows as 2 entities by default.. an IASZone and a Button.

The IASZone shows 'clear' and I was hoping it would update on smoke detected, but in my testing it does not.
The 'Identify' button makes the LED on the unit flash once.

[image]

javicalle, 2022-10-21:
For the IASZone I believe that can be done with that quirk:
```python
class SmokeDetector(XiaomiQuickInitDevice):
    """Aqara smoke detector custom device."""

    def __init__(self, *args, **kwargs):
        """Init method."""
        # self.battery_size = 8  # CR123a
        super().__init__(*args, **kwargs)

    signature = {
        # NodeDescriptor(
        #     logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0,
        #     frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>,
        #     manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100,
        #     descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False,
        #     *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, 
        #     *is_security_capable=False
        # )
        NODE_DESCRIPTOR: XIAOMI_NODE_DESC,  # not sure if needed, also don't match the device one
        MODELS_INFO: ((LUMI, "lumi.sensor_smoke.acn03"),),
        ENDPOINTS: {
            #  input_clusters=["0x0000", "0x0001", "0x0003", "0x0500"]
            #  output_clusters=["0x0019"]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    BasicCluster.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    BasicCluster,
                    XiaomiPowerConfiguration,
                    Identify.cluster_id,
                    XiaomiSmokeIASCluster,
                    DeviceTemperatureCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }
```

You will need to enable your local quirk configuration. There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Copy in your local quirks folder the current quirk:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/mija/smoke.py

Edit the file, adding the proposed quirk at the end of the file.

Restart HA, remove and pair you device again. Check that the device is loading the quirk and that the device signatura has changed.

As side comments, the quirk can configure the battery type. Can you tell which type is?
Not sure if the `NODE_DESCRIPTOR` is needed here and also don't match. If the device (after a couple of tries) don't load the quirk, try to comment or remove this part.

wishie, 2022-10-21:
I will try shortly.

The battery is a CR17450

wishie, 2022-10-21:
Quirk appears to have loaded. Have not tested the "IASZone" yet (its loud).

[image]

MattWestb, 2022-10-21:
@javicalle The `CR17450` is not in ZCL R8 / Zigpy:
https://github.com/zigpy/zigpy/blob/e0faf074f6b26e638003b62180e2854a23e5b092/zigpy/zcl/clusters/general.py#L245-L255

But some "creative" person have adding little extra in ZHA for getting nice Xiaomi sensors displaying other battery formats:
https://github.com/home-assistant/core/blob/4abe5aec6cffa174065c4f8865f091aed871eb19/homeassistant/components/zha/sensor.py#L81-L95
(type 9 - 11 is the extras and wold being better have using type number 20X if ZCL is adding new types in the low range)

Possible using `2: "Other"` or patching ZHA with `CR17450` as done in the last example.

wishie, 2022-10-21:
I guess I will have to set something on fire in the morning to see if the state of the IASZone entity updates when the alarm sounds.

wishie, 2022-10-22:
I set off the alarm, but the state of the entity did not seem to update.

javicalle, 2022-10-22:
Any logs related?
Is the temperature updating? 

Please, enable debug logs and attach the relevant info.

wishie, 2022-10-23:
I have enabled debug logs for 'zigpy' and 'zigpy_znp'.

I have restarted HA. The 'device temperature' entity has always been 'Unknown' and the IASZone has always been 'clear' (as far as I am aware).

I guess I will try to set something on fire again, and see if it updates.

Can I then just use 'Download Diagnostics' or do I have to check in the log file itself and find the relevant parts?

wishie, 2022-10-23:
`2022-10-23 17:07:29.022 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=64704, SrcAddr=0x4D7E, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=94, SecurityUse=<Bool.false: 0>, TimeStamp=10459478, TSN=0, Data=b'\x18\x48\x0A\x3A\x01\x21\x01\x00\x3B\x01\x20\x0A', MacSrcAddr=0x4D7E, MsgResultRadius=29)
2022-10-23 17:07:29.022 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4D7E), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=64704, data=Serialized[b'\x18H\n:\x01!\x01\x00;\x01 \n'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=94, rssi=None)
2022-10-23 17:07:29.022 DEBUG (MainThread) [zigpy.device] Ignoring message (b'18480a3a012101003b01200a') on cluster 64704: unknown endpoint or cluster id: 'No cluster ID 0xfcc0 on (54:ef:44:10:00:48:2d:c1, 1)'
2022-10-23 17:07:36.474 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x15A6, Relays=[0x487F])
2022-10-23 17:07:36.475 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1280, SrcAddr=0x15A6, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=43, SecurityUse=<Bool.false: 0>, TimeStamp=10924815, TSN=0, Data=b'\x19\x77\x00\x00\x00\x00\x00\x00\x00', MacSrcAddr=0x487F, MsgResultRadius=28)
2022-10-23 17:07:36.475 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x15A6), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=1280, data=Serialized[b'\x19w\x00\x00\x00\x00\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=43, rssi=None)
2022-10-23 17:07:36.475 DEBUG (MainThread) [zigpy.zcl] [0x15A6:1:0x0500] Received ZCL frame: b'\x19w\x00\x00\x00\x00\x00\x00\x00'
2022-10-23 17:07:36.475 DEBUG (MainThread) [zigpy.zcl] [0x15A6:1:0x0500] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=119, command_id=0, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-23 17:07:36.476 DEBUG (MainThread) [zigpy.zcl] [0x15A6:1:0x0500] Decoded ZCL frame: IasZone:status_change_notification(zone_status=<ZoneStatus.0: 0>, extended_status=<bitmap8.0: 0>, zone_id=0, delay=0)
2022-10-23 17:07:36.476 DEBUG (MainThread) [zigpy.zcl] [0x15A6:1:0x0500] Received command 0x00 (TSN 119): status_change_notification(zone_status=<ZoneStatus.0: 0>, extended_status=<bitmap8.0: 0>, zone_id=0, delay=0)
2022-10-23 17:07:51.902 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=64704, SrcAddr=0x4D7E, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=43, SecurityUse=<Bool.false: 0>, TimeStamp=11889524, TSN=0, Data=b'\x18\x49\x0A\x3B\x01\x20\x05', MacSrcAddr=0x4D7E, MsgResultRadius=29)
2022-10-23 17:07:51.902 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4D7E), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=64704, data=Serialized[b'\x18I\n;\x01 \x05'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=43, rssi=None)
2022-10-23 17:07:51.902 DEBUG (MainThread) [zigpy.device] Ignoring message (b'18490a3b012005') on cluster 64704: unknown endpoint or cluster id: 'No cluster ID 0xfcc0 on (54:ef:44:10:00:48:2d:c1, 1)'
2022-10-23 17:07:53.083 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2022-10-23 17:07:53.088 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)
2022-10-23 17:08:00.143 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=64704, SrcAddr=0x4D7E, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=65, SecurityUse=<Bool.false: 0>, TimeStamp=12404630, TSN=0, Data=b'\x18\x4A\x0A\x3B\x01\x20\x00', MacSrcAddr=0x4D7E, MsgResultRadius=29)
2022-10-23 17:08:00.143 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4D7E), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=64704, data=Serialized[b'\x18J\n;\x01 \x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=65, rssi=None)
2022-10-23 17:08:00.143 DEBUG (MainThread) [zigpy.device] Ignoring message (b'184a0a3b012000') on cluster 64704: unknown endpoint or cluster id: 'No cluster ID 0xfcc0 on (54:ef:44:10:00:48:2d:c1, 1)'
2022-10-23 17:08:03.952 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x15A6, Relays=[0x487F])
[... cut ...] eply=True), tsn=66, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-23 17:08:36.198 DEBUG (MainThread) [zigpy.zcl] [0x4D7E:1:0x0003] Decoded ZCL frame: Identify:Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-10-23 17:08:53.094 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2022-10-23 17:08:53.099 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)`

wishie, 2022-10-23:
```
2022-10-23 17:07:29.022 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=64704, SrcAddr=0x4D7E, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=94, SecurityUse=<Bool.false: 0>, TimeStamp=10459478, TSN=0, Data=b'\x18\x48\x0A\x3A\x01\x21\x01\x00\x3B\x01\x20\x0A', MacSrcAddr=0x4D7E, MsgResultRadius=29)
2022-10-23 17:07:29.022 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4D7E), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=64704, data=Serialized[b'\x18H\n:\x01!\x01\x00;\x01 \n'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=94, rssi=None)
2022-10-23 17:07:29.022 DEBUG (MainThread) [zigpy.device] Ignoring message (b'18480a3a012101003b01200a') on cluster 64704: unknown endpoint or cluster id: 'No cluster ID 0xfcc0 on (54:ef:44:10:00:48:2d:c1, 1)'
2022-10-23 17:07:36.474 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x15A6, Relays=[0x487F])
2022-10-23 17:07:36.475 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1280, SrcAddr=0x15A6, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=43, SecurityUse=<Bool.false: 0>, TimeStamp=10924815, TSN=0, Data=b'\x19\x77\x00\x00\x00\x00\x00\x00\x00', MacSrcAddr=0x487F, MsgResultRadius=28)
2022-10-23 17:07:36.475 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x15A6), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=1280, data=Serialized[b'\x19w\x00\x00\x00\x00\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=43, rssi=None)
2022-10-23 17:07:36.475 DEBUG (MainThread) [zigpy.zcl] [0x15A6:1:0x0500] Received ZCL frame: b'\x19w\x00\x00\x00\x00\x00\x00\x00'
2022-10-23 17:07:36.475 DEBUG (MainThread) [zigpy.zcl] [0x15A6:1:0x0500] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=119, command_id=0, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-23 17:07:36.476 DEBUG (MainThread) [zigpy.zcl] [0x15A6:1:0x0500] Decoded ZCL frame: IasZone:status_change_notification(zone_status=<ZoneStatus.0: 0>, extended_status=<bitmap8.0: 0>, zone_id=0, delay=0)
2022-10-23 17:07:36.476 DEBUG (MainThread) [zigpy.zcl] [0x15A6:1:0x0500] Received command 0x00 (TSN 119): status_change_notification(zone_status=<ZoneStatus.0: 0>, extended_status=<bitmap8.0: 0>, zone_id=0, delay=0)
2022-10-23 17:07:51.902 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=64704, SrcAddr=0x4D7E, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=43, SecurityUse=<Bool.false: 0>, TimeStamp=11889524, TSN=0, Data=b'\x18\x49\x0A\x3B\x01\x20\x05', MacSrcAddr=0x4D7E, MsgResultRadius=29)
2022-10-23 17:07:51.902 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4D7E), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=64704, data=Serialized[b'\x18I\n;\x01 \x05'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=43, rssi=None)
2022-10-23 17:07:51.902 DEBUG (MainThread) [zigpy.device] Ignoring message (b'18490a3b012005') on cluster 64704: unknown endpoint or cluster id: 'No cluster ID 0xfcc0 on (54:ef:44:10:00:48:2d:c1, 1)'
2022-10-23 17:07:53.083 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2022-10-23 17:07:53.088 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)
2022-10-23 17:08:00.143 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=64704, SrcAddr=0x4D7E, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=65, SecurityUse=<Bool.false: 0>, TimeStamp=12404630, TSN=0, Data=b'\x18\x4A\x0A\x3B\x01\x20\x00', MacSrcAddr=0x4D7E, MsgResultRadius=29)
2022-10-23 17:08:00.143 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4D7E), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=64704, data=Serialized[b'\x18J\n;\x01 \x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=65, rssi=None)
2022-10-23 17:08:00.143 DEBUG (MainThread) [zigpy.device] Ignoring message (b'184a0a3b012000') on cluster 64704: unknown endpoint or cluster id: 'No cluster ID 0xfcc0 on (54:ef:44:10:00:48:2d:c1, 1)'
2022-10-23 17:08:03.952 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x15A6, Relays=[0x487F])
[... cut ...] y=True), tsn=66, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-23 17:08:36.198 DEBUG (MainThread) [zigpy.zcl] [0x4D7E:1:0x0003] Decoded ZCL frame: Identify:Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-10-23 17:08:53.094 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2022-10-23 17:08:53.099 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)
```

javicalle, 2022-11-02:
A couple of observations.

Logs shows 2 devices: `0x15A6` and `0x98BB` (these are the device Nwk values, you can check it in the device view, 'Zigbee info'). None of them seems to be your device (from the screenshot)

There is at least one device reporting values from the `XiaomiAqaraE1Cluster`. We can add it to the quirk and see what is telling to us.
Add the import at the beggining of file:
```python
from zhaquirks.xiaomi import (
    LUMI,
    XIAOMI_NODE_DESC,
    BasicCluster,
    DeviceTemperatureCluster,
    XiaomiAqaraE1Cluster,
    XiaomiPowerConfiguration,
    XiaomiQuickInitDevice,
)
```

And in the replacement part:
```python
    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    BasicCluster,
                    XiaomiPowerConfiguration,
                    Identify.cluster_id,
                    XiaomiSmokeIASCluster,
                    DeviceTemperatureCluster,
                    XiaomiAqaraE1Cluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }
```

Remove any `__pycache__` folder from your local quirk folder, restart HA and re pair the device.

Check the logs to see if there are any regular reports from device. But you will need to repeat the test to see if the quirk is working.

gus1492022, 2022-12-18:
Hi, I have the same problem.  I have bought these sensors, I connect them without problem to zigbee2mqtt.  It has several sensors and options (alarm, mute) turn the light on or off every minute... it works fine when I change the option in home assistant, the problem happens when I set off the smoke alarm.  the alarm sounds but does not send any message to the home assistant.  For it to work I have to enter the expose option and update the status so that it detects the alarm!!  I have tried to do what you say here, but everything continues the same.  how can i check if it is picking up the quirks?  Thanks a lot

javicalle, 2022-12-21:
If you are using zigbee2mqtt, you are not using that code.
You must ask for help in the Z2M repository.

ronirozen, 2022-12-31:
> היי, יש לי אותה בעיה. קניתי את החיישנים האלה, אני מחבר אותם בלי בעיה ל-zigbee2mqtt. יש לו מספר חיישנים ואפשרויות (אזעקה, השתקה) להדליק או לכבות את האור כל דקה... זה עובד מצוין כשאני משנה את האופציה ב-home assistant, הבעיה מתרחשת כשאני מכבה את אזעקת העשן. האזעקה נשמעת אך לא שולחת שום הודעה לעוזרת הביתית. כדי שזה יעבוד אני צריך להיכנס לאפשרות החשיפה ולעדכן את הסטטוס כך שיזהה את האזעקה!! ניסיתי לעשות את מה שאתה אומר כאן, אבל הכל ממשיך אותו דבר. איך אני יכול לבדוק אם זה קולט את המוזרויות? תודה רבה

[gus1492022](https://github.com/gus1492022) Hi, I have the same problem. Did you find a solution?

ronirozen, 2022-12-31:
> Hi, I have the same problem. I have bought these sensors, I connect them without problem to zigbee2mqtt. It has several sensors and options (alarm, mute) turn the light on or off every minute... it [...]

[gus1492022](https://github.com/gus1492022) Hi, I have the same problem. Did you find a solution?

RSDynamics, 2023-02-04:
Als found this definition with ZHA Toolkit.

{
  "ieee": "54:ef:44:10:00:5a:7d:xx",
  "nwk": "0x22bf",
  "model": "lumi.sensor_smoke.acn03",
  "manufacturer": "LUMI",
  "manufacturer_id": "0x4151",
  "endpoints": [
    {
      "id": 1,
      "device_type": "0x0402",
      "profile": "0x0104",
      "in_clusters": {
        "0x0000": {
          "cluster_id": "0x0000",
          "title": "BasicCluster",
          "name": "basic",
          "attributes": {
            "0x0000": {
              "attribute_id": "0x0000",
              "attribute_name": "zcl_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "READ",
              "access_acl": 1,
              "attribute_value": 3
            },
            "0x0001": {
              "attribute_id": "0x0001",
              "attribute_name": "app_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "READ",
              "access_acl": 1,
              "attribute_value": 17
            },
            "0x0002": {
              "attribute_id": "0x0002",
              "attribute_name": "stack_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "READ",
              "access_acl": 1,
              "attribute_value": 2
            },
            "0x0003": {
              "attribute_id": "0x0003",
              "attribute_name": "hw_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "READ",
              "access_acl": 1,
              "attribute_value": 1
            },
            "0x0004": {
              "attribute_id": "0x0004",
              "attribute_name": "manufacturer",
[... 513 lines cut ...]
              "command_id": "0x03",
              "command_name": "3",
              "command_args": "not_in_zcl"
            },
            "0x04": {
              "command_id": "0x04",
              "command_name": "4",
              "command_args": "not_in_zcl"
            },
            "0x06": {
              "command_id": "0x06",
              "command_name": "6",
              "command_args": "not_in_zcl"
            },
            "0x08": {
              "command_id": "0x08",
              "command_name": "8",
              "command_args": "not_in_zcl"
            }
          }
        }
      }
    }
  ]
}

johnboiles, 2023-02-22:
I picked up a few of these and am happy to try things out. To summarize so far:
* Zigbee2MQTT seems to support this device (based on this [post](https://community.home-assistant.io/t/aqara-smoke-detector-linkage-alarm/517656/5) and the Zigbee2MQTT [docs](https://www.zigbee2mqtt.io/devices/JY-GZ-01AQ.html)
* ZHA exposes a functional 'Identify' button and a binary sensor that never updates.
* The quirks proposed so far have not yet worked.

The way forward here is to keep trying to iterate on a quirk until it works right?

TheJulianJES, 2023-02-22:
Relevant Z2M code:

device definition:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/261066095215898c006e7c9c390c677129fae630/devices/xiaomi.js#L2118-L2154

converting messages from the device -> Z2M:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/261066095215898c006e7c9c390c677129fae630/lib/xiaomi.js#L377-L414
https://github.com/Koenkk/zigbee-herdsman-converters/blob/261066095215898c006e7c9c390c677129fae630/lib/xiaomi.js#L464-L473
https://github.com/Koenkk/zigbee-herdsman-converters/blob/261066095215898c006e7c9c390c677129fae630/lib/xiaomi.js#L479-L504
https://github.com/Koenkk/zigbee-herdsman-converters/blob/261066095215898c006e7c9c390c677129fae630/lib/xiaomi.js#L539-L548

messages from Z2M -> device:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/261066095215898c006e7c9c390c677129fae630/converters/toZigbee.js#L2667-L2738

... so it's quite a bit to do.

The framework in ZHA to parse the special Xiaomi messages is here:
https://github.com/zigpy/zha-device-handlers/blob/f6ab87e1c48ccb6db262e8732ca6b63d67fb6252/zhaquirks/xiaomi/__init__.py#L355-L365

johnboiles, 2023-02-22:
All of these transformations should be possible to do in a quirk right?

TheJulianJES, 2023-02-22:
Yeah. The most important part is probably this: https://github.com/Koenkk/zigbee-herdsman-converters/blob/261066095215898c006e7c9c390c677129fae630/lib/xiaomi.js#L380-L382

```python
        elif self.endpoint.device.model == "lumi.sensor_smoke.acn03":
            attribute_names.update({160: "smoke_detected"})
```

The device would need a `LocalDataCluster` for smoke

Then, the "smoke_detected" in the Xiaomi message could be used to write to that "fake" `LocalDataCluster` endpoint (example from another device):
https://github.com/zigpy/zha-device-handlers/blob/f6ab87e1c48ccb6db262e8732ca6b63d67fb6252/zhaquirks/xiaomi/__init__.py#L288-L291

This is just for basic functionality (smoke detected or not).

If I have some time, I could try to actually implement this, but I don't have a sensor myself.
Can anyone who would test this (I'd guess you have to cause some smoke -- not sure the test button is enough) upvote this comment?

johnboiles, 2023-02-22:
I'm definitely in to test this. The other bells and whistles are nice to have someday but getting smoke detection working is certainly the main thing.

I haven't written a quirk before so I asked ChatGPT to take a stab at it and here's what it came up with. Not sure if any of it's useful or not:

<details>
  <summary>ChatGPT's Quirk Attempt</summary>

  ```python
  from zigpy import types as t
  from zigpy.profiles import zha
  from zigpy.quirks import CustomDevice
  from zigpy.zcl import foundation
  from zigpy.zcl.clusters import general, measurement, security
  from zigpy.zcl.clusters.general import Basic, Ota, PowerConfiguration, Identify
  from zigpy.zcl.clusters.security import IasZone
  from zigpy.zcl.clusters.measurement import DeviceTemperature
  
  from zhaquirks import Bus, LocalDataCluster, ZigbeeCustomCluster, EventableCluster
  from zhaquirks.const import (
      BUTTON_1, BUTTON_2, BUTTON_BOTH, COMMAND,
      COMMAND_OFF, COMMAND_ON, DEVICE_TYPE, ENDPOINTS,
      INPUT_CLUSTERS, MODELS_INFO, OUTPUT_CLUSTERS, PROFILE_ID, STATE,
  )
  from zhaquirks.xiaomi import (
      LUMI, XiaomiCustomDevice, XiaomiPowerConfiguration,
      XiaomiQuickInitDevice, XiaomiSmokeIASCluster,
  )
  
  AQARA_CLUSTER_ID = 0xFD00
  AQARA_MANUFACTURER_CODE = 0x115F
  
  
  class JYGZ01AQ(XiaomiCustomDevice):
      """Aqara smoke detector custom device."""
  
      signature = {
          MODELS_INFO: [(LUMI, 'lumi.sensor_smoke.acn03')],
          ENDPOINTS: {
              1: {
                  PROFILE_ID: zha.PROFILE_ID,
                  DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                  INPUT_CLUSTERS: [
                      Basic.cluster_id,
                      PowerConfiguration.cluster_id,
                      Identify.cluster_id,
                      IasZone.cluster_id,
                  ],
                  OUTPUT_CLUSTERS: [Ota.cluster_id],
              },
              2: {
                  PROFILE_ID: zha.PROFILE_ID,
                  DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                  INPUT_CLUSTERS: [
                      Basic.cluster_id,
                      PowerConfiguration.cluster_id,
                      Identify.cluster_id,
                      AQARA_CLUSTER_ID,
                  ],
                  OUTPUT_CLUSTERS: [AQARA_CLUSTER_ID, Ota.cluster_id],
              },
          },
      }
  
      replacement = {
          ENDPOINTS: {
              1: {
                  INPUT_CLUSTERS: [
                      Basic.cluster_id,
                      XiaomiPowerConfiguration.cluster_id,
                      Identify.cluster_id,
                      XiaomiSmokeIASCluster.cluster_id,
                      DeviceTemperature.cluster_id,
                      AQARA_CLUSTER_ID,
                  ],
                  OUTPUT_CLUSTERS: [Ota.cluster_id],
              },
              2: {
                  INPUT_CLUSTERS: [
                      Basic.cluster_id,
                      XiaomiPowerConfiguration.cluster_id,
                      Identify.cluster_id,
                      AQARA_CLUSTER_ID,
                  ],
                  OUTPUT_CLUSTERS: [AQARA_CLUSTER_ID, Ota.cluster_id],
              },
          },
      }
  
      device_automation_triggers = {
          (COMMAND_ON, BUTTON_1): {COMMAND: 'click'},
          (COMMAND_OFF, BUTTON_1): {COMMAND: 'double_click'},
          (COMMAND_ON, BUTTON_2): {COMMAND: 'hold'},
          (COMMAND_OFF, BUTTON_2): {COMMAND: 'release'},
          (COMMAND_ON, BUTTON_BOTH): {COMMAND: 'triple_click'},
      }
  
      def __init__(self, *args, **kwargs):
          """Init method."""
          self.battery_size = 8  # CR123a
          super().__init__(*args, **kwargs)
  
  class AQARA_CLUSTER(XiaomiCustomDevice.cluster_class):
      """Aqara cluster implementation."""
  
      cluster_id = 0xFD00
      ep_attribute = 'aqara_cluster'
      manufacturer_code = 0x115F
  
      attributes = {
          0x0000: ('smoke', t.uint8_t),
          0x0001: ('smoke_density', t.uint8_t),
          0x0002: ('selftest', t.uint8_t),
          0x0003: ('buzzer', t.uint8_t),
          0x0004: ('buzzer_manual_alarm', t.uint8_t),
          0x0005: ('buzzer_manual_mute', t.uint8_t),
          0x0006: ('heartbeat_indicator', t.uint8_t),
          0x0007: ('linkage_alarm', t.uint8_t),
          0x0008: ('linkage_alarm_state', t.uint8_t),
          0x8002: ('battery', t.uint8_t),
          0x8003: ('voltage', t.uint8_t),
          0x8004: ('power_outage_count', t.uint16_t),
      }
  
      COMMANDS = {
          0x00: ('aqara_alarm', (t.uint8_t,), False),
          0x01: ('aqara_density', (t.uint8_t,), False),
          0x02: ('aqara_selftest', (t.uint8_t,), False),
          0x03: ('aqara_buzzer', (t.uint8_t,), False),
          0x04: ('aqara_buzzer_manual', (t.uint8_t,), False),
          0x05: ('JYGZ01AQ_heartbeat_indicator', (t.uint8_t,), False),
          0x06: ('aqara_linkage_alarm', (t.uint8_t,), False),
      }
  
      def handle_cluster_request(self, hdr, args):
          """Handle cluster request."""
          tsn = hdr.sequence_number
          cmd = args[0]
          if cmd not in self.COMMANDS:
              return super().handle_cluster_request(hdr, args)
  
          cmd, args, response_required = self.COMMANDS[cmd]
          if response_required:
              self.debug("Received command %s with response", cmd)
          else:
              self.debug("Received command %s without response", cmd)
  
          return self.endpoint.device.bus.issue_cluster_command(
              hdr,
              self.endpoint.endpoint_id,
              self.cluster_id,
              cmd,
              args,
              manufacturer=self.manufacturer_code,
              expect_reply=response_required,
              tsn=tsn,
          )
  ```
</details>

TheJulianJES, 2023-02-22:
Take a look at: https://github.com/TheJulianJES/zha-device-handlers/tree/tjj/aqara_smoke_detector
If the entity doesn't appear (when removing + re-pairing device), try to see if something appears in the logs when the sensor detects smoke.

You either need to switch `zhaquirks` to `debug` level or do the following:
```python
_LOGGER.debug(
    "%s - Attribute report. attribute_id: [%s] value: [%s]",
```
replace `debug` with `warning` in the above code (in the Xiaomi ``__init__.py`` file. That way, you'll easier see all custom Xiaomi reports (from the smoke sensor)).

To put this in a custom quirk, you have to put the added file and the Xiaomi ``__init__.py`` file in your custom quirk folder and then replace this import in the added smoke quirk file:
```python
from zhaquirks.xiaomi import DeviceTemperatureCluster, XiaomiAqaraE1Cluster
```

with
```python
from . import DeviceTemperatureCluster, XiaomiAqaraE1Cluster
```
(If I'm correct)

---

I'll have another look at this later. No clue if this works at all.

johnboiles, 2023-02-22:
Ok! I'm away from the smoke alarms today but I'll give this a try tomorrow! To confirm, my `custom_zha_quirks` dir should look like this:

```
config/custom_zha_quirks/
∟xiaomi/
  ∟__init__.py
  ∟aqara/
    ∟smoke.py
```

Using [xiaomi/__init__.py](https://github.com/TheJulianJES/zha-device-handlers/blob/5ec4ac7dd39ab60164169ddaa6f7ca00e66cabd3/zhaquirks/xiaomi/__init__.py) (with the `from . import DeviceTemperatureCluster, XiaomiAqaraE1Cluster` change and the `_LOGGER.debug...` change) and [xiomi/aqara/smoke.py](https://github.com/TheJulianJES/zha-device-handlers/blob/5ec4ac7dd39ab60164169ddaa6f7ca00e66cabd3/zhaquirks/xiaomi/aqara/smoke.py). Is that correct?

TheJulianJES, 2023-02-22:
With the `.` import, both files need to be in the same folder (so no aqara folder).

Rest should be good I think.

johnboiles, 2023-02-22:
Ok great, I'll give it a shot tomorrow. Thank you!

johnboiles, 2023-02-23:
Nope did not have an effect. I removed the device. Re-paired it. Then made some smoke to set the alarm off. I didn't get any change or event in Home Assistant. The `Quirk: xiaomi.smoke.LumiSensorSmokeAcn03` line shows that the quirk was loaded correct?

[image]

I'm seeing `AttributeError: 'LumiSensorSmokeAcn03' object has no attribute 'battery_bus'` in the logs as an obvious potential problem. But not sure if that'd affect the smoke things too. I'm also not seeing any temperature entity show up which looks like it maybe should from reading your quirk.

What should I try next?

Here are my full logs:
<details>

<summary>Logs</summary>

    ```
    File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 423, in handle_cluster_general_request
      self._update_attribute(attr.attrid, value)
    File "/config/custom_zha_quirks/xiaomi/__init__.py", line 258, in _update_attribute
      self.endpoint.device.battery_bus.listener_event(
    AttributeError: 'LumiSensorSmokeAcn03' object has no attribute 'battery_bus'
    2023-02-23 10:05:47.610 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=161, profile_id=0, cluster_id=<ZDOCmd.Mgmt_Permit_Joining_req: 0x0036>, data=Serialized[b'\xa1\xfe\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
    2023-02-23 10:05:47.611 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendBroadcast: (0xfffc, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=161), 0, 162, b'\xa1\xfe\x00')
    2023-02-23 10:05:47.614 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'27b321a9622ae94d59947c25aa5592489c4e86ab4fcdc675fdf9477e'
    2023-02-23 10:05:47.617 DEBUG (MainThread) [bellows.ezsp.protocol] Send command addTransientLinkKey: (ff:ff:ff:ff:ff:ff:ff:ff, [90, 105, 103, 66, 101, 101, 65, 108, 108, 105, 97, 110, 99, 101, 48, 57])
    2023-02-23 10:05:47.623 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'73b3a1a9622a15ed60997e'
    2023-02-23 10:05:47.623 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
    2023-02-23 10:05:47.625 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'30b021a9fb2aea4da66bb5da55aac820fb0c42ceaca20be29ca800eccc47cf527e'
    2023-02-23 10:05:47.627 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received sendBroadcast: [<EmberStatus.SUCCESS: 0>, 95]
    2023-02-23 10:05:47.630 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'03b3b5a94d2a89be187e'
    2023-02-23 10:05:47.630 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
    2023-02-23 10:05:47.637 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'13b3b1a9112a10b259a24a25aa5593499c11d8abedce9874fe679d896ee77e'
    2023-02-23 10:05:47.637 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
    2023-02-23 10:05:47.639 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'24b0a1a9fb2a15847c7e'
    2023-02-23 10:05:47.639 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
    2023-02-23 10:05:47.640 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received stackStatusHandler: [<EmberStatus.NETWORK_OPENED: 156>]
    2023-02-23 10:05:47.640 DEBUG (MainThread) [bellows.zigbee.application] Received stackStatusHandler frame with [<EmberStatus.NETWORK_OPENED: 156>]
    2023-02-23 10:05:47.655 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=95), 255, 0, 0x0000, 255, 255, b'\xa1\xfe\x00']
    2023-02-23 10:05:47.656 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=95), 255, 0, 0x0000, 255, 255, b'\xa1\xfe\x00']
    2023-02-23 10:05:47.656 DEBUG (MainThread) [bellows.zigbee.application] Ignoring message type: <EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>
    2023-02-23 10:05:47.656 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received addTransientLinkKey: [<EmberStatus.SUCCESS: 0>]
    2023-02-23 10:05:47.678 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setPolicy: (<EzspPolicyId.TRUST_CENTER_POLICY: 0>, <EzspDecisionBitmask.ALLOW_UNSECURED_REJOINS|ALLOW_JOINS: 3>)
    2023-02-23 10:05:47.681 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'43b121a9012a15b15952fb7e'
    2023-02-23 10:05:47.687 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'35b1a1a9012a156a557e'
    2023-02-23 10:05:47.687 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
    2023-02-23 10:05:47.692 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received setPolicy: [<EzspStatus.SUCCESS: 0>]
[... cut ...] D = 0, NWK_DECRYPTION_FAILURE = 0, APS_DECRYPTION_FAILURE = 0, ALLOCATE_PACKET_BUFFER_FAILURE = 0, RELAYED_UNICAST = 0, PHY_TO_MAC_QUEUE_LIMIT_REACHED = 0, PACKET_VALIDATE_LIBRARY_DROPPED_COUNT = 0, TYPE_NWK_RETRY_OVERFLOW = 0, PHY_CCA_FAIL_COUNT = 0, BROADCAST_TABLE_FULL = 0, PTA_LO_PRI_REQUESTED = 0, PTA_HI_PRI_REQUESTED = 0, PTA_LO_PRI_DENIED = 0, PTA_HI_PRI_DENIED = 0, PTA_LO_PRI_TX_ABORTED = 0, PTA_HI_PRI_TX_ABORTED = 0, ADDRESS_CONFLICT_SENT = 0, EZSP_FREE_BUFFERS = 153]
    ```
</details>

johnboiles, 2023-02-23:
I also triggered the smoke alarm with the logs open. I didn't see a log pop up when the alarm was triggered -- or at least it popped up kinda late. There were some attributes logged:

<details>
<summary>Logs</summary>

    ```
    2023-02-23 10:13:13.689 DEBUG (MainThread) [bellows.ezsp.protocol] Send command permitJoining: (254,)
    2023-02-23 10:13:13.694 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'762421a9762aeb93077e'
    2023-02-23 10:13:13.701 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'6024a1a9762a1529587e'
    2023-02-23 10:13:13.701 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
    2023-02-23 10:13:13.705 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received permitJoining: [<EmberStatus.SUCCESS: 0>]
    2023-02-23 10:15:18.916 ERROR (MainThread) [custom_components.ultimaker.sensor]  Timeout error occurred while polling ultimaker printer using url http://192.168.2.60/api/v1/printer
    2023-02-23 10:15:18.920 ERROR (MainThread) [custom_components.ultimaker.sensor] Cannot parse data received from Ultimaker printer local variable 'response' referenced before assignment
    2023-02-23 10:15:30.571 WARNING (MainThread) [xiaomi] 54:ef:44:10:00:71:1d:00 - Attribute report. attribute_id: [247] value: [{'battery_voltage_mV': 3214, 'temperature': 25, 'X-attrib-4': 5032, 'X-attrib-5': 19, 'X-attrib-6': 6, '0xff01-8': 273, 'path': 0, '0xff01-12': 1, '0xff01-102': 3, '0xff01-103': 1, '0xff01-104': 168, 'smoke': 1, '0xff01-161': 4, '0xff01-162': 0, '0xff01-163': 1, '0xff01-164': 0, '0xff01-165': 0}]
    2023-02-23 10:15:30.575 ERROR (MainThread) [bellows.ezsp] Exception running handler
    Traceback (most recent call last):
      File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py", line 351, in handle_callback
        handler(*args)
      File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 470, in ezsp_callback_handler
        self._handle_frame(*args)
      File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 515, in _handle_frame
        self.packet_received(
      File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 873, in packet_received
        self.handle_message(
      File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 379, in handle_message
        return sender.handle_message(
      File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 370, in handle_message
        self.endpoints[src_ep].handle_message(
      File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 224, in handle_message
        handler(hdr, args, dst_addressing=dst_addressing)
      File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 377, in handle_message
        self.handle_cluster_general_request(hdr, args, dst_addressing=dst_addressing)
      File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 423, in handle_cluster_general_request
        self._update_attribute(attr.attrid, value)
      File "/config/custom_zha_quirks/xiaomi/__init__.py", line 258, in _update_attribute
        self.endpoint.device.battery_bus.listener_event(
    AttributeError: 'LumiSensorSmokeAcn03' object has no attribute 'battery_bus'
    2023-02-23 10:15:54.961 WARNING (MainThread) [xiaomi] 54:ef:44:10:00:71:1d:00 - Attribute report. attribute_id: [247] value: [{'battery_voltage_mV': 3214, 'temperature': 25, 'X-attrib-4': 5032, 'X-attrib-5': 19, 'X-attrib-6': 4, '0xff01-8': 273, 'path': 0, '0xff01-12': 1, '0xff01-102': 3, '0xff01-103': 1, '0xff01-104': 168, 'smoke': 1, '0xff01-161': 1, '0xff01-162': 0, '0xff01-163': 1, '0xff01-164': 0, '0xff01-165': 0}]
    2023-02-23 10:15:54.965 ERROR (MainThread) [bellows.ezsp] Exception running handler
    Traceback (most recent call last):
      File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py", line 351, in handle_callback
        handler(*args)
      File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 470, in ezsp_callback_handler
        self._handle_frame(*args)
      File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 515, in _handle_frame
        self.packet_received(
      File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 873, in packet_received
        self.handle_message(
      File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 379, in handle_message
        return sender.handle_message(
      File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 370, in handle_message
        self.endpoints[src_ep].handle_message(
      File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 224, in handle_message
        handler(hdr, args, dst_addressing=dst_addressing)
      File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 377, in handle_message
        self.handle_cluster_general_request(hdr, args, dst_addressing=dst_addressing)
      File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 423, in handle_cluster_general_request
        self._update_attribute(attr.attrid, value)
      File "/config/custom_zha_quirks/xiaomi/__init__.py", line 258, in _update_attribute
        self.endpoint.device.battery_bus.listener_event(
    AttributeError: 'LumiSensorSmokeAcn03' object has no attribute 'battery_bus'
    ```
</details>

[... cut ...]
2023-02-23 10:16:44.266 WARNING (MainThread) [xiaomi] 54:ef:44:10:00:71:1d:00 - Attribute report. attribute_id: [247] value: [{'battery_voltage_mV': 3195, 'temperature': 25, 'X-attrib-4': 5032, 'X-attrib-5': 20, 'X-attrib-6': 1, '0xff01-8': 273, 'path': 0, '0xff01-12': 1, '0xff01-102': 3, '0xff01-103': 1, '0xff01-104': 168, 'smoke': 0, '0xff01-161': 0, '0xff01-162': 0, '0xff01-163': 0, '0xff01-164': 0, '0xff01-165': 0}]
```

So I guess that's the thing that needs to be hooked up to a HA entity.

TheJulianJES, 2023-02-23:
Hmm, I pushed another commit to add the ``battery_bus`` for now.

I'd have expected to have `'smoke': 1` when there's smoke, but that doesn't look like it's the case?

johnboiles, 2023-02-23:
That fixes the `battery_bus` error! No change in smoke behavior though. But strangely even when I trigger the smoke alarm I'm not getting any logs.

I'm literally right next to my Home Assistant Yellow making smoke in the server closet so it shouldn't be a signal strength thing :) Home assistant shows it last heard from the device the same minute I made smoke (~10:45a), but yet I'm not seeing updates in the logs from that time.

[image]

<details>
<summary>Logs</summary>

    ```
    2023-02-23 10:31:44.668 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received getValue: [<EzspStatus.SUCCESS: 0>, b'\x99']
    2023-02-23 10:31:44.672 DEBUG (MainThread) [bellows.zigbee.application] Free buffers status EzspStatus.SUCCESS, value: 153
    2023-02-23 10:31:44.673 DEBUG (MainThread) [bellows.zigbee.application] ezsp_counters: [MAC_RX_BROADCAST = 2, MAC_TX_BROADCAST = 28, MAC_RX_UNICAST = 257, MAC_TX_UNICAST_SUCCESS = 33, MAC_TX_UNICAST_RETRY = 8, MAC_TX_UNICAST_FAILED = 0, APS_DATA_RX_BROADCAST = 7, APS_DATA_TX_BROADCAST = 7, APS_DATA_RX_UNICAST = 24, APS_DATA_TX_UNICAST_SUCCESS = 11, APS_DATA_TX_UNICAST_RETRY = 0, APS_DATA_TX_UNICAST_FAILED = 0, ROUTE_DISCOVERY_INITIATED = 0, NEIGHBOR_ADDED = 2, NEIGHBOR_REMOVED = 2, NEIGHBOR_STALE = 0, JOIN_INDICATION = 2, CHILD_REMOVED = 2, ASH_OVERFLOW_ERROR = 0, ASH_FRAMING_ERROR = 0, ASH_OVERRUN_ERROR = 0, NWK_FRAME_COUNTER_FAILURE = 0, APS_FRAME_COUNTER_FAILURE = 0, UTILITY = 0, APS_LINK_KEY_NOT_AUTHORIZED = 0, NWK_DECRYPTION_FAILURE = 0, APS_DECRYPTION_FAILURE = 0, ALLOCATE_PACKET_BUFFER_FAILURE = 0, RELAYED_UNICAST = 0, PHY_TO_MAC_QUEUE_LIMIT_REACHED = 0, PACKET_VALIDATE_LIBRARY_DROPPED_COUNT = 0, TYPE_NWK_RETRY_OVERFLOW = 0, PHY_CCA_FAIL_COUNT = 0, BROADCAST_TABLE_FULL = 0, PTA_LO_PRI_REQUESTED = 0, PTA_HI_PRI_REQUESTED = 0, PTA_LO_PRI_DENIED = 0, PTA_HI_PRI_DENIED = 0, PTA_LO_PRI_TX_ABORTED = 0, PTA_HI_PRI_TX_ABORTED = 0, ADDRESS_CONFLICT_SENT = 0, EZSP_FREE_BUFFERS = 153]
    2023-02-23 10:31:52.469 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xBC5C](lumi.sensor_smoke.acn03): Device seen - marking the device available and resetting counter
    2023-02-23 10:31:52.470 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xBC5C](lumi.sensor_smoke.acn03): Update device availability -  device available: True - new availability: True - changed: False
    ```
</details>

TheJulianJES, 2023-02-23:
> Edit edit: Nevermind the logging is in __init__.py which I didn't change. Strange I'm not seeing any Attribute report. attribute_id when I trigger the alarm

Did you change the import in the ``smoke.py`` file again?

johnboiles, 2023-02-23:
Good catch I did not. I need to probably pull these files into git so i'm not forgetting manual changes.

johnboiles, 2023-02-23:
Also. This should be enough to enable debug logging for quirks in general right?
```
logger:
  default: info
  logs:
    zhaquirks: debug
    custom_zha_quirks: debug
```

TheJulianJES, 2023-02-23:
``zhaquirks: debug`` will put all official quirks into debug logger mode, yes.
For the custom quirks, it depends on the folder structure. Looking at your logs, it looks like the logger is called `xiaomi`:
> WARNING (MainThread) [xiaomi]

So ``xiaomi: debug`` would change that logger to debug. I guess you have `xiaomi` -> `smoke` folder -> both files?

(The import in ``smoke.py`` still needs to be changed to use the custom quirk ``__init__.py`` file as long as it's a custom quirk).

johnboiles, 2023-02-23:
I have both files in a dir named `xiaomi`

```
config/custom_zha_quirks/
∟xiaomi/
  ∟__init__.py
  ∟smoke.py
```

So I would do this?

```
logger:
  default: info
  logs:
    zhaquirks: debug
    custom_zha_quirks: debug
    xiaomi: debug
```

johnboiles, 2023-02-23:
Ok changers are made and I've restarted HA and re-paired the detector. Here are the logs from pairing:
<details>
<summary>Logs during discovery</summary>

    ```
    2023-02-23 11:20:20.296 INFO (SyncWorker_1) [homeassistant.loader] Loaded iqvia from homeassistant.components.iqvia
    2023-02-23 11:20:20.298 INFO (SyncWorker_1) [homeassistant.loader] Loaded hisense_aehw4a1 from homeassistant.components.hisense_aehw4a1
    2023-02-23 11:20:20.299 INFO (SyncWorker_1) [homeassistant.loader] Loaded solaredge from homeassistant.components.solaredge
    2023-02-23 11:20:20.300 INFO (SyncWorker_1) [homeassistant.loader] Loaded mikrotik from homeassistant.components.mikrotik
    2023-02-23 11:20:20.302 INFO (SyncWorker_1) [homeassistant.loader] Loaded samsungtv from homeassistant.components.samsungtv
    2023-02-23 11:20:44.372 INFO (MainThread) [homeassistant.components.zha.api] Removing node 54:ef:44:10:00:71:1d:00
    2023-02-23 11:20:44.375 INFO (MainThread) [zigpy.application] Removing device 0x9441 (54:ef:44:10:00:71:1d:00)
    2023-02-23 11:24:47.807 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=11, profile_id=0, cluster_id=<ZDOCmd.Mgmt_Permit_Joining_req: 0x0036>, data=Serialized[b'\x0b\xfe\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
    2023-02-23 11:24:47.810 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendBroadcast: (0xfffc, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=11), 0, 12, b'\x0b\xfe\x00')
    2023-02-23 11:24:47.813 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'229321a9622ae94d59947c25aa5592489c4e2cabe1cd6c75fd93747e'
    2023-02-23 11:24:47.817 DEBUG (MainThread) [bellows.ezsp.protocol] Send command addTransientLinkKey: (ff:ff:ff:ff:ff:ff:ff:ff, [90, 105, 103, 66, 101, 101, 65, 108, 108, 105, 97, 110, 99, 101, 48, 57])
    2023-02-23 11:24:47.825 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'2393a1a9622a15f446137e'
    2023-02-23 11:24:47.825 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
    2023-02-23 11:24:47.828 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'339021a9fb2aea4da66bb5da55aac820fb0c42ceaca20be29ca800eccc473b4c7e'
    2023-02-23 11:24:47.830 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received sendBroadcast: [<EmberStatus.SUCCESS: 0>, 70]
    2023-02-23 11:24:47.831 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'3393b5a94d2a89d29d7e'
    2023-02-23 11:24:47.832 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
    2023-02-23 11:24:47.834 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'4393b1a9112a10b259a24a25aa5593499c08d8abedce9874fecd9d896ae97e'
    2023-02-23 11:24:47.834 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
    2023-02-23 11:24:47.839 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received stackStatusHandler: [<EmberStatus.NETWORK_OPENED: 156>]
    2023-02-23 11:24:47.840 DEBUG (MainThread) [bellows.zigbee.application] Received stackStatusHandler frame with [<EmberStatus.NETWORK_OPENED: 156>]
    2023-02-23 11:24:47.841 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=70), 255, 0, 0x0000, 255, 255, b'\x0b\xfe\x00']
    2023-02-23 11:24:47.841 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=70), 255, 0, 0x0000, 255, 255, b'\x0b\xfe\x00']
    2023-02-23 11:24:47.842 DEBUG (MainThread) [bellows.zigbee.application] Ignoring message type: <EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>
    2023-02-23 11:24:47.847 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'5490a1a9fb2a1535157e'
    2023-02-23 11:24:47.848 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
    2023-02-23 11:24:47.850 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received addTransientLinkKey: [<EmberStatus.SUCCESS: 0>]
    2023-02-23 11:24:47.855 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setPolicy: (<EzspPolicyId.TRUST_CENTER_POLICY: 0>, <EzspDecisionBitmask.ALLOW_UNSECURED_REJOINS|ALLOW_JOINS: 3>)
    2023-02-23 11:24:47.859 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'469121a9012a15b1595e5f7e'
    2023-02-23 11:24:47.867 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'6591a1a9012a15b5ca7e'
    2023-02-23 11:24:47.867 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
    2023-02-23 11:24:47.869 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received setPolicy: [<EzspStatus.SUCCESS: 0>]
    2023-02-23 11:24:48.453 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
[... cut ...]
    2023-02-23 11:27:27.430 WARNING (MainThread) [xiaomi] 54:ef:44:10:00:71:1d:00 - Attribute report. attribute_id: [247] value: [{'battery_voltage_mV': 3183, 'temperature': 25, 'X-attrib-4': 424, 'X-attrib-5': 27, 'X-attrib-6': 1, '0xff01-8': 273, 'path': 0, '0xff01-12': 1, '0xff01-102': 3, '0xff01-103': 1, '0xff01-104': 168, 'smoke': 0, '0xff01-161': 0, '0xff01-162': 0, '0xff01-163': 0, '0xff01-164': 0, '0xff01-165': 0}]
    ```
</details>

Is there something the quirk would need to do to tell the smoke alarm to update its attributes when triggered?

TheJulianJES, 2023-02-23:
> can it be that you need to update the 0x0002 attribute (ZONE_STATUS) and not the ZONE_STATE?

I think that we need to send a "Zone Status Change Notification" command for the IasZone stuff.
Here's how the motion sensor quirks do it:
https://github.com/zigpy/zha-device-handlers/blob/f6ab87e1c48ccb6db262e8732ca6b63d67fb6252/zhaquirks/__init__.py#L287-L288

@johnboiles I've added another commit: https://github.com/zigpy/zha-device-handlers/pull/2219/commits/4d5116a9679c623298ce9b21c901055d4768e59a (modified ``__init__.py`` file)

johnboiles, 2023-02-24:
Heyyy that's some progress!

[image]

But unfortunately this doesn't actually happen until someone hits the mute button. Then it seems to push the state out to the device. In these logs, I triggered the smoke alarm at ~16:38:00 then hit the mute button at 16:38:47. The state did not update until I hit the mute button.

```
2023-02-23 16:21:40.600 WARNING (MainThread) [xiaomi] 54:ef:44:10:00:71:1c:25 - Attribute report. attribute_id: [247] value: [{'battery_voltage_mV': 3229, 'temperature': 25, 'X-attrib-4': 5032, 'X-attrib-5': 11, 'X-attrib-6': 1, '0xff01-8': 273, 'path': 0, '0xff01-12': 1, '0xff01-102': 3, '0xff01-103': 1, '0xff01-104': 168, 'smoke': 0, '0xff01-161': 0, '0xff01-162': 0, '0xff01-163': 0, '0xff01-164': 0, '0xff01-165': 0}]
2023-02-23 16:38:47.880 WARNING (MainThread) [xiaomi] 54:ef:44:10:00:71:1d:00 - Attribute report. attribute_id: [247] value: [{'battery_voltage_mV': 3195, 'temperature': 25, 'X-attrib-4': 5032, 'X-attrib-5': 28, 'X-attrib-6': 3, '0xff01-8': 273, 'path': 0, '0xff01-12': 1, '0xff01-102': 3, '0xff01-103': 1, '0xff01-104': 168, 'smoke': 1, '0xff01-161': 10, '0xff01-162': 0, '0xff01-163': 1, '0xff01-164': 0, '0xff01-165': 0}]
2023-02-23 16:40:20.634 WARNING (MainThread) [xiaomi] 54:ef:44:10:00:71:1d:00 - Attribute report. attribute_id: [247] value: [{'battery_voltage_mV': 3195, 'temperature': 25, 'X-attrib-4': 5032, 'X-attrib-5': 28, 'X-attrib-6': 8, '0xff01-8': 273, 'path': 0, '0xff01-12': 1, '0xff01-102': 3, '0xff01-103': 1, '0xff01-104': 168, 'smoke': 0, '0xff01-161': 0, '0xff01-162': 0, '0xff01-163': 0, '0xff01-164': 0, '0xff01-165': 0}]
```

So the remaining question is, how do we get the device to update attributes when the smoke alarm is triggered.

TheJulianJES, 2023-02-24:
Hmm, so the log message containing `'smoke': 0` did only arrive when you hit the mute button (and not when the alarm was actually triggered)?

And there was nothing in the logs when the alarm triggered, right?

Z2M does some attribute reads when pairing the device -- maybe that's related(?)

johnboiles, 2023-02-24:
Correct on both counts: nothing logs at all when the alarm is triggered. It's not until I hit the mute button that the attribute gets logged (and the binary sensor switches to 'detected'). Generally I can get the device to send attributes anytime I want by hitting the button. Here's a video demonstration: https://youtu.be/Lmj14oUMV3o

I'm new to Zigbee. Is it typical to need to subscribe or bind to certain events in Zigbee?

TheJulianJES, 2023-02-24:
Interesting, thanks for the video.

> Is it typical to need to subscribe or bind to certain events in Zigbee?

Somewhat. Generally, you need to set up reporting for an attribute in order to get "attribute reports". However, Xiaomi/Aqara doesn't follow Zigbee spec at all and they set up their own little messaging system. Most of the time, their devices send these messages automatically (when anything changes). I'll look at what Z2M does that could solve this.

TheJulianJES, 2023-02-24:
Also, what coordinator (and what coordinator firmware) are you using?

johnboiles, 2023-02-24:
I have a Home Assistant Yellow on HA 2023.2.5 
[image]

I think on the Yellow the firmware comes with HA but I'm not sure. Let me know if there's a place to see the firmware version.

I'm out of the office now (where the smoke alarms are) but I'll add logging and try tomorrow! Thanks for all your help on this!

TheJulianJES, 2023-02-24:
Pushed some more changes: https://github.com/zigpy/zha-device-handlers/pull/2219/files

Just to explain it a bit, the added ``attribute_names.update(...)`` adds a proper name for the "special Xiaomi attribute reports" (the ones you see when you press the button). What's weird is that there are many duplicated ones (in Z2M's code).

-> Try to grab a log message when triggering the alarm (maybe you need to press the button for now) (`'smoke': 1`)
-> Try to grab one where it switches back to `'smoke': 0`

---

I've also added the "normal special" Zigbee attributes on the ``OppleCluster``. I'm not sure the types I've given them are correct, because Z2M doesn't really type them.

I'm also unsure why Z2M reads all those attributes when "configuring"/pairing(?)

This time, please delete and re-pair the sensor again before trying anything (possibly so ZHA binds + sets up attribute reporting for those reports?)
~~I'm not (yet) doing anything with possible "normal" attribute reports for the attributes on the `OppleCluster`. They should produce a log message (with the updated logger settings) if there is an attribute report though.~~
I've tried to add some code to also parse normal attribute reports (if they exist) for the `smoke` attribute on the `OppleCluster`, so the entity gets updated. Still, please post the log messages when you trigger the alarm (and when you press the button).

You can also try to read/set those attributes on the `OppleCluster` using the clusters UI in HA.
(Device page -> three dots -> Manage Zigbee device -> clusters, then select the `OppleCluster` and in the second drop down, you can select the attributes and read/set them. Do note that you always want to leave the second text field empty.
If you want to write an attribute, put the text in the first text field (where it's also put if you read an attribute).)

johnboiles, 2023-02-24:
I've applied the new changes, and the logging changes remotely and restarted HA.

I can read the OppleCluster things. I'm getting the following values:
* `buzzer (id: 0x013e)`: `None`
* `buzzer_manual_alarm (id: 0x013d)`: `0`
* `buzzer_manual_mute (id: 0x0126)`: `0`
* `heartbeat_indicator (id: 0x013c)`: `0`
* `linkage_alarm (id: 0x014b)`: `0`
* `selftest (id: 0x0127)`: `None`
* `smoke (id: 0x013a)`: `0`
* `smoke_density (id: 0x013b)`: `0`

Sometimes the attributes would fail to read with some logs like below. So I had to try several times on a few of the attributes. But I expect that's a normal thing when a zigbee device doesn't wake up at the right time right?

<details>
<summary>Logs when reading OppleCluster attributes fail</summary>

```
2023-02-23 19:50:27.493 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received messageSentHandler: [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 37953, EmberApsFrame(profileId=260, clusterId=64704, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=147), 28, <EmberStatus.DELIVERY_FAILED: 102>, b'']
2023-02-23 19:50:27.494 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 37953, EmberApsFrame(profileId=260, clusterId=64704, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=147), 28, <EmberStatus.DELIVERY_FAILED: 102>, b'']
2023-02-23 19:50:27.496 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547977646416] Error handling message: Unknown error (unknown_error) from 127.0.0.1 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/16.3 Safari/605.1.15)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 27, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 820, in websocket_read_zigbee_cluster_attributes
    success, failure = await cluster.read_attributes(
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 471, in read_attributes
    result = await self.read_attributes_raw(to_read, manufacturer=manufacturer)
  File "/usr/local/lib/python3.10/site-packages/zigpy/quirks/__init__.py", line 219, in read_attributes_raw
    return await super().read_attributes_raw(
  File "/usr/local/lib/python3.10/site-packages/zigpy/quirks/__init__.py", line 285, in _read_attributes
    return await super()._read_attributes(
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 324, in request
    return await self._endpoint.request(
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 237, in request
    return await self.device.request(
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 297, in request
    await self._application.request(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 690, in request
    await self.send_packet(
  File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 807, in send_packet
    raise zigpy.exceptions.DeliveryError(
zigpy.exceptions.DeliveryError: Failed to deliver message: 
2023-02-23 19:50:27.511 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingRouteErrorHandler: [<EmberStatus.MAC_INDIRECT_TIMEOUT: 66>, 0x9441]
2023-02-23 19:50:27.512 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteErrorHandler frame with [<EmberStatus.MAC_INDIRECT_TIMEOUT: 66>, 0x9441]
2023-02-23 19:50:27.512 DEBUG (MainThread) [bellows.zigbee.application] Processing route error: status=EmberStatus.MAC_INDIRECT_TIMEOUT, nwk=0x9441
```
</details>

Ooooh if I write a `1` to `buzzer_manual_alarm (id: 0x013d)` I can turn on the alarm (as verified via audio from my security cameras).

Tomorrow I will
* Start collecting logs in a separate window
* Remove and re-pair the smoke alarm (I've been doing that every time anyways just in case -- it's pretty quick)
* Trigger the alarm
* Read the `smoke`, `smoke_density`, `buzzer` attritbutes
* Try to write a 1 to `buzzer_manual_mute`
* (If I don't get a `'smoke':1` message) hit the mute button to send an attribute update
* Cancel the alarm (long press the mute button)
* (if I don't get a `'smoke':0` message) hit the mute button to send an attribute update

Let me know if there's anything else you want me to try!

TheJulianJES, 2023-02-24:
Sounds good!

Actually, I saw that a later Z2M commit removed the device temperature sensor due to being fixed at 25°C like you mentioned, so I'll probably remove it in the next commit too.

johnboiles, 2023-02-25:
It worked! The binary sensor showed `Detected` as soon as the alarm fired! Did you expect that it would work this time? I'm also wondering if setting `linkage_alarm` to `1` in the `OppleCluster` could have had an effect on the real-time alarm reporting. [This post](https://community.home-assistant.io/t/aqara-smoke-detector-linkage-alarm/517656/11?u=johnboiles) says `linkage_alarm` is needed to be able to manually trigger the alarm, perhaps it's also needed to automatically report?

<details>
<summary>Logs from discovery and re-pairing</summary>

```
2023-02-24 16:21:21.220 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received childJoinHandler: [1, <Bool.false: 0>, 0x9441, 54:ef:44:10:00:71:1d:00, <EmberNodeType.SLEEPY_END_DEVICE: 4>]
2023-02-24 16:21:21.221 DEBUG (MainThread) [bellows.zigbee.application] Received childJoinHandler frame with [1, <Bool.false: 0>, 0x9441, 54:ef:44:10:00:71:1d:00, <EmberNodeType.SLEEPY_END_DEVICE: 4>]
2023-02-24 16:21:24.260 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=143, profile_id=0, cluster_id=<ZDOCmd.Mgmt_Permit_Joining_req: 0x0036>, data=Serialized[b'\x8f\xfe\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-02-24 16:21:24.261 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendBroadcast: (0xfffc, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=143), 0, 144, b'\x8f\xfe\x00')
2023-02-24 16:21:24.263 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'75e421a9622ae94d59947c25aa5592489c4ea8ab7d5dcde875fd5b7d387e'
2023-02-24 16:21:24.266 DEBUG (MainThread) [bellows.ezsp.protocol] Send command addTransientLinkKey: (ff:ff:ff:ff:ff:ff:ff:ff, [90, 105, 103, 66, 101, 101, 65, 108, 108, 105, 97, 110, 99, 101, 48, 57])
2023-02-24 16:21:24.276 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'50e4a1a9622a156f919d7e'
2023-02-24 16:21:24.276 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2023-02-24 16:21:24.278 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'06e521a9fb2aea4da66bb5da55aac820fb0c42ceaca20be29ca800eccc4769ec7e'
2023-02-24 16:21:24.282 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'60e4b5a94d2a896d557e'
2023-02-24 16:21:24.282 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2023-02-24 16:21:24.286 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'70e4b1a9112a10b259a24a25aa5593499c93d8abedce9874fe499d89f4fe7e'
2023-02-24 16:21:24.286 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2023-02-24 16:21:24.289 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received sendBroadcast: [<EmberStatus.SUCCESS: 0>, 221]
2023-02-24 16:21:24.290 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received stackStatusHandler: [<EmberStatus.NETWORK_OPENED: 156>]
2023-02-24 16:21:24.290 DEBUG (MainThread) [bellows.zigbee.application] Received stackStatusHandler frame with [<EmberStatus.NETWORK_OPENED: 156>]
2023-02-24 16:21:24.290 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=221), 255, 0, 0x0000, 255, 255, b'\x8f\xfe\x00']
2023-02-24 16:21:24.291 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=221), 255, 0, 0x0000, 255, 255, b'\x8f\xfe\x00']
2023-02-24 16:21:24.291 DEBUG (MainThread) [bellows.zigbee.application] Ignoring message type: <EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>
2023-02-24 16:21:24.294 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'01e5a1a9fb2a15a0b87e'
2023-02-24 16:21:24.294 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2023-02-24 16:21:24.306 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received addTransientLinkKey: [<EmberStatus.SUCCESS: 0>]
2023-02-24 16:21:24.313 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setPolicy: (<EzspPolicyId.TRUST_CENTER_POLICY: 0>, <EzspDecisionBitmask.ALLOW_UNSECURED_REJOINS|ALLOW_JOINS: 3>)
2023-02-24 16:21:24.315 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'7d31ea21a9012a15b15985707e'
2023-02-24 16:21:24.323 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'12eaa1a9012a15aed17e'
2023-02-24 16:21:24.324 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2023-02-24 16:21:24.326 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received setPolicy: [<EzspStatus.SUCCESS: 0>]
2023-02-24 16:21:25.211 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'22eab1a96b2a134ea6944a13aa5592499d4e27767dce6742017e'
2023-02-24 16:21:25.211 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
[... cut ...]
</details>

I also tried some of the `OppleCluster` things. I was able to read `smoke`:`1` and `smoke_density`:`3` (and `10` at one point). And I was able to send `buzzer_manual_mute`:`1` to mute the alarm. Very cool.

Probably the next step would be to expose a few of these `OppleCluster` entities:
* Numeric sensor for `smoke_density`
* Button for `buzzer_manual_mute`
* Button for `buzzer_manual_alarm`

@TheJulianJES do you want me to send one of these to you to be able to land the plane on this?

johnboiles, 2023-02-25:
Ok I tested with another smoke alarm on which I hadn't touched `linkage_alarm` and it was still reporting smoke immediately on the `binary_sensor` :tada:

TheJulianJES, 2023-02-25:
Awesome!

So the sensor didn't fire a "Xiaomi attribute report", but a normal Zigbee attribute report for the `smoke` and `smoke_density` attributes.

Could you press the button on the smoke alarm to generate one of the "Xiaomi attribute reports" again? (which includes a bunch of data that should now be properly named)
Should start with this ``[xiaomi] 54:ef:44:10:00:71:1c:25 - Attribute report.``
That's what all of these are for: https://github.com/zigpy/zha-device-handlers/blob/23994028004c49b1379706c8f681af5ed11ff89e/zhaquirks/xiaomi/__init__.py#L383-L397
I'd like to see if `smoke` and `smoke2` are both present and work (or if just one works) and that for each of the duplicated entries.

> Probably the next step would be to expose a few of these OppleCluster entities:

Yep, we could probably do this, but that would need to be done in ZHA (HA Core) itself, as they're custom attributes. (I'll track the progress for this here when I get to it)

The "Xiaomi attribute reports" should probably also update the cache for the attributes on the `OppleCluster` for that (for now, that's only implemented for `smoke` (but for `smoke` that uses a normal attribute report apparently)).

Axionkt, 2023-02-25:
Here is the result with JY-GZ-03AQ
Reading and writing values to the `OppleCluster` seems to work, they report the correct value and they seem to react appropriately when writing the attributes.
`buzzer_manual_alarm` to `1`: Sounds the alarm
`buzzer_manual_mute` to `1`: Mutes the alarm

When I put the smoke detector in smoke it changes `smoke` to `1` and it clears to `0` after 10 min
`smoke_density` seem to be either `0` (when `smoke=0`) or `10` (when `smoke=1`) this seems to be the case whether I introduct a small or a large amount of smoke. Although this his quite hard to tell as I have no idea how sensitive the detector is.
It is probably behaving as it should but manually setting `smoke` to `0` when it is `1` doesn't work (read only)

The binary_sensor.lumi_lumi_sensor_smoke_acn03_iaszone updates when smoke is detected and returns to cleared after 10minutes

<details><summary>Button press</summary>
<p>

```
2023-02-25 15:32:19.715 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=50), 255, -34, 0xeaf0, 255, 255, b'\x18)\n\x05\x00B\x17lumi.sensor_smoke.acn03']
2023-02-25 15:32:19.716 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=50), 255, -34, 0xeaf0, 255, 255, b'\x18)\n\x05\x00B\x17lumi.sensor_smoke.acn03']
2023-02-25 15:32:19.716 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xEAF0), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=50, profile_id=260, cluster_id=0, data=Serialized[b'\x18)\n\x05\x00B\x17lumi.sensor_smoke.acn03'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-34)
2023-02-25 15:32:19.717 DEBUG (MainThread) [zigpy.zcl] [0xEAF0:1:0x0000] Received ZCL frame: b'\x18)\n\x05\x00B\x17lumi.sensor_smoke.acn03'
2023-02-25 15:32:19.717 DEBUG (MainThread) [zigpy.zcl] [0xEAF0:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=41, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-02-25 15:32:19.718 DEBUG (MainThread) [zigpy.zcl] [0xEAF0:1:0x0000] Decoded ZCL frame: Basic:Report_Attributes(attribute_reports=[Attribute(attrid=0x0005, value=TypeValue(type=CharacterString, value='lumi.sensor_smoke.acn03'))])
2023-02-25 15:32:19.718 DEBUG (MainThread) [zigpy.zcl] [0xEAF0:1:0x0000] Received command 0x0A (TSN 41): Report_Attributes(attribute_reports=[Attribute(attrid=0x0005, value=TypeValue(type=CharacterString, value='lumi.sensor_smoke.acn03'))])
2023-02-25 15:32:19.719 DEBUG (MainThread) [zigpy.zcl] [0xEAF0:1:0x0000] Attribute report received: model='lumi.sensor_smoke.acn03'
2023-02-25 15:32:19.723 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=64704, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=51), 255, -33, 0xeaf0, 255, 255, b'\x1c_\x11*\n\xf7\x00A>\x01!\xfe\x0b\x03(\x19\x04!\xa8\x13\x05!+\x00\x06$\x01\x00\x00\x00\x00\x08!\x11\x01\n!\x00\x00\x0c \x01f \x03g \x01h!\xa8\x00\xa0!\x00\x00\xa1 \x00\xa2 \x00\xa3 \x00\xa4 \x00\xa5 \x00']
2023-02-25 15:32:19.723 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=64704, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=51), 255, -33, 0xeaf0, 255, 255, b'\x1c_\x11*\n\xf7\x00A>\x01!\xfe\x0b\x03(\x19\x04!\xa8\x13\x05!+\x00\x06$\x01\x00\x00\x00\x00\x08!\x11\x01\n!\x00\x00\x0c \x01f \x03g \x01h!\xa8\x00\xa0!\x00\x00\xa1 \x00\xa2 \x00\xa3 \x00\xa4 \x00\xa5 \x00']
2023-02-25 15:32:19.723 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xEAF0), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=51, profile_id=260, cluster_id=64704, data=Serialized[b'\x1c_\x11*\n\xf7\x00A>\x01!\xfe\x0b\x03(\x19\x04!\xa8\x13\x05!+\x00\x06$\x01\x00\x00\x00\x00\x08!\x11\x01\n!\x00\x00\x0c \x01f \x03g \x01h!\xa8\x00\xa0!\x00\x00\xa1 \x00\xa2 \x00\xa3 \x00\xa4 \x00\xa5 \x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-33)
2023-02-25 15:32:19.724 DEBUG (MainThread) [zigpy.zcl] [0xEAF0:1:0xfcc0] Received ZCL frame: b'\x1c_\x11*\n\xf7\x00A>\x01!\xfe\x0b\x03(\x19\x04!\xa8\x13\x05!+\x00\x06$\x01\x00\x00\x00\x00\x08!\x11\x01\n!\x00\x00\x0c \x01f \x03g \x01h!\xa8\x00\xa0!\x00\x00\xa1 \x00\xa2 \x00\xa3 \x00\x [... cut ...]
2023-02-25 15:44:16.570 DEBUG (MainThread) [zigpy.zcl] [0xEAF0:1:0xfcc0] Decoded ZCL frame: OppleCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0126, value=TypeValue(type=uint8_t, value=0))])
2023-02-25 15:44:16.570 DEBUG (MainThread) [zigpy.zcl] [0xEAF0:1:0xfcc0] Received command 0x0A (TSN 51): Report_Attributes(attribute_reports=[Attribute(attrid=0x0126, value=TypeValue(type=uint8_t, value=0))])
2023-02-25 15:44:16.571 DEBUG (MainThread) [zigpy.zcl] [0xEAF0:1:0xfcc0] Attribute report received: buzzer_manual_mute=0
```
</p>
</details>

johnboiles, 2023-02-25:
> Could you press the button on the smoke alarm to generate one of the "Xiaomi attribute reports" again? (which includes a bunch of data that should now be properly named)

Looks like @Axionkt got one! Thanks @Axionkt. I can do this on Monday also when I'm back with the alarms (in my office).

> Yep, we could probably do this, but that would need to be done in ZHA (HA Core) itself

Cool no rush, obviously smoke detection is the most important function of a smoke detector :) Thanks for all your work on this so far!

> I'd love to have one, but I'm based in Germany, so I guess shipping will be a bit difficult.

Yeah for sure. Happy to send you some $$ on PayPal or something for one if you want!

TheJulianJES, 2023-02-26:
Nice! The logs are all really helpful. It looks like attribute reporting is properly set up for `buzzer_manual_mute` for example. So that would be that HA entity possible relatively easily.

I do wonder if we need to parse these "Xiaomi attribute reports" at all:
```python
[xiaomi] 54:ef:44:10:00:5a:6f:93 - Attribute report. attribute_id: [247] value: [{'battery_voltage_mV': 3070, 'temperature': 25, 'X-attrib-4': 5032, 'X-attrib-5': 43, 'X-attrib-6': 4, '0xff01-8': 273, 'path': 0, '0xff01-12': 1, '0xff01-102': 3, '0xff01-103': 1, '0xff01-104': 168, 'smoke': 1, 'smoke_density': 10, 'test': 0, 'buzzer_manual_mute': 1, 'heartbeat_indicator': 0, 'linkage_alarm': 0}]
```
Currently, only the `smoke` attribute is parsed from that (but only a normal attribute report is sent when smoke is detected, so the above doesn't seem to really matter?).
I wonder if these Xiaomi attribute reports are really only sent when the button is pressed or periodically?

> Looks like @Axionkt got one! Thanks @Axionkt. I can do this on Monday also when I'm back with the alarms (in my office).

@johnboiles Yep, it would be nice if you could also get one of these Xiaomi attribute reports then (so the `[xiaomi]` log lines when you press the button). I wonder if there are model differences.
Z2M had a bunch of duplicated Xiaomi attributes in those reports, but they don't seem to be in the Xiaomi attribute report from @Axionkt

> Yeah for sure. Happy to send you some $$ on PayPal or something for one if you want!

I've had a look on AliExpress and they seem to go for about 42€ (44 USD). Shipping will probably take a bit, but it would make the entity stuff in HA easier for me if I had one.
If you want, you can contact me at julian.tjj@proton.me and I can send you my PayPal address (different from that email).
(No need/pressure for you to send me anything of course)

johnboiles, 2023-02-27:
This is what I got when I pushed the button!

```python
{
    'battery_voltage_mV': 3202,
    'temperature': 25,
    'X-attrib-4': 5032,
    'X-attrib-5': 15,
    'X-attrib-6': 1,
    '0xff01-8': 273,
    'path': 0,
    '0xff01-12': 1,
    '0xff01-102': 3,
    '0xff01-103': 1,
    '0xff01-104': 168,
    'smoke': 0,
    'smoke_density': 0,
    'test': 0,
    'buzzer_manual_mute': 0,
    'heartbeat_indicator': 0,
    'linkage_alarm': 0
}
```

<details>
<summary>Full logs from pushing the button</summary>

```
2023-02-27 13:20:23.091 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=96), 156, -61, 0xbc5c, 255, 255, b'\x18x\n\x05\x00B\x17lumi.sensor_smoke.acn03']
2023-02-27 13:20:23.092 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=96), 156, -61, 0xbc5c, 255, 255, b'\x18x\n\x05\x00B\x17lumi.sensor_smoke.acn03']
2023-02-27 13:20:23.092 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xBC5C), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=96, profile_id=260, cluster_id=0, data=Serialized[b'\x18x\n\x05\x00B\x17lumi.sensor_smoke.acn03'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=156, rssi=-61)
2023-02-27 13:20:23.093 DEBUG (MainThread) [zigpy.zcl] [0xBC5C:1:0x0000] Received ZCL frame: b'\x18x\n\x05\x00B\x17lumi.sensor_smoke.acn03'
2023-02-27 13:20:23.094 DEBUG (MainThread) [zigpy.zcl] [0xBC5C:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=120, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-02-27 13:20:23.097 DEBUG (MainThread) [zigpy.zcl] [0xBC5C:1:0x0000] Decoded ZCL frame: Basic:Report_Attributes(attribute_reports=[Attribute(attrid=0x0005, value=TypeValue(type=CharacterString, value='lumi.sensor_smoke.acn03'))])
2023-02-27 13:20:23.098 DEBUG (MainThread) [zigpy.zcl] [0xBC5C:1:0x0000] Received command 0x0A (TSN 120): Report_Attributes(attribute_reports=[Attribute(attrid=0x0005, value=TypeValue(type=CharacterString, value='lumi.sensor_smoke.acn03'))])
2023-02-27 13:20:23.099 DEBUG (MainThread) [zigpy.zcl] [0xBC5C:1:0x0000] Attribute report received: model='lumi.sensor_smoke.acn03'
2023-02-27 13:20:23.115 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=64704, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=97), 156, -61, 0xbc5c, 255, 255, b'\x1c_\x11y\n\xf7\x00A>\x01!\x82\x0c\x03(\x19\x04!\xa8\x13\x05!\x0f\x00\x06$\x01\x00\x00\x00\x00\x08!\x11\x01\n!\x00\x00\x0c \x01f \x03g \x01h!\xa8\x00\xa0!\x00\x00\xa1 \x00\xa2 \x00\xa3 \x00\xa4 \x00\xa5 \x00']
2023-02-27 13:20:23.116 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=64704, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=97), 156, -61, 0xbc5c, 255, 255, b'\x1c_\x11y\n\xf7\x00A>\x01!\x82\x0c\x03(\x19\x04!\xa8\x13\x05!\x0f\x00\x06$\x01\x00\x00\x00\x00\x08!\x11\x01\n!\x00\x00\x0c \x01f \x03g \x01h!\xa8\x00\xa0!\x00\x00\xa1 \x00\xa2 \x00\xa3 \x00\xa4 \x00\xa5 \x00']
2023-02-27 13:20:23.116 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xBC5C), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=97, profile_id=260, cluster_id=64704, data=Serialized[b'\x1c_\x11y\n\xf7\x00A>\x01!\x82\x0c\x03(\x19\x04!\xa8\x13\x05!\x0f\x00\x06$\x01\x00\x00\x00\x00\x08!\x11\x01\n!\x00\x00\x0c \x01f \x03g \x01h!\xa8\x00\xa0!\x00\x00\xa1 \x00\xa2 \x00\xa3 \x00\xa4 \x00\xa5 \x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=156, rssi=-61)
2023-02-27 13:20:23.118 DEBUG (MainThread) [zigpy.zcl] [0xBC5C:1:0xfcc0] Received ZCL frame: b'\x1c_\x11y\n\xf7\x00A>\x01!\x82\x0c\x03(\x19\x04!\xa8\x13\x05!\x0f\x00\x06$\x01\x00\x00\x00\x00\x08!\x11\x01\n!\x00\x00\x0c \x01f \x03g \x01h!\xa8\x00\xa0!\x00\x00\xa1 \x00\xa2 \x00\xa3 \x00\xa4 \x00\xa5 \x00'
2023-02-27 13:20:23.119 DEBUG (MainThread) [zigpy.zcl] [0xBC5C:1:0xfcc0] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), manufacturer=4447, tsn=121, comman [... cut ...]
2023-02-27 13:20:23.122 WARNING (MainThread) [xiaomi] 54:ef:44:10:00:71:1c:25 - Attribute report. attribute_id: [247] value: [{'battery_voltage_mV': 3202, 'temperature': 25, 'X-attrib-4': 5032, 'X-attrib-5': 15, 'X-attrib-6': 1, '0xff01-8': 273, 'path': 0, '0xff01-12': 1, '0xff01-102': 3, '0xff01-103': 1, '0xff01-104': 168, 'smoke': 0, 'smoke_density': 0, 'test': 0, 'buzzer_manual_mute': 0, 'heartbeat_indicator': 0, 'linkage_alarm': 0}]
2023-02-27 13:20:23.136 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xBC5C:1:0x0500]: Updated alarm state: 0
```
</details>

TheJulianJES, 2023-03-06:
Got the sensor today and I'm currently working on it. Was able to clean up the quirk code a bit and got entities in HA. The only entity that is currently missing is the "smoke density" one. The 0 to 10 for smoke density could be easily exposed, but Z2M does some internal conversion from Xiaomi's scale to dB/m. I don't think exposing both is great, so we could either (1) just expose the raw value, (2) do the conversion in zha-quirks and then update it as a fake attribute, or (3) do the conversion in HA Core itself. Have to think about this.

Due to how ZHA currently works, the other entities will be exposed as a configuration entities, but I think that should be fine (not in area calls and not exposed to cloud/Alexa/Google Assistant/... by default).

Although I've read contradicting information, I think the linkage alarm works like this (have yet to confirm this).
When there's smoke without linkage alarm, the "linkage alarm state" binary sensor does NOT turn on.
When there's smoke with linkage alarm enabled, the "linkage alarm state" binary sensor turns on for until how long the smoke is present OR the sensor is muted (either via UI or by a button press/hold).

The manual "buzzer manual alarm" works regardless of if linkage alarm is enabled for me.

When triggering it manually, the manual alarm seems to take priority over the manual mute.
I think it's best if these are just kept as two switches instead of doing anything more complicated.

Z2M implemented some of these features in a bit of a weird way from what I can see. Write only and then another sensor entity for reading. 

This is how it currently looks in HA:

[image]

If you have different ideas for default mdi icons, please let me know. (https://pictogrammers.com/library/mdi/)

Currently, I'm also not actually parsing the Xiaomi attribute reports (except for `smoke`, but I could probably drop that too).
It looks like these reports are only sent when pressing the button (and maybe once every couple of hours?)
I don't really see a benefit, as the custom attributes themselves report properly by themselves.

RSDynamics, 2023-03-07:
Thanks for your great work!
I copied your smoke.py and __init__.py from the xiomi folder but I only see the Iaszone and the diagnostics.
I have three of these detectors and I removed them all, deleted the pycache but the end resuld stays the same.
Is there something else I should do to get the rest of the entities?
Thnx!

TheJulianJES, 2023-03-07:
The zigbee.db thing isn't the issue. Due to how ZHA works at the moment, the entities need to be created in Home Assistant Core. There's no good way to test that. (I'm also not sure if I already pushed those changes, but I might have)

javicalle, 2023-03-07:
You've probably thought of this already, but would it make sense to set `Buzzer manual alarm` and `Buzzer manual mute` as buttons instead of switches?
In the case of `Buzzer manual alarm` how/when is the value reset to off?

TheJulianJES, 2023-03-08:
The smoke detector itself can reset that and you can also disable the mute during an active alarm again and the alarm resumes (as long as there's smoke). It's also toggled (and reflected in HA UI) if it's changed by clicking the button on the sensor during an alarm.

RSDynamics, 2023-03-10:
Hi @TheJulianJES 
I don't know if this is to be expected in the current state but I have two things that don't work with this quirk in my environment.

On the the Device Information tab of the device the powersource the keeps stating "Mains" and no battery battery entities are created. When I read the values with the Manage Zigbee Device function, in the power_source (id: 0x0007) in the Basic cluster states PowerSource.Battery. I don't know if this is important but when I read the battery_size (id: 0x0031) from the PowerConfiguration cluster it states CR2 but the actual battery size is CR17450.
In the PowerConfiguration cluster I can get the battery_voltage (id: 0x0020) and other battery values but the battery_percentage_remaining (id: 0x0021) states None.

Also when I try to trigger the self_test with value 1 from the Manage Zigbee Device function the write seems to succeed but the self_test is not triggered.
When I use the zha_toolkit service and send attr_type 16 with attr_val 1 to the attribute 295 the self_test does fire.
Maybe the attr_type is wrong?

Thanks for al the great work!

[24 more replies]

closed 2023-10-18 by github-actions[bot]