zigpy/zha-device-handlers #1829: miboxer fut089z integration

mattiabrunello5, 2022-10-15:
**Is your feature request related to a problem? Please describe.**
hi i wish to integrate the color response on Home Assistant (if i change the color via color wheel on the remote the controller fire it but HA does not.
the on/off toggle and brightness level works fine

**Describe the solution you'd like**
integrate the color change
<details>

<summary>{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x1000",
        "0xe002",
        "0xef00"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0019",
        "0x1000",
        "0xe002"
      ]
    }
  },
  "manufacturer": "_TZ3000_xwh1e22x",
  "model": "TS1002",
  "class": "zigpy.device.Device"
}</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Logger: zigpy.zdo
Source: runner.py:119
First occurred: 11:09:48 (221 occurrences)
Last logged: 17:25:10

[0x0000:zdo] Unknown ZDO cluster 0x0300
[0x0000:zdo] Unknown ZDO cluster 0x0008
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-10-15:
We need that you enable the debug logs and attach the relevant info:
* https://www.home-assistant.io/integrations/zha/#debug-logging

It is also necesary that you can correlate button events with your logs.

According to `herdsman-converters` it seems to be an `EnchantedDevice`:
* https://github.com/Koenkk/zigbee-herdsman-converters/pull/4153/files

mattiabrunello5, 2022-10-15:
these are the event of zigbee device

Evento 17 generato 0:37:
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
event_type: zha_event
data:
  device_ieee: 54:0f:57:ff:fe:19:ed:24
  unique_id: 54:0f:57:ff:fe:19:ed:24:1:0x0008
  device_id: 2d2248ee911555f4c29e83f1281376d3
  endpoint_id: 1
  cluster_id: 8
  command: move_to_level
  args:
    - 254
    - 0
  params:
    level: 254
    transition_time: 0
    options_mask: null
    options_override: null
origin: LOCAL
time_fired: "2022-10-15T22:37:16.522826+00:00"
context:
  id: 01GFEWPQVAC1CMA14SWEYHYHT9
  parent_id: null
  user_id: null

Evento 16 generato 0:37:
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
[... 707 lines cut ...]
event_type: zha_event
data:
  device_ieee: 54:0f:57:ff:fe:19:ed:24
  unique_id: 54:0f:57:ff:fe:19:ed:24:1:0x0008
  device_id: 2d2248ee911555f4c29e83f1281376d3
  endpoint_id: 1
  cluster_id: 8
  command: move_to_level
  args:
    - 254
    - 0
  params:
    level: 254
    transition_time: 0
    options_mask: null
    options_override: null
origin: LOCAL
time_fired: "2022-10-15T22:36:52.745692+00:00"
context:
  id: 01GFEWP0M9CXRYR23M8XYZ29VA
  parent_id: null
  user_id: null

and these are the debug log
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/9793250/home-assistant.log)

javicalle, 2022-10-16:
Please, put all the events log between ` ``` ` marks to make it readable.

What are all these events?
Except an on/off event I just see dimming events to the max brightness.

I don't know the device, I don't know what tests are you doing and what are you expecting from the device. 

Help me to help you.

mattiabrunello5, 2022-10-16:
the event pop out when i press the touch button on the remote (put a photo below),even if i press the brightness, color saturation  or the color wheel
[image]

REMEMBER: this remote control the light via RGBW controller, this two devices talk perfect to each other, the problem is when i change the color via this remote home assistant via ZHA don't update the color. i need to resolve this  

this is the log  
'''
2022-10-16 13:38:53.281 DEBUG (MainThread) [zigpy_deconz.zigbee.application] device: 0xb6c7 - _TZ3000_xwh1e22x TS1002, FFD=False, Rx_on_when_idle=False
2022-10-16 13:38:53.281 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Restoring 54:0f:57:ff:fe:19:ed:24/0xb6c7 device as direct child
2022-10-16 13:38:53.289 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_indication[30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0xcccf>, 10, 260, 6, b'\x08E\n\x00\x00\x10\x00', 0, 175, 135, 174, 248, 30, 0, -77]
2022-10-16 13:38:53.289 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0xcccf>, 10, 260, 6, b'\x08E\n\x00\x00\x10\x00', 0, 175, 135, 174, 248, 30, 0, -77]
2022-10-16 13:38:53.289 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xCCCF), src_ep=10, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=6, data=Serialized[b'\x08E\n\x00\x00\x10\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=135, rssi=-77)
2022-10-16 13:38:53.290 DEBUG (MainThread) [zigpy.zcl] [0xCCCF:10:0x0006] Received ZCL frame: b'\x08E\n\x00\x00\x10\x00'
2022-10-16 13:38:53.291 DEBUG (MainThread) [zigpy.zcl] [0xCCCF:10:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=69, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-16 13:38:53.291 DEBUG (MainThread) [zigpy.zcl] [0xCCCF:10:0x0006] Decoded ZCL frame: OnOff:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-10-16 13:38:53.291 DEBUG (MainThread) [zigpy.zcl] [0xCCCF:10:0x0006] Received command 0x0A (TSN 69): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-10-16 13:38:53.292 DEBUG (MainThread) [zigpy.zcl] [0xCCCF:10:0x0006] Attribute report received: on_off=<Bool.false: 0>
2022-10-16 13:38:53.292 DEBUG (MainThread) [zigpy.zcl] [0xCCCF:10:0x0006] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=69, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-16 13:38:53.293 DEBUG (MainThread) [zigpy.zcl] [0xCCCF:10:0x0006] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2022-10-16 13:38:53.293 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=10, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xCCCF), dst_ep=10, source_route=None, extended_timeout=False, tsn=69, profile_id=260, cluster_id=6, data=Serialized[b'\x18E\x0b\n\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-10-16 13:38:53.293 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=AddressMode.NWK address=0xcccf>, ep: 10, profile: 0x0104, cluster_id: 0x0006, data: b'08450a00001000'
2022-10-16 13:38:53.294 DEBUG (MainThread) [zigpy_deconz.api] Command Command.add_neighbour (12, 1, 0xB6C7, 54:0f:57:ff:fe:19:ed:24, <MACCapabilityFlags.AllocateAddress: 128>)
2022-10-16 13:38:53.297 DEBUG (MainThread) [zigpy_deconz.api] Received command add_neighbour[12, 1, 0xb6c7, 54:0f:57:ff:fe:19:ed:24, 128]
2022-10-16 13:38:53.297 DEBUG (MainThread) [zigpy_deconz.api] add neighbour response: [12, 1, 0xb6c7, 54:0f:57:ff:fe:19:ed:24, 128]
2022-10-16 13:38:53.297 DEBUG (MainThread) [zigpy_deconz.zigbee.application] device: 0x82a7 - _TZ3210_ttkgurpb TS0504B, FFD=True, Rx_on_when_idle=True
2022-10-16 13:38:53.298 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=2, profile_id=0, cluster_id=<ZDOCmd.Mgmt_Permit_Joining_req: 0x0036>, data=Serialized[b'\x02\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
[... cut ...] ned_0x00: 0>))])
2022-10-16 13:38:53.334 DEBUG (MainThread) [zigpy.zcl] [0xF5AF:1:0x0201] Received command 0x0A (TSN 7): Report_Attributes(attribute_reports=[Attribute(attrid=0x0029, value=TypeValue(type=bitmap16, value=<bitmap16.0: 0>)), Attribute(attrid=0x001C, value=TypeValue(type=enum8, value=<enum8.undefined_0x00: 0>))])
2022-10-16 13:38:53.335 DEBUG (MainThread) [zigpy.zcl] [0xF5AF:1:0x0201] Attribute report received: running_state=<bitmap16.0: 0>, system_mode=<enum8.undefined_0x00: 0>
'''

javicalle, 2022-10-16:
The format marks are these ` ``` ` not these ` ''' `.
Your device seems to be the `0xb6c7` but there are no logs related to that device. They would be something like:
```
2022-10-16 13:38:53.307 DEBUG (MainThread) [zigpy.zcl] [0xb6c7:1:0x0006] Received ZCL frame: b'\x08F\n\x00\x00\x10\x00'
2022-10-16 13:38:53.307 DEBUG (MainThread) [zigpy.zcl] [0xb6c7:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=70, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-16 13:38:53.307 DEBUG (MainThread) [zigpy.zcl] [0xb6c7:1:0x0006] Decoded ZCL frame: OnOff:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-10-16 13:38:53.308 DEBUG (MainThread) [zigpy.zcl] [0xb6c7:1:0x0006] Received command 0x0A (TSN 70): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-10-16 13:38:53.308 DEBUG (MainThread) [zigpy.zcl] [0xb6c7:1:0x0006] Attribute report received: on_off=<Bool.false: 0>
2022-10-16 13:38:53.309 DEBUG (MainThread) [zigpy.zcl] [0xb6c7:1:0x0006] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=70, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-16 13:38:53.309 DEBUG (MainThread) [zigpy.zcl] [0xb6c7:1:0x0006] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
```

What would help would be to correlate and event with its log. If you don't say what is happening I can't guess from the logs.

For example:

When I push the button on, the logs show:
```
2022-10-12 21:07:09.852 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=239), 200, -50, 0xb4e7, 255, 255, b'\te\x01\x00\t\x02\x04\x00\x01\x00']
2022-10-12 21:07:09.853 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB4E7), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=239, profile_id=260, cluster_id=61184, data=Serialized[b'\te\x01\x00\t\x02\x04\x00\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=200, rssi=-50)
2022-10-12 21:07:09.854 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Received ZCL frame: b'\te\x01\x00\t\x02\x04\x00\x01\x00'
2022-10-12 21:07:09.855 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=101, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-12 21:07:09.860 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Decoded ZCL frame: AtlanticSwitchManufCluster:get_data(data=TuyaCommand(status=0, tsn=9, datapoints=[TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-10-12 21:07:09.861 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Received command 0x01 (TSN 101): get_data(data=TuyaCommand(status=0, tsn=9, datapoints=[TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-10-12 21:07:09.862 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] No datapoint handler for TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
2022-10-12 21:07:09.864 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=101, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-12 21:07:09.865 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>)
2022-10-12 21:07:09.866 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB4E7), dst_ep=1, source_route=None, extended_timeout=False, tsn=101, profile_id=260, cluster_id=61184, data=Serialized[b'\x18e\x0b\x01\x86'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
[... cut ...]
2022-10-12 21:07:01.631 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
2022-10-12 21:07:01.632 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB4E7), dst_ep=1, source_route=None, extended_timeout=False, tsn=99, profile_id=260, cluster_id=61184, data=Serialized[b'\x18c\x0b\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
```

mattiabrunello5, 2022-10-17:
when i change the color to yellow i get this on the log
```2022-10-17 10:12:45.818 DEBUG (MainThread) [zigpy_deconz.api] Received command device_state_changed[<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-10-17 10:12:45.818 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-10-17 10:12:45.819 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, <DataIndicationFlags.Always_Use_NWK_Source_Addr: 1>)
2022-10-17 10:12:45.823 DEBUG (MainThread) [zigpy_deconz.api] Received command device_state_changed[<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-10-17 10:12:45.823 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-10-17 10:12:45.823 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_indication[30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, <DeconzAddress address_mode=AddressMode.GROUP address=0xa001>, 0, <DeconzAddress address_mode=AddressMode.NWK address=0xb6c7>, 1, 260, 768, b'\x01_\x06&\xfe\x00\x00', 0, 175, 255, 244, 32, 31, 0, -22]
2022-10-17 10:12:45.823 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, <DeconzAddress address_mode=AddressMode.GROUP address=0xa001>, 0, <DeconzAddress address_mode=AddressMode.NWK address=0xb6c7>, 1, 260, 768, b'\x01_\x06&\xfe\x00\x00', 0, 175, 255, 244, 32, 31, 0, -22]
2022-10-17 10:12:45.823 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB6C7), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.Group: 1>, address=0xA001), dst_ep=0, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=768, data=Serialized[b'\x01_\x06&\xfe\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-22)
2022-10-17 10:12:45.823 WARNING (MainThread) [zigpy.zdo] [0x0000:zdo] Unknown ZDO cluster 0x0300
2022-10-17 10:12:45.824 DEBUG (MainThread) [zigpy.device] Ignoring message (b'015f0626fe0000') on cluster 768: unknown endpoint or cluster id: 'No cluster ID 0x0300 on (54:0f:57:ff:fe:19:ed:24, 1)'
2022-10-17 10:12:45.825 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=AddressMode.NWK address=0xb6c7>, ep: 1, profile: 0x0104, cluster_id: 0x0300, data: b'015f0626fe0000'
2022-10-17 10:12:45.825 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, <DataIndicationFlags.Always_Use_NWK_Source_Addr: 1>)
2022-10-17 10:12:45.829 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_indication[29, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.GROUP address=0xa001>, 0, <DeconzAddress address_mode=AddressMode.NWK address=0xb6c7>, 1, 260, 8, b'\x01`\x00\xfe\x00\x00', 0, 175, 255, 178, 32, 31, 0, -22]
2022-10-17 10:12:45.829 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [29, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.GROUP address=0xa001>, 0, <DeconzAddress address_mode=AddressMode.NWK address=0xb6c7>, 1, 260, 8, b'\x01`\x00\xfe\x00\x00', 0, 175, 255, 178, 32, 31, 0, -22]
2022-10-17 10:12:45.829 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB6C7), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.Group: 1>, address=0xA001), dst_ep=0, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=8, data=Serialized[b'\x01`\x00\xfe\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-22)
2022-10-17 10:12:45.829 WARNING (MainThread) [zigpy.zdo] [0x0000:zdo] Unknown ZDO cluster 0x0008
2022-10-17 10:12:45.830 DEBUG (MainThread) [zigpy.zcl] [0xB6C7:1:0x0008] Received ZCL frame: b'\x01`\x00\xfe\x00\x00'
2022-10-17 10:12:45.830 DEBUG (MainThread) [zigpy.zcl] [0xB6C7:1:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=96, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-10-17 10:12:45.831 DEBUG (MainThread) [zigpy.zcl] [0xB6C7:1:0x0008] Decoded ZCL frame: LevelControl:move_to_level(level=254, transition_time=0, options_mask=None, options_override=None)
2022-10-17 10:12:45.831 DEBUG (MainThread) [zigpy.zcl] [0xB6C7:1:0x0008] Received command 0x00 (TSN 96): move_to_level(level=254, transition_time=0, options_mask=None, options_override=None)
2022-10-17 10:12:45.831 DEBUG (MainThread) [zigpy.zcl] [0xB6C7:1:0x0008] No explicit handler for cluster command 0x00: move_to_level(level=254, transition_time=0, options_mask=None, options_override=None)
2022-10-17 10:12:45.831 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=54:0f:57:ff:fe:19:ed:24, unique_id=54:0f:57:ff:fe:19:ed:24:1:0x0008, device_id=2d2248ee911555f4c29e83f1281376d3, endpoint_id=1, cluster_id=8,  [... cut ...]
2022-10-17 10:14:24.429 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=AddressMode.NWK address=0xafe2>, ep: 1, profile: 0x0104, cluster_id: 0x0201, data: b'185e0a110029280a130029280a140029d00702001801'

2022-10-17 10:14:32.761 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xB6C7](TS1002): Device seen - marking the device available and resetting counter
2022-10-17 10:14:32.761 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xB6C7](TS1002): Update device availability -  device available: True - new availability: True - changed: False

```

mattiabrunello5, 2022-10-19:
hi, did you have any news ?

Taddeus01, 2022-12-24:
> 
> **Is your feature request related to a problem? Please describe.** hi i wish to integrate the color response on Home Assistant (if i change the color via color wheel on the remote the controller fire it but HA does not. the on/off toggle and brightness level works fine

Hi Mattia,
I paired this remote on HA, using ZHA and SONOFF ZigBee 3.0 ZBDongle-P but, I only viewing two entities sensor.tz3000_xwh1e22x_ts1002_battery and button.tz3000_xwh1e22x_ts1002_identify (I can't understand what's is this...)

How did you get the on/off keys and brightness slider to works?

Thanks in advance.

mattiabrunello5, 2022-12-24:
Hi I use the “combee ii ”dongle  paired the remote with ZHA , with the
lights I use the compatible mi boxer led controller. The entities of the
remote shows are only battery and identity

SciLor, 2023-05-05:
I am also unable to use this remote with a rooted tuya based gateway by silvercrest (Lidl Silvercrest Gateway Zigbee).
I receive no zha_events on button press. The only thing that works is battery and the signal quality / strength. It will also update as soon I remove and replace the batteries in the remote.

SciLor, 2023-11-01:
Not stale

zeljkopopovic, 2023-12-31:
Hi friends,
I paired this remote on HA, using ZHA and SONOFF ZigBee 3.0 ZBDongle but same like others , I only viewing two entities sensor.
I found solution for using  zigbee2mqtt
https://www.zigbee2mqtt.io/devices/FUT089Z.html
But still they mention that they didn't implemented color wheel
Brightness (%)
Color Temperature (mireds)
Color (not implemented yet)
[image]

mycanaletto, 2024-01-11:
With the latest z2m update (1.35.1-1), the extension is no longer required. However, what I had done on the basis of what is suggested [here](https://www.zigbee2mqtt.io/devices/FUT089Z.html) no longer works (although it’s still in the doc).

https://community.home-assistant.io/t/how-to-use-miboxer-rgb-cct-remote-zigbee-3-0-fut089z-with-hass-getting-no-zha-event/567991/10?u=canaletto

SciLor, 2024-07-09:
Not stale

SciLor, 2025-01-05:
Not stale

SciLor, 2025-08-12:
Not stale

SciLor, 2026-02-09:
Not stale