zigpy/zha-device-handlers #1830: [Device Support Request] TS0042 by _TZ3000_5e235jpa

Quelbazar, 2022-10-15:
**Is your feature request related to a problem? Please describe.**
Trying to add a "Moes zigbee Scene switch". Seems to be added as "TS0042 by _TZ3000_5e235jpa" and "Quirk: zhaquirks.tuya.ts0042.TuyaSmartRemote0042TO". However, only Battery, Lqi and Rssi entities appear.

**Describe the solution you'd like**
Have "buttons" entities (short press, long press / button 1, button 2)

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_5e235jpa",
  "model": "TS0042",
  "class": "zhaquirks.tuya.ts0042.TuyaSmartRemote0042TO"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.10.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Zurich",
    "os_name": "Linux",
    "os_version": "5.15.61-v7l+",
    "supervisor": "2022.10.0",
    "host_os": "Raspbian GNU/Linux 11 (bullseye)",
    "docker_version": "20.10.18",
    "chassis": "",
    "run_as_root": true
  },
  "custom_components": {
[... 415 lines cut ...]
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4B04), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=1, data=Serialized[b'\x18`\x01 \x00\x00 \x1e!\x00\x00 \xc8'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-49)
[0x4B04:1:0x0001] Received ZCL frame: b'\x18`\x01 \x00\x00 \x1e!\x00\x00 \xc8'
[0x4B04:1:0x0001] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=96, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x4B04:1:0x0001] Decoded ZCL frame: PowerConfiguration:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0020, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint8_t, value=30)), ReadAttributeRecord(attrid=0x0021, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint8_t, value=200))])
[0x4B04:1:0x0001]: Performing channel specific initialization: ['battery_voltage', 'battery_percentage_remaining']
[0x4B04:1:0x0001]: Reading attributes in chunks: ['battery_size', 'battery_quantity']
[0x4B04:1:0x0001] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=99, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0x4B04:1:0x0001] Sending request: Read_Attributes(attribute_ids=[49, 51])
[0x4b04] Extending timeout for 0x63 request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4B04), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=1, data=Serialized[b'\x18c\x011\x00\x863\x00\x86'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-49)
[0x4B04:1:0x0001] Received ZCL frame: b'\x18c\x011\x00\x863\x00\x86'
[0x4B04:1:0x0001] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=99, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x4B04:1:0x0001] Decoded ZCL frame: PowerConfiguration:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0031, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>), ReadAttributeRecord(attrid=0x0033, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>)])
[0x4B04:1:0x0001]: finished channel initialization
[0x4B04:1:0x0000]: 'async_initialize' stage succeeded
[0x4B04:1:0x0001]: 'async_initialize' stage succeeded
[0x4B04:1:0x0019]: 'async_initialize' stage succeeded
[0x4B04](TS0042): power source: Battery or Unknown
[0x4B04](TS0042): completed initialization
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2022-10-15:
Do you getting short press, long press / button 1, button 2 events ?
Is the device automation for the device working ?

Quelbazar, 2022-10-15:
> Do you getting short press, long press / button 1, button 2 events ?

Yes, I can see that on the "device" page :

```
_TZ3000_5e235jpa TS0042 Remote Button Long Press - Button 2 event was fired
9:09:54 PM - Now
_TZ3000_5e235jpa TS0042 Remote Button Long Press - Button 1 event was fired
9:09:49 PM - 5 seconds ago
_TZ3000_5e235jpa TS0042 Remote Button Short Press - Button 2 event was fired
9:09:43 PM - 11 seconds ago
_TZ3000_5e235jpa TS0042 Remote Button Short Press - Button 1 event was fired
9:09:38 PM - 16 seconds ago
```

> Is the device automation for the device working ?

Yes : Automation is working with both buttons / with short click, double click, long click.
So I can use this device properly...

.... Now I'm afraid I'm wrong... I expected to see "buttons" entities for the type of device. 
But am I wrong ?...
Sorry I'm newbee....

javicalle, 2022-10-15:
Remote devices are not intended to have entities.
The way to integrate them is through the events generated by the buttons. Events like `short`/`long`/`double_press` for both buttons are available as triggers when you create a new 'Device' type automation and selecting your remote as device.

Quelbazar, 2022-10-15:
OK. Sorry for that. 
This is a misunderstanding on my part. Sorry for the waste of your time and THANK YOU for your explanations.

closed 2022-10-15 by Quelbazar

javicalle, 2022-10-15:
Remotes are stateless while switches are stateful.

The remote send signal but don't have control of "whatever it does" status (for example a wired switch), thats the reason to be stateless.
A wired switch does know the state of the device and can show its status despite are turned of by a remote.