zigpy/zha-device-handlers #1834: [Device Support Request] Tuya Model CX-7026 ID: TS0601 | _TZE200_znbl8dj5 recognized but all 3 sensors readings are 0

bbK1ngSrb, 2022-10-16:
**Is your feature request related to a problem? Please describe.**

I have this device
https://github.com/blakadder/zigbee/blob/3fd28d4a941cd4709cd0760edcbf4b59b93cde05/_zigbee/Tuya_CX-7026.md

It is recognized by ZHA but all 3 sensors reading remind on 0 (zero) value.

**Describe the solution you'd like**
Sensor should report theirs real value to HA via ZHA integration

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0402",
        "0x0405"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_znbl8dj5",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.10.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Belgrade",
    "os_name": "Linux",
    "os_version": "5.10.103-v7l+",
    "supervisor": "2022.10.0",
    "host_os": "Raspbian GNU/Linux 10 (buster)",
    "docker_version": "20.10.19",
    "chassis": "",
    "run_as_root": true
  },
  "custom_components": {
    "worlds_air_quality_index": {
      "version": "0.3.3",
      "requirements": []
[... 432 lines cut ...]
</details>

<details>
<summary>Additional logs</summary>

```
Device info
ConBee II
by dresden elektronik
Connected via [dresden elektronik ConBee II](http://192.168.10.10:8123/config/devices/device/0832ad00afc52c1d735e9cdc35008577)
Zigbee info
IEEE: 00:21:2e:ff:ff:04:ec:7f
Nwk: 0x0000
Device Type: Coordinator
LQI: 69
RSSI: 27
Last Seen: 2022-10-16T21:59:55
Power Source: Mains
```

</details>

**Additional context**

Reading on device screen are normal, ie. device is in working condition.

javicalle, 2022-10-21:
Can you try with that quirk?:
```python
"""Tuya temp and humidity sensor with e-ink screen."""

from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.measurement import RelativeHumidity, TemperatureMeasurement

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaPowerConfigurationCluster2AAA
from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaDPType, TuyaMCUCluster

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

class TuyaRelativeHumidity(RelativeHumidity, TuyaLocalCluster):
    """Tuya local RelativeHumidity cluster."""

class TemperatureHumidityManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with Temperature and Humidity data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaTemperatureMeasurement.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 10,  # decidegree to centidegree
        ),
        2: DPToAttributeMapping(
            TuyaRelativeHumidity.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 100,  # decipercent to centipercent
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster2AAA.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 2,  # double reported percentage
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
    }

class TuyaTempHumiditySensor(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=81
        # device_version=1
        # input_clusters=[4, 5, 61184, 0]
        # output_clusters=[25, 10]>
        MODELS_INFO: [
            ("_TZE200_znbl8dj5", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA,
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    TemperatureHumidityManufCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
        },
    }
```

Restart HA and pair you device again. Check that the device is loading the quirk and that the device signature has changed.

Double check if reports are fine (values) and that there is no picks or 0 values in reports.

bbK1ngSrb, 2022-10-24:
HI. Thanks for reply.

Could you explain me or point me to tutorial how to force this quirk in device pairing process?

So far I tried following.

Created folder custom_zha_quirks in HA config folder.
Created custom.py file and copy/paste your quirk in that new file.

Added following in configuration.yaml and restarted HA.

```

zha:
  enable_quirks: true
  custom_quirks_path: /config/custom_zha_quirks

```

Started pairing process again, but device pickup original config.

Thanks in advance.

javicalle, 2022-10-24:
There is a guide about enabling custom quirks, but your steps are fine to me:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Check if there is any error in your logs, during the HA start up or in the pairing process.

javicalle, 2022-11-01:
Related to: #1670

bbK1ngSrb, 2022-11-07:
To confirm that device is fully recognized and in working condition.

[image]

PR #1938 (2022-11-19): Add `_TZE200_znbl8dj5` sensor
Fixes: #1834
Related to: https://github.com/blakadder/zigbee/issues/766
commit 4f3130da (2022-11-19): Add `_TZE200_znbl8dj5` sensor (#1938)

Fixes: #1834
Fixes: #766
closed 2022-11-07 by bbK1ngSrb