zigpy/zha-device-handlers #1835: [Device Support Request]

TannerBrunscheon, 2022-10-17:
**Is your feature request related to a problem? Please describe.**
[These cheap](https://www.amazon.com/dp/B0BF9W3WMK?psc=1&ref=ppx_yo2ov_dt_b_product_details) Third Reality Temperature and Humidity sensors pair nicely with ZHA except they expose no entities to Home Assistant

**Describe the solution you'd like**
I should be able to get relative humidity and temperature from these guys.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4659, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0402",
        "0x0405"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Third Reality, Inc",
  "model": "3RTHS24BZ",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Chicago",
    "os_name": "Linux",
    "os_version": "5.15.72",
    "supervisor": "2022.10.0",
    "host_os": "Home Assistant OS 9.2",
    "docker_version": "20.10.17",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "holidays": {
      "version": "1.8.0",
      "requirements": [
        "python-dateutil>=2.8.2",
        "holidays>=0.14.2"
      ]
    },
    "garbage_collection": {
      "version": "4.8.3",
[... 310 lines cut ...]
            "attributes": {
              "0x0000": {
                "attribute_name": "measured_value",
                "value": 5100
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

**Additional context**

javicalle, 2022-10-19:
That seems like a very straight zigbee device.
Even in Z2M don't seems to have too much to quirk (just battery_voltage?):
* https://github.com/Koenkk/zigbee-herdsman-converters/blob/328e2a362e1f67cfc13dc91e1d939e2c5edd75a1/devices/third_reality.js#L139-L147

It can be a pairing issue?
Battery powered devices must be kept awake during the pairing procedure.
To do this, while the pairing lasts, you have to briefly press the button to keep the device awake.

dmulcahey, 2022-10-20:
The data is there to expose entities. Are you sure there are no entities? May I have a screenshot of the entire device page?

tube0013, 2022-10-20:
Pairing log: https://dpaste.org/Ovq4E
 

[image]

MattWestb, 2022-10-20:
Try unbinding all cluster (from device card > manage zigbee device) both to device and groups and bound the temperature cluster to the coordinator and see if its start reporting.

Bojack84, 2022-10-21:
I just got the same device and I am getting Temp and the Battery % in but not the Humidity. 

[image]

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "5.15.72",
    "supervisor": "2022.10.0",
    "host_os": "Home Assistant OS 9.2",
    "docker_version": "20.10.17",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "bhyve": {
      "version": "3.0.0",
      "requirements": []
    },
    "localtuya": {
      "version": "4.1.1",
      "requirements": []
    },
    "nfl": {
      "version": "0.1",
      "requirements": [
        "arrow"
      ]
    },
    "anniversaries": {
      "version": "4.5.0",
      "requirements": [
        "python-dateutil>=2.8.1",
        "integrationhelper>=0.2.2",
        "voluptuous>=0.12.1"
      ]
    },
    "deebot": {
      "version": "1.4.1",
      "requirements": [
        "deebot-client==1.5.1",
        "numpy>=1.21.2"
      ]
    },
    "fullykiosk": {
      "version": "1.1.0",
      "requirements": [
        "python-fullykiosk==0.0.11"
      ]
    },
    "google_home": {
      "version": "1.9.15",
      "requirements": [
        "glocaltokens==0.6.8"
      ]
    },
    "alexa_media": {
      "version": "4.1.2",
      "requirements": [
        "alexapy==1.26.3",
        "packaging>=20.3",
[... 319 lines cut ...]
            "endpoint_attribute": "humidity",
            "attributes": {
              "0x0000": {
                "attribute_name": "measured_value",
                "value": 3600
              }
            },
            "unsupported_attributes": {
              "0x0000": {
                "attribute_name": "measured_value"
              }
            }
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

TannerBrunscheon, 2022-10-23:
These are working for me now. If you guys want to leave this open for discussion thats fine otherwise close it.

TannerBrunscheon, 2022-10-26:
Small update, I have since purchased 2 more and the new 2 are showing the same thing as before, no home assistant entities.  The previous one worked after a day but its been 24 hours and its not working. 

> Try unbinding all cluster (from device card > manage zigbee device) both to device and groups and bound the temperature cluster to the coordinator and see if its start reporting.

Could you expand on this?

TannerBrunscheon, 2022-10-26:
I did find this in the logs. Is it concerning that the measure_value attributes also show up in the unsupported_attributes 

```json
"cluster_details": {
      "1": {
        "device_type": {
          "name": "TEMPERATURE_SENSOR",
          "id": 770
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "Third Reality, Inc"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "3RTHS24BZ"
              },
              "0x4000": {
                "attribute_name": "sw_build_id",
                "value": "1.00.20"
              }
            },
            "unsupported_attributes": {}
          },
          "0x0001": {
            "endpoint_attribute": "power",
            "attributes": {
              "0x0020": {
                "attribute_name": "battery_voltage",
                "value": 31
              },
              "0x0021": {
                "attribute_name": "battery_percentage_remaining",
                "value": 200
              }
            },
            "unsupported_attributes": {
              "0x0020": {
                "attribute_name": "battery_voltage"
              },
              "0x0021": {
                "attribute_name": "battery_percentage_remaining"
              },
              "0x0033": {
                "attribute_name": "battery_quantity"
              },
              "0x0031": {
                "attribute_name": "battery_size"
              }
            }
          },
          "0x0402": {
            "endpoint_attribute": "temperature",
            "attributes": {
              "0x0000": {
                "attribute_name": "measured_value",
                "value": 2680
              }
            },
            "unsupported_attributes": {
              "0x0000": {
                "attribute_name": "measured_value"
              }
            }
          },
          "0x0405": {
            "endpoint_attribute": "humidity",
            "attributes": {
              "0x0000": {
                "attribute_name": "measured_value",
                "value": 3900
              }
            },
            "unsupported_attributes": {
              "0x0000": {
                "attribute_name": "measured_value"
              }
            }
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
```

dmulcahey, 2022-10-30:
This device needs a firmware update. We have let them know

3reality-support, 2022-11-02:
A related problem has been found and is working hard to solve it, we will fix it as soon as possible!

dmulcahey, 2022-11-07:
Not sure you are correct. We set reporting parameters for the amount of change needed to trigger a report. This is standard ZCL and has not changed in ZHA.

puddly, 2022-11-07:
@xudo-github The `measured_value` attribute of both the Temperature and Humidity clusters being readable and reportable is aligned with the ZCL R8 specification:

[image]

> This behavior is only introduced in latest ZHA, the old version actually donot need this.

ZHA has never polled Zigbee 3.0 temperature sensors. Can you clarify which version you're referring to?

xudo-github, 2022-11-07:
You are right, it is a bug in 3R side. Somehow the previous version of ZHA tolerant it, but the latest version is not happy with this. The fixed firmware worked well with multi-round of testing.

xudo-github, 2022-11-07:
[image]

This is the missing report capture

xudo-github, 2022-11-07:
And yes the fix is actually made the values reportable which aligns with the spec

dmulcahey, 2022-11-07:
Yep so straight up firmware issue. Nothing has changed with ZHA with regards to cluster binding or configuration and the sensor classes for ZHA haven’t been touched in forever. The history is all there. Feel free to look.

If you’re positive that this worked at any point I’m going to guess it was a fluke as you have proven that the firmware was not specification compliant 🤷🏻‍♂️

Glad you have it sorted out so that it will work now for your customers.

dmulcahey, 2022-11-07:
@xudo-github please also ensure that the binding table is large enough to accommodate binding the 3 clusters that support attribute reporting.

TheJulianJES, 2023-01-04:
OTA updates for ThirdReality devices are now natively available in Home Assistant (https://github.com/zigpy/zigpy/pull/1096).
They can be activated by adding this to the ``configuration.yaml``:

```yaml
zha:
  zigpy_config:
    ota:
      thirdreality_provider: true
```
An "update notify" command also needs to be send to the device.

Are all bugs fixed with the latest firmware available now?

xudo-github, 2023-01-16:
> Assistant (zigpy/zigpy#1096 <https://github.com/zigpy/zigpy/pull/1096>).

Yes, we have all latest firmware updates uploaded.

closed 2023-07-22 by github-actions[bot]