zigpy/zha-device-handlers #1838: [Device Support Request] Tuya Smoke Detector TS0601 (_TZE200_uebojraa)

marcelkous, 2022-10-18:
I am looking to help add support for this smoke detector into ZHA.

I found that it is supported by Zigbee2MQTT (https://zigbee.blakadder.com/Smart9_S9ZB-SA02.html, https://www.zigbee2mqtt.io/devices/TS0601_smoke.html)

Hopefully, it can be added quite easily to ZHA also. 

<details><summary>Device signature</summary>
<p>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_uebojraa",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</p>
</details>

<details><summary>Device diagnostics</summary>
<p>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "5.15.72",
    "supervisor": "2022.10.0",
    "host_os": "Home Assistant OS 9.2",
    "docker_version": "20.10.17",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.2",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
[... 427 lines cut ...]
[0xAAB2:1:0xef00] Unknown cluster command 2 b'\x00>\x01\x04\x00\x01\x01'
[0xAAB2:1:0xef00] Received command 0x02 (TSN 116): b'\x00>\x01\x04\x00\x01\x01'
[0xAAB2:1:0xef00] No explicit handler for cluster command 0x02: b'\x00>\x01\x04\x00\x01\x01'
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xAAB2), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=51, profile_id=260, cluster_id=61184, data=Serialized[b'\tt\x02\x00>\x01\x04\x00\x01\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-27)
[0xAAB2:1:0xef00] Received ZCL frame: b'\tt\x02\x00>\x01\x04\x00\x01\x01'
[0xAAB2:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=116, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0xAAB2:1:0xef00] Unknown cluster command 2 b'\x00>\x01\x04\x00\x01\x01'
[0xAAB2:1:0xef00] Received command 0x02 (TSN 116): b'\x00>\x01\x04\x00\x01\x01'
[0xAAB2:1:0xef00] No explicit handler for cluster command 0x02: b'\x00>\x01\x04\x00\x01\x01'
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xAAB2), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=52, profile_id=260, cluster_id=61184, data=Serialized[b'\tt\x02\x00>\x01\x04\x00\x01\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-27)
[0xAAB2:1:0xef00] Received ZCL frame: b'\tt\x02\x00>\x01\x04\x00\x01\x01'
[0xAAB2:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=116, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0xAAB2:1:0xef00] Unknown cluster command 2 b'\x00>\x01\x04\x00\x01\x01'
[0xAAB2:1:0xef00] Received command 0x02 (TSN 116): b'\x00>\x01\x04\x00\x01\x01'
[0xAAB2:1:0xef00] No explicit handler for cluster command 0x02: b'\x00>\x01\x04\x00\x01\x01'
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xAAB2), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=53, profile_id=260, cluster_id=61184, data=Serialized[b'\tu\x02\x00?\x0e\x04\x00\x01\x02'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-27)
[0xAAB2:1:0xef00] Received ZCL frame: b'\tu\x02\x00?\x0e\x04\x00\x01\x02'
[0xAAB2:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=117, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0xAAB2:1:0xef00] Unknown cluster command 2 b'\x00?\x0e\x04\x00\x01\x02'
[0xAAB2:1:0xef00] Received command 0x02 (TSN 117): b'\x00?\x0e\x04\x00\x01\x02'
[0xAAB2:1:0xef00] No explicit handler for cluster command 0x02: b'\x00?\x0e\x04\x00\x01\x02'
```

</p>
</details>

javicalle, 2022-10-19:
Probably it can match with the existing quirks.

There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Copy in your local quirks folder the current quirk:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_smoke.py

Edit the file, adding your device like this:
```python
        MODELS_INFO: [
            ("_TZE200_aycxwiau", "TS0601"),
            ("_TZE200_ntcy3xu1", "TS0601"),
            ("_TZE200_uebojraa", "TS0601")
            ("_TZE200_vzekyi4c", "TS0601"),
        ],
```

Restart HA and pair you device again. Check that the device is loading the quirk and that the device signatura has changed.

marcelkous, 2022-10-19:
@javicalle 
Thank you very much. I'm gonna try this.

marcelkous, 2022-10-19:
It is in a 'better' state than before but not completely.

This is what I did to add the local quirk.

I added the **custom_zha_quirks** folder in the **config** folder

> /config/custom_zha_quirks

In the **configuration.yaml** file i've added:

```
zha: 
  custom_quirks_path: /config/custom_zha_quirks/
```

Then I saved the file from https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_smoke.py into that folder and added the models info like @javicalle described above. 

Then a restart of Home Assistant.

After the restart, I removed the 'faulty' device and re-added it again.
I saw the following happening in the "Add Device" section.
[image]
and then:
[image]
After that nothing happened. Normally you see a green box that tells you that the device is added and ready to use.

I can see ZHA used the custom quirk
[image]

Also, a sensor appeared:
[image]
And when I press the test button on the device the logbook shows the activity.
[image]

So far, so good :-) 
The only thing that's missing is the **Diagnostic** section what should show the **Battery** state and maybe the **Identifybutton** property.

Like in this example of another brand smoke detector.
[image]

The Device diagnostics shows this.
<details><summary>Device diagnostics</summary>
<p>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "5.15.72",
    "supervisor": "2022.10.0",
    "host_os": "Home Assistant OS 9.2",
    "docker_version": "20.10.17",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
[... 242 lines cut ...]
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</p>
</details>

marcelkous, 2022-10-19:
I tried paring several times already. I didn't have this issue before, it always ended up green.

But, I tried it again and then pressed the test button several times (auw, sharp beep sound every time) and now it ended 'green' indeed. 
[image]
There are now three 'properties', the first is the **Iaszone** (Smoke sensor), and the two others say **'null'**.
[image] [image]
One of the 'null' is probably the battery and the other is maybe link quality?

dmulcahey, 2022-10-19:
The triangles are how disabled entities show by default. Those are LQI and RSSI

marcelkous, 2022-10-19:
I see this in the Core logs around the time I added the device again. 

```
Logger: homeassistant.components.zha.core.channels.base
Source: components/zha/core/channels/base.py:486
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 12:38:39 PM (12 occurrences)
Last logged: 5:00:38 PM

[0x7B67:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0x7B67:1:0x0702]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0xAAB2:1:0x0500]: 'async_configure' stage failed:
[0xAAB2:1:0x0500]: async_initialize: all attempts have failed: [TimeoutError(), TimeoutError(), TimeoutError(), TimeoutError()]
[0xAAB2:1:0x0500]: 'async_initialize' stage failed: Failed to convert status=Command(status=0, tsn=103, command_id=1025, function=0, data=[1, 1]) from type <class 'zhaquirks.tuya.TuyaManufCluster.Command'> to <enum 'Status'>
```

javicalle, 2022-10-19:
The quirk don't have a 'battery' cluster.
Are you sure that the device reports its battery percentage?
Blakadder mention just `battery_low` but Z2M tells abou battery percentage. I think that probably the device will report just the low battery status.

marcelkous, 2022-10-20:
Yes, I saw too that it hadn't a PowerConfiguration setting in the quirk.

the Z2M page says that it has a numeric value (0-100) with the unit being %
https://www.zigbee2mqtt.io/devices/TS0601_smoke.html#battery-numeric
[image]

I'm looking for TS0205 (Smoke detector) quirk or the TS0212 (CO detector) quirk to see how the PowerConfiguration is handled in there (but I can't find those files) because they are working correct in ZHA.

marcelkous, 2022-10-23:
Hi,

I've ordered a tuya hub to test and see what all will be visible in there.
Will update here what i find out.

marcelkous, 2022-10-26:
Hi all,

I've received the zigbee hub and added the smoke sensor to it.
At the device level it shows this

javicalle, 2022-10-27:
If it doesn't show any numerical value for battery, my guess is that the device just has a low battery alert as indicated by Blakadder.
I bet that Z2M creates a fake cluster where if there is no low battery report the value is 100% and when the report arrives it changes the value to 0%.

marcelkous, 2022-10-27:
I have a CO sensor that shows op like this in HA.

[image]

And like this in the Tuya app.

[image]

I'm not convinced that it can't show the battery level. 🙂

(Only time will tell, when the battery drains)

javicalle, 2022-10-27:
Well, if it is the case, the device must report the values in any way.
Check the HA logs for any `unhandled attribute:` message from your device.
Without a `PowerConfiguration` cluster, the device only could report the value trought the manufacturer cluster. As, currently, only `TUYA_SMOKE_DETECTED_ATTR (0x0401)` is checked, any other report will generate a warning log:

https://github.com/zigpy/zha-device-handlers/blob/e8a1570203012ba12777fecd18df82ecd1ff4883/zhaquirks/tuya/ts0601_smoke.py#L35-L51

marcelkous, 2022-10-27:
I've enabled again all debug logging but I can't find any entry that says something like **unhandled attribute** or any other warning or error around this device. 
🤷‍♂️

marcelkous, 2022-12-12:
I've tested the Smoke Sensor with a desktop power supply and turned the power down to 2.5V with steps of 0.1V from 3.0V. 
It shows only "Low" as shown in the image below. So @javicalle, you are probably right it only works with High/Low. But at least something should be possible. 
[image]

One of my colleagues uses HA with Z2M. It was recognized but not functional, so with ZHA and the quirk I'm further getting it to work than with Z2M. 
He also didn't see any 'warnings in the logs' ...

javicalle, 2022-12-12:
Doesn't report any `unhandled attribute:` when the power drops to 2.5V?
Device would report the 'low battery' in some way. It can be a new `attrid != TUYA_SMOKE_DETECTED_ATTR` (which would generate a warning `... unhandled attribute: attrid` or it could be that report the `TUYA_SMOKE_DETECTED_ATTR` with a diferent `value`.

From Z2M I would say that must be from attrid `0x040F` or `0x040E`.

morenod, 2023-02-15:
Please add also this model of fire alarm:

            ("_TZE200_m9skfctm", "TS0601"),

[image]

[image]

javicalle, 2023-02-15:
> Please add also this model of fire alarm:
> 
> ```
>         ("_TZE200_m9skfctm", "TS0601"),
> ```

Which is the working quirk for that device?

morenod, 2023-02-15:
> > Please add also this model of fire alarm:
> > ```
> >         ("_TZE200_m9skfctm", "TS0601"),
> > ```
> 
> Which is the working quirk for that device?

https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_smoke.py

git-peterk, 2023-09-10:
I just ordered 2 of [these sensors](https://www.aliexpress.com/item/1005005863519099.html), which I think are using the same data structrure for communication as used by the devices mentioned in this thread.

Looks like mine came with yet another manufacturer info: "TS0601" from "_TZE204_ntcy3xu1"
The device was working fine with the released ts0601_smoke.py quirk after adding my own device's model to it, but I did not like it was not reporting battery status, so I created this custom quirk which adds both the battery status and the tampered state:

```python
"""
DY-YG400A Smoke Sensor, TS0601 from _TZE204_ntcy3xu1
https://www.aliexpress.com/item/1005005863519099.html
https://www.aliexpress.com/item/1005005854203557.html
"""

import logging

import zigpy.profiles.zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Ota,
    Scenes,
    Time,
    BinaryInput,
    BatterySize,
)
from zigpy.zcl.clusters.security import IasZone
from zhaquirks import Bus
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZONE_TYPE,
)
from zhaquirks.tuya import (
    TuyaLocalCluster,
    TuyaManufCluster,
    TuyaManufClusterAttributes,
    TuyaPowerConfigurationCluster,
)

_LOGGER = logging.getLogger(__name__)

TUYA_SMOKE_DETECTED_ATTR = 0x0401  # [0]/[1] [Detected]/[Clear]!
TUYA_SMOKE_TAMPERED_ATTR = 0x0104  # [0]/[1] [Clear]/[Tampered]!
TUYA_SMOKE_BATTERY_ATTR = 0x040e  # [0]/[1]/[2] [Low]/[Med]]/[Full]!

class TuyaSmokeDetectorCluster(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the TS0601 smoke detector."""

    attributes = {
        TUYA_SMOKE_DETECTED_ATTR: ("smoke_detected", t.uint8_t, True),
        TUYA_SMOKE_TAMPERED_ATTR: ("tampered_device", t.uint8_t, True),
        TUYA_SMOKE_BATTERY_ATTR: ("battery_status", t.uint8_t, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == TUYA_SMOKE_DETECTED_ATTR:
            status = IasZone.ZoneStatus.Alarm_1 if value == 0 else 0
            self.endpoint.device.ias_bus.listener_event("update_attribute", "zone_status", status)
        elif attrid == TUYA_SMOKE_TAMPERED_ATTR:
            self.endpoint.device.tamper_detection_bus.listener_event(
                "update_attribute", "present_value", bool(value)
            )
        elif attrid == TUYA_SMOKE_BATTERY_ATTR:
            batt = 5 if value == 0 else 40 if value == 1 else 100
[... 98 lines cut ...]
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }

```

The only "downside" was that due to how the BinaryInput (that I'm using to represent the Tampered input) is implemented in Home Assistant, it will be added as an entity named "Binary input".
[image]

So I had to manually fix this by clicking "Settings" on the "Binary input" entity:
[image]

and changing its "Name", "Show As" and "Entity ID" as follows:
[image]

Now all enetities show up correctly in Home Assistant:
[image]

__Note:__
- I removed the existing models from the `MODELS_INFO` list of the `signature` dictionary, so you should edit it as needed depending on your own device's model/manufacturer.
- After adding the smoke detector to Home Assistant, it is recommended to remove and reinstall the base plate of the detector to trigger an update, as the Battery and Tamper status will be incorrect until receiving an update message from the device.

git-peterk, 2023-10-03:
@erkr, I did not create a PR with this change, as:
- I wasn't sure if the Tampered state is available on all models, and
- I think it would be strage if the sensor shows up with the default `Binary input` entity in Home Assistant. Without the description it would not be straightforward that it is representing the `Tampered` input. Unfortunately this information cannot be relayed to Home Assistant (or zha) at the moment. Maybe zha could be updated to create the "Binary input" entity with the name from the `description` attribute of the `BinaryInput` cluster if it is not `None`?

erkr, 2023-10-03:
I'm trying to adopt your code for another version (I received two models in the same batch :-(  ). The other version is ("_TZ3210_up3pngle", "TS0205"). That one was default reporting an opening binary sensor (that naming doesn't help either what it is.), That tampering sensor doesn't work, so I adopt your solution to test and I will report back if tampering works.

erkr, 2023-10-03:
Update, that device has a different signature, and tampering still doesn't work on that one:
```
    signature = {
        MODELS_INFO: [
            ("_TZ3210_up3pngle", "TS0205"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zigpy.profiles.zha.PROFILE_ID,
                DEVICE_TYPE: zigpy.profiles.zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Time.cluster_id,
                    OnOff.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id, 
                ],
            },
        },
    }
```

liouma, 2023-11-08:
Hello,
I bought 5 of those smoke sensors TZE200_m9skfctm
My version does not have the tamper switch (triggered when the sensor is detached from the ceiling).
I tried both quirks:

- The one working for @morenod:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_smoke.py
- And the one posted by @git-peterk 

In all cases the smoke signal is working but I can not get any battery information.

Does anyone have the battery signal working ?
Any help will be really welcome.
Thanks !!

harrlih, 2023-12-13:
@git-peterk - great work! many thanks - finally got the battery status!! 👍

liouma, 2023-12-13:
> @git-peterk - great work! many thanks - finally got the battery status!! 👍

Hello, can you please show me the exact model of your smoke sensor ? 
Unfortunately it doesn't work with mine

harrlih, 2023-12-13:
> Hello, can you please show me the exact model of your smoke sensor ? Unfortunately it doesn't work with mine

Hi @liouma, I have a TZE204_ntcy3xu1 device - unfortunateley I can't be of any further help - too inexperienced - too inexperienced with the tuya quirks...

n0tOdd, 2023-12-20:
> TZE204_ntcy3xu1

Worked perfectly for " TZE200_ntcy3xu1" to, i just needed to add the model info

marcelkous, 2024-01-19:
in the end, I used **Quirk: ts0601_smoke_custom.TuyaSmokeDetector0601** and that works perfectly.

[image]

hellebauer, 2024-02-28:
_TZE200_aycxwiau   (Woox Smoke Detector R7049)

No luck with above model.  Pairing works, but nothing works beyond that.

closed 2024-09-05 by github-actions[bot]