zigpy/zha-device-handlers #1839: [Device Support Request] MOES TRV TV02 (TS0601 - _TZE200_kds0pmmv)

matlar83, 2022-10-18:
**Is your feature request related to a problem? Please describe.**
I have the MOES TV02 TRV. I added to my Zigbee network with ZHA but no entities are created in HA. Please add support for this tuya smart valve.

**Describe the solution you'd like**
Fully support the valve in ZHA

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_kds0pmmv",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Rome",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.10.0",
    "host_os": "Home Assistant OS 9.2",
    "docker_version": "20.10.17",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "watchman": {
      "version": "0.5.1",
      "requirements": [
        "prettytable==3.0.0"
      ]
    },
    "hpprinter": {
      "version": "1.0.8",
[... 257 lines cut ...]
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-10-19:
I would try to add it to the `ZonnsmartTV01_ZG` quirk.

There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Copy in your local quirks folder the current quirk:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_trv.py

Edit the file, adding your device like this:
```python
        MODELS_INFO: [
            ("_TZE200_7yoranx2", "TS0601"),  # MOES TRV TV01 ZTRV-ZX-TV01-MS
            ("_TZE200_e9ba97vf", "TS0601"),
            ("_TZE200_hue3yfsn", "TS0601"),  # TV02-ZG
            ("_TZE200_husqqvux", "TS0601"),
            ("_TZE200_kds0pmmv", "TS0601"),
            ("_TZE200_kly8gjlz", "TS0601"),
            ("_TZE200_lnbfnyxd", "TS0601"),
            ("_TZE200_mudxchsu", "TS0601"),
        ],
```

Restart HA, remove and pair you device again. Check that the device is loading the quirk and that the device signatura has changed.

matlar83, 2022-10-19:
Done.
I confirm it loads the quirk:
[image]

This is the device signature:
`
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000d",
        "0x000f",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0007",
      "in_clusters": [
        "0x0006",
        "0x000d"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0007",
      "in_clusters": [
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZE200_kds0pmmv",
  "model": "TS0601",
  "class": "ts0601_trv.ZonnsmartTV01_ZG"
}

These are the exposed entities:
[image]

I don't know shta the 3 switches are.

And I'd like to have the valve position too, ie the percentage of opening valve.

Thank you
`

javicalle, 2022-10-19:
I'm not familiar with the TRV implementations. Maybe someone else could help here.

The 2 firsts controls (level bar) I would say that are the "Temperature Offset" and "Window Openend Temperature".
The switches must be "Boost function", "Child Lock" and "Online Mode" (whatever they are).

The valve percentage don't seems to have any entity in HA, but i can't be of help in that regard.

Do you mind to create the PR to add your device to the mainstream?

Just edit from the web the file content with the detected changes and follow the instructions:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_trv.py

This way other users will be able to use that device in ZHA thanks to you!

javicalle, 2022-10-19:
Thanks!

Can you add in the fisrt comment of your PR the reference to this issue? Something like:
```
Fixes: #1839
```

PR #1843 (2022-10-21): Add support to Moes TRV TV02
Added the model in MODELS_INFO signature to apply the quirk also to Moes TV02 TRV

Fixes: #1839
closed 2022-10-21 by dmulcahey