zigpy/zha-device-handlers #1840: [Device Support Request] Lonsonho 2CH Zigbee Dimmer Module (_TZ3210_wdexaypg)

gregoryduckworth, 2022-10-18:
**Is your feature request related to a problem? Please describe.**
Support new device

**Describe the solution you'd like**
Support for this device including the ability to change how it toggles on/off at the switch

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_wdexaypg",
  "model": "TS110E",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.4",
[... 444 lines cut ...]
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=257, device_version=1, input_clusters=[4, 5, 3, 6, 8, 61184, 0], output_clusters=[25, 10])
Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=257, device_version=1, input_clusters=[4, 5, 3, 6, 8, 61184], output_clusters=[])
Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33])
```

</details>

javicalle, 2022-10-20:
It will be nice if you can include a link to the device manufacturer or to the site vendor to be able to identify the device.

A little more info about what is working and what no, or what do you expect from the device is necessary to try to 'quirk' it.

gregoryduckworth, 2022-10-20:
Hi @javicalle,

I've updated the main issue to include a link to the product itself. 

As for the device itself, it works through HA without any problems with typically dimming, on/off functionality, but when using it at the wall switch (which is a rocker), I cannot turn it on and end up resetting the device. I'm hoping that there is the ability to update this, similar to https://www.zigbee2mqtt.io/devices/TS110E_2gang.html#lonsonho-ts110e_2gang.

javicalle, 2022-10-20:
Ok, the it seems that you need just to configure the `external_switch_type` for your device.

2 steps: configure local quirk and configure your device.

```python
"""Tuya Dimmer TS110E."""
from typing import Any, Optional, Union

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaDimmerSwitch,
    TuyaZBExternalSwitchTypeCluster,
)

TUYA_LEVEL_ATTRIBUTE = 0xF000
TUYA_BULB_TYPE_ATTRIBUTE = 0xFC02
TUYA_MIN_LEVEL_ATTRIBUTE = 0xFC03
TUYA_MAX_LEVEL_ATTRIBUTE = 0xFC04
TUYA_CUSTOM_LEVEL_COMMAND = 0x00F0

class TuyaLevelPayload(t.Struct):
    """Tuya Level payload."""

    level: t.uint16_t
    transtime: t.uint16_t

class TuyaBulbType(t.enum8):
    """Tuya bulb type."""

    LED = 0x00
    INCANDESCENT = 0x01
    HALOGEN = 0x02

class F000LevelControlCluster(NoManufacturerCluster, LevelControl):
    """LevelControlCluster that reports to attrid 0xF000."""

    server_commands = LevelControl.server_commands.copy()
    server_commands[TUYA_CUSTOM_LEVEL_COMMAND] = foundation.ZCLCommandDef(
        "moveToLevelTuya",
        (TuyaLevelPayload,),
        is_manufacturer_specific=False,
    )

    attributes = LevelControl.attributes.copy()
    attributes.update(
        {
            # 0xF000
[... 197 lines cut ...]
                    F000LevelControlCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

```

There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Create a new `ts110e.py` file in your local quirks folder.
Put the proposed code in the new file.
Restart HA and pair you device again. Check that the device is loading the quirk and that the device signature has changed.

Once the device has loaded the quirk, you could change the `external_switch_type` from cluster `TuyaZBExternalSwitchTypeCluster`.
First get the current value and then change it to a new one. With a rocker a would say that could be `0`, but you can try any value `0`/`1`/`2`.

gregoryduckworth, 2022-10-24:
Thanks for your help with this @javicalle, I've loaded up a custom quirk and according to the diagnostics it is loading correctly, however I don't seem to have the `TuyaZBExternalSwitchTypeCluster` option.

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_wdexaypg",
  "model": "TS110E",
  "class": "zigpy.device.Device"
}
```

gregoryduckworth, 2022-11-07:
On further inspection as I've had a few goes now, it looks like the setting isn't coming through, the device diagnostics come back as:
```
"0xef00": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          },
```

javicalle, 2022-11-07:
Hi, your device is NOT loading the quirk.
If done, the device signature would be changed to something like `ts110e. DimmerSwitchWithNeutral2Gang`, but don't seems to be the case.

Check the HA startup log for any error related to the quirk file.

gregoryduckworth, 2022-11-07:
Thanks for your help with this, from the looks of it I've set something up wrong but can't see the wood for the trees. 

I've got my zha config here:
[image]

I've created the file, like so:
[image]

then copied in the code you provided above but it still doesn't seem to be picking up the quirk:
[image]

gregoryduckworth, 2022-11-07:
Ahh, I've found more information:
```
2022-11-07 20:35:29.882 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts110e.DimmerSwitchWithNeutral2Gang'>
2022-11-07 20:35:29.882 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint
```

Will investigate this a little more.

javicalle, 2022-11-07:
My bad, I have mistaken the cluster 🤦🏻‍♂️ 
Give me a minute.

gregoryduckworth, 2022-11-07:
I've figured that bit out now, with your help 👍 

Just need to hopefully get the right settings...

```"""Tuya Dimmer TS110E."""
from typing import Any, Optional, Union

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaZBExternalSwitchTypeCluster,
    TuyaDimmerSwitch,
)
from zhaquirks.tuya.mcu import (
    TuyaLevelControlManufCluster,
)

TUYA_LEVEL_ATTRIBUTE = 0xF000
TUYA_BULB_TYPE_ATTRIBUTE = 0xFC02
TUYA_MIN_LEVEL_ATTRIBUTE = 0xFC03
TUYA_MAX_LEVEL_ATTRIBUTE = 0xFC04
TUYA_CUSTOM_LEVEL_COMMAND = 0x00F0

class TuyaLevelPayload(t.Struct):
    """Tuya Level payload."""

    level: t.uint16_t
    transtime: t.uint16_t

class TuyaBulbType(t.enum8):
    """Tuya bulb type."""

    LED = 0x00
    INCANDESCENT = 0x01
    HALOGEN = 0x02

class F000LevelControlCluster(NoManufacturerCluster, LevelControl):
    """LevelControlCluster that reports to attrid 0xF000."""

    server_commands = LevelControl.server_commands.copy()
    server_commands[TUYA_CUSTOM_LEVEL_COMMAND] = foundation.ZCLCommandDef(
        "moveToLevelTuya",
        (TuyaLevelPayload,),
        is_manufacturer_specific=False,
    )

    attributes = LevelControl.attributes.copy()
    attributes.update(
[... 184 lines cut ...]
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    F000LevelControlCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

gregoryduckworth, 2022-11-07:
It's looking better! Just need to get it into the right state:
[image]

javicalle, 2022-11-07:
That seems that will require some extra work:
* https://github.com/Koenkk/zigbee2mqtt/issues/12952

gregoryduckworth, 2022-11-07:
Aww, that is a shame! My WAF is going through the floor 😉 

Thanks for all your help with this. Looks like I will have to play a little more.

javicalle, 2022-11-07:
Look at the `F000LevelControlCluster` of that device.
There must be a `bulb_type` attribute (ignore the name).
Get its value (probably `0`).
Try to update to another value: `0`/`1`/`2`/`3`

gregoryduckworth, 2022-11-07:
The `bulb_type` comes back as `TuyaBulbType.LED` on both

javicalle, 2022-11-07:
I'm pretty sure that this is the attribute but I don't know why you can't update its value.
I'll take another look, but would be tomorrow.

References:
* https://github.com/Koenkk/zigbee-herdsman-converters/pull/4393/files
* https://github.com/gorbunovav/zigbee-herdsman-converters/blob/master/devices/lonsonho.js

javicalle, 2022-11-11:
I can't find any problem in the code. Can you enable the debug log and attach them?
* https://www.home-assistant.io/integrations/zha/#debug-logging

Please attach the logs from changing the value in HA and also the logs from interacting with the phisycal device (switching, dimming, etc)

gregoryduckworth, 2022-11-12:
Hopefully I've got everything in this log, there is a fair old amount of information coming through!

<details>

<summary>All debug Log</summary>

```
2022-11-12 14:44:52.708 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=light, service=turn_on, service_data=entity_id=light.lounge_light_light>
2022-11-12 14:44:52.710 DEBUG (MainThread) [homeassistant.components.zha.entity] light.lounge_light_light: setting transitioning flag to True
2022-11-12 14:44:52.710 DEBUG (MainThread) [zigpy.zcl] [0xF9DE:1:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=23, command_id=1, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-11-12 14:44:52.711 DEBUG (MainThread) [zigpy.zcl] [0xF9DE:1:0x0006] Sending request: on()
2022-11-12 14:44:52.711 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xF9DE), dst_ep=1, source_route=None, extended_timeout=False, tsn=23, profile_id=260, cluster_id=6, data=Serialized[b'\x01\x17\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-11-12 14:44:52.713 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xF9DE), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=6, TSN=23, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=0, Data=b'\x01\x17\x01')
2022-11-12 14:44:52.732 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-11-12 14:44:52.737 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=23)
2022-11-12 14:44:52.829 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xF9DE, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=21, SecurityUse=<Bool.false: 0>, TimeStamp=8871441, TSN=0, Data=b'\x18\x17\x0B\x01\x00', MacSrcAddr=0xFB19, MsgResultRadius=28)
2022-11-12 14:44:52.830 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xF9DE, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=21, SecurityUse=<Bool.false: 0>, TimeStamp=8871744, TSN=0, Data=b'\x08\xDA\x0A\x00\x00\x10\x01', MacSrcAddr=0xFB19, MsgResultRadius=28)
2022-11-12 14:44:52.830 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xF9DE), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=6, data=Serialized[b'\x18\x17\x0b\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=21, rssi=None)
2022-11-12 14:44:52.830 DEBUG (MainThread) [zigpy.zcl] [0xF9DE:1:0x0006] Received ZCL frame: b'\x18\x17\x0b\x01\x00'
2022-11-12 14:44:52.831 DEBUG (MainThread) [zigpy.zcl] [0xF9DE:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=23, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-11-12 14:44:52.831 DEBUG (MainThread) [zigpy.zcl] [0xF9DE:1:0x0006] Decoded ZCL frame: OnOff:Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
2022-11-12 14:44:52.831 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xF9DE), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=6, data=Serialized[b'\x08\xda\n\x00\x00\x10\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=21, rssi=None)
2022-11-12 14:44:52.832 DEBUG (MainThread) [zigpy.zcl] [0xF9DE:1:0x0006] Received ZCL frame: b'\x08\xda\n\x00\x00\x10\x01'
2022-11-12 14:44:52.832 DEBUG (MainThread) [zigpy.zcl] [0xF9DE:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=218, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-11-12 14:44:52.832 DEBUG (MainThread) [zigpy.zcl] [0xF9DE:1:0x0006] Decoded ZCL frame: OnOff:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.true: 1>))])
2022-11-12 14:44:52.833 DEBUG (MainThread) [zigpy.zcl] [0xF9DE:1:0x0006] Received command 0x0A (TSN 218): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.true: 1>))])
2022-11-12 14:44:52.833 DEBUG (MainThread) [zigpy.zcl] [0xF9DE:1:0x0006] Attribute report received: on_off=<Bool.true: 1>
[... cut ...]
2022-11-12 14:53:46.096 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-11-12 14:53:46.096 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x6832), DstEndpoint=2, DstPanId=0x0000, SrcEndpoint=1, ClusterId=6, TSN=164, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=0, Data=b'\x00\xA4\x00\x00\x00')
2022-11-12 14:53:46.104 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
```
</details>

closed 2023-07-10 by github-actions[bot]