zigpy/zha-device-handlers #1841: iTead Sonoff ZBMini-L not leasing pairing mode after pairing

veitw, 2022-10-18:
I have several Sonoff ZBMini-L which paired and worked just fine using a Sonoff ZBBridge. Now I am migrating to ZHA using a Sonoff ZB Dongle Plus-E as coordinator but I am experiencing problems pairing the ZBMini-L, as it seems to not recognise the pairing to have finished successfully, while ZHA states otherwise:

After pairing the device with ZHA it states that the device is now ready for use, but the device stays in pairing mode (indicated by the green LED blinking at ~1 Hz). After about a minute, the device seems to restart pairing mode, drops out of the Zigbee network, reconnects, and drops out of the network again but is then never seen again until pairing in ZHA is started again. For the seconds the device is part of the Zigbee network, it can be controlled just fine. All the time the green LED stays flashing until at some point the ZBMini-L gives up and completely stops the pairing.

Pairing the device to the Sonoff ZBBridge, the "healthy" behaviour is as follows:
Putting the device into pairing mode, the green LED flashes. After pairing the device using the app, the green LED stops flashing and instead stays on or off continuously according to the state of the switch relay. The device can be controlled and does not drop out of the network.

All devices are on the latest software
* ZBMini-L runs firmware 1.1.1
* ZBDongle-E runs firmware 6.10.3-v1.0.1
* Home Assistant is at version 2022.10.4, running on HAOS 9.2

veitw, 2022-10-19:
@Hedda Thank you very much for these hints. I have updated the description according to the bug report template.

I also have enabled debug logging as described and performed the pairing, but within these few minutes this produced 15k+ log lines including quite personal information that should not belong here. Do you have any further hints on what to grep from that log that is actually required for debugging such an issue, so I can filter out and redact these lines?

The method of contantly pressing the button while pairing is something I already tried, without success. Also I do not think this is a low energy/power saving issue, as due to its capacitors the device even continues to pair several times for up to about a minute even after being disconnected from power (breaker off, bulb unscrewed or plug pulled).

veitw, 2022-10-19:
I found this in the forums:
https://community.home-assistant.io/t/endless-pairing-loops-for-some-devices/370602

It describes exactly the issue that I have, just for Innr and Ikea Zigbee LED bulbs instead of the Sonoff ZBMini-L.

Unfortunately, the workaround described (switching off/powering off the device after first successful pairing) does not work -- I tried this about 15 times today.

orestesmas, 2022-10-25:
For your information, I have 3 of these devices working in my house and I'm not experiencing the issue you report.

I'm using a Conbee II coordinator with ZHA, as people you mentiones in your last message.

Nevertheless, I have other issues with these ZBMini-L devices: One of them is controlling a LED light in the porch, and I set up to turn on the light at sunset. Well, this works ok but I noticed that after a while (variable between 30 and 90 minutes approx.) the switch disconnects itself. I'm totally confused about that, because the logs doesn't show anything strange. I've to investigate this in depth, but I've read that these "no neutral required" devices need a load of around 3-4W minimum to work properly, and perhaps my LED bulb isn't an strong enough load and for that reason the relay disconnects.

I don't know if this behavior may be related to your problem, but I say it here just in case...

veitw, 2022-10-25:
@orestesmas : Thanks for these infos. Are your ZBMini-L on firmware version 1.1.1, too?

Regarding the minimum load, the ZBMini-L is very humble compared to other no-neutral actuators, iTead even advertises that it does not require a bypass/capacitor being installed for low loads. In fact this is also true for my device, it works regardless whether only a single LED bulb is driven or an old 40W incandescent light bulb. I also measured the device to consume less than 0.5W (which is the lower limit my power meter does measure accurately) chained with either bulb.

stperin, 2022-11-02:
Hi,

I have same issue on ZMINI-L firmware version 1.1.1

Did somebody found any solution?
Is there any alternative firmware to install to try to fix this?

MattWestb, 2022-11-02:
@veitw Your EZSP version is one of the best and is not having problem joining "normal" devices (some extreme strange devices is all coordinators is having problems with).

Sonoff is having firmware problems with some new devices but i dont knowing is its related with this device and Z2M have digging deep for finding the problem and itead have being involved for workaround (not having the network open for joining that Z2M is having = not Zigbee 3 compliant and large security risk) but the work around is not working with ZHA then ZHA is only open the network for limited time  as shall being made for good Zigbee 3 coordinators.

https://github.com/Koenkk/zigbee2mqtt/issues/10282

I don have the device so i dont knowing what is happening but if some can sniffing one paring i can look if the coordinator and the device is doing all OK or if one part is not.

thomasfr, 2022-11-02:
I can also confirm the behaviour. I am using Home Assistant Yellow (ezsp), bought 2 ZB Mini-L, both with the same issue. 
Pairing Mode (green led flashes), Home Assistant can find it, initializes, and says that the configuration is done and the device is ready to be used, while the green led keeps on flashing. When i change to the Devices section of Home Assistant it sometimes shows as available and sometimes it works to switch it on/off 2-3 times, but shortly after the logs say that the device is not available anymore.

MattWestb, 2022-11-02:
Servus @thomasfr.
Do you have the possibility sniffing the paring with Yellow ?

If not I can doing it if can having one device for some days (time i cant guarantee ten its little chaotic in private for the moment).

Im in W1110 so shall being easy arranging form other all of Wien.

thomasfr, 2022-11-02:
Oh servus! Nice to meet someone from vienna here :) 
Sure, i can.... if you can tell me how.

MattWestb, 2022-11-02:
Im pinging you from my GITHUB for more info (I deleting my e-mail address later sent so its floating around).

I think its easy only some U3 stations and little walking :-))

thomasfr, 2022-11-02:
I am not sure what you mean. 
How do sniff the pairing on home assistant?

MattWestb, 2022-11-02:
I have pinging you in one new issue in my github in the message is my email but deleted in the issue now.
Better explaining way i have time problems for the moment.

For sniffing you need one CC2532 with sniff firmware or one normal EZSP coordinator but i can doing it for you.

thomasfr, 2022-11-02:
I sent you an E-Mail

MattWestb, 2022-11-02:
I have getting it and replayed.

Software for sniffing with EZSP https://github.com/zsmartsystems/com.zsmartsystems.zigbee.sniffer
that can being tricky getting it working (i think some modules or somthing is missing but i have it working on my laptop) plus wireshark.
All is explained in the link also links to guides and examples.

Hedda, 2022-11-03:
Also see https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

etienn01, 2022-11-17:
@veitw @stperin @thomasfr did you try with your ZigBee dongle on an USB extension cord? It fixed the problem for me with my ConBee II.

veitw, 2022-11-17:
Hi @etienn01,

thanks for that hint. My Sonoff ZB Dongle Plus V2 (ZBDongle-E) is already offset from my HA box using a 3m USB 2.0 extension cord, is also connected to a USB 2.0 port instead of USB 3.x, and the HA box is wired by Ethernet with WiFi and BT disabled. 

Best regards,
// Veit

SalN85, 2023-01-04:
Today I reported the same issue, it happened aftear i replaced my old cc2531 with a brand new sonoff usb dogle 3.0 (ZBDongle-E).
Note. In my network there is another Zbimi-l the works fine
Does anyone has solved the issue?

roygeraets, 2023-01-24:
Hi Guys, 

I've got the same issue, ZHA and Z2MQTT. Same ZBDongle-E.
With Zigbee2MQTT it is possible to downgrade the firmware with the "Force" tag. 

But the Device itself does not allow the downgrade functionality.

Hope Itead will come with some decent firmware changes. I'm going to return my ZBMini-L devices as this is not the solution I'm hoping for.

veitw, 2023-01-24:
As I also had problems with ZHA regarding my Osram/Ledvance A60 bulbs (in another ticket), which restart over and over again until eventually becoming stable after sometimes hours (which was quite annoying esp. at night in the bedroom, when after a short power outage the light switched on and on again every few seconds for hours), I switched to Z2M and the problems were instantly gone, both regarding my ZBMini-Ls as well as my Osram/Ledvance bulbs. It just worked as expected and I am so happy now.

All my ZBMini-L work well in Z2M with both the ZBDongle-E and also a new ZBDongle-P (CC2652P) that I bought as EZSP support is still experimental in Z2M and EZSP coordinator backup is not supported yet.

But I am still able to test for any updates as the ZBDongle-E is still connected to ZHA and I have one unused ZBMini-L left, though I have switched the ZBDongle-E to firmware version 7 to try ZHA+OpenThread in parallel but I could still switch back to firmware version 6 for testing until I get my first Matter-Thread devices. I have not yet tried the ZBMini-L with firmware version 7 though...

I read in a thread that people were successful pairing ZBMini-L after they paired a ZBDongle-P in router mode with Koenkk's router firmware and told ZHA to pair the problematic devices via the ZBDongle-P instead of the coordinator. I had not tried this as I read this after I already switched to Z2M with ZBDongle-P, but I had no success trying to pair the ZBMini-L via Ledvance bulbs and some Tuya plugs.

roygeraets, 2023-01-24:
@veitw Can you post a link of ZBDongle-E firmware version 7.
The official version is 6.10.3 https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/Router

Would like to check if this maybe fixes the issue. 

> But I am still able to test for any updates as the ZBDongle-E is still connected to ZHA and I have one unused ZBMini-L left, though I have switched the ZBDongle-E to firmware version 7 to try [...]

veitw, 2023-01-24:
@roygeraets : This is the MultiPAN firmware that I installed:
https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/RCPMultiPAN

This is not an official firmware, nor is it even signed, but I do not think that there is anything special about the ZBDongle-E that would prevent a generic EFR32MG21 firmware from working properly.

As I have no more Zigbee devices paired to ZHA, I cannot tell whether it actually works, I can only tell that ZHA says EZSP protocol is up, OpenThread seems to handle this well and that Z2M (stable) does not work with it beyond initial step of pairing as it does not support EZSP v7 yet. Did not try Z2M (dev) though.

Be warned that I am positive that you will have to re-pair devices or restore a coordinator backup when upgrading to EZSP v7 MultiPAN.

MattWestb, 2023-01-24:
The xps1989 is / was from Sonoff devs but its not updated then the dev have moving on.
Signing is only for unhacked Sonoff ZBB and the unsignes is for the rest Sonofs EZSP.

Latest RCP / GSDK 7.20 is having one bug with direct children but its working good i have 30 mixed devices up and running with thread parallel running tuya MG21 and IKEA MG1P modules with ZHA. Then Silabs is fixing the bug HA is going Open thread with boarder router and matter support parallel with Zigbee on the same chip = all in.

kylehowells, 2023-01-30:
Chiming in to say I've also been experiencing this with `Aqara smart wall switch H1 EU (with neutral, single rocker)`  [`WS-EUK03`](https://www.zigbee2mqtt.io/devices/WS-EUK03.html) device.

It pairs almost instantly and is controllable in ZHA HA, but the wall switch itself stays flashing it's LED in pairing mode and doesn't seem to think it is connected.
It works is is controllable, but every few seconds the device reboots, and then after about 5-10mins it exists pairing mode and shows the red "failed to pair" led color instead, and is no longer discoverable in HA.

It seems like the device is pairing, but then it isn't being sent whatever signal the device is expecting to tell it that it has worked.

etmmafl, 2023-06-15:
Same problem here

Zigbee Sonoff dongle 3.0
ESMLFzm_w6_TW by AwoX

closed 2023-12-19 by github-actions[bot]