zigpy/zha-device-handlers #1842: [Device Support Request]TS0601 by _TZE200_qoy0ekbd temperature/humidity sensor

FS1961, 2022-10-19:
**Is your feature request related to a problem? Please describe.**
The device connects to ZHA but has a static 32 degree and 0% humidity status. I've removed and re-added the device to no avail

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

<details>

<summary>Device signature</summary>

```yaml
Device info
TS0601
by _TZE200_qoy0ekbd
Connected via [Zigbee Coordinator](https://ha-erdes-fs61.duckdns.org:8123/config/devices/device/48526946c36575235f266a2ab7ef7fd3)
Zigbee info

IEEE: a4:c1:38:a2:9d:1a:12:50
Nwk: 0x570e
Device Type: EndDevice
LQI: 7
RSSI: Unknown
Last Seen: 2022-10-19T12:28:05
Power Source: Battery or Unknown
```

</details>

<details>
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.72",
    "supervisor": "2022.10.0",
    "host_os": "Home Assistant OS 9.2",
    "docker_version": "20.10.17",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.28.2",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "smartthinq_sensors": {
      "version": "0.24.2",
      "requirements": [
        "pycountry>=20.7.3",
        "xmltodict>=0.12.0",
        "cchardet>=2.1.7"
      ]
    },
    "garbage_collection": {
      "version": "4.8.3",
      "requirements": [
        "python-dateutil>=2.8.2"
      ]
    },
    "localtuya": {
[... 297 lines cut ...]
        }
      }
    }
  }
}

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

FS1961, 2022-10-20:
Sorry, I didn't see it because it was buried under Manage Device>Signature.  Thanks!

codyhackw, 2022-10-20:
Does the physical device list a manufacturer, etc?

And if you go into that menu again and Manage Clusters > "TemperatureMeasurement (Endpoint id: 1, Id: 0x0402, Type: in)" > "measured_value" and then do a Get Zigbee Attribute do you see anything come through?

FS1961, 2022-10-21:
Ordered via Amazon.de and it describes the manufacturer as FENGCHUANG TUYA. The write-up says it requires a Tuya hub. No markings on the physical device.

codyhackw, 2022-10-21:
Did you see anything in the Manage Clusters? Or guessing it's giving you the 32 degrees and 0% humidity that you see depending on which you're looking at?

codyhackw, 2022-10-21:
Actually, can you try this quirk?

```
"""Tuya temp and humidity sensor with e-ink screen."""

from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.measurement import RelativeHumidity, TemperatureMeasurement

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaPowerConfigurationCluster2AAA
from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaDPType, TuyaMCUCluster

# NOTES:
# The data comes in as a string on cluster, if there is nothing set up you may see these lines in the logs:
# Unknown message (b'19830100a40102000400000118') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:d0:18:8b:64:aa, 1)'
#                                          28.0 degrees
# Unknown message (b'19840100a5020200040000022c') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:d0:18:8b:64:aa, 1)'
#                                          55.6% humid
# Unknown message (b'19850100a60402000400000064') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:d0:18:8b:64:aa, 1)'
#                                          100% battery

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

class TuyaRelativeHumidity(RelativeHumidity, TuyaLocalCluster):
    """Tuya local RelativeHumidity cluster."""

class TemperatureHumidityManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with Temperature and Humidity data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaTemperatureMeasurement.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 10,  # decidegree to centidegree
        ),
        2: DPToAttributeMapping(
            TuyaRelativeHumidity.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 100,  # decipercent to percent
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster2AAA.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: x * 2,  # double reported percentage
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
    }

class TuyaTempHumiditySensor(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=81
        # device_version=1
        # input_clusters=[4, 5, 61184, 0]
        # output_clusters=[25, 10]>
        MODELS_INFO: [("_TZE200_bjawzodf", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TemperatureHumidityManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    TemperatureHumidityManufCluster,  # Single bus for temp, humidity, and battery
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    TuyaPowerConfigurationCluster2AAA,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }
    
class TuyaTempHumiditySensor_Square(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=0x0302
        # input_clusters=[0x0000, 0x0001, 0x0402, 0x0405]
        # output_clusters=[0x000a, 0x0019]>
        MODELS_INFO: [("_TZE200_qoy0ekbd", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA,
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    TemperatureHumidityManufCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }
```
Just add it to your custom quirk folder as "TS0601_sensoroverride.py" or similar?

FS1961, 2022-10-21:
So just copy & paste into a file?

codyhackw, 2022-10-21:
Yeah so you'll need to go into your HA configuration.yaml and add the "custom_quirks_path:" similar to the below - 

```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```

and then you'll create a folder "custom_zha_quirks" in the same directory and then create that "TS0601_sensoroverride.py" file within that new folder and then restart and see if it applies to that device in the information tab of the device.

FS1961, 2022-10-22:
Everything worked great. Thank you.

closed 2022-10-22 by FS1961

codyhackw, 2022-10-23:
Noting this was related to - https://github.com/zigpy/zha-device-handlers/issues/1670 and the solution offered by @javicalle