zigpy/zha-device-handlers #1844: [BUG]

mospaeda, 2022-10-19:
**Describe the bug**
In home assistant, the firmware seems reported wrong in the display
ZLinky_TIC
par LiXee
Connecté via [Zigbee Coordinator]
Firmware: 0x00000001
I don't know from where the value 0x00000001 is coming but it is probably not pointing to right register

**To Reproduce**
Steps to reproduce the behavior:
go the ZHA device page : LiXee ZLinky_TIC
look at the firmware version : Firmware: 0x00000001

But From the cluster details it seems that it is different (in my case firmware 9 as I received last week)
reference from offcial doc : https://github.com/fairecasoimeme/Zlinky_TIC#clusters

Basic (Endpoint id: 1, Id: 0x0000, Type: in)
Valeur 9

sw_build_id (id: 0x4000)
4000-0009
date_code (id: 0x0006)
20220930

**Expected behavior**
Should be Firmware: 0x00000009 

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=200, server_mask=11264, maximum_outgoing_transfer_size=200, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0053",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0702",
        "0x0b01",
        "0x0b04",
        "0xff66"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LiXee",
  "model": "ZLinky_TIC",
  "class": "zhaquirks.lixee.zlinky.ZLinkyTIC"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

codyhackw, 2022-10-20:
Can you pull up the device, manage clusters > OTA and then under Cluster Attributes > "current_file_version" and do a "Get Zigbee Attribute" and check what that reports?

mospaeda, 2022-10-20:
Detail is as below :

Ota (Endpoint id: 1, Id: 0x0019, Type: out)
current_file_version (id: 0x0002)
Value : 1

So value is 1

Hedda, 2022-10-21:
Maybe you should also report as issue to https://github.com/fairecasoimeme/Zlinky_TIC/issues for attention of @fairecasoimeme

mospaeda, 2022-10-21:
For record
I received the LiXee ZLinky_TIC already with the firmware 9 installed 
I didn't update the received Zlinky by OTA to version 9

Basic -Endpoint id: 1, Id: 0x0000 - Value is 9
Basic - sw_build_id (id: 0x4000) - Value is 4000-0009
Basic - date_code (id: 0x0006) - Value is 20220930.

For some reason, the OTA - current_file_version is reported as 1

mospaeda, 2022-10-21:
@codyhackw 
Can you confirm the cluster used for firmware version ?
 
https://github.com/fairecasoimeme/Zlinky_TIC/issues/115#issuecomment-1286718366

> I didn't know that on ZHA the firmware version was related to the OTA cluster
> For me the version firmware is in Basic not in OTA
> I'll see to update also the version in OTA cluster
> That would be fixed in next version ...

codyhackw, 2022-10-21:
I know it uses the OTA cluster with other devices, which is why I asked you to check there. Not certain if that's only the case if/when the OTA cluster is used or how the determination is made though.

Hedda, 2023-02-27:
Not sure if this can be fixed on the zha-device-handlers or zigpy side without the user updating device firmware?

For reference, the workaround, for now at least, is for the user to update ZLinky TIC v13.0 device firmware (or later)

https://github.com/fairecasoimeme/Zlinky_TIC

https://github.com/fairecasoimeme/Zlinky_TIC/releases/tag/v13.0

https://github.com/fairecasoimeme/Zlinky_TIC/issues

 ZLinky TIC v13.0 firmware is supported in the upcoming Home Assistant 2023.3 release as it will have zha-device-handlers v0.0.93:

https://github.com/zigpy/zha-device-handlers/releases/tag/0.0.93

mospaeda, 2023-09-04:
Not possible in my case to update to firmware 13.
I seem something is missing or not working (Zigate and Zlinky on Home assistant latest version )
Still need to investigate why (and especilally if Zigate can push update)

Hedda, 2023-09-05:
> Not possible in my case to update to firmware 13.
> I seem something is missing or not working (Zigate and Zlinky on Home assistant latest version )
> Still need to investigate why (and especilally if Zigate can push update)

I do not own one but believe that there are two types of firmware variants with different modes from them and you can only perform Zigbee OTA update if the mode of the OTA file is the same as the previous one.

https://github.com/fairecasoimeme/Zlinky_TIC/blob/master/README.md#ota

That means you can not change flash an image over OTA that has a different mode, so if you want to change mode then you must update the firmware manually by flashing via a TTL Serial to USB adapter or similar, see "Non OTA" section in their readme file:

https://github.com/fairecasoimeme/Zlinky_TIC/blob/master/README.md#non-ota

https://zigate.fr/documentation/mise-a-jour-de-la-zigate-2/

Regardless, suggest that you ask them for help about firmware upgrade issues as that part is really off-topic here:

https://github.com/fairecasoimeme/Zlinky_TIC/issues

closed 2024-03-10 by github-actions[bot]