zigpy/zha-device-handlers #1846: [BUG] Ikea Starkvind air purifier & air purifier table sensors missing

willliamchan, 2022-10-20:
**Describe the bug**
All sensors are missing, zha pick up the device as fan only.

**To Reproduce**
N/A

**Expected behavior**
Expecting zha to pick up sensor like PM25, status light etc.

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0007",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0202",
        "0xfc57",
        "0xfc7c",
        "0xfc7d"
      ],
      "out_clusters": [
        "0x0019",
        "0x0400",
        "0x042a"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "STARKVIND Air purifier table",
  "class": "zigpy.device.Device"
}</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/London",
[... 601 lines cut ...]
        }
      }
    }
  }
}</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

javicalle, 2022-10-22:
Seems to be another version of the current quirk.

Enable the local custom quirk:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Copy in your local quirks folder the current quirk:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/ikea/starkvind.py

Add at the end of file this quirk:
```python
class IkeaSTARKVIND_v2(CustomDevice):
    """STARKVIND Air purifier by IKEA of Sweden."""

    def __init__(self, *args, **kwargs):
        """Init."""
        self.pm25_bus = Bus()
        self.change_fan_mode_bus = Bus()
        self.change_fan_mode_ha_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=7 (0x0007)
        # device_version=0
        # input_clusters=[0, 3, 4, 5, 514, 64599, 64637] output_clusters=[25, 1024, 1066]>
        MODELS_INFO: [
            (IKEA, "STARKVIND Air purifier"),
            (IKEA, "STARKVIND Air purifier table"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,  # 0
                    Identify.cluster_id,  # 3
                    Groups.cluster_id,  # 4
                    Scenes.cluster_id,  # 5
                    Fan.cluster_id,  # 514    0x0202
                    WWAH_CLUSTER_ID,  # 64599  0xFC57
                    0xFC7C,
                    IkeaAirpurifier.cluster_id,  # 64637  0xFC7D
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,  # 25      0x0019
                    IlluminanceMeasurement.cluster_id,  # 1024    0x0400
                    PM25.cluster_id,  # 1066    0x042A PM2.5 Measurement Cluster
                ],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=0
            # input_clusters=[33] output_clusters=[33]>
            242: {
                PROFILE_ID: 0xA1E0,  # 41440 (dec)
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,  # 0x0021 = GreenPowerProxy.cluster_id
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,  # 0
                    Identify.cluster_id,  # 3
                    Groups.cluster_id,  # 4
                    Scenes.cluster_id,  # 5
                    WWAH_CLUSTER_ID,  # 64599  0xFC57
                    0xFC7C,
                    IkeaAirpurifier,  # 64637  0xFC7D control air purifier with manufacturer-specific attributes
                    PM25Cluster,  # 1066    0x042A PM2.5 Measurement Cluster
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,  # 25      0x0019
                    IlluminanceMeasurement.cluster_id,  # 1024    0x0400
                ],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=0
            # input_clusters=[33] output_clusters=[33]>
            242: {
                PROFILE_ID: 0xA1E0,  # 41440 (dec)
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,  # 0x0021 = GreenPowerProxy.cluster_id
                ],
            },
        },
    }
```

Restart HA, remove and pair you device again. Check that the device is loading the quirk and that the device signature has changed.

willliamchan, 2022-10-22:
> Seems to be another version of the current quirk.
> 
> Enable the local custom quirk:
> 
> * [Editing (ZHA) quirks in a (HA) container or local. #693 (comment)](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274)
> 
> Copy in your local quirks folder the current quirk:
> 
> * https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/ikea/starkvind.py
> 
> Add at the end of file this quirk:
> 
> ```python
> class IkeaSTARKVIND_v2(CustomDevice):
>     """STARKVIND Air purifier by IKEA of Sweden."""
> 
>     def __init__(self, *args, **kwargs):
>         """Init."""
>         self.pm25_bus = Bus()
>         self.change_fan_mode_bus = Bus()
>         self.change_fan_mode_ha_bus = Bus()
>         super().__init__(*args, **kwargs)
> 
>     signature = {
>         # <SimpleDescriptor endpoint=1 profile=260 device_type=7 (0x0007)
>         # device_version=0
>         # input_clusters=[0, 3, 4, 5, 514, 64599, 64637] output_clusters=[25, 1024, 1066]>
>         MODELS_INFO: [
>             (IKEA, "STARKVIND Air purifier"),
>             (IKEA, "STARKVIND Air purifier table"),
>         ],
>         ENDPOINTS: {
>             1: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,  # 0
>                     Identify.cluster_id,  # 3
>                     Groups.cluster_id,  # 4
>                     Scenes.cluster_id,  # 5
>                     Fan.cluster_id,  # 514    0x0202
>                     WWAH_CLUSTER_ID,  # 64599  0xFC57
>                     0xFC7C,
>                     IkeaAirpurifier.cluster_id,  # 64637  0xFC7D
>                 ],
>                 OUTPUT_CLUSTERS: [
>                     Ota.cluster_id,  # 25      0x0019
>                     IlluminanceMeasurement.cluster_id,  # 1024    0x0400
>                     PM25.cluster_id,  # 1066    0x042A PM2.5 Measurement Cluster
>                 ],
>             },
>             # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
>             # device_version=0
>             # input_clusters=[33] output_clusters=[33]>
>             242: {
>                 PROFILE_ID: 0xA1E0,  # 41440 (dec)
>                 DEVICE_TYPE: 0x0061,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [
>                     GreenPowerProxy.cluster_id,  # 0x0021 = GreenPowerProxy.cluster_id
>                 ],
>             },
>         },
>     }
> 
>     replacement = {
>         ENDPOINTS: {
>             1: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,  # 0
>                     Identify.cluster_id,  # 3
>                     Groups.cluster_id,  # 4
>                     Scenes.cluster_id,  # 5
>                     WWAH_CLUSTER_ID,  # 64599  0xFC57
>                     0xFC7C,
>                     IkeaAirpurifier,  # 64637  0xFC7D control air purifier with manufacturer-specific attributes
>                     PM25Cluster,  # 1066    0x042A PM2.5 Measurement Cluster
>                 ],
>                 OUTPUT_CLUSTERS: [
>                     Ota.cluster_id,  # 25      0x0019
>                     IlluminanceMeasurement.cluster_id,  # 1024    0x0400
>                 ],
>             },
>             # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
>             # device_version=0
>             # input_clusters=[33] output_clusters=[33]>
>             242: {
>                 PROFILE_ID: 0xA1E0,  # 41440 (dec)
>                 DEVICE_TYPE: 0x0061,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [
>                     GreenPowerProxy.cluster_id,  # 0x0021 = GreenPowerProxy.cluster_id
>                 ],
>             },
>         },
>     }
> ```
> 
> Restart HA, remove and pair you device again. Check that the device is loading the quirk and that the device signature has changed.

Worked like a charm! PM2.5 sensor is still missing however the thread owner said it will reappear in a few days. I'll report if it's not showing up on next Monday.

closed 2022-10-22 by willliamchan

MattWestb, 2022-10-22:
I think this is related to the firmware update in the device:
- Release Version: 1.22.0 GA
  - 6th Sep 2022
  - STARKVIND Air Purifier (v1.1.001)
  - Upgrade to stack v6.10.2. 

I dont have one of this then the price was 99€ and now its around 150€ and its out off my tech budget :-((

I was getting one recycled TREDANSEN without all needed things only the battery and after little talking i was paying 13.99€ = - 90% very good and its working.
I shall try finding the same person and asking if i can getting one STARKVIND returned that not possible selling and have working Zigbee module and getting one good price.

The Satrkvind shall having more sensors like filter change and runtime left for the filter filter time reset command and more but that need sniffing the IKEA Hub but shall not being any problem if i can getting one working device.

javicalle, 2022-10-22:
There are a few configuration entities in ZHA for this device:
* https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/core/channels/manufacturerspecific.py#L240-L250