zigpy/zha-device-handlers #1848: [Device Support Request] GLEDOPTO GL-S-004Z Dual White and Color

aquemy, 2022-10-21:
Hello,

I have two Gledopto GL-S-004Z Dual White and Color (details [here](https://zigbee.blakadder.com/Gledopto_GL-S-004Z.html)).
They are perfectly recognized by ZHA. However, I cannot access the RGB capabilities, only WW. 

**Describe the solution you'd like**

Technically, these bad boys can handle both White and Colors at the same time so the perfect solution would be to have a way to control simultaneously both.    
However, I would already be happy with RGB support and a way to switch between WW and RGB as I have with other GLEDOPTO device (e.g. [GL-MC-001P](https://zigbee.blakadder.com/Gledopto_GL-MC-001P.html) controller).

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "11": {
      "profile_id": 49246,
      "device_type": "0x0220",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300"
      ],
      "out_clusters": []
    },
    "13": {
      "profile_id": 49246,
      "device_type": "0x0220",
      "in_clusters": [
        "0x1000"
      ],
      "out_clusters": [
        "0x1000"
      ]
    }
  },
  "manufacturer": "GLEDOPTO",
  "model": "GL-S-004Z",
  "class": "zigpy.device.Device"
}
 ```
</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Warsaw",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.10.0",
    "host_os": "Home Assistant OS 9.0",
[... 383 lines cut ...]
        },
        "profile_id": 49246,
        "in_clusters": {
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

Happy to help with anything if needed!

javicalle, 2022-10-21:
It appears to be a slightly different version of the current quirk.
For some reason, all quirks for the manufacturer are based on remove some endpoint, so I suggest the same pattern.

Can you try with that one?:
```python
class SoposhMR16(CustomDevice):
    """GLEDOPTO Soposh Dual White and color 4W MR16 250lm."""

    signature = {
        MANUFACTURER: GLEDOPTO,
        ENDPOINTS: {
            11: {
                PROFILE_ID: zll.PROFILE_ID,
                DEVICE_TYPE: DeviceType.COLOR_TEMPERATURE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            13: {
                PROFILE_ID: zll.PROFILE_ID,
                DEVICE_TYPE: DeviceType.COLOR_TEMPERATURE_LIGHT,
                INPUT_CLUSTERS: [LightLink.cluster_id],
                OUTPUT_CLUSTERS: [LightLink.cluster_id],
            },
        }
    }
```

There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Copy in your local quirks folder the current quirk:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/gledopto/soposhgu10.py

Edit the file, adding the proposed quirk at the end of the file.

Restart HA, remove and pair you device again. Check that the device is loading the quirk and that the device signatura has changed.

aquemy, 2022-10-24:
Hi @javicalle,

Thank you for your time and help.
(Un)fortunately, the problem was between the chair and the keyboard. This particular model does not have RGB but only WW/CW.
I've been mistaken while ordering it because of the name and because there is a RGB model with roughly the [same name](https://zigbee.blakadder.com/Gledopto_GL-S-005Z.html).

The good news is that this particular light bulb is fully supported by ZHA. I opened a [PR](https://github.com/blakadder/zigbee/pull/783) on [https://zigbee.blakadder.com](https://zigbee.blakadder.com) to reflect this

closed 2022-10-24 by aquemy