zigpy/zha-device-handlers #1849: [Device Support Request] _TZE200_4hbx5cvx TS0601

olepw, 2022-10-21:
Can you add support for this thermostat: https://www.connecte.no/termostat
From what I can see, this seems to be a Tuya device. The device is discovered by ZHA and shows as a thermostat. Both room temperature and the set thermostat temperature are read correctly, but I'm not able to change the temperature or turn it on or off.

I would like to be able to control the temperature on the thermostat and turn the heat on and off.

Also, this device has been added to zigbee2mqtt:
https://github.com/Koenkk/zigbee2mqtt/issues/11895
Technical documentation:
[For.utviklere.og.avanserte.smarthus.systemer.som.krever.koding.av.Connecte.termostat.docx](https://github.com/zigpy/zha-device-handlers/files/9838153/For.utviklere.og.avanserte.smarthus.systemer.som.krever.koding.av.Connecte.termostat.docx)

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0201",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_4hbx5cvx",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Oslo",
    "os_name": "Linux",
    "os_version": "5.15.61-v7l",
    "supervisor": "2022.10.0",
    "host_os": "Home Assistant OS 9.2",
    "docker_version": "20.10.17",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
[... 352 lines cut ...]
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

[image]

Let me know if you need more information!

MattWestb, 2023-01-17:
This device is needing tuya magic spell.
https://github.com/zigpy/zha-device-handlers/discussions/2032#discussioncomment-4709357

closed 2023-09-10 by github-actions[bot]

eivindor, 2024-08-20:
Did anyone solve this? I have the same device

gilrim, 2024-08-20:
nah.. ZHA tends to have limited support for devices, and figuring out how to make things work can be quite challenging. I've switched to using Z2M instead, and so far, I haven't run into any compatibility issues.