zigpy/zha-device-handlers #1851: pzem-004t  not working with zha

delta010, 2022-10-21:
**Describe the bug**
I am using cc2530 with pzem-004t v3 with PTVO configurable firmware in the home assistant  using this custom quirk  
[pzem004t.py.txt](https://github.com/zigpy/zha-device-handlers/files/9839085/pzem004t.py.txt) but no entites showing in zha dashboard

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
power monitering entites,the screem short show the expected entites
[image]

**Screenshots**
my zha dashbord with no entites
[image]
PTVO firmware configeration
[image]

<details>

<summary>Device Signature </summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle: 136>, manufacturer_code=4447, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0xfffe",
      "in_clusters": [
        "0x0000",
        "0x000c",
        "0x0014"
      ],
      "out_clusters": [
        "0x0000"
      ]
    }
  },
  "manufacturer": "ptvo.info",
  "model": "ptvo.switch",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.9.7",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "UTC",
    "os_name": "Linux",
    "os_version": "5.18.0-kali7-amd64",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.33.1",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.79",
      "zigpy-deconz==0.18.1",
      "zigpy==0.50.3",
      "zigpy-xbee==0.15.0",
      "zigpy-zigate==0.9.2",
      "zigpy-znp==0.8.2"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "1A86",
        "pid": "55D4",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus v2"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*zigstar*",
        "known_devices": [
          "ZigStar Coordinators"
        ]
      },
      {
        "vid": "1CF1",
        "pid": "0030",
        "description": "*conbee*",
        "known_devices": [
          "Conbee II"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8A2A",
        "description": "*zigbee*",
        "known_devices": [
          "Nortek HUSBZB-1"
        ]
      },
      {
        "vid": "0403",
        "pid": "6015",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate+"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8B34",
        "description": "*bv 2010/10*",
        "known_devices": [
          "Bitron Video AV2010/10"
        ]
      }
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly"
    ],
    "zeroconf": [
      {
        "type": "_esphomelib._tcp.local.",
        "name": "tube*"
      },
      {
        "type": "_zigate-zigbee-gateway._tcp.local.",
        "name": "*zigate*"
      }
    ],
    "dependencies": [
      "file_upload"
    ],
    "after_dependencies": [
      "onboarding",
      "usb",
      "zeroconf"
    ],
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp"
    ],
    "is_built_in": true
  },
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 34,
    "manufacturer": "ptvo.info",
    "model": "ptvo.switch",
    "name": "ptvo.info ptvo.switch",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4447,
    "power_source": "Battery or Unknown",
    "lqi": 84,
    "rssi": null,
    "last_seen": "2022-10-21T18:21:57",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle: 136>, manufacturer_code=4447, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0xfffe",
          "in_clusters": [
            "0x0000",
            "0x000c",
            "0x0014"
          ],
          "out_clusters": [
            "0x0000"
          ]
        }
      }
    },
    "active_coordinator": false,
    "entities": [],
    "neighbors": [],
    "endpoint_names": [
      {
        "name": "undefined_0xfffe"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "049629de53bc6753e6e007adbc3e7238",
    "area_id": null,
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "undefined_0xfffe",
          "id": 65534
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "ptvo.info"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "ptvo.switch"
              }
            },
            "unsupported_attributes": {}
          },
          "0x0014": {
            "endpoint_attribute": "multistate_value",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000c": {
            "endpoint_attribute": "analog_input",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```

```
[log.txt](https://github.com/zigpy/zha-device-handlers/files/9839255/log.txt)

</details>

**Additional context**
Add any other context about the problem here.

glcos, 2022-10-21:
Looks like you are using the quirk from my repository https://github.com/glcos/pzem004t
Please notice that in order for this quirk to work you need to use the exact PTVO GUI settings as shown in my repository, otherwise the quirk signature is not recognized by ZHA and the quirk ignored.
I have uploaded the latest version of the quirk I'm using with current Home Assistant release, please use that one instead.
Also notice that my device is running a PTVO firmare which is quite old, the latest version might have other differences that require adjusting the quirk.

delta010, 2022-10-21:
I updated the quirk and changed the settings but still got the same result no entities. do I need to use the same PTVO software

glcos, 2022-10-21:
The guys at PTVO have made considerable improvement and bug fixes to their firmware since I downloaded the release that is currently running on my device, therefore I would not recommend downgrading to that.
You can try to enable additional logging on your system and see if you can figure out what's going wrong.

```
# Logger
logger:
  default: warning
  logs:
    homeassistant.components.zha: debug
    zigpy.zcl: debug
    zhaquirks: debug
```

delta010, 2022-10-22:
this is my home assistant log file
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/9843453/home-assistant.log)

glcos, 2022-10-22:
Today I upgraded HA to the latest version and took the chance to capture the log for you.
This is what you are expected to see if the device is paired and the quirk is loaded properly

```
2022-10-22 14:46:49.995 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'ElectricalMeasurement' using ['electrical_measurement']
2022-10-22 14:46:49.995 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'ElectricalMeasurementApparentPower' using ['electrical_measurement']
2022-10-22 14:46:49.995 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'ElectricalMeasurementRMSCurrent' using ['electrical_measurement']
2022-10-22 14:46:49.996 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'ElectricalMeasurementRMSVoltage' using ['electrical_measurement']
2022-10-22 14:46:49.996 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'ElectricalMeasurementFrequency' using ['electrical_measurement']
2022-10-22 14:46:49.996 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'ElectricalMeasurementPowerFactor' using ['electrical_measurement']
2022-10-22 14:46:49.996 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'Temperature' using ['temperature']
2022-10-22 14:46:49.996 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'RSSISensor' using ['basic']
2022-10-22 14:46:49.996 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'LQISensor' using ['basic']
2022-10-22 14:46:49.996 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'SmartEnergyMetering' using ['smartenergy_metering']
2022-10-22 14:46:49.996 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'SmartEnergySummation' using ['smartenergy_metering']
2022-10-22 14:46:50.006 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] [0x4C57](CSRLabs pzem004t) restored as 'available', last seen: 0:03:30 ago, consider_unavailable_time: 21600 seconds

```
Even in the case the quirk is not working you should see in the log that the device is actually sending its values to the coordinator, something like this

```
2022-10-22 14:48:46.264 DEBUG (MainThread) [zigpy.zcl] [0x4C57:2:0x000c] Received ZCL frame: b'\x089\nU\x009\x00\xeb,Jo\x00\x18\x00\x1c\x00B\x02Wh'
2022-10-22 14:48:46.265 DEBUG (MainThread) [zigpy.zcl] [0x4C57:2:0x000c] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=57, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-22 14:48:46.266 DEBUG (MainThread) [zigpy.zcl] [0x4C57:2:0x000c] Decoded ZCL frame: PtvoAnalogInputCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=2833088.0)), Attribute(attrid=0x006F, value=TypeValue(type=bitmap8, value=<bitmap8.0: 0>)), Attribute(attrid=0x001C, value=TypeValue(type=CharacterString, value='Wh'))])
2022-10-22 14:48:46.267 DEBUG (MainThread) [zigpy.zcl] [0x4C57:2:0x000c] Received command 0x0A (TSN 57): Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=2833088.0)), Attribute(attrid=0x006F, value=TypeValue(type=bitmap8, value=<bitmap8.0: 0>)), Attribute(attrid=0x001C, value=TypeValue(type=CharacterString, value='Wh'))])
2022-10-22 14:48:46.267 DEBUG (MainThread) [zigpy.zcl] [0x4C57:2:0x000c] Attribute report received: present_value=2833088.0, status_flags=<bitmap8.0: 0>, description='Wh'
2022-10-22 14:48:46.271 DEBUG (MainThread) [zigpy.zcl] [0x4C57:2:0x000c] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=57, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-22 14:48:46.271 DEBUG (MainThread) [zigpy.zcl] [0x4C57:2:0x000c] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2022-10-22 14:48:46.331 DEBUG (MainThread) [zigpy.zcl] [0x4C57:2:0x000c] Received ZCL frame: b'\x088\nU\x009\xd1"\xbb?o\x00\x18\x00\x1c\x00B\x01A'
2022-10-22 14:48:46.332 DEBUG (MainThread) [zigpy.zcl] [0x4C57:2:0x000c] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=56, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-22 14:48:46.334 DEBUG (MainThread) [zigpy.zcl] [0x4C57:2:0x000c] Decoded ZCL frame: PtvoAnalogInputCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=1.4620000123977661)), Attribute(attrid=0x006F, value=TypeValue(type=bitmap8, value=<bitmap8.0: 0>)), Attribute(attrid=0x001C, value=TypeValue(type=CharacterString, value='A'))])
[... cut ...]
2022-10-22 14:48:46.456 DEBUG (MainThread) [zigpy.zcl] [0x4C57:2:0x000c] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=59, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-22 14:48:46.456 DEBUG (MainThread) [zigpy.zcl] [0x4C57:2:0x000c] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)

```
I'm sorry, apart from this I don't know how to help further.

delta010, 2022-10-23:
to make everything clear the RX pin of CC2530 P0.2 and TX is  CC2530 P0.3  why in the software p00 is enabled

glcos, 2022-10-23:
P00 is enabled because I connected a pushbutton for pairing.
I used CC2530 for my very early experiment just to evaluate the PZEM module, then I moved to EBYTE E72-2G4M20S1F module which is based on CC2652RB for the final project.
I'm attaching an old picture that might help.
[image]

This is the final project with CC2652RB schematic and PCB

[image]

[image]

delta010, 2022-10-23:
can you please provide your firmware(.hex file)? for some reason, no entities are discovered by zha or by the home assistant

```
2022-10-22 14:46:49.995 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'ElectricalMeasurement' using ['electrical_measurement']
2022-10-22 14:46:49.995 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'ElectricalMeasurementApparentPower' using ['electrical_measurement']
2022-10-22 14:46:49.995 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'ElectricalMeasurementRMSCurrent' using ['electrical_measurement']
2022-10-22 14:46:49.996 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'ElectricalMeasurementRMSVoltage' using ['electrical_measurement']
2022-10-22 14:46:49.996 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'ElectricalMeasurementFrequency' using ['electrical_measurement']
2022-10-22 14:46:49.996 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'ElectricalMeasurementPowerFactor' using ['electrical_measurement']
2022-10-22 14:46:49.996 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'Temperature' using ['temperature']
2022-10-22 14:46:49.996 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'SmartEnergyMetering' using ['smartenergy_metering']
2022-10-22 14:46:49.996 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'SmartEnergySummation' using ['smartenergy_metering']
2022-10-22 14:46:50.006 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] [0x4C57](CSRLabs pzem004t) restored as 'available', last seen: 0:03:30 ago, consider_unavailable_time: 21600 seconds
```
only Link quality and rssi are  discovered by the home assistant

glcos, 2022-10-23:
Unfortunately I no longer have the firmware that I used on CC2530.
But, again, you may want to double check why the quirk is not being loaded, that's why you are not seing the other entities.

delta010, 2022-10-23:
i think the quirk is loaded
```
2022-10-21 17:24:52.802 DEBUG (MainThread) [zhaquirks] Loading custom quirks from /config/quirks
2022-10-21 17:24:52.805 DEBUG (MainThread) [zhaquirks] Loading custom quirks module pzem004t
```

MattWestb, 2022-10-23:
Posting the diagnostic from the device card so can see the endpoints and cluster the device is using and also if the quirk is being loaded (likely not then the cluster setting is changed).

delta010, 2022-10-23:
device diagnostic 

```
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.10.5",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "UTC",
    "os_name": "Linux",
    "os_version": "5.18.0-kali7-amd64",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.2",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.83",
      "zigpy-deconz==0.19.0",
      "zigpy==0.51.3",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.2",
      "zigpy-znp==0.9.1"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "1A86",
        "pid": "55D4",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus v2"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
[... 172 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x0007": {
            "endpoint_attribute": "on_off_config",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0012": {
            "endpoint_attribute": "multistate_input",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

MattWestb, 2022-10-23:
The:
```
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
```
is saying the system have not loading our local quirk that the system have found https://github.com/zigpy/zha-device-handlers/issues/1851#issuecomment-1288053535.
So some (on or more) cluster on one or more endpoints is not the same as in the quirk signature.
Fixing the signature and the system shall loading the quirk and you shall getting the functionality that is added in the quirk for your device.

delta010, 2022-10-23:
I am sorry  to ask this how I fix the  signature

MattWestb, 2022-10-23:
The quirk is having one signature part with much information the device is having.
If all information is the same as the device the system is loading the quirk and if one parameter is not OK its not loading it.
One example for one easy device class:
https://github.com/zigpy/zha-device-handlers/blob/f2822031a464bfdf766dfccebf667a502622ebbb/zhaquirks/ikea/opencloseremote.py#L79-L116
Then have the signature OK and the system is loading the quirk you can doing changes in the next part  `replacement` that is what the system shall doing the device.

If you is adding one new endpoint with one new cluster you is adding it the the signature and then also in the replacement with changes if you like.

delta010, 2022-10-23:
actually i don't know what to replace

glcos, 2022-10-25:
@delta010 looking again at what you have provided it seems that your device is not sending the chip temperature value.
This is expected by the quirk, therefore the signature will not match if the temperature value is not sent by the device.
You can try to enable the temperature on the PTVO firmware GUI and see if the issue persists

[image]

In addition, if you are using the latest version of the quirk, please set "CSRLabs" as the manufacturer name otherwise it will not match

```
"attributes": {
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "CSRLabs"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "pzem004t"
              }

```

delta010, 2022-10-27:
I enable the internal temperature in the ptvo GUI, but no luck same as above 
[image]

I think the internal temperature is now meshing 

```
2022-10-27 12:40:10.218 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Decoded ZCL frame: AnalogInput:Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=63.0)), Attribute(attrid=0x001C, value=TypeValue(type=CharacterString, value='Wh'))])
2022-10-27 12:40:10.218 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Received command 0x0A (TSN 254): Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=63.0)), Attribute(attrid=0x001C, value=TypeValue(type=CharacterString, value='Wh'))])
2022-10-27 12:40:10.219 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Attribute report received: present_value=63.0, description='Wh'
2022-10-27 12:40:10.229 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Received ZCL frame: b'\x18\xff\nU\x009\x00\x00\x00\x00\x1c\x00B\x01W'
2022-10-27 12:40:10.229 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=255, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-27 12:40:10.230 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Decoded ZCL frame: AnalogInput:Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=0.0)), Attribute(attrid=0x001C, value=TypeValue(type=CharacterString, value='W'))])
2022-10-27 12:40:10.231 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Received command 0x0A (TSN 255): Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=0.0)), Attribute(attrid=0x001C, value=TypeValue(type=CharacterString, value='W'))])
2022-10-27 12:40:10.231 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Attribute report received: present_value=0.0, description='W'
2022-10-27 12:40:10.238 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Received ZCL frame: b'\x18\x00\nU\x00933GB\x1c\x00B\x02Hz'
2022-10-27 12:40:10.238 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=0, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-27 12:40:10.239 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Decoded ZCL frame: AnalogInput:Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=49.79999923706055)), Attribute(attrid=0x001C, value=TypeValue(type=CharacterString, value='Hz'))])
2022-10-27 12:40:10.239 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Received command 0x0A (TSN 0): Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=49.79999923706055)), Attribute(attrid=0x001C, value=TypeValue(type=CharacterString, value='Hz'))])
2022-10-27 12:40:10.239 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Attribute report received: present_value=49.79999923706055, description='Hz'
2022-10-27 12:40:10.247 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Received ZCL frame: b'\x18\x01\nU\x009\x00\x00\x00\x00\x1c\x00B\x02pf'
2022-10-27 12:40:10.248 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=1, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-27 12:40:10.248 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Decoded ZCL frame: AnalogInput:Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=0.0)), Attribute(attrid=0x001C, value=TypeValue(type=CharacterString, value='pf'))])
2022-10-27 12:40:10.248 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Received command 0x0A (TSN 1): Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=0.0)), Attribute(attrid=0x001C, value=TypeValue(type=CharacterString, value='pf'))])
2022-10-27 12:40:10.248 DEBUG (MainThread) [zigpy.zcl] [0xAB58:1:0x000c] Attribute report received: present_value=0.0, description='pf'
2022-10-27 12:40:10.259 DEBUG (MainThread) [zigpy.zcl] [0xAB58:2:0x000c] Received ZCL frame: b'\x18\x02\nU\x009\x00\x00&C\x1c\x00B\x01C'
2022-10-27 12:40:10.259 DEBUG (MainThread) [zigpy.zcl] [0xAB58:2:0x000c] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=2, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-27 12:40:10.259 DEBUG (MainThread) [zigpy.zcl] [0xAB58:2:0x000c] Decoded ZCL frame: AnalogInput:Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=166.0)), Attribute(att [... cut ...] g, value='C'))])
2022-10-27 12:40:10.259 DEBUG (MainThread) [zigpy.zcl] [0xAB58:2:0x000c] Received command 0x0A (TSN 2): Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=166.0)), Attribute(attrid=0x001C, value=TypeValue(type=CharacterString, value='C'))])
2022-10-27 12:40:10.260 DEBUG (MainThread) [zigpy.zcl] [0xAB58:2:0x000c] Attribute report received: present_value=166.0, description='C'

```
is the internal temperature pin matter I use ```P01```

delta010, 2022-10-29:
also i find this,

```
2022-10-29 12:04:58.222 DEBUG (MainThread) [zigpy.zcl] [0x0B6B:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=13, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-29 12:04:58.223 DEBUG (MainThread) [zigpy.zcl] [0x0B6B:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='CSRLabs')), ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='pzem004t'))])
2022-10-29 12:04:58.225 INFO (MainThread) [zigpy.device] [0x0b6b] Read model 'pzem004t' and manufacturer 'CSRLabs' from <Endpoint id=1 in=[basic:0x0000, multistate_value:0x0014, analog_input:0x000C, on_off_config:0x0007] out=[basic:0x0000, multistate_input:0x0012] status=<Status.ZDO_INIT: 1>>
2022-10-29 12:04:58.225 INFO (MainThread) [zigpy.device] [0x0b6b] Discovered basic device information for <Device model='pzem004t' manuf='CSRLabs' nwk=0x0B6B ieee=00:12:4b:00:18:62:26:90 is_initialized=True>
2022-10-29 12:04:58.226 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='pzem004t' manuf='CSRLabs' nwk=0x0B6B ieee=00:12:4b:00:18:62:26:90 is_initialized=True>
2022-10-29 12:04:58.227 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for CSRLabs pzem004t (00:12:4b:00:18:62:26:90)
2022-10-29 12:04:58.227 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'pzem004t.pzem004t'>
2022-10-29 12:04:58.228 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 3} {1, 2}
2022-10-29 12:04:58.228 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace2'>
2022-10-29 12:04:58.228 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 242} {1, 2}
2022-10-29 12:04:58.228 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1'>
2022-10-29 12:04:58.228 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 41, 42, 242, 51, 21, 31} {1, 2}
2022-10-29 12:04:58.228 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2022-10-29 12:04:58.229 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1, 2}
2022-10-29 12:04:58.229 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2022-10-29 12:04:58.229 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1, 2}
2022-10-29 12:04:58.229 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts0201.MoesTemperatureHumidtySensorWithScreen'>
2022-10-29 12:04:58.229 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2}
2022-10-29 12:04:58.229 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
2022-10-29 12:04:58.230 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2}
2022-10-29 12:04:58.230 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
2022-10-29 12:04:58.230 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2}
2022-10-29 12:04:58.230 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
2022-10-29 12:04:58.231 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2}
2022-10-29 12:04:58.231 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2022-10-29 12:04:58.231 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {11, 13} {1, 2}

```
this log says ```2022-10-29 12:04:58.227 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for CSRLabs pzem004t (00:12:4b:00:18:62:26:90)```
```2022-10-29 12:04:58.227 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'pzem004t.pzem004t'>```
```2022-10-29 12:04:58.228 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 3} {1, 2}``` I think that's why the quirks not work  

how to solve the ```endpoint list mismatch```problem

delta010, 2022-11-02:
in my  cluster attributes, there is no cluster for ```TemperatureMeasurementCluster,SmartEnergyCluster,ElectricalMeasurementCluster``` is that ok
[image]

can someone share you're device diagnostic for reference

glcos, 2022-11-02:
This is my device diagnostic, you can compare it to yours

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle: 136>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=11264, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0xfffe",
      "in_clusters": [
        "0x0000",
        "0x0014",
        "0x0402",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x0000"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0053",
      "in_clusters": [
        "0x000c",
        "0x0014"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0053",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "CSRLabs",
  "model": "pzem004t",
  "class": "CSRLabs.pzem004t.pzem004t"
}
```

delta010, 2022-11-02:
not all 3 endpoints are  discoverd in my case, also not all cluster ids

your endpoints
```
"endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0xfffe",
      "in_clusters": [
        "0x0000",
        "0x0014",
        "0x0402", Temperature Measurement
        "0x0702", Simple Metering Cluster
        "0x0b04"  Electrical Measurement
      ],
      "out_clusters": [
        "0x0000"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0053", 
      "in_clusters": [
        "0x000c",  Analogue Input
        "0x0014"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0053",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    }

```

my endpoints
```
 "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0xfffe",
          "in_clusters": [
            "0x0000", 
            "0x0007",  On/Off Switch Configuration
            "0x000c",   Analogue Input
            "0x0014"
          ],
          "out_clusters": [
            "0x0000",
            "0x0012"   Multistate Input (Basic)
          ]
        },
        "2": {
          "profile_id": 260,
          "device_type": "0xfffe",
          "in_clusters": [
            "0x000c"
          ],
          "out_clusters": []
        }
      }
    },
``` 
in my case, there is no cluster id for  ```Temperature Measurement, Simple Metering Cluster,Electrical Measurement```
also, the device type is wrong, what is this ```0x0053``` device type? I tried all possible firmware configurations but had no luck.

Does anyone know why the temperature and metering clusters are not initializing?

MattWestb, 2022-11-02:
Found device type `0x0053`
https://github.com/zigpy/zigpy/blob/fc8e923e2f2482810b9643c0b2bde149a56a1930/zigpy/profiles/zha.py#L25.

delta010, 2022-11-04:
there is something fundamentally wrong with my GUI configuration which I don't know. no necessary cluster ids are generated according to my configuration.
[image]
[image]

glcos, 2022-11-04:
Are you sure that your CC2530 is properly communicating with the PZEM-004T module? You should see the RX and TX LEDs on pzem blinking at the polling frequency you set in the GUI.

delta010, 2022-11-04:
yes, the pzem communicated with cc2530 I can see the value on the home assistant log also the led blink every 5 sec

glcos, 2022-11-04:
Then the only thing I can think of is the newer version of the PTVO firmware. The guys over there keep changing things. As soon as I have time I'll try to install the latest version and check. In the meantime you can try to edit the quirk and remove the entities that are no longer being sent by yout device and see if it matches and is loaded by HA.

delta010, 2022-11-07:
do you have ptvo firmware version 1.7.2.0, if so please share it .no other versions are working with the quirks

delta010, 2022-11-12:
> Then the only thing I can think of is the newer version of the PTVO firmware. The guys over there keep changing things. As soon as I have time I'll try to install the latest version and check. In [...]

did you try the latest version of PTVO firmware?

delta010, 2022-11-22:
Does anyone know why the appropriate clusterID is not generated in ZHA  . in my case no clusterid related to electricity measurement is generated .

closed 2023-05-28 by github-actions[bot]