zigpy/zha-device-handlers #1852: [Device Support Request] Immax Neo Smart Keypad "_TZE200_n9clpsht"

PonyOny, 2022-10-21:
Immax Neo Keypad Zigbee 3.0 / MaxSmart MS-K1AZ, is discovered fine by ZHA, but unfortunately no entities show up.
It seems a Quirk is needed...

**Describe the solution you'd like**
Entities for armed/disarmed/armed-home, battery, SOS and others if possible to show up in HA

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_n9clpsht",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.10.5",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Madrid",
    "os_name": "Linux",
    "os_version": "5.10.103-v7l+",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.28.2",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "zha_toolkit": {
      "version": "v0.8.20",
      "requirements": [
        "packaging>=20.8"
      ]
    },
    "imou_life": {
      "version": "1.0.2",
      "requirements": [
        "imouapi==1.0.2"
      ]
    },
    "alarmo": {
      "version": "v1.9.6",
      "requirements": []
    },
    "smartir": {
      "version": "1.17.6",
      "requirements": [
        "aiofiles==0.6.0"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.2",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.83",
      "zigpy-deconz==0.19.0",
      "zigpy==0.51.3",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.2",
      "zigpy-znp==0.9.1"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "1A86",
        "pid": "55D4",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus v2"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*zigstar*",
        "known_devices": [
          "ZigStar Coordinators"
        ]
      },
      {
        "vid": "1CF1",
        "pid": "0030",
        "description": "*conbee*",
        "known_devices": [
          "Conbee II"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8A2A",
        "description": "*zigbee*",
        "known_devices": [
          "Nortek HUSBZB-1"
        ]
      },
      {
        "vid": "0403",
        "pid": "6015",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate+"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8B34",
        "description": "*bv 2010/10*",
        "known_devices": [
          "Bitron Video AV2010/10"
        ]
      }
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly"
    ],
    "zeroconf": [
      {
        "type": "_esphomelib._tcp.local.",
        "name": "tube*"
      },
      {
        "type": "_zigate-zigbee-gateway._tcp.local.",
        "name": "*zigate*"
      },
      {
        "type": "_zigstar_gw._tcp.local.",
        "name": "*zigstar*"
      }
    ],
    "dependencies": [
      "file_upload"
    ],
    "after_dependencies": [
      "onboarding",
      "usb",
      "zeroconf"
    ],
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp"
    ],
    "is_built_in": true
  },
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 25691,
    "manufacturer": "_TZE200_n9clpsht",
    "model": "TS0601",
    "name": "_TZE200_n9clpsht TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4098,
    "power_source": "Battery or Unknown",
    "lqi": 68,
    "rssi": -83,
    "last_seen": "2022-10-21T18:56:12",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0051",
          "in_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "out_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      }
    },
    "active_coordinator": false,
    "entities": [],
    "neighbors": [],
    "endpoint_names": [
      {
        "name": "SMART_PLUG"
      }
    ],
    "user_given_name": "Alarm Keypad",
    "device_reg_id": "29d983dbf0e41066045e1ef60dcc2199",
    "area_id": null,
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "SMART_PLUG",
          "id": 81
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "_TZE200_n9clpsht"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0601"
              }
            },
            "unsupported_attributes": {}
          },
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xef00": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

**Additional context**
[image]

PonyOny, 2022-10-22:
There is a custom component for Zigbee2MQTT that is working for this device, could someone help out converting this to a ZHA quirk?

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_n9clpsht'
        },
    ],
    model: '07505L', // Vendor model number, look on the device for a model number
    vendor: 'Immax', // Vendor of the device (only used for documentation and startup logging)
    description: 'Immax Neo Smart Keypad', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [
		fz.ignore_basic_report,
		{
		  cluster: "manuSpecificTuya",
		  type: [
			"commandSetDataResponse",
			"commandGetData",
			"commandActiveStatusReport",
			"commandDataResponse"
		  ],
		  convert: (model, msg, publish, options, meta) => {
			const dp = msg.data.dpValues[0].dp;
			const armModeImmax = {
				26: 'disarm',
				27: 'arm_away',
				28: 'arm_home',
				29: 'sos',
			};

			return {action: armModeImmax[dp]};
//			return {data: msg.data};
		  },
		},
	],
    exposes: [e.action(['disarm', 'arm_home', 'arm_away', 'sos'])],
    toZigbee: [],
};

module.exports = definition;
```

MrAdam, 2022-10-27:
I've requested support for this in DeCONZ as well, and have purchased a Tuya Zigbee Gateway to map out all the DP ID's.
You can find the DP ID's here if it helps with the integration: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6277#issuecomment-1260484312

PonyOny, 2022-10-27:
> I've requested support for this in DeCONZ as well, and have purchased a Tuya Zigbee Gateway to map out all the DP ID's. You can find the DP ID's here if it helps with the integration: [dresden-elektronik/deconz-rest-plugin#6277 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6277#issuecomment-1260484312)

Wow amazing work, I was going to buy a tuya hub just to be able to get those values, that is amazing, thanks for your hard work.

MrAdam, 2022-10-28:
> Wow amazing work, I was going to buy a tuya hub just to be able to get those values, that is amazing, thanks for your hard work.

If I'm missing any values, please tell, and I'll try and get them :)

PonyOny, 2022-11-01:
That wood be amazing, as now I have had to add Zigbee2MQTT just to use this keypad, it would be amazing if i could use it with ZHA, I have been trying with Quirks, but i don't know how to process the messages received from the device in ZHA, I can see them on the debug log but can't process them as in Z2M.

PonyOny, 2022-11-01:
Lets see if we can get something to work with ZHA, and get a working quirk so we can start using this keypad 😊

MrAdam, 2022-11-06:
I got some time to work on the quirk this weekend, but am far from finished.
So far, I am able to control the configuration (enabling/disabling settings, and setting arm delay) of the device using ZHA, and I am also able to reset the device from ZHA.
I still need to figure out how to map the state events, and if it is possible to set the armed/unarmed more using ZHA.
```python
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.security import IasAce
from zigpy.zcl.clusters.general import PowerConfiguration

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)

from zhaquirks.tuya import TuyaLocalCluster, TUYA_CLUSTER_ID
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
)

TUYA_DP_ID_BATTERY_LEVEL_PERCENTAGE = 3  # Percentage
TUYA_DP_ID_SOS_ALARM = 23
TUYA_DP_ID_ANTI_REMOVE_ALARM = 24
TUYA_DP_ID_BATTERY_LEVEL_ENUM = 25  # "Low", "Middle", "High"
TUYA_DP_ID_DISARMED = 26
TUYA_DP_ID_ARMED = 27
TUYA_DP_ID_ARMED_HOME = 28
TUYA_DP_ID_SOS = 29
TUYA_DP_ID_ARM_DELAY_TIME = 103  # Seconds
TUYA_DP_ID_KEYPAD_BEEPS = 104  # Boolean
TUYA_DP_ID_QUICK_SOS = 105  # Boolean
TUYA_DP_ID_QUICK_DISARM = 106  # Boolean
TUYA_DP_ID_QUICK_ARM = 107  # Boolean
TUYA_DP_ID_ADMIN_CODE = 108
TUYA_DP_ID_USER_CODE = 109
TUYA_DP_ID_RESET = 110
TUYA_DP_ID_ARM_DELAY_BEEPS = 111  # Boolean
TUYA_DP_ID_UNKNOWN = 112

class TuyaPowerConfigurationCluster3AAA(PowerConfiguration, TuyaLocalCluster):
    """PowerConfiguration cluster for devices with 3 AAA."""

    BATTERY_SIZE = 0x0031
    BATTERY_QUANTITY = 0x0033
    BATTERY_RATED_VOLTAGE = 0x0034

    _CONSTANT_ATTRIBUTES = {
        BATTERY_SIZE: PowerConfiguration.BatterySize.AAA,
        BATTERY_QUANTITY: 3,
        BATTERY_RATED_VOLTAGE: 15,
    }

class WirelessZigbeeKeypadManufCluster(TuyaMCUCluster):
    """Tuya manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            TUYA_CLUSTER_ID + TUYA_DP_ID_ARM_DELAY_TIME: ("arm_delay_time", t.uint32_t),
            TUYA_CLUSTER_ID + TUYA_DP_ID_KEYPAD_BEEPS: ("keypad_beeps", t.Bool),
[... 94 lines cut ...]
                    Ota.cluster_id,
                    Time.cluster_id,
                ],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ANCILLARY_CONTROL,
                INPUT_CLUSTERS: [
                    WirelessZigbeeKeypadManufCluster,
                    TuyaPowerConfigurationCluster3AAA,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                    Time.cluster_id,
                ],
            }
        },
    }
```

dmulcahey, 2022-11-06:
Have these conversations in PR’s just mark them draft until you’re ready

PonyOny, 2022-11-06:
@MrAdam Amazing work 👌🏻 I will give it a try.

pimseb, 2023-06-17:
I've moved from zigbee2mqtt to ZHA and I own this keypad.
I'm not able to make it work with ZHA. Is there any update on this ?
Thank you

MrAdam, 2023-06-19:
I stopped working on this and moved to zigbee2mqtt, as I had to make changes in both home assistant repository, and this repository, but weren't able to get any feedback on which direction to go with the code.
See comment here: https://github.com/zigpy/zha-device-handlers/pull/1908#issuecomment-1453438851

closed 2023-12-23 by github-actions[bot]