zigpy/zha-device-handlers #1853: [BUG] Tuya 2 gang switch (TS0002 _TZ3000_zmy4lslw) always switches all relays

BenPru, 2022-10-21:
**Describe the bug**
When I turn on switch 1 in ha its automatically turn on switch 2 together. Turning off the same and turn off or on switch 2 switches relay 1 together.
When I switch the hardware switch 1 only relay 1 is switched. Same with 2.
Hardware works as exected, switching over software not!

Electricalmeasurement is also not working. Values are always zero.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect TS0002 _TZ3000_zmy4lslw
2. Switch 1 or 2

**Expected behavior**
Switch every relay alone.

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_zmy4lslw",
  "model": "TS0002",
  "class": "ts000x_own.Switch_2G_GPP"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

javicalle, 2022-10-21:
Your device is loading a local quirk.
How is the device signature withot the quirk? Please, post it.
Can be the behavior related to the local quirk?

josecefe, 2022-10-24:
Hi,
I have the same problem with this device, and this is the device signature:

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_zmy4lslw",
  "model": "TS0002",
  "class": "zigpy.device.Device"
}
```

The problem is that switches are working as a kind of group when they are switch on/off in HA, but the work well when are operated with the physical switches.

MattWestb, 2022-10-24:
@javicalle The @josecefe device s not loading the ZHA quirk and its needed for getting it not putting the relays of by it self.
New cluster config i the device ?

Also its need the tuya magic cast for fixing the endpoint working OK and not being connected together (i think you have putting it in the latest quirk).

Its looks like one updated version of https://github.com/zigpy/zha-device-handlers/blob/5abcf7fa6cb75f0f703a6c5c6f14ed3fb236e2ff/zhaquirks/tuya/ts000x.py#L88 with some power cluster added:
```
0x0B04 = ElectricalMeasurement
0x0702 = Simple Metering
```

Mvh  MW

javicalle, 2022-10-24:
I believe that #1860 would fit here.
Not sure about metrics configuration (multipliers / divisors)

PR #1860 (2022-10-25): Add new TS000X metering devices
New quirks for the TS000X metering devices.

Fixes: #1821
Fixes: #1853
closed 2022-10-25 by dmulcahey