zigpy/zha-device-handlers #1854: [BUG] Moes 2 Gang Dimmer TS0601 | _TZE200_e3oitdyu issues

dobromir-hristov, 2022-10-22:
**Describe the bug**
The [2 Gang Moes dimmer -_TZE200_e3oitdyu](https://zigbee.blakadder.com/Moes_MS-105B.html) when connected allows turning on/off via UI, but when using sliders to dim, it works randomly.

If you toggle both lights quickly, it rarely toggles both.

If you use the physical switch, it does not get reflected in the dashboard

**To Reproduce**
Steps to reproduce the behavior:

1. Try using the UI, by toggling on/off the lighs.
2. Try using the UI,  by dragging the dimming sliders.
3. Try using the physical buttons connected to the dimmer, and check if those are reflected in HA. 

**Expected behavior**
All of the above should work as you would expect. Physical changes get reflected in the dashboard. Turning both lights through the dimmer should work as expected. Dimming should work as expected.

**Screenshots**
NA

<details>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0006",
        "0x0008"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZE200_e3oitdyu",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_dimmer.TuyaDoubleSwitchDimmer"
}

```

</details>

<details>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.5",
    "dev": false,
    "hassio": true,
[... 365 lines cut ...]
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

dobromir-hristov, 2022-10-22:
Might be related to:
- https://github.com/zigpy/zha-device-handlers/issues/1554

javicalle, 2022-10-22:
Let's continue the conversation in #1554

dobromir-hristov, 2022-10-23:
I am not sure if it's the same cause but sure.

TheJulianJES, 2023-01-04:
Please check if this is fixed with 2023.1.0b5 or later (full release comes out today).

dobromir-hristov, 2023-01-04:
Keeping an eye out for the release. Thanks!

dobromir-hristov, 2023-01-04:
Just updated, the issue persists. I looked at the logs and all I see related to zigpy is:

```
Logger: zigpy.device
Source: runner.py:120
First occurred: 11:44:13 PM (129 occurrences)
Last logged: 11:46:40 PM

Failed to parse message (b'095202007301') on cluster 61184, because Data is too short to contain 1 bytes
Failed to parse message (b'095302007301') on cluster 61184, because Data is too short to contain 1 bytes
Failed to parse message (b'095502007501') on cluster 61184, because Data is too short to contain 1 bytes
Failed to parse message (b'095602007501') on cluster 61184, because Data is too short to contain 1 bytes
Failed to parse message (b'095702007501') on cluster 61184, because Data is too short to contain 1 bytes
```

and 
```
2023-01-04 23:50:36.569 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x4C55, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=78, SecurityUse=<Bool.false: 0>, TimeStamp=13496902, TSN=0, Data=b'\x09\x22\x02\x00\x77\x01', MacSrcAddr=0x4C55, MsgResultRadius=29)
2023-01-04 23:50:36.569 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4C55), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\t"\x02\x00w\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=78, rssi=None)
2023-01-04 23:50:36.570 DEBUG (MainThread) [zigpy.zcl] [0x4C55:1:0xef00] Received ZCL frame: b'\t"\x02\x00w\x01'
2023-01-04 23:50:36.570 DEBUG (MainThread) [zigpy.zcl] [0x4C55:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=34, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-01-04 23:50:36.571 ERROR (MainThread) [zigpy.device] Failed to parse message (b'092202007701') on cluster 61184, because Data is too short to contain 1 bytes
```

pyrexfm, 2023-01-09:
Just want to +1 this! This is the only decent looking multi gang dimmer and, as a HA Yellow user, would be great to be able to use ZHA at my home.

dobromir-hristov, 2023-01-09:
> Just want to +1 this! This is the only decent looking multi gang dimmer and, as a HA Yellow user, would be great to be able to use ZHA at my home.

Unfortunately there isn't much choice for a behind the switch dimmer yes... In Europe there isn't much at all 😂 

I wonder if the other Moes dimmer with the rotating knob works decent? I read reviews they should be the same but yeah... Who knows

javicalle, 2023-01-13:
The code complains deserializing the response, probably here:
https://github.com/zigpy/zha-device-handlers/blob/cec721eceb85fb7c680b6f468c53610b606355d7/zhaquirks/tuya/__init__.py#L1410-L1415

That `b'\t"\x02\x00w\x01'` frame is:
```
set_data_response(data=TuyaCommand(status=0, tsn=119, datapoints=[TuyaDatapointData(dp=1)]))
```

The `TuyaCommand`:
https://github.com/zigpy/zha-device-handlers/blob/cec721eceb85fb7c680b6f468c53610b606355d7/zhaquirks/tuya/__init__.py#L258-L263

* `status`: 0
* `tsn`: 119

The `TuyaDatapointData`:
https://github.com/zigpy/zha-device-handlers/blob/cec721eceb85fb7c680b6f468c53610b606355d7/zhaquirks/tuya/__init__.py#L251-L255

* `dp`: 1
* but no `data`  (`b''`)

Not sure if it can be some new Tuya message, a faulty device or a chopped frame. Also, not sure if related to the issue.

Can you post a full log with the faulty operation?

dobromir-hristov, 2023-01-13:
@javicalle can you let me know from where I should get the full log? I went to the full log `config/logs` its 800+ messages of the same thing when I filter by `zigpy`

javicalle, 2023-01-15:
I was expecting all the debug logs for a few operations.
Here are instructions to enable the debug logs:
* https://www.home-assistant.io/integrations/zha/#debug-logging

It can be very verbose. Try to catch the operation part for your device.

closed 2023-07-21 by github-actions[bot]

dobromir-hristov, 2023-10-28:
There is still an issue in the latest HA - 2023.10.5.

1. Dimmer mostly works, when controlling bulbs one by one via the UI.
2. If you use a card to turn off the two bulbs at once, one hangs in the ON state.
3. If you use a hardware switch to change brightness or toggle bulbs, that is rarely reflected in the UI, usually with a delay.

If there was a setting to set, to ping the device for status or something, I guess 3) would be fixable.

Not sure why a bulk turn OFF command leaves one hanging (always the same one), but switching both hardware switches works fine.

I imagine staggering the OFF might fix it :D