zigpy/zha-device-handlers #1863: Tuyda TS0601 MmwRadarMotion does not support any configuration

aquemy, 2022-10-24:
Hi,

**Is your feature request related to a problem? Please describe.**

I have a Tuya TS0601 mmWave and light motion detector. It is paired properly and works OK out of the box.
However, there is no way to control anything that the sensor is supposed to expose (see [here](https://www.zigbee2mqtt.io/devices/TS0601_smart_human_presense_sensor.html)).

I have seen that there is an issue opened for the same sensor here https://github.com/zigpy/zha-device-handlers/issues/1850 but in my case, the quirk is properly loaded: 
```
TS0601 by _TZE200_ztc6ggyl 
...
Quirk: zhaquirks.tuya.ts0601_motion.MmwRadarMotion
```

**Describe the solution you'd like**
Being able to access the attributes/entities in such a way:
[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x000c",
        "0x0400",
        "0x0406",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_ztc6ggyl",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_motion.MmwRadarMotion"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Warsaw",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
[... 503 lines cut ...]
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

**Additional context**

It looks like the actual [quirk](https://github.com/zigpy/zha-device-handlers/blob/a652d86078fbf9e790b3c1067e7aa8401cd43474/zhaquirks/tuya/ts0601_motion.py#L351) is missing these features.    
I've never touched quirk and Zigbee dev but I'd be happy to participate if anyone can guide me.

TheJulianJES, 2022-10-25:
It's not ideal but you might be able to use the "manage clusters" dialog to change the attributes on one of these clusters at the moment:
- MmwRadarManufCluster
- TuyaOccupancySensing
- (TuyaAnalogInput)
- (TuyaIlluminanceMeasurement)

javicalle, 2022-10-25:
There are no stupid questions, but we can't always give the expected answers.
We are working on improving the interaction through the UI, but on some devices it is still quite limited, and some manufacturers do not help us in this regard.

The steps you have taken are correct. 
If the device does not accept the values, it is possible that the quirk requires some more adjustment.

Some people has been working in another implementation for this devices.
If you want to try, here is the implementation:
* https://github.com/zigpy/zha-device-handlers/issues/1645#issuecomment-1227366645

You will need to enable the local quirk in your instalation. There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Create a `ts0601_motion.py` in your local quirk folder and put the suggested quirk in the file.

Restart HA, remove and pair you device again. Check that the device is loading the quirk and that the device signature has changed.

TheJulianJES, 2022-10-25:
Also, make sure to leave the "manufacturer override code" field completely empty:
[image]

Only change the value in the upper field and then press "write attribute". You can check if it wrote successfully by checking "read attribute". But again, make sure the lower field is completely empty all the time.

sparkydave1981, 2022-10-28:
As I've mentioned in issue #1645 I also can't get the new values to stay set.  Hitting 'read value' shows the original/factory setting again.

aquemy, 2022-10-31:
I confirm that the custom quirk in #1645 works perfectly.

aquemy, 2022-12-13:
I am re-opening with a strange issue.

I now have `_TZE200_ztc6ggyl` but also `_TZE204_ztc6ggyl`. For some time the quirk specified in #1645 worked perfectly, so I adjusted the settings and forgot about the sensor that works perfectly since then,

Yesterday I added the new  `_TZE204_ztc6ggyl` to ZHA with the exact same quirk. I was surprised that I could not modify any setting although the entities/sliders were there. Whenever I tried to modify one of these sliders, it was reset to the original position after the sensor got triggered. I thought this was due to this particular model but I then tried to modify the setting of the other one ( `_TZE200_ztc6ggyl`) and got the exact same result.

[image]

What is crazy is that I can set the Sensitivity parameters, but none of the others. I clearly see that when I modify the sensitivity, the service is called (see below), but not for any other parameter.

See the logbook: we clearly see the service `number.set_value` triggered by myself, but the slider min range moved to the max did not trigger anything (and 2 sec later was reset to 0s).
[image]

My quirk for both:
```python
"""Tuya mmw radar occupancy sensor."""

import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaLocalCluster,
    TuyaNewManufCluster,
)
from zhaquirks.tuya.mcu import (
    TuyaDPType,
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

class TuyaMmwRadarSelfTest(t.enum8):
    """Mmw radar self test values."""
    TESTING = 0
    TEST_SUCCESS = 1
    TEST_FAILURE = 2
    OTHER = 3
[... 770 lines cut ...]
            },
            "unsupported_attributes": {
              "0x0000": {
                "attribute_name": "measured_value"
              }
            }
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

closed 2023-07-10 by github-actions[bot]