zigpy/zha-device-handlers #1865: [Device Support Request] Aqara door sensor recognized as switch

jclsn, 2022-10-25:
**Is your feature request related to a problem? Please describe.**
I have five Aqara door sensors of [this kind](https://zigbee.blakadder.com/Aqara_MCCGQ11LM.html). Four of them are recognized correctly, but for one the quirk is missing and it is recognized as a switch.

**Describe the solution you'd like**
I would like it to be recognized correctly.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x5f01",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0006",
        "0xffff"
      ],
      "out_clusters": [
        "0x0000",
        "0x0004",
        "0xffff"
      ]
    }
  },
  "manufacturer": "unk_manufacturer",
  "model": "lumi.sensor_magnet.aq2",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.10.0",
    "host_os": "Home Assistant OS 9.2",
    "docker_version": "20.10.17",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.27.2",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "openweathermap_all": {
[... 246 lines cut ...]
        },
        "out_clusters": {
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xffff": {
            "endpoint_attribute": "manufacturer_specific",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

TheJulianJES, 2022-10-25:
A manufacturer of ``unk_manufacturer`` generally means that the pairing didn't complete successfully. In that case, I'd remove the device from ZHA completely, then allow joins again (click on "Add devices") and hold down the reset button on the Aqara sensor until it re-joins the network.

javicalle, 2022-10-25:
And keep the device awake short pushing the button every second until pairing finish.

dmulcahey, 2022-10-25:
It was a bad pairing this is all set

closed 2022-10-25 by dmulcahey