zigpy/zha-device-handlers #1866: [Device Support Request] Add ExternalSwitchType property to tuya.ts130f.TuyaTS130Double_GP (TS130F _TZ3000_j1xl73iw)

BenPru, 2022-10-25:
**Is your feature request related to a problem? Please describe.**
I have a TS130F _TZ3000_j1xl73iw Girier Tuya Smart Blind 2 Gang. My problem is that they act as ExternalSwitchType.Momentary / button switch.
I connect it to my existing hardware switches that stays pressed.
Now I press my hardware switch for down, but nothing happens. Only after I push the hardware down switch to neutral position back, the tuya switch acts and shut down the blind. So it act as button switch.
I have to change the default config from ExternalSwitchType.Momentary to ExternalSwitchType.Toggle.
I have create a quirk with TuyaZBExternalSwitchTypeCluster with attribute external_switch_type (id: 0xd030) but it doesn't work.
I have other single blind switches TS130F _TZ3000_e3vhyirx which has no TuyaZBExternalSwitchTypeCluster but works fine with my hardware switches because they act as ExternalSwitchType.Toggle by default.

**Describe the solution you'd like**
Allow TuyaZBExternalSwitchTypeCluster with attribute external_switch_type (id: 0xd030) to change the external switch behavior.

<details>

<summary>Device signature</summary>

```yaml

  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_j1xl73iw",
  "model": "TS130F",
  "class": "zhaquirks.tuya.ts130f.TuyaTS130Double_GP"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
[... 610 lines cut ...]
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

Konstel, 2023-06-12:
what quirks did you use to get them to work? 

I tried with "ts130f.py" but it seems that the quirk is not loaded by HA for this device.

ErKoldun, 2023-06-18:
I've installed this switch and I'm having a lot of compatibility issues. I'm unable to write almost any of the cluster properties and device doesn't work most of the time. I don't know if it can be related with this issue (I'm pretty new in this)

sanchosk, 2023-11-30:
I bought this switch and it seems to be quite reliable, BUT I cannot do the calibration, so it only works for 10 seconds up/down and then stops. Any idea how can I set the number of seconds for full open/close?

closed 2024-06-04 by github-actions[bot]