zigpy/zha-device-handlers #1868: TS011F by _TZ3000_typdpbpg  smart plug shuts down goes offline

TommisGR, 2022-10-27:
**Describe the bug**
I have several TS011F by _TZ3000_typdpbpg  smart plugs that use zhaquirks.tuya.ts011f_plug.Plug
The work oK except from the issue on the title,
At random the go offline and shutdown but work normaly if u turn them on again via button of HA

**To Reproduce**
Steps to reproduce the behavior:
1. Add to zha
2. Leave it on for some time
3. They  turn off without automation or user access
4. Other times "became unavailable"

**Expected behavior**
When on to remain ON until action is called

**Screenshots**
[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_typdpbpg",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Athens",
[... 602 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

MattWestb, 2022-10-27:
Its 110% one undocumented future in the firmware (=bug).
You can read little more here https://github.com/Koenkk/zigbee2mqtt/issues/11648.

cloudbr34k84, 2022-11-05:
If you have a tuya zigbee hub just add it to that, then run the firmware upgrade. then re add to HA. Works for me

KaherdinTristan, 2022-11-07:
And if I only have a Conbee 2 and running ZHA, how can I update tuya devices? 👍 👎 ?

> If you have a tuya zigbee hub just add it to that, then run the firmware upgrade. then re add to HA. Works for me

MattWestb, 2022-11-07:
@KaherdinTristan Grab your firmware (if you is finding one) and configure OTA updates in ZHA https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates.

closed 2023-07-10 by github-actions[bot]