zigpy/zha-device-handlers #1870: [Device Support Request]

kimme1024, 2022-10-28:
**Is your feature request related to a problem? Please describe.**
I've bought a garage door opener (_TZE200_wfxuhoea) which isn't supported in ZHA.
I managed to get it working by using the quirk shared in the following issue: https://github.com/zigpy/zha-device-handlers/issues/1260 by changing the name in the quirk.
This way I can use the garage door opener as intended and I also can retrieve the status of the door sensor.

However there are 2 huge issues that make the system unusable:
1) On multiple times of the day (sometimes hourly) it sends an "open"-command trough HA. There aren't any automations that may cause this, and even when I disconnect the device from power, HA still tries to open it. I guess this is related to the unadapted quirk.
2) When the power has been cut for some time, the relay is triggered when the device is plugged in again. So when the power recovers after a power outage, the garage door will open.

Obviously those 2 issues make the system completely unsafe as the garage door opens constantly.

**Describe the solution you'd like**
I guess the quirk has to be adapted to this specific device to be able to work correctly.

<details>
```
<!-- Device signature can be acquired by clicking on the "Zigbee Device Signature" button in the device settings view -->
<summary>Device signature</summary>
```

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0500"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_wfxuhoea",
  "model": "TS0601",
  "class": "ts0601_garage.TuyaGarageSwitchTO"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
[... 442 lines cut ...]
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
In the attached image you can see that the device is turned on (ingeschakeld) automatically at 1:47am and again at 2:47. At 1:47 I woke up from the sound of the garage door. I went down and disconnected the device from the mains power. However at 2:47 it got triggered again. Obviously the garage didn't open but this means that HA causes it to open, not the device itself. As there are no automations running to open the garage door, the only thing that may cause this is the quirk (I suppose).
[image]

javicalle, 2022-10-28:
The quirk from #1260 has not been merged.
Even no PR created for the quirk, so better to ask for support in the issue than splitting the info in other issue.

kimme1024, 2022-10-28:
> The quirk from #1260 has not been merged.
> Even no PR created for the quirk, so better to ask for support in the issue than splitting the info in other issue.

As I’m having another device than referred to in issue #1260 I’ve made this new one so I won’t interfere with the support of the other device.

I only referred to #1260 as the solution there already works partially. This way it might save some time.

crydante, 2022-11-08:
I'm waiting too for the support of my _TZE200_wfxuhoea Garage Door Opener

LarrxX, 2022-11-08:
Same problem, same device. Works correctly with Zigbee2MQTT though...

MyKapacidade, 2023-03-31:
Any new?

closed 2023-10-04 by github-actions[bot]