zigpy/zha-device-handlers #1872: [Device Support Request] SmartThings Power Outlet UK reports some values incorrectly

jamesgreeley, 2022-10-30:
**Is your feature request related to a problem? Please describe.**

[SmartThings Power Outlet UK](https://zigbee.blakadder.com/SmartThings_F-APP-UK-V2.html) F-APP-UK-V2 reports incorrect values

The entities reporting incorrect values appear to be (plug named smart_thing_plug_1_dehumidifier)

sensor.smart_thing_plug_1_dehumidifier_electricalmeasurement
sensor.smart_thing_plug_1_dehumidifier_electricalmeasurementrmscurrent

electricalmeasurement in watts appears to be roughly 10x higher than the real value
dehumidifier_electricalmeasurementrmscurrent is out by a much larger factor.

**Describe the solution you'd like**
Correct values to be reported

[image]

<details>

<summary>Device signature</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4174, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0b04",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "CentraLite",
  "model": "3200-Sgb",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.10.0",
    "host_os": "Home Assistant OS 9.3",
[... 653 lines cut ...]
<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.3
-- | --
update_channel | stable
supervisor_version | supervisor-2022.10.0
agent_version | 1.4.1
docker_version | 20.10.18
disk_total | 30.8 GB
disk_used | 19.4 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok

bmccluskey, 2022-12-05:
I have seen this issue with devices on the same firmware version 0x30015310. The 2 devices I have on the same version report approx 10x the real value for electrical measurement.  I have 4 other devices using version 0x30065310 (Note the 6 instead of 1) that are reporting the value correctly.

Im attempting to upgrade the firmware via Smartthings. Well let you know if I have any success.

TheJulianJES, 2023-01-05:
So for the power consumption, we could just make a quirk that overrides the ``AC_POWER_DIVISOR`` on the ``ElectricalMeasurement`` cluster to ``10``.

However, I'm not sure by what value the current is off. It looks like it should be ``1,263 A * 241 V = 304.4 W``

closed 2023-07-11 by github-actions[bot]