zigpy/zha-device-handlers #1875: [BUG] Mijia Honeywell Gas Detector (JTQJ-BF-01LM/BW) gas density and detector sensitivity support

avbor, 2022-10-31:
Hi.
Is any way to add gas density sensor and detector sensitivity switch for this device?
z2m and some other software already supported this. https://www.zigbee2mqtt.io/devices/JTQJ-BF-01LM_BW.html

For now i have only binary_sensor for gas and sensor for device temperature (which shows complete nonsense):
[image]

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x000a",
        "0x0500"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.sensor_natgas",
  "class": "zigpy.device.Device"
}
```

cvocvo, 2023-03-23:
Were you able to get this working? deCONZ also seems to support it but ZHA would be ideal: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1182

avbor, 2023-03-23:
> Were you able to get this working? deCONZ also seems to support it but ZHA would be ideal: [dresden-elektronik/deconz-rest-plugin#1182](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1182)

Yeah, I use this sensor through https://github.com/AlexxIT/XiaomiGateway3 integration.
But, it would be desirable through zha...

cvocvo, 2023-03-23:
Thanks for the quick reply; unfortunately I don't have that gateway hardware, but I've had one of these gas detectors in the box for a few years. (I bought it but couldn't ever get it configured with the app.) It only recently dawned on me to see if it was Zigbee. A binary sensor may be better than nothing; I'll give it a go with ZHA later and see what I find.

avbor, 2023-03-23:
There is also a problem with this binary sensor - after detecting gas it never turns off again. only removing and adding the sensor to zha helps to reset it.

avbor, 2023-03-23:
So now he is absolutely useless in the zha...

cvocvo, 2023-03-23:
> There is also a problem with this binary sensor - after detecting gas it never turns off again. only removing and adding the sensor to zha helps to reset it.

Ahh that's non-ideal. There's some complex way to evidently change the state of a binary sensor (https://community.home-assistant.io/t/how-to-manually-set-state-value-of-sensor/43975).

Do you know if the device itself reports the sensor state as 'ON' basically forever? Or is it an issue on the HA end? Sort of a dumb work around would be that if it is ever triggered, after some period of time I could reset the sensor state to 'OFF'.

avbor, 2023-03-23:
This is definitely a problem on the zha side, but I have never been able to reset the sensor state by hand.

user514, 2023-06-06:
Would like to comment that I have the same exact problem described by avbor. No density value on this sensor under ZHA (I have it ok under zigbee2mqtt). Also, the thing about temperature which set always at 0.4ºC which is absurd, of course. Nothing about temperature on this sensor.

The only difference is that today, june-6-2023, I have had a gas detection  for some seconds, and it cleared perfectly fine, so it is working for me:

[image]

 ...not as commented, that the sensor kept on 'Detected' state forever. Not to me.

  So from now on, I can use it with automations to alert on trigger by the 'Clear' vs 'Detected' state. It will be better to extract from ZHA the density value as we did with zigbee2mqtt, to have more precise information.

  Just to let you know.

closed 2023-12-10 by github-actions[bot]