zigpy/zha-device-handlers #1876: [BUG] Tuya TS601 Saswell TRV resets its heating setpoint to 16°C

jclsn, 2022-10-31:
**Describe the bug**
I have the above mentioned TRV and the issues that it automatically sets its heating setpoint to 16°C. I also had this issue with Zigbee2MQTT and could manage it there by setting the setpoint to 30°C every time I turn them on. This does unfortunately not work with ZHA. I am not sure whether this is a ZHA issue or an issue with the TRV itself.

**To Reproduce**
Steps to reproduce the behavior:
1. Add the above mentioned TRV with ZHA
2. Set the HVAC mode to "heat" and setpoint to 30°C
3. The setpoint sets itself back to 16°C

**Expected behavior**
It shouldn't reset the setpoint.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0201",
        "0xef00"
      ],
      "out_clusters": [
        "0x0001",
        "0x000a",
        "0x0019",
        "0x0201"
      ]
    }
  },
  "manufacturer": "_TZE200_c88teujp",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_trv_sas.Thermostat_TZE200_c88teujp"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.10.0",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
[... 350 lines cut ...]
                "value": 2170
              },
              "0x0012": {
                "attribute_name": "occupied_heating_setpoint",
                "value": 3000
              },
              "0x001c": {
                "attribute_name": "system_mode",
                "value": 0
              },
              "0x001e": {
                "attribute_name": "running_mode",
                "value": 0
              }
            },
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

javicalle, 2022-11-01:
Duplicate of #1815

jclsn, 2022-11-19:
@javicalle Yes, it is a duplicate. Very annoying issue. Hoping it will get fixed soon!

javicalle, 2022-11-21:
Fixed by: https://github.com/zigpy/zha-device-handlers/pull/1816

Will be included in next version of HA.

closed 2022-11-21 by javicalle