zigpy/zha-device-handlers #1880: Aqara plug not updating consumption and power reading out 10x

mynameisdaniel32, 2022-11-01:
I think I'm seeing the same issues worked through here, but maybe with a different variant of the plug?
https://github.com/zigpy/zha-device-handlers/issues/1337

[image]

I'm using Home Assistant 2022.10.5 and have removed and repaired this device following the advice from the thread.

I'm in Australia using a version of this plug and this is how it's populated sensors:
[image]

The power reading is 10x what it actually is (or I accidentally bought a 20kW heater...). Using this tip fixes that but is lost again after pairing:
https://github.com/zigpy/zha-device-handlers/issues/1337#issuecomment-1230276786

I'm also not getting any consumption measurements.

Please let me know if any other info is needed.

Bucolo, 2022-11-01:
I have a similar issue, I have an aquara plug MAEU01 with the new firmware 0x00000029. It was working fine before, but since HA 2022.10.5, the consumption graph is less detailed and after a new pairing, the consumption is not updated. I installed a zha custom quirk with version 0.0.82 and it's working properly as before. However there is still an issue with the temperature that display an incorrect value.

TheJulianJES, 2022-11-03:
@mynameisdaniel32 The fix was indeed only for EU plugs. I can't easily test stuff with Australian/EU plugs, as I don't have them and they seem to work differently compared to the EU plugs.
The "10x reading fix" randomly "disappearing" should not happen at all. (If it does, it's likely not a bug in zha-quirks but rather zigpy, as it's a LocalDataCluster with a fake attribute).
It's possible that the plug has multiple different signatures causing this though.

Can you remove the plug from ZHA, pair it again, and then check if the quirk applies (and if the correct consumption is correct)?
Do that multiple times please. If, after pairing, the device doesn't pick-up the quirk, please upload the signature here.

@Bucolo That's a different issue. Please remove all custom quirks and update to Home Assistant 2022.11.0 or later.
After removing the quirks and having updated HA, remove and then pair the plug again.
If there are any issues with the power consumption then, please create a new issue on this repo (fill out the template) and tag me.

Device temperature occasionally seems to be reported incorrectly by the plug. I've considered removing it. This is not a bug in zha quirks.

MattWestb, 2022-11-03:
Xaomi/Aqara device temperature (not thermometer sensor temperature) was removed then it was false added ads one temperature sensor in the system and normally is very un-acculturate.

By user request it was added back as one diagnostic temperature sensor with the option its its **very** bad reading from more or less all devices but its normally working but its not one ZHA bug its one device future that is not good implemented in the device and cant being fixed in ZHA.

So the option is removing it for all devices from the ZHA and making many user sad or accepting you is getting what you is baying from China.

TheJulianJES, 2022-11-03:
Yeah, it's at least marked as "device temperature" here, but the issue isn't that it's only an inaccurate sensor for these plugs, but rather that the reported values are often just ~25°C or 0°C (see the drop after 20:00):
[image]

TheJulianJES, 2023-01-04:
Similar to https://github.com/zigpy/zha-device-handlers/issues/1135

TheJulianJES, 2023-02-04:
Should be fixed with 2023.2.0. You likely need to remove the device from ZHA, wait a minute, and then re-pair the device.
If it doesn't work, let me know.

closed 2023-02-04 by TheJulianJES