zigpy/zha-device-handlers #1881: [Device Support Request] CH/Socket/2 by Schneider Electric

woodsb02, 2022-11-01:
**Is your feature request related to a problem? Please describe.**
Device is a dual electrical power socket, and is Zigbee 3.0 certified.
Whilst it joins the network it shows the domain types of the 2 switches as light when it should be switch (any electrical device could be connected to this socket, not just lights).

Manufacturer: Clipsal by Schneider Electric
Model: 3025CSGZ
Name: Wiser Iconic Connected Socket
Website: https://www.clipsal.com/products/detail?CatNo=3025CSGZ&itemno=3025CSGZ

**Describe the solution you'd like**
Please change device type to switch.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4190, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000f",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000f",
        "0x0b05"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Schneider Electric",
  "model": "CH/Socket/2",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>
[... 602 lines cut ...]
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```

```

</details>

**Additional context**
This quirk has been added to Zigbee2MQTT here:
https://github.com/Koenkk/zigbee-herdsman-converters/commit/f62e9b2b413778e60b9c525685c4b9c0621ff67c

The Zigbee2MQTT project documentation on this device is here:
https://www.zigbee2mqtt.io/devices/3025CSGZ.html

woodsb02, 2022-11-01:
This may be best integrated with #1705 or added after that lands?

axellebot, 2022-11-06:
Feel free to contribute to the PR or create your own PR if you already have a working code.

closed 2023-05-12 by github-actions[bot]