zigpy/zha-device-handlers #1884: [Device Support Request] Configurable Z-Stack Router firmware radio `transmit_power`

ruimarinho, 2022-11-02:
**Is your feature request related to a problem? Please describe.**
I'm not entirely sure this requires a quirk, but Koenkk has just released a version of the Z-Stack router firmware with support for remotely setting the CC2652P radio amplifier gain (https://github.com/Koenkk/Z-Stack-firmware/commit/3d43a50acdc09d6ff12605de4a5d104389c69e85#diff-f4872ebd913c23b0dc0d04319df88a7d1d21179f1ef005e0b9d5500f890fef0fR186-R201). It is part of the basic cluster.

**Describe the solution you'd like**
It would be great to be able to set the transmit power of the radio via a slider input.

TheJulianJES, 2022-11-03:
This should be available in the December HA 2022.12.0 release through the "Manage Zigbee Device" -> "Clusters" UI.
It's possible that a config entity in ZHA could also be added, but there are some things to consider with older firmwares.

If you need it now, you should be able to add the ``zhaquirks/texasinstruments/router.py`` file from https://github.com/zigpy/zha-device-handlers/pull/1886 as a custom quirk. But please remove it when the change is available in HA (but that'll still take some time). I'll try to update this issue when it gets merged into HA.

Hedda, 2022-11-04:
FYI, the premium version of PTVO Zigbee Configurable Firmware has the same function but via its ”RF signal level” sensor -> https://ptvo.info/zigbee-configurable-firmware-features/rf-signal-level-control/

> [RF signal level control](https://ptvo.info/zigbee-configurable-firmware-features/rf-signal-level-control/)
> 
> The firmware can control RF signal strength for all chips with a power amplifier. This feature allows you to find tune your mesh network and exclude situations when a device constantly jumps from one router to another because a signal level from both router is almost identical.
> 
> The firmware also allows you to control signal level remotely in run-time.
> 
> ⚠️ THIS FEATURE IS AVAILABLE IN THE [PREMIUM VERSION](https://ptvo.info/zigbee-configurable-firmware-features/premium/)
> 
> To control signal level, you should add the “RF signal level” sensor on any output, using any [virtual pin](https://ptvo.info/zigbee-configurable-firmware-features/gpio/#virtual). The “Signal strength level” parameter defines an initial value after flashing or resetting, where 0 is the lowest signal level and 20 is the highest signal level. Please note, this number is not a precise number of signal power in dBm.
> 
> RF signal level configuration
> To control signal strength remotely, you should write the corresponding level value to the configured endpoint (e.g., output 5 will work on endpoint 5).
> 
> ⚠️ Please be careful because this action has an immediate effect, and your device may become unresponsive if a signal strength is too small.
> 
> You can also create a [custom converter](https://ptvo.info/zigbee-configurable-firmware-features/expert/) for Z2M on the “Expert” tab that adds the corresponding option field to exposes.
> 
> The firmware does not send the configured signal strength value with periodic reports because this value is constant. If you need to get this value, you may use the “Refresh” button in exposes or the “get” via MQTT.
> 
> Zigbee2MQTT commands
```
read – immediately reads the signal strength value

Topic: zigbee2mqtt/[friedly_name]/get/[channel]
Channel: l1, l2, l3 … l16
Payload: 1
Returns the channel state: {"[channel]": value}
write – writes the signal strength value

Topic: zigbee2mqtt/[friedly_name]/set
Payload: {"[channel]": value}
Channel: l1, l2, l3 … l16
```

closed 2022-11-04 by dmulcahey