zigpy/zha-device-handlers #1885: [BUG] Zigbee devices stopped working after HA update

YewTreeWeb, 2022-11-03:
Hi,

I upgraded my Home Assistant to version 2022.11.0 and now all my Zigbee devices have stopped working. I’m using a Combee II and after unplugging it I got ZHA to reload as that originally had the status of failed. Once reloaded if I try and turn off or on my devices I get the following errors:

[0x0503:11:0x0300]: async_initialize: all attempts have failed: [TimeoutError(), DeliveryError('Failed to deliver packet: <TXStatus.MAC_CHANNEL_ACCESS_FAILURE: 225>'), DeliveryError('Failed to deliver packet: <TXStatus.MAC_CHANNEL_ACCESS_FAILURE: 225>'), DeliveryError('Failed to deliver packet: <TXStatus.MAC_CHANNEL_ACCESS_FAILURE: 225>')]
[0x33F1:1:0x0406]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver packet: <TXStatus.MAC_CHANNEL_ACCESS_FAILURE: 225>'), DeliveryError('Failed to deliver packet: <TXStatus.MAC_CHANNEL_ACCESS_FAILURE: 225>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>')]
[0x2BFB:11:0x0008]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver packet: <TXStatus.MAC_CHANNEL_ACCESS_FAILURE: 225>'), TimeoutError(), DeliveryError('Failed to deliver packet: <TXStatus.MAC_CHANNEL_ACCESS_FAILURE: 225>'), DeliveryError('Failed to deliver packet: <TXStatus.MAC_CHANNEL_ACCESS_FAILURE: 225>')]
[0xF7F2:1:0x0102]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver packet: <TXStatus.MAC_CHANNEL_ACCESS_FAILURE: 225>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.MAC_CHANNEL_ACCESS_FAILURE: 225>')]
[0x1E20:1:0x1000]: Couldn't get list of groups: Failed to deliver packet: <TXStatus.MAC_CHANNEL_ACCESS_FAILURE: 225>

Everything was working fine before upgrading to 2022.11.0.

Any help will be great. Thanks

dmulcahey, 2022-11-03:
MAC_CHANNEL_ACCESS_FAILURE Has nothing to do with the upgrade. It means you have significant interference and the zigbee stick can’t communicate. Ensure your coordinator is not near an SSD, Make sure it is on an extension cord and plugged into usb 2 port if possible.

YewTreeWeb, 2022-11-03:
Hi dmulcahey, thanks for commenting. I appreciate your feedback. I have already done those things and as I mentioned in my post everything was working until I upgraded. I have now rolled back to 2022.10.5 without changing anything and my ZHA is working again. So I can only think it must be the upgrade to 2022.11.0 as why was it broken with the new version and works with 10.5?

MattWestb, 2022-11-03:
`TXStatus.MAC_CHANNEL_ACCESS_FAILURE` = the coordinator is trying to sending but the radio channel is blocked (Zigbee device is **not** allowed sending if some other device is sending and is first listening before try sending, therefor you is getting it the `TXStatus`).

If your coordinator is supporting radio scan to see how its looking in the air from the radio view and also one WiFi scan with your mobile with one app and you can  see how its in the real.  One thing WiFi and Zigbee channels numbers is not the same so need matching frequencies if its overlapping.

YewTreeWeb, 2022-11-04:
> `TXStatus.MAC_CHANNEL_ACCESS_FAILURE` = the coordinator is trying to sending but the radio channel is blocked (Zigbee device is **not** allowed sending if some other device is sending and is first [...]

Hi MattWebstb, thank you for your reply and that is interesting. I have now changed my Wifi channels as I noticed my Zigbee channel is 11. 

But can you see why I’m thinking it’s been caused by the update when it only started happening after updating and now I’ve rolled back to 10.5 it’s working again?

MattWestb, 2022-11-04:
If Zigbee is on 11 (the lowest channel) your WiFi shall being oi the middle and upper part of the band for not blocking the Zigbee coordinator.

Also downloading one WiFi scanner for your mobile is good then can see if you have some neighbors spamming the air.I cant telling if its somthing in the code that is triggering the log message but if its true sent from the coordinator its what is see in the "air" and if the radio channel is blocked it cant sending any thing and we cant forcing it to do (its the standard for this devices and is one must for certifying the Zigbee stack that is controlling all things in the device).

TheJulianJES, 2023-01-04:
Closing, as ``MAC_CHANNEL_ACCESS_FAILURE`` is a not a zha-quirks issue.

If you still need support, I'd recommend joining the [HA Discord](https://discord.com/invite/home-assistant) and asking in the #zigbee channel.
Otherwise, you can also try the forums but they're not as actively monitored by the devs.

closed 2023-01-04 by TheJulianJES