zigpy/zha-device-handlers #1888: [BUG] Aqara Smart plug (maeu01) doesn't show energy usage

WilliamHusum, 2022-11-04:
**Describe the bug**
The smart plug doesn't show the energy usage in Home Assistant. The switch works though.
I have tried resetting and re-adding maybe 10 times now.
[image]

**To Reproduce**
Unknown

**Screenshots**
[image]

**Signature**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug.maeu01",
  "class": "zhaquirks.xiaomi.aqara.plug_eu.PlugMAEU01"
}
```
@TheJulianJES

TheJulianJES, 2022-11-04:
What firmware are you running on your Conbee? (and try updating it)

WilliamHusum, 2022-11-04:
Hi, thanks for the quick response. 

Updating the Conbee seemed to work, thanks! I was using an old version.

closed 2022-11-04 by WilliamHusum