zigpy/zha-device-handlers #1889: [BUG] Schneider Exxact Wiser outlet reporting W & Wh a factor 1000 off

ptrrkssn, 2022-11-04:
**Describe the bug**
Schneider Exxact Wiser outlet seems to report W and Wh a factor 1000 off (or HA/ZHA seems to think so at least). Ie a 900W load is reported as 900kW

**To Reproduce**
Steps to reproduce the behavior:
Plug in an 600-900W electrical element into the outlet, check the power usage in Home Assistant for the device...

**Expected behavior**
W & Wh numbers displayed a factor 1000 less.. In the screenshot below it should be 881W not 881840W...

**Screenshots**
[image]

<details>

<summary>Device signature</summary>

```yaml
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
32
33
34
35
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4190, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "6": {
      "profile_id": 260,
      "device_type": "0x0009",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0708",
        "0x0b04",
        "0x0b05",
        "0xfc04"
[... 803 lines cut ...]
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

javicalle, 2022-11-04:
Can you try to read from the cluster `MeteringCluster (Endpoint id: 1, Id: 0x0702, Type: in)` the attributes `divisor (id: 0x0302)` and `multiplier (id: 0x0301)`?

You can do it from the device view -> Admin Zigbee device --> Clusters 
Select the cluster and attribute and push the "get attribute" button.

Once you get the values try to increasse the divisor value x1000.

ptrrkssn, 2022-11-04:
Tried it now - divisor was set to 1000. multiplier to 1. Tried to change it but it seems to just revert back to the same values.

javicalle, 2022-11-05:
My bad, the attributes are read only:
[image]

It seems extrange to me because divisor already has the 1000 divisor value. Which Zigbee gateway are you using?

ptrrkssn, 2022-11-05:
I'm using a ConBee II with the latest available firmware (0x26780700). (Connected via an extension cable to an Intel NUC running the updated HassOS/HA :-)

HA 2022.11.1
SUP 2022.10.2
OS 9.3
Frontend 20221102.1

javicalle, 2022-11-05:
Have you tried to remove and pair the device again?
It seems that the value is already here so no clear to me if the quirk would work.

ptrrkssn, 2022-11-05:
The device (devices actually, I have 3 of them, one I've had some time, and two new ones I just got) was recently paired, but I guess I can try to re-pair it.

javicalle, 2022-11-05:
Is the same behavior in all three devices?

ptrrkssn, 2022-11-05:
Yes, same on all three.

javicalle, 2022-11-06:
I still believe that the problem must be elsewhere but if you want to try, there is a quirk for your device:
<details open>
<summary>schneider_plug.py</summary>

```python
"""Schneider Electric plugs quirks."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
)
from zigpy.zcl.clusters.homeautomation import Diagnostic, ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import DeviceManagement, Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class MeteringCluster(CustomCluster, Metering):
    """Fix the Metering divisor."""

    MULTIPLIER = 0x0301
    DIVISOR = 0x0302
    _CONSTANT_ATTRIBUTES = {MULTIPLIER: 1, DIVISOR: 1000000}

class SocketOutlet2(CustomDevice):
    """Schneider Electric Socket outlet 2."""

    signature = {
        MODELS_INFO: [("Schneider Electric", "SOCKET/OUTLET/2")],
        ENDPOINTS: {
            6: {
                # "profile_id": 260,
                # "device_type": "0x0009",
                # "in_clusters": ["0x0000","0x0003","0x0004","0x0005","0x0006","0x0702","0x0708","0x0b04","0x0b05","0xfc04"],
                # "out_clusters": ["0x0019"]
                PROFILE_ID: 0x0104,
                DEVICE_TYPE: zha.DeviceType.MAIN_POWER_OUTLET,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    DeviceManagement.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    Diagnostic.cluster_id,
                    0xFC04,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            242: {
                # "profile_id": 41440,
                # "device_type": "0x0061",
                # "in_clusters": [],
                # "out_clusters": ["0x0021"]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            6: {
                PROFILE_ID: 0x0104,
                DEVICE_TYPE: zha.DeviceType.MAIN_POWER_OUTLET,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    MeteringCluster,
                    DeviceManagement.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    Diagnostic.cluster_id,
                    0xFC04,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

You will need to enable the local quirk in your instalation. There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Create in your local quirk folder a new `schneider_plug.py` file and put the suggested quirk in file.

Restart HA, remove and pair you device again. 
Check that the device is loading the quirk and that the device signature has changed.

javicalle, 2022-11-06:
Maybe related to: #1705

ptrrkssn, 2022-11-07:
With that local quirk installed and the devices re-paired the Instantaneous Demand (W) figure is right (881W), however I wonder if not the Summation Delivered now becomes a factor 1000 too low instead - which was correct before. 

I wonder if we're chasing the wrong thing and perhaps it's the ZCL_ATTR_METERING_SUMMATION_FORMATTING and ZCL_ATTR_METERING_DEMAND_FORMATTING attributes that isn't used correctly somewhere? I'm no Zigbee expert so this might be totally wrong but...

Summation formatting has the value "bitmap8.64|8|2|1" and Demand Formatting has "none" when I try to read them.

If I read the zcl metering cluster documentation correctly that should mean 01001011 = 
0 = do not suppress leading zeros
1 = ?
001 = 1 digit to the right of decimal point
011 = 3 digits to the left of the decimal point
 
The instantaneous_demand attribute value is 873311 and current_summ_delivered is 1200 right now.

A reasonable interpretation of those numbers at 873.311 W and 1.2 kWH (seems reasonable since I re-pair:ed it)

javicalle, 2022-11-07:
> With that local quirk installed and the devices re-paired the Instantaneous Demand (W) figure is right (881W), however I wonder if not the Summation Delivered now becomes a factor 1000 too low instead - which was correct before.

That makes sense because `instantaneous_demand` and `current_summ_delivered` are afected by the same multiplier and divisor:
* https://github.com/zigpy/zigpy/blob/8a2c7548065367d7e94ddf5857652ca1cd087e70/zigpy/zcl/clusters/smartenergy.py#L190-L191

> If I read the zcl metering cluster documentation correctly that should mean 01001011 =
> 0 = do not suppress leading zeros
> 1 = ?
> 001 = 1 digit to the right of decimal point
> 011 = 3 digits to the left of the decimal point

According to (i.e.) ZHA:
* https://github.com/home-assistant/core/blob/bc146a09dba6df6eaec429039f6e1efcfbd1123f/homeassistant/components/zha/core/channels/smartenergy.py#L194-L200

That would be like (but don't sure because not make sense to me with the result 🤷🏻‍♂️):
0 = do not suppress leading zeros
1001 = 7 digits to the **left** of decimal point (bits 3-6)
011 = 3 digits to the **right** of the decimal point (bits 0-2)

There are a few examples here:
* https://zigbeealliance.org/wp-content/uploads/2019/11/docs-07-5356-19-0zse-zigbee-smart-energy-profile-specification.pdf#page=236&zoom=100,90,402

So, do you think we could have fixed it by just setting the `demand_formatting`?

Maybe with a little of trial and error we can get the right 'format'.

You can try with:
```python
class MeteringCluster(CustomCluster, Metering):
    """Fix the Metering divisor."""

    MULTIPLIER = 0x0301
    DIVISOR = 0x0302
    DEMMAND_FORMATTING = 0x0304

    # _CONSTANT_ATTRIBUTES = {MULTIPLIER: 1, DIVISOR: 1000000}
    _CONSTANT_ATTRIBUTES = {DEMMAND_FORMATTING: 75}
```

ptrrkssn, 2022-11-08:
Hmm.. Nope, doesn't work. It seems to ignore the demand_formatting stuff. I've been reading the core/homeassistant/components/zha/sensor.py code a bit to try to understand what's happening. A couple of observations:

class SmartEnergySummation, starting at line 472, unit_of_measure_map: 0x00 = ENERGY_KILO_WATT_HOUR
...
   def formatter(self, value: int) -> int | float:
        """Numeric pass-through formatter."""
        if self._channel.unit_of_measurement != 0:
            return self._channel.summa_formatter(value)

        cooked = float(self._channel.multiplier * value) / self._channel.divisor
        return round(cooked, 3)

class SmartEnergyMetering, starting at line 424, unit_of_measure_map: 0x00 = POWER_WATT
...
    def formatter(self, value: int) -> int | float:
        """Pass through channel formatter."""
        return self._channel.demand_formatter(value)

Notice here: no cooked-stuff with the multiplier/divisor! And unit_of_measurement is 0x00 so for Summation it will execute the cooked stuff and ignore the channel.summa_formatter, but for the Instantaneous Demand stuff it will call the demand_formatter.

Looking at the formatter code in smartenergy.py, line 213, it seems to assume that unit_of_measurement always means kW:

    def _formatter_function(
        self, selector: FormatSelector, value: int
    ) -> int | float | str:
        """Return formatted value for display."""
        value_float = value * self.multiplier / self.divisor
        if self.unit_of_measurement == 0:
            # Zigbee spec power unit is kW, but we show the value in W
            value_watt = value_float * 1000
            if value_watt < 100:
                return round(value_watt, 1)
            return round(value_watt)
        if selector == self.FormatSelector.SUMMATION:
            assert self._summa_format
            return self._summa_format.format(value_float).lstrip()
        assert self._format_spec
        return self._format_spec.format(value_float).lstrip()

This feels a bit wrong and I think that multiplication of value_float -> value_watt by 1000 should be removed, or the value sent to it divided by 1000 in sensor.py. I'm no Python programmer so I'm a bit unsure how to implement a quirk that overrides this though...

ptrrkssn, 2022-11-15:
I modified the quirk with this and then re-pair:ed the outlet and now I get sane values for both Instantaneous Demand and Summation Delivered...

However I still think that the code in core/homeassistant/components/zha/sensor.py at line 424 is suspect and possibly should be fixed similar to the code at line 472.

```
class MeteringCluster(CustomCluster, Metering):
    """Fix the Instataneous Demand value."""

    INSTANTANEOUS_DEMAND = 0x0400

    def _update_attribute(self, attrid, value):
        if attrid == self.INSTANTANEOUS_DEMAND:
            value = value / 1000.0
        super()._update_attribute(attrid, value)
        

```

dmulcahey, 2022-11-15:
What are you looking at in ZHA? https://github.com/home-assistant/core/blob/7614aba401ffd4041d9dac09344d37b7525c4685/homeassistant/components/zha/sensor.py#L424

ptrrkssn, 2022-11-15:
Yes, my code is from the quirk we discussed previously in this thread. I just removed the code we were testing there.

But what I suspect is the real problem (also written about above) is in core/homeassistant/components/zha/sensor.py 
where:

class SmartEnergySummation, starting at line 472, unit_of_measure_map: 0x00 = ENERGY_KILO_WATT_HOUR
...

   def formatter(self, value: int) -> int | float:
        """Numeric pass-through formatter."""
        if self._channel.unit_of_measurement != 0:
            return self._channel.summa_formatter(value)

        cooked = float(self._channel.multiplier * value) / self._channel.divisor
        return round(cooked, 3)

but:

class SmartEnergyMetering, starting at line 424, unit_of_measure_map: 0x00 = POWER_WATT
...

    def formatter(self, value: int) -> int | float:
        """Pass through channel formatter."""
        return self._channel.demand_formatter(value)

Notice here: no cooked-stuff with the multiplier/divisor! And unit_of_measurement is 0x00 so for Summation it will execute the cooked stuff and ignore the channel.summa_formatter, but for the Instantaneous Demand stuff it will call the demand_formatter. So perhaps add the same " if self._channel.unit_of_measurement != 0:" test to the this code? 

I'd test this on my HA installation but I don't know how to do that since I'm running HassIO on my system.

dmulcahey, 2022-11-15:
No the quirk should be setting the correct values for multiplier and divisor.

dmulcahey, 2022-11-15:
The logic is all in the channel: https://github.com/home-assistant/core/blob/ae7272575a30e73fc9d6cecb0217cd15a4ba0d59/homeassistant/components/zha/core/channels/smartenergy.py#L230

ptrrkssn, 2022-11-15:
> No the quirk should be setting the correct values for multiplier and divisor.

Sure, but how? If we set divisor to 1000 (which it already is by default from the device btw) then Summation is correct, but Instantaneous becomes 1000 times to high. If we set divisor to 1000000 then Instantaneous is correct, but Summation gets 1000 times too low...

javicalle, 2022-11-15:
I believe that there is something at least 'extrange' to me.

In Zigbee specification I can read:
```
D.3.2.2.4Formatting
The following set of attributes provides the ratios and formatting hints required to transform the
received summations, consumptions, temperatures, or demands/ rates into displayable values.
If the Multiplier and Divisor attribute values are non-zero, they are used in conjunction with the
SummationFormatting, ConsumptionFormatting, DemandFormatting, and
TemperatureFormatting attributes.
```

As I understand, if there are a multiplier/divisor it aplies to all measurements.

But, in ZHA there is something diferent in how is applied in `SmartEnergyMetering` and `SmartEnergySummation`. Shouldn't it be applied in the same way in both cases?

dmulcahey, 2022-11-15:
Full transparency I responded the first time to just the last comment without reading the thread. Let me look through the entire thing later tonight to see what the deal is.

ptrrkssn, 2022-11-15:
I wish I had some other zigbee devices with power metering that provides these attributes, but my only other (zigbee) device is an Aqara Plug, and it doesn't support them (at least not at the firmware level that one is at :-). Would have been interesting to compare things :-)

dmulcahey, 2022-11-15:
I may have some that support both. I’ll try to investigate tonight/tomorrow depending on availability

javicalle, 2022-11-15:
> I'd test this on my HA installation but I don't know how to do that since I'm running HassIO on my system.

You can copy all the zha folder as a `custom_components` in HA and edit the desired class.

krlssn, 2022-11-16:
Hello.
I have installed 2 WDE002172 (Wiser 1 way socket), same as @ptrrkssn but with one socket. Exactly the same problem. Following this and hope you find a solution, if I can help with anything please tell me. Home assistant on NUC, Conbee II stick, FW on my sockets is 0x020612ff
Picture below is with washing machine, even I think kWh is correct and W is wrong.  
[image]

krlssn, 2022-11-16:
I'm sorry, don't understand what info you want. The problem we have is that the sensor named Instant... is reporting energy x1000, in my example above the correct value should be 2053,132W or even better 2053W.

javicalle, 2022-11-16:
@krlssn please, attach the device diagnostic with a screenshoot for the current values.

dmulcahey, 2022-11-16:
Instantaneous is current power draw for the device… meaning what it is consuming at that instant

javicalle, 2022-11-16:
Yes, but also the device diagnostics (to the Right of "KONFIGURERA OM", the 3 dots, 'download diagnostics').

It will be better if there are some load in the device that will give a "instantaneous demand"

krlssn, 2022-11-16:
[image]
This is with 754W load (235Vx3,3A=about 750W
And this was much code😊

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Stockholm",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.10.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.2",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.85",
      "zigpy-deconz==0.19.0",
      "zigpy==0.51.5",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.3",
      "zigpy-znp==0.9.1"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "1A86",
        "pid": "55D4",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus v2"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
[... 481 lines cut ...]
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

javicalle, 2022-11-16:
Please, edit your comment and put the diagnostics inside ` ``` ` marks like:
````
```
All the diagnostics here
```
````

dmulcahey, 2022-11-16:
So is the device doing it’s own conversion to watts on the instantaneous demand attribute?

krlssn, 2022-11-16:
@dmulcahey 
I don't know, but Current value is correct and fair, these stuff is new to me, the problem is "Instantaneous demmand" is not correct, should be divided with 1000. Earlier in the thread someone tested to change something, then the kWh values in Summation became incorrect instead. Excuse my english, I think I got the code above correct after a few attempts😊

javicalle, 2022-11-16:
Looking at the ZHA part:

* at the smartenergy channel, the value multiplied/divided:
https://github.com/home-assistant/core/blob/bc146a09dba6df6eaec429039f6e1efcfbd1123f/homeassistant/components/zha/core/channels/smartenergy.py#L217

* and then in the ZHA sensor part there are 2 behaviors:
  * for the `SmartEnergyMetering` the value is just formatted:
https://github.com/home-assistant/core/blob/bc146a09dba6df6eaec429039f6e1efcfbd1123f/homeassistant/components/zha/sensor.py#L448-L450
  * for the `SmartEnergySummation` (if self._channel.unit_of_measurement == 0) the value again is multiplied/divided:
https://github.com/home-assistant/core/blob/bc146a09dba6df6eaec429039f6e1efcfbd1123f/homeassistant/components/zha/sensor.py#L496-L502

Sorry if I'm making unrelated/unnecessary noise.

krlssn, 2022-11-16:
I'm happy with all help I can get, for me this coding stuff is totally new, feels like the only problem is this sensor is showing the correct value except it is somewhere multiplied with 1000. You cant draw 767 189 W from a normal outlet but 767,189 is no problem. In a perfect world the sensor have given me 767W. If I test the same sensor in for example zigbee2mqtt could it possibly make any diffenerce, or is this device info picked from the same place regardless which integration I use? I guess you understand now that I'm new to Home assistant 😉

dmulcahey, 2022-11-16:
> Ummmm, the formatter is the same for both measurements: https://github.com/home-assistant/core/blob/bc146a09dba6df6eaec429039f6e1efcfbd1123f/homeassistant/components/zha/core/channels/smartenergy.py#L230-L231
> 
> And the formatter does the kW (from Zigbee standar) to W (the uom that HA uses) here: https://github.com/home-assistant/core/blob/bc146a09dba6df6eaec429039f6e1efcfbd1123f/homeassistant/components/zha/core/channels/smartenergy.py#L218-L220
> 
> Then in the sensors part, the `unit_of_measure_map` uses:
> 
> * `0x00: POWER_WATT` for the `SmartEnergyMetering`
> * `0x00: ENERGY_KILO_WATT_HOUR` for the `SmartEnergySummation`
> 
> Or I missing something or that can't be fine because the formatter makes the kW --> W conversion always that `uom == 0`, and that is true in both cases. If the conversion is done always, the UI oum must be in W (or W per hour), isn't it?

Summation only uses the formatted if the UOM is NOT 0: https://github.com/home-assistant/core/blob/bc146a09dba6df6eaec429039f6e1efcfbd1123f/homeassistant/components/zha/sensor.py#L498 

and I have non quirked devices that this is working correctly for…

javicalle, 2022-11-16:
> Summation only uses the formatted if the UOM is NOT 0

🤦🏻‍♂️, right

David, do you mind share the diagnostic from your device?

I am out of ideas. Maybe the device need to quirk the value?

ptrrkssn, 2022-11-16:
Hmm.. Looking at the diagnostic output there is a n interesting difference between summation and demand:

```
         "0x0702": {
            "endpoint_attribute": "smartenergy_metering",
            "attributes": {
              "0x0000": {
                "attribute_name": "current_summ_delivered",
                "value": 6931
              },
...              
              "0x0303": {
                "attribute_name": "summation_formatting",
                "value": 75
              },
...
              "0x0400": {
                "attribute_name": "instantaneous_demand",
                "value": 872121
              }
            },
            "unsupported_attributes": {
              "0x0304": {
                "attribute_name": "demand_formatting"
              }
            }
```

Notice that "demand_formatting" is listed under "unsupported_attributes" from the Schneider devices (both). Is that handled as if it has the value 0?

In the Sengled bulbs - what does they report for "demand_formatting"?

dmulcahey, 2022-11-16:
```
{
  "home_assistant": {
    "installation_type": "Home Assistant Core",
    "version": "2022.12.0.dev0",
    "dev": true,
    "hassio": false,
    "virtualenv": true,
    "python_version": "3.10.2",
    "docker": false,
    "arch": "x86_64",
    "timezone": "UTC",
    "os_name": "Darwin",
    "os_version": "12.6.1",
    "run_as_root": false
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.2",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.86",
      "zigpy-deconz==0.19.0",
      "zigpy==0.51.5",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.3",
      "zigpy-znp==0.9.1"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "1A86",
        "pid": "55D4",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus v2"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
[... 280 lines cut ...]
              "0x0400": {
                "attribute_name": "instantaneous_demand",
                "value": 31
              }
            },
            "unsupported_attributes": {}
          },
          "0x0b05": {
            "endpoint_attribute": "diagnostic",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

dmulcahey, 2022-11-16:
> Hmm.. Looking at the diagnostic output there is a n interesting difference between summation and demand:
>
> ```
>          "0x0702": {
>             "endpoint_attribute": [...]

See above… 33

krlssn, 2022-11-16:
Tried to compare the code with my zero experience so I dont know what it changes, but:
My current is correct value in sensor. When I read the code it seems for me that it reports 3269 (mA), then 0x0603 divides it with 1000 to 3,269A which is the value I get in HA (3,3A). Then I look at row "0x0400" and it reports 755113 (mW?), but under there multiplier and divisor lines is missing compared to these lines in current section. Power multiplier and power divisor is listed under unsupported attributes. Could it be that easy that if this lines were like in the current section everything would be ok? Like power multiplier=1 and power divisor=1000. I dont know if i can test this by myself by changing something in the 

```
                "attribute_name": "rms_current",
                "value": 3269
              },
              "0x0602": {
                "attribute_name": "ac_current_multiplier",
                "value": 1
              },
              "0x0603": {
                "attribute_name": "ac_current_divisor",
                "value": 1000
              }
```
```
  },
              "0x0400": {
                "attribute_name": "instantaneous_demand",
                "value": 755113
              }
```
```
  },
            "unsupported_attributes": {

              "0x0403": {
                "attribute_name": "power_divisor"

              },
              "0x0402": {
                "attribute_name": "power_multiplier"
              }
            }
```

krlssn, 2022-11-16:
Compared with current divisor I get "1000".
[image]

ptrrkssn, 2022-11-16:
I've now tried various values for "demand_formatting" in the quirk. So far I've not found any value that seems to make any difference what so ever... Tried 75, 0 & 33 so far. 

```
class MeteringCluster(CustomCluster, Metering):
    """Fix the Metering divisor."""

    INSTANTANEOUS_DEMAND = 0x0400
    DEMAND_FORMATTING = 0x0304

    _CONSTANT_ATTRIBUTES = { DEMAND_FORMATTING: 33 }

    def _update_attribute(self, attrid, value):
        if attrid == self.INSTANTANEOUS_DEMAND:
           value = value / 1000.0
        super()._update_attribute(attrid, value)

```

Also include the full quirk code (that works for me) for reference. (And yes, I've deleted the __pycache__, restarted HA and removed and re-pair:ed the device again every time. For it to work with the other type of outlet it probably needs to be adjusted to include it's signature or something... 

[schneider_plug.py.txt](https://github.com/zigpy/zha-device-handlers/files/10026128/schneider_plug.py.txt)

krlssn, 2022-11-16:
@ptrrkssn 
Do you know what happens if I try to write value "1000" for "power_divisor" in my pic above? Is that read only and impossible to change? If I can do it, does it write to the outlet or somewhere in "HA software"? Is it restorable from backup? Any thoughts about it? I want to think that it is that easy when I compare current and power, but maybe that's what ypu tried earlier with wrong kWh (summary) readings as result...?

javicalle, 2022-11-16:
The multiplier/divisor attributes are read only and couldn't be changed from UI.
Usually the values you change, you can restore... if you now the previous value.
I appreciate the efforts, but we have the necessary info from the diagnostics.

javicalle, 2022-11-17:
Looking at the data I have come to the same conclusion as David (but much slower):

> So is the device doing it’s own conversion to watts on the instantaneous demand attribute?

Putting 2 plugs in serial (the affected one and another well working one) would confirm the suspicions.

So probably the solution goes through the quirk that Peter proposes.
However, I am concerned that since it is a bug in the device, in an updated version the problem may be resolved and that it may present conflicts.

For these attributes, the formatting don't play any role in the code (also, didn't would change the value, just the format), so not relevant.

krlssn, 2022-11-17:
So you mean that if Schneider fix a FW update of the device it would be fixed? Does this only affect HA (ZHA) connected devices or devices connected through Wisers own app too?
I don't have other Zigbee devices with metering, but I have Fibro wall plugs if it should help (to confirm this issue)?

javicalle, 2022-11-17:
What I mean is, if that is a firmware bug as it seems, probably SE will patch it. Then it could be that we can find devices with new firmware version (because SE sells the devices with the new version or because users update its devices through SE gateway).
If we patch the same in ZHA (with a quirk) I am not sure if we can check the device firmware version to apply or not the quirk.

I don't know if Wiser app exposes the values correctly or not, but every manufacturer can do a particular interpretation from the Zigbee specifications, so maybe they in its app are showing the values in other way (and fine with the metering values) despite the device don't follow the standar.

> I don't have other Zigbee devices with metering, but I have Fibro wall plugs if it should help (to confirm this issue)?

My point is to compare between 2 devices with 'instantaneous demmand' metering to see its cluster attribute values.

krlssn, 2022-11-17:
Ok, then I guess FB wallplug don't talk "the same language" and it won't help here. No idea if Schneider will fix fw, if it is no problem in their own app I guess their priority is lower and a fix from you would be awesome. Understand the problem if/when different FW appear.

javicalle, 2022-11-17:
And here it is...
> Elko EKO09738 and EKO09716 reports power in mW, scale to W

https://github.com/Koenkk/zigbee-herdsman-converters/blob/081c81ef0131c5cd7dae342d3784e501e1c0cb68/converters/fromZigbee.js#L712-L725

That quirk is applied to the SE devices:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/081c81ef0131c5cd7dae342d3784e501e1c0cb68/devices/schneider_electric.js#L762-L779

We don't thank Z2M enough for the work they do there.

@ptrrkssn if you don't mind you can create the PR yourself, if possible also adding the signature for the `SOCKET/OUTLET/1`.

krlssn, 2022-11-17:
Sounds perfect😄 What do I need to do/what should I add to my system to get the correct scaled values?

javicalle, 2022-11-17:
You can wait to the next HA release to see if the quirk has been added to the version.
If you don't want/can't wait until next month, you can put the proposed quirk as a local quirk in your instalation (and remember to remove it when updating HA).

The quirk would be the one from https://github.com/zigpy/zha-device-handlers/issues/1889#issuecomment-1317754162

There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Create in your local quirk folder a new `schneider_plug.py` file and put the suggested quirk in file.

Restart HA, remove and pair you device again. 
Check that the device is loading the quirk and that the device signature has changed.

krlssn, 2022-11-17:
Thank you! If I wait until it comes in future HA release, do I have to unpair/pair devices to get correct values or do it "autoupdate"?

javicalle, 2022-11-17:
I can't tell you for sure, but as a standar procedure I will suggest you to perform the unpair/pair to be sure.

krlssn, 2022-11-27:
Hi
Stupid question again... How can I find if this has been fixed? Have got some updates to my system, but I dont know in whick of the updates bugs like this is fixed, and how I find it (how it's named) in release notes...? Is it Home assistant, supervisor or operating system that includes these uptades?
[image]

javicalle, 2022-11-27:
There's no an easy way to know it.
You can check if the HA version includes any zha handler update. Then you can look at the version changelog if includes the fix. But...

The first thing will be to have a PR/fix/quirk to look for. And there's no PR yet for this issue. So, until someone creates a PR and that would be merged the fix don't be available to all the users.

krlssn, 2022-12-02:
Update:
I managed to insert the local quirk @ptrrkssn (🙏👍) made for SOCKET/OUTLET/2 (I bought a 2way socket too) with success! I have also made a new quirk file for SOCKET/OUTLET/1 (just replaced "2" with "1" in the code. Everything seems to work. I didn't need to unpair/reinclude the devices, HA restart was enough. Maybe I re-include just to get correct graphs in the history because now it is 0-2500000 so hard to read when ~0-2500 looks better, maybe there is a simpler way...
How is the "correct" way to to this, can multiple devices be included in the same quirk file or is 1 file/device better? If it can be done in the same file, is it copy paste all code after each other, or can I just add SOCKET/OUTLET/2 and SOCKET/OUTLET/1 after each other on the places they are used in the code?
@javicalle Does your text in https://github.com/zigpy/zha-device-handlers/issues/1889#issuecomment-1318480151 mean that this "PR" covers the Wiser sockets too, or do they need a new "PR"?

[image]

[image]

javicalle, 2022-12-02:
I would suggest to create a new `schneider_electric` or `schneiderelectric` (like in https://github.com/zigpy/zha-device-handlers/pull/1705/files)

I would put all together in the same file and the same quirk, probably in a `plug.py`.
Like both devices have the same signature it will be better to put in the same quirk:
```python
    signature = {
        MODELS_INFO: [
            ("Schneider Electric", "SOCKET/OUTLET/1"),
            ("Schneider Electric", "SOCKET/OUTLET/2"),
        ],
        ENDPOINTS: {
            .../...
```

krlssn, 2022-12-05:
New *py file if someone needs it. Working for me with both WDE002182 (2way) and WDE002172 (1way). Gives me correct values in "instantaneous_demand".

Checked the site [https://github.com/zigpy/zha-device-handlers/compare](url) but not sure what to do, needs comparing to open PR and this stuff is new to me so I dont know what to "compare" to open a new pull request.

[socketoutlet.py.txt](https://github.com/zigpy/zha-device-handlers/files/10154387/socketoutlet.py.txt)

krlssn, 2023-11-30:
> I just added my 2-way outlet and ran into this very issue. I am using the latest version of Home Assistant (running on a HA Green). Using SkyConnect and ZHA.
> So is there a way to get this merged into an actual release or how does that work?
>
> I just added my 2-way outlet and ran into this very issue. I am using the latest version of Home Assistant (running on a HA Green). Using SkyConnect and ZHA.
> So is there a way to get this merged into an actual release or how does that work?
>
> I just added my 2-way outlet and ran into this very issue. I am using the latest version of Home Assistant (running on a HA Green). Using SkyConnect and ZHA.
> So is there a way to get this merged into an actual release or how does that work?
>
> I just added my 2-way outlet and ran into this very issue. I am using the latest version of Home Assistant (running on a HA Green). Using SkyConnect and ZHA.
> So is there a way to get this merged into an actual release or how does that work?
>
> I just added my 2-way outlet and ran into this very issue. I am using the latest version of Home Assistant (running on a HA Green). Using SkyConnect and ZHA.
> So is there a way to get this merged into an actual release or how does that work?
>
> I just added my 2-way outlet and ran into this very issue. I am using the latest version of Home Assistant (running on a HA Green). Using SkyConnect and ZHA.
> So is there a way to get this merged into an actual release or how does that work?
>
> I just added my 2-way outlet and ran into this very issue. I am using the latest version of Home Assistant (running on a HA Green). Using SkyConnect and ZHA.
> So is there a way to get this merged into an actual release or how does that work?
>
> [... cut ...]
> So is there a way to get this merged into an actual release or how does that work?
>
> I just added my 2-way outlet and ran into this very issue. I am using the latest version of Home Assistant (running on a HA Green). Using SkyConnect and ZHA.
> So is there a way to get this merged into an actual release or how does that work?

As I understand the fix (quirk) needs to be added as a pull request first, then it can be added into future release of ZHA integration in HA. I didnt manage how to do this. I still need to "inactivate" and reactivate the quirk every time I update HA (and some other steps). Had been very nice to get this fixed. I think it's ZHA people who need to merge this when the PR is created. I read somewhere that zigbee2mqtt integration has fixed this, but i'm not very happy to rebuild my entire Zigbee network. Feel free to try opening a pull request, this stuff is new to me, I don't think it should be that hard to do 😏

[4 more replies]

closed 2024-09-21 by github-actions[bot]