zigpy/zha-device-handlers #1891: [Device Support Request] TS11F _TZ3000_c7nc9w3c

1zun4, 2022-11-05:
**Is your feature request related to a problem? Please describe.**
[image]

It detects 3 switches (for the 3 outlets) but these switches will turn on/off all outlets at once, not seperate.
It also does not report the current and consumption correctly. It shows a whopping 1.587 A, which is most likely supposed to be 1587 mA (1.58A) and crazy 54,0 kWh.

**Describe the solution you'd like**
I would really like to have a working support for this smart power strip.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_c7nc9w3c",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
nothin
```

</details>

**Additional context**
[image]

javicalle, 2022-11-05:
Is a 3 AC outlet device or 4 AC outlet? (the image don't match the description)
Do the device have also any USB port? For the device signature I will say that don't have.

Can you add any link to the manufacturer or vendor product?

1zun4, 2022-11-05:
> 

It is a 4 AC outlet but only 3 can be toggled. It has 2 USB ports.

https://www.aliexpress.com/item/4000372097774.html?spm=a2g0o.order_list.0.0.3abd5c5fGRpxIp&gatewayAdapt=glo2deu

javicalle, 2022-11-05:
That is my proposal for your device:
```python
class Plug_4AC_2USB_Metering(EnchantedDevice):
    """Tuya Tuya 4 outlet + 2 USB with power metering."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            1: {
                # "profile_id": 260,
                # "device_type": "0x010a",
                # "in_clusters": ["0x0000","0x0003","0x0004","0x0005","0x0006","0x0702","0x0b04","0xe000","0xe001"],
                # "out_clusters": ["0x000a","0x0019"]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                # "profile_id": 260,
                # "device_type": "0x010a",
                # "in_clusters": ["0x0004","0x0005","0x0006","0xe001"],
                # "out_clusters": []
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                # "profile_id": 260,
                # "device_type": "0x010a",
                # "in_clusters": ["0x0004","0x0005","0x0006","0xe001"],
                # "out_clusters": []
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                # "profile_id": 260,
                # "device_type": "0x010a",
                # "in_clusters": ["0x0004","0x0005","0x0006","0xe001"],
                # "out_clusters": []
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
[... 85 lines cut ...]
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

You will need to enable the local quirk in your instalation. There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Copy in your local quirk folder the current `ts011f_plug.py` file and put the suggested quirk in the bottom of file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py

Restart HA, remove and pair you device again. 
Check that the device is loading the quirk and that the device signature has changed.

1zun4, 2022-11-06:
Thank you very much. I added your snippet to the `ts011f_plug.py`: https://gist.githubusercontent.com/1zun4/c430ef5797bde2364c05aa39fd5de0e1/raw/f1aa8bac245e9a27aac64846b1d961f549429284/ts011f_plug.py and placed it into my custom_quirks folder 
[image]
and enabled custom quirks in the configuration.yaml 
[image]

but somehow it is not being picked up your custom quirk for the device.
[image]

and still shows the data wrong. Is there anything I have done wrong?

1zun4, 2022-11-06:
Thank YOU VERY MUCH!

[image]

I have done a edit to your snippet. I noticed that the signature has no endpoint number 4 and 5, so I decided to remove them. Now it is being detected correctly.

https://gist.githubusercontent.com/1zun4/18f6f4c687d5084792a924bc05eb601a/raw/22e05c6ff9890094ef68d5e3e076e970992dda35/ts011f_plug.py

javicalle, 2022-11-06:
> Thank you very much. I added your snippet to the `ts011f_plug.py`: https://gist.githubusercontent.com/1zun4/c430ef5797bde2364c05aa39fd5de0e1/raw/f1aa8bac245e9a27aac64846b1d961f549429284/ts011f_plug.py and placed it into my custom_quirks folder

My bad. The copy&paste always leads me to some error. I have fixed the proposed code.

> I noticed that the signature has no endpoint number 4 and 5, so I decided to remove them. Now it is being detected correctly.

But you will only control the firsts 3 outlet, don't you?
Maybe the new version fix it.

1zun4, 2022-11-06:
> My bad. The copy&paste always leads me to some error. I have fixed the proposed code.
>
>
> But you will only control the firsts 3 outlet, don't you? Maybe the new version fix it.

Yes. Only the 3 first outlets are able to be controlled. All others are always on by design.

javicalle, 2022-11-06:
> Only the 3 first outlets are able to be controlled. All others are always on by design.

Now I understant the device behavior. It would be better if I could understant from the beginning.

That picture is most illustrative:

[image]

If the device is working as expected, could you create a PR with the new quirk?
You can edit the file from the web and just follow the instructions.

1zun4, 2022-11-07:
Yes, sure. I make a pull request later this day.

KaherdinTristan, 2022-11-08:
@1zun4 & @javicalle 
Hello. My name is Tristan, and I have a humble request for You both; 
Would You care to take a look at my two Device Requests linked below?

They are both "zigbee in wall switch" used behind outlets and ordinary wall switches. One has power metering and the other does not. The issue I am facing is that they randomly turn off. One thought is that they do this when the voltage abruptly changes. 

I don't have the knowledge to even try to build the code necessary to remedy these issues, but I have understood that You know at least how to create quirks for ZHA. 

I can step by step give You all the data needed and I would greatly appreciate you helping me (and others) out. 

https://github.com/zigpy/zha-device-handlers/issues/1904 
https://github.com/zigpy/zha-device-handlers/issues/1905

Best regards
Tristan

closed 2022-11-09 by dmulcahey