zigpy/zha-device-handlers #1892: [Device Support Request] Tuya Zigbee Irrigation ControllerQT-06Z-01

cloudbr34k84, 2022-11-05:
Hi im hoping to have this [zigbee irrigation controller](https://www.aliexpress.com/item/1005004222098040.html?spm=a2g0o.order_list.0.0.f7d718021TVlwf) added please. 

Zigbee Signature
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_a7sghmms",
  "model": "TS0601",
  "class": "zigpy.device.Device"
```
[zha-d536ea29b443ee58a5d09b996a6a4554-_TZE200_a7sghmms TS0601-0a475f32ab61d66a0fd89f23957da342.json.txt](https://github.com/zigpy/zha-device-handlers/files/9943693/zha-d536ea29b443ee58a5d09b996a6a4554-_TZE200_a7sghmms.TS0601-0a475f32ab61d66a0fd89f23957da342.json.txt)

mbo18, 2022-11-05:
Check here: https://github.com/zigpy/zha-device-handlers/issues/1622#issuecomment-1214705293

hkoci, 2022-11-05:
I used it on this variant by just changing the manufacturer code in the code provided by @infinitejest https://github.com/zigpy/zha-device-handlers/issues/1622#issuecomment-1214475892 to:

MODELS_INFO: [("_TZE200_sh1btabb", "TS0601"),("_TZE200_a7sghmms", "TS0601")],

My current file is essentially the same, have used it since last summer when on vacation to water the flowers. Enable ZHA custom quirks and place the configuration file in the folder you located for the quirks.

I have used it and find that it works well - the valve turns on and doesn't turn off until you toggle the light entity again.

Though there were some connection issues with when the toggle does not actually toggle but this has not happened frequently.

```
"""Tuya valve devices."""

import logging
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import NoManufacturerCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaDPType,
    TuyaLevelControl,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaOnOffNM,
    TuyaPowerConfigurationCluster,
)

_LOGGER = logging.getLogger(__name__)

class TuyaLevelControlNM(NoManufacturerCluster, TuyaLevelControl):
    """Tuya LevelControl cluster with NoManufacturerID."""

class TuyaValveManufCluster(TuyaMCUCluster):
    """Tuya valve manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF03: ("dp_3", t.uint32_t, True),
            0xEF65: ("dp_101", t.uint32_t, True),
            0xEF66: ("dp_102", t.uint32_t, True),  # <-- new cluster attribute with fake IDs (102=0x66).Format= ID: ("name", type, True)
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        2: DPToAttributeMapping(
            TuyaOnOffNM.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
        1: DPToAttributeMapping(
            TuyaLevelControlNM.ep_attribute,
            "current_level",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: (x * 255) // 100,
            dp_converter=lambda x: (x * 100) // 255,
        ),
        3: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_3",
            dp_type=TuyaDPType.VALUE,
        ),
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_101",
            dp_type=TuyaDPType.VALUE,
        ),
        102: DPToAttributeMapping(        # <-- device DP102
            TuyaMCUCluster.ep_attribute,  # <-- reference to the cluster which has the attribute
            "dp_102",                     # <-- attribute "name"
            dp_type=TuyaDPType.VALUE,     # <-- DP Type it is related to the attribute type
        ),
        108: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
        ),
    }

    data_point_handlers = {
        2: "_dp_2_attr_update",
        1: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
        102: "_dp_2_attr_update",
        108: "_dp_2_attr_update",
    }

    async def write_attributes(self, attributes, manufacturer=None):
        """Overwrite to force manufacturer code."""

        return await super().write_attributes(
            attributes, manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID
        )

class TuyaValve(CustomDevice):
    """Tuya valve device."""

    signature = {
        MODELS_INFO: [("_TZE200_sh1btabb", "TS0601"),("_TZE200_a7sghmms", "TS0601")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0051
            # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
            # output_clusters=[0x000a, 0x0019]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaValveManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaValveManufCluster,
                    TuyaOnOffNM,
                    TuyaLevelControlNM,
                    TuyaPowerConfigurationCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

```

closed 2022-11-05 by cloudbr34k84