zigpy/zha-device-handlers #1894: [Device Support Request] - ts0601_air_quality.py - No Particulate matter 2.5 µm entity

cloudbr34k84, 2022-11-05:
This devices reports Particulate matter 2.5 µm, any reason why the quick does not have that included?

javicalle, 2022-11-05:
The short answer is because Tuya devices are a mess and most of maintainers / collaborators don't have every device that is implemented.

We do the best we can with the information provided. For example, `ts0601_air_quality.py` implements 3 quirks for 7 identified devices. 

And someone can walk by and say something like 'my device isn't working'. And we try to help.

Other times, someone comes and say something like 'my "_TZE200_dwcarsat" according to manufacturer can report PM25 as you can see in this screenshoot, but don't show anything in HA. There is also this messages in my logs refering my device'. And we try to help.

But it is very difficult to keep track of all the issues of all the devices.
Sometimes the tests keeps halfway done and the quirk is not implemented.
Sometimes you have conversations with 2 or 3 people who say they have the same device and who give you contradictory information.

In conclusion, there is no reason why that functionality has not been implemented. We may not have been able to perform the necessary tests or obtain conclusive information to get it working, but if you are interested and with the correct information, we can try 😉

cloudbr34k84, 2022-11-05:
> The short answer is because Tuya devices are a mess and most of maintainers / collaborators don't have every device that is implemented.
>
> We do the best we can with the information provided. For [...]

Hey, cheers for the response and the explanation, i was just wondering that's all. I have the devices fully working through the Tuya Integration for now.  ill close this down :)

closed 2022-11-05 by cloudbr34k84