zigpy/zha-device-handlers #1896: [Device Support Request] ("_TZE200_3towulqd", "TS0601") Motion and Illuminance Sensor

cloudbr34k84, 2022-11-05:
I tried to get the ts0601_motion.py  to work with this sensor.

This devices does motion and Lux sensor? 

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0500"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_3towulqd",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

javicalle, 2022-11-06:
Duplicate of #1620

javicalle, 2022-11-12:
Can you try with the quirk from https://github.com/zigpy/zha-device-handlers/issues/1599#issuecomment-1219634462?

javicalle, 2022-11-21:
Let's keep the conversation inside the issue https://github.com/zigpy/zha-device-handlers/issues/1620

closed 2022-11-21 by javicalle