zigpy/zha-device-handlers #1897: [BUG] Aqara lumi.plug.maeu01 no power measurements

Mariuss811, 2022-11-06:
**Describe the bug**
Aqara Smart Plug won't show power measurements but device Temperature.

**To Reproduce**
Steps to reproduce the behavior:
1. Remove Plug 
2. Restarted Hassio 
3. Set Plug in Pair mode 
4. Pair it with ZHA
5. See error

**Expected behavior**2
Get the power mesurements.

**Screenshots**
[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug.maeu01",
  "class": "zhaquirks.xiaomi.aqara.plug_eu.PlugMAEU01"
}
```

</details>

[... 544 lines cut ...]
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>

**Additional context**
As requestes in [https://github.com/zigpy/zha-device-handlers/issues/1337]

@TheJulianJES

TheJulianJES, 2022-11-06:
It's likely that your Raspbee coordinator isn't on the latest firmware. It needs to be on a somewhat recent version in order to send Xiaomi manufacturing codes during pairing.

So, try to update your Raspbee firmware and then pair the plug again.

kimbirkelund, 2022-11-06:
I have the same issue. 

I just got two plug and they both have this issue. Interestingly however I have an identical device that I set up in December last year which worked immediately back then and continues to work. I'll attach info for the working one and one of the ones that doesn't.

### Info for working device

<details>
<summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug.maeu01",
  "class": "zhaquirks.xiaomi.aqara.plug_eu.PlugMAEU01"
}
``` 
</details>

<details>
<summary>Diagnostics information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
[... 1375 lines cut ...]
        "out_clusters": {}
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

Hope someone can figure the issue.

TheJulianJES, 2022-11-06:
There are two firmware versions for the plug. The quirk that's in 2022.10.5 and later works with both firmware versions.
However, the newer firmware versions requires the coordinator to identify itself as a Xiaomi hub during pairing.
This is supported by TI/ZNP coordinators (firmware update needed), EZSP coordinator (no firmware update needed), and also Conbee/Raspbee (firmware update needed).

So, what coordinator are you running and what firmware is on it?

(Otherwise, it's also possible to downgrade the plug: https://github.com/zigpy/zha-device-handlers/pull/1656#issuecomment-1244750465)

kimbirkelund, 2022-11-06:
I found that out. Upgrading the firmware of my ConBeeII fixed it 👍

Mariuss811, 2022-11-08:
Alright, after updating my Raspee II with the latest Firmware (deCONZ_RaspBeeII_0x26780700.bin.GCF) and repairing the Plug it works! 
But I assumed I got the latest Firmware according to the DECONZ Add-on but for some reason they disabled automatic updates some time ago.

Thank you for solving this!

closed 2022-11-08 by Mariuss811