zigpy/zha-device-handlers #1898: Issues with Tuya TS0012 2 Gang

mbrihed, 2022-11-06:
Hi!

I have several TS0012 Tuna 2 gang, TS001 1 gang and also TS0013 3 gang devices. All of them are "No Neutral"

On the TS0011 I have no issues, they load and also load the quirk as seen here:

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_ji4araar",
  "model": "TS0011",
  "class": "zhaquirks.tuya.ts001x.Tuya_Single_No_N"
}

On the 2-gang I do not get them to lad and they are found by the coordinator (TubeZB) but it is not working, they either go back to the same status when turning the switch in Home Automation (but the light does not turn on or off) or they flick both switches as if it was a 1gang switch.

I have been searching and reading and also followed the post Issue #1301 https://github.com/zigpy/zha-device-handlers/issues/1301 but can't get this working.

Home Assistant:

Home Assistant 2022.11.1
Supervisor 2022.10.2
Operating System 9.3
Frontend 20221102.1 - latest

mbrihed, 2022-11-06:
This is the diagnostic file for one of them:

[zha-09251d82decb68e2a0992ae516deadf1-_TZ3000_jl7qyupf TS0012-6d0aa46e4ab6ecf655105a4e3af71b8f.json.txt](https://github.com/zigpy/zha-device-handlers/files/9946483/zha-09251d82decb68e2a0992ae516deadf1-_TZ3000_jl7qyupf.TS0012-6d0aa46e4ab6ecf655105a4e3af71b8f.json.txt)

mbrihed, 2022-11-06:
This part is something I don't get, it is not loading quirks

  "data": {
    "ieee": "**REDACTED**",
    "nwk": 49942,
    "manufacturer": "_TZ3000_jl7qyupf",
    "model": "TS0012",
    "name": "_TZ3000_jl7qyupf TS0012",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",

javicalle, 2022-11-06:
If they are battery powered devices, the pairing process can be tricky. You must keep short pushing buttons until the pairing process ends to keep the device awake.

If not working, add the signature from the devices to see if they need a new quirk.

mbrihed, 2022-11-06:
Hi!

The device adds and becomes nice and free in ZHA when adding them. Once they are added they do not work. They are switches in the light socket behind the light wall switch. No Neutral devices but powered devices.

javicalle, 2022-11-06:
It seems to be another f****ng tuya variant.

The proposed quirk for the `TS0012` would be:
```python
class Tuya_Double_Var05(TuyaSwitch):
    """Tuya 2 gang no neutral light switch (variant 05)."""

    signature = {
        # NodeDescriptor(
        #     logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, 
        #     frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, 
        #     maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, 
        #     descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, 
        #     *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False
        MODEL: "TS0012",
        ENDPOINTS: {
            1: {
                # "profile_id": 260,
                # "device_type": "0x0100",
                # "in_clusters": ["0x0000","0x0003","0x0004","0x0005","0x0006","0xe000","0xe001"],
                # "out_clusters": ["0x000a","0x0019"]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
            2: {
                # "profile_id": 260,
                # "device_type": "0x0100",
                # "in_clusters": ["0x0004","0x0005","0x0006"],
                # "out_clusters": []
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```

You will need to enable the local quirk in your instalation. There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Copy in your local quirk folder the current `ts001x.py` file and put the suggested quirk in the bottom of file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts001x.py

Restart HA, remove and pair you device again. 
Check that the device is loading the quirk and that the device signature has changed.

mbrihed, 2022-11-06:
Hi!
I have enabled it and other devices pick it up but not the TS0012. It refuses to pick it up. I have tried the custom file in issue #1301 as well as the ts001x.py file you refer to but none of them works and the devices refuse to pick them up.

mbrihed, 2022-11-06:
With the file you shared it picks up the quirk :-). Removed the device, restarted HA and then re-added the device.

Still not working, when flicking the switch in HA to on it goes off after a few seconds. Try both switches but the same. The lights do not turn on.

mbrihed, 2022-11-06:
The other switch I have installed works different even though it is the same switch from the same vendor. When I flick one of the switches it flicks both of them. It sees it as if it would be a 1-gang switch even though it shows the two switches.

javicalle, 2022-11-06:
Can we go one by one, please.

Share one device signature and behavior and we will try to fix it, but one by one.

mbrihed, 2022-11-06:
And yes, I agree on the comment, another *****ng Tuna device. Useless but now I have more than 25 of them for the entire house and I have no neutral wire in the switch boxes. If not I would have thrown them out. I also have several of the switches in 2-way wiring so it limits my choices.

mbrihed, 2022-11-06:
This one puts the switch back to what it was after a couple of seconds, any of the switches but treats them as two switches 

Signature file:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_jl7qyupf",
  "model": "TS0012",
  "class": "ts0012.Tuya_Double_Var05"
}

javicalle, 2022-11-06:
Not sure if I understand which is the device's behavior. But can be related to the Tuya `EnchantedDevice`.

I have updated the quirk signature. Just the `class Tuya_Double_Var05(EnchantedDevice, TuyaSwitch):` part.

Update the file, remove any `__pycache__` folder and restart HA.

You will need to remove the device and pair it again.

mbrihed, 2022-11-07:
Tried the steps you provided but no change.
And to explain the two different switches: 
They are both 2 gang and TS0012 from the same vendor.
The one we are looking at first is the signature file I uploaded last. The behavior of that device is that whenever I flick any of the two gangs it just goes back to the previous state after a couple of seconds. The lights will not turn on.

The second device (TS0012) behaves different, it treats the two gangs as the same so which ever I flick it will flick both of them. The light do turn on but both do regardless which one I switch.

Hope this explains :-)

mbrihed, 2022-11-07:
These are the signatures of the two devices I mentioned:

First device (that puts back the switch to the previous state after a few seconds):

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_jl7qyupf",
  "model": "TS0012",
  "class": "ts001x.Tuya_Double_Var05"
}

Second device (that treats the gangs as the same):

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_jl7qyupf",
  "model": "TS0012",
  "class": "ts001x.Tuya_Double_Var05"
}

javicalle, 2022-11-07:
For its behavior, the second device will need the "Tuya spell".
This means that the device needs the updated version of the quirk. And once updated the code, you need to pair the device again (to apply the new code version).
Be sure that there is no `__pycache__` folder before restart HA.

MattWestb, 2022-11-07:
Both device is loading the same quirk and the same device classes and have the same manufacture name "_TZ3000_jl7qyupf".

Do the things @javicalle have suggesting in https://github.com/zigpy/zha-device-handlers/issues/1898#issuecomment-1304922525 and remove both devices from ZHA and wait one minute. Then restart HA and then its stable adding the device one by one and look if they is behaving the same.

The reason we need to this is that we is casting one "tua magic spell" then the device is joining the network and its needed for getting the device working OK and must / cant being done then the device is doing one new joining and is not working if the device is known of the system.

mbrihed, 2022-11-07:
OK, did the following:

1) Removed both devices
2) Removed __pycache__ directory
3) Restarted HA
4) Waited 10 min
5) Re-added one of the devices (the one I mentioned first in above post)
6) When adding the device the two gangs are grayed out
7) Turning on the light manually with the rocker switch, enables the switches in HA and they become "enabled"
8) Still the same behavior, I turn on either of the two switches in HA and they go back to previous state after a few seconds.

End result, nothing changed :-(

javicalle, 2022-11-07:
Are you sure that your local quirk is updated like thst?
```python
class Tuya_Double_Var05(EnchantedDevice, TuyaSwitch):
```

javicalle, 2022-11-07:
Please, attach the full class (the new added part)
Put the code inside ` ``` ` marks (in a new line at the beginning and the end of the code)

javicalle, 2022-11-07:
Any related error in your logs?

mbrihed, 2022-11-07:
```class Tuya_Double_Var05(EnchantedDevice, TuyaSwitch):
    """Tuya 2 gang no neutral light switch (variant 05)."""

    signature = {
        # NodeDescriptor(
        #     logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, 
        #     frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, 
        #     maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, 
        #     descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, 
        #     *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False
        MODEL: "TS0012",
        ENDPOINTS: {
            1: {
                # "profile_id": 260,
                # "device_type": "0x0100",
                # "in_clusters": ["0x0000","0x0003","0x0004","0x0005","0x0006","0xe000","0xe001"],
                # "out_clusters": ["0x000a","0x0019"]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
            2: {
                # "profile_id": 260,
                # "device_type": "0x0100",
                # "in_clusters": ["0x0004","0x0005","0x0006"],
                # "out_clusters": []
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```

mbrihed, 2022-11-07:
This could be an error in the log but not sure it is related:

Logger: homeassistant
Source: components/zha/core/channels/base.py:78 
First occurred: 11:59:30 (6 occurrences) 
Last logged: 12:18:34

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 943, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 715, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 581, in async_handle_light_off_service
    await light.async_turn_off(**filter_turn_off_params(light, params))
  File "/usr/src/homeassistant/homeassistant/components/zha/light.py", line 416, in async_turn_off
    result = await self._on_off_channel.off()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/base.py", line 78, in wrapper
    result = await command(*args, **kwds)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 324, in request
    return await self._endpoint.request(
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 237, in request
    return await self.device.request(
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 297, in request
    await self._application.request(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 659, in request
    await self.send_packet(
  File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 742, in send_packet
    status, _ = await self._ezsp.sendUnicast(
AttributeError: 'NoneType' object has no attribute 'sendUnicast'

mbrihed, 2022-11-07:
I am starting to think I will redo my entire network but the problem is that I have 100 devices and it will be painful with all automations etc that needs to be updated.
I migrated from a PoE TubeZB coordinator to a new a week ago and the migration seemed to be ok but maybe I am suffering from something.
Want to avoid that and since everything else seems to work fine

MattWestb, 2022-11-07:
@mbrihed The devs was saying:
https://github.com/zigpy/zha-device-handlers/issues/1898#issuecomment-1305607155

and you was posing some parts of the quirk and left the most important out, the first line of the device class !!!

Pleas post the complete quirk you is using / testing in ` ``` ` or as attachment as one Zipfile.

Without that information or the complete HA long from your  100 device system is not possible if the device is getting the tuya magic spell.

mbrihed, 2022-11-07:
```
class Tuya_Double_Var05(EnchantedDevice, TuyaSwitch):
    """Tuya 2 gang no neutral light switch (variant 05)."""

    signature = {
        # NodeDescriptor(
        #     logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, 
        #     frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, 
        #     maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, 
        #     descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, 
        #     *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False
        MODEL: "TS0012",
        ENDPOINTS: {
            1: {
                # "profile_id": 260,
                # "device_type": "0x0100",
                # "in_clusters": ["0x0000","0x0003","0x0004","0x0005","0x0006","0xe000","0xe001"],
                # "out_clusters": ["0x000a","0x0019"]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
            2: {
                # "profile_id": 260,
                # "device_type": "0x0100",
                # "in_clusters": ["0x0004","0x0005","0x0006"],
                # "out_clusters": []
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```

MattWestb, 2022-11-07:
Thanks the interesting was the `EnchantedDevice` :-)))

javicalle, 2022-11-07:
All the parts seem to be fine.

Please try to repeat the remove and pair device. Some users report to had tried twice or even 3 times before they work.

mbrihed, 2022-11-07:
Shall I then remove them, delete the __pychache__ dir, restart HA every time or just remove and then re-add.

javicalle, 2022-11-07:
Just remove the device from HA and pair it again.

mbrihed, 2022-11-08:
I have tried to remove it and the re-add it 10 times but no change. It does not even notice when I manually turn on the light on the rocker switch. It does not turn on the lights when using the switch in HA, it still just goes back to off after turning it on but no reaction or sign of light on the lights.

javicalle, 2022-11-10:
Please, enable the logs and attache the logs from the ha commands and from the physical device.:
* https://www.home-assistant.io/integrations/zha#debug-logging

mbrihed, 2022-11-24:
Hi!
sorry for the delay but been overloaded with work. 

I have made some progress that at least now I can control it and lights switch on and off. The problem is now that when I switch one gang both reacts. So if I turn on light 1 it also switches on light 2. And when they are on I switch off one of them it switches both off.

javicalle, 2022-11-24:
Is the device loading the quirk? (Check it at the device's signature)
That behavior is typical of enchanted devices, but the quirk is already patched.

What modifications or changes have you made to be able to progress?

mbrihed, 2022-11-24:
Hi!
Yes it loads the quirk.
Modifications is that I started from the beginning with my entire zigbee network. I have been running HA and zigbee for years and also migrated from one coordinator to a new so was suspecting a mess :-).
So decided to start from scratch and migrate over from the old instance to the new.

mbrihed, 2022-11-24:
Forgot to mention that the two different devices that was behaving different before now behave the same. As described above that it treats the two gangs as one.

mbrihed, 2022-11-24:
Wow, some magic happened! Suddenly they started to work. I am so happy! Now I can continue my installation of the remaining 20+ devices haha

Thanks a million for all the help! Happy thanksgiving

closed 2022-11-29 by dmulcahey

guidocioni, 2023-01-12:
I just installed one of these devices and I've been having the same problem: device is correctly recognized and the latest version of the quirk is loaded.
Unfortunately whenever I try to turn on one of the two lights in HA both turn on; same when I turn them off. The physical switches work as intended. 
The quirk that is loaded is `zhaquirks.tuya.ts001x.Tuya_Double_Var05` which according to https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts001x.py should be up to date.
I tried to restart HA, remove the device and add it again...any other suggestion?

MattWestb, 2023-01-12:
The device need "tuya magic spell" being cast then its joining the network.
That only happening then the device is not known to the system so restarting HA is not helping then the device is not in INIT mode.
For getting it working (then its loading the quirk OK in your system) you need deleting the device and restarting HA and then its stable adding the device new and it shall casting the spell then the quirk is being loaded and the device is in INIT mode.

guidocioni, 2023-01-12:
> The device need "tuya magic spell" being cast then its joining the network. That only happening then the device is not known to the system so restarting HA is not helping then the device is not in [...]

Actually I realized that the quirk loaded from `zha-quirks` did not have the `EnchantedDevice` class. Somehow my `zha-quirks` in the `deps` folder is not up to date... I thought I already had this PR merged in my version but actually that was not the case. I modified the file manually, re-added to the network and now it's working.

MattWestb, 2023-01-12:
## OPS !!
I was thinking the new quirk was having it (its have https://github.com/zigpy/zha-device-handlers/pull/2049) but then looking it not having released in the last HA (quirk ver 0.0.90 is having it).
You can download the current wuirk and putting in custom quirks or patching the HA container https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts001x.py.

If doing container patch its being no problem with next HA update with it in but if making custom qilrs you shall deleting the custom quirk then its being in HA so its not overriding the HA ones that is getting fixes in the future.

guidocioni, 2023-01-12:
> ## OPS !!
> I was thinking the new quirk was having it (its have #2049) but then looking it not having released in the last HA (quirk ver 0.0.90 is having it). You can download the current quirk and putting in custom quirks or patching the HA container https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts001x.py.
> 
> If doing container patch its being no problem with next HA update with it in but if making custom quikrs you shall deleting the custom quirk then its being in HA so its not overriding the HA ones that is getting fixes in the future.

yeah I did patch it manually.
I have HA core 2022.12.3... in which version is the pr already implemented? As you said it shouldn't be a problem when I'll update now

MattWestb, 2023-01-12:
I have looking and its in HA 2023.1 https://github.com/home-assistant/core/pull/85004.
`class Tuya_Double_Var05(EnchantedDevice, TuyaSwitch):`
Updating HA and it shall working but you need re adding all switches new for it to working OK.
If you have one custom quirk installed deleting it or its overriding the HA ones.

guidocioni, 2023-01-12:
> I have looking and its in HA 2023.1 [home-assistant/core#85004](https://github.com/home-assistant/core/pull/85004). `class Tuya_Double_Var05(EnchantedDevice, TuyaSwitch):` Updating HA and it shall working but you need re adding all switches new for it to working OK. If you have one custom quirk installed deleting it or its overriding the HA ones.

mmm I'm not sure I'll have to re add the switches because the code will basically stay the same so it should work as it works now, but let's see what happens 😃

MattWestb, 2023-01-12:
Its depends of the device.
Some need the spell then the device is on INIT mode (like TS004F remotes) that is the first mlli seconds its joining the network and other is happy is getting it later like tuy / LIDL power strips.
Test if its working and if not  testing deleting one switch and adding it new.

guidocioni, 2023-01-17:
@MattWestb do you have these switches installed at your place? They've been working fine but what I've noticed is that, when pressing the physical switch, there is some delay before the light turns on (like almost a second). Is that normal? It happens on all the ones that I own and I'm pretty sure it's not related to the ZHA integration

MattWestb, 2023-01-17:
I only having TS0004 (quad switch) and one TS110F (dual dimmer module).
If you is switching the switch local and its delaying its not from the Zigbee network or host system it must being lazy device that is sleeping or doing somthing else.

Have you getting the switching from ZHA working with only switching the right gang or is it still all together ?

PR #1967 (2022-11-29): New `TS0012` variation
New quirk for `TS0012` devices.
Will support `_TZ3000_jl7qyupf`
Fixes: #1898
commit 05ee5195 (2022-11-29): New `TS0012` variation (#1967)

New quirk for `TS0012` devices.
Will support `_TZ3000_jl7qyupf`
Fixes: #1898