zigpy/zha-device-handlers #1900: [BUG] Saswell SEA802-Zigbee only showing off button

jzielke84, 2022-11-06:
**Describe the bug**
After adding my thermostat the climate card - aside of the temperature control - only shows the `off`-button. When I click on it, the thermostat is turned off and I can't control it anymore from HA. I have to manually set the thermostat to a specific temperature instead to switch the `hvac_mode` to `heat`. I tried overriding this with a customized yaml but then I get the following error while clicking on the `heat` button:

```shell
Nov  6 14:03:46 homeassistant homeassistant[909]: #033[33m2022-11-06 14:03:46.293 WARNING (MainThread) [homeassistant.components.zha.entity] climate.thermostat_thermostat: can't set 'heat' mode. Supported modes are: [<HVACMode.OFF: 'off'>]#033[0m
````

**To Reproduce**
Steps to reproduce the behavior:
1. Pair thermostat
2. set temperature
3. click`off`-button
4. try to set temperature again. It won't change.

**Expected behavior**
Offer a working `off` and `heat` button.

**Screenshots**
[image]

<details>

<summary>Device signature</summary>

```yaml
IEEE: 84:ba:20:ff:fe:3c:e6:02
Nwk: 0xc598
Device Type: EndDevice
LQI: Unbekannt
RSSI: Unbekannt
Zuletzt gesehen: 2022-11-06T13:54:12
Energiequelle: Battery or Unknown
Eigenart: zhaquirks.tuya.ts0601_trv_sas.Thermostat_TZE200_c88teujp ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.11.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.10.0-19-amd64",
    "supervisor": "2022.11.dev0401",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "20.10.21",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.28.3",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    }
  },
  "integration_manifest": {
[... 314 lines cut ...]
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.

Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

MattWestb, 2022-11-06:
The Saswell SEA802 is not updated with more function then basic from it was added for nearly 2 years ago.
Our users have trying fixing it and getting all function from this devices and its one working custom quirk for it but is not being merged in ZHA then the code is failing some test (code).

Configure local quirk in (Z)HA and adding the quirk from the post and restart HA
https://github.com/zigpy/zha-device-handlers/issues/1189#issuecomment-1287512365.
Also reading the thread for more information.

jzielke84, 2022-11-06:
@MattWestb Thanks for pointing that out. I've read the whole thread and the quick seems to be broken. After adding it and re-pairing my thermostat, now I see a bunch of switches where I can only guess what they do. Why is this so difficult where zigbee2mqtt has no problems with that at all?
[image]

MattWestb, 2022-11-06:
´ZHA can only using real Zigbee devices that is Zigbee compatible so we can only using the standard things and not investing new one then we is braking  the system.
Its possible writing custom GUI things but no user have not doing it for tuya TRVs.
TRVs that is more standard is no problem in ZHA but tuya is zero Zoigbee standard and we is trying doing the best of it.

If you is reading the first post in the issue you is having the matrix wot all switches is and can renaming them.
```
Since currently we are not able to change names of on_off entities below is the description:

Window Detection
Child Lock
Anti-Freeze
Limescale Protection
Schedule Mode On/Off
Away Mode
```
Have you testing if the TRV functionality is OK ?

jzielke84, 2022-11-06:
@MattWestb 
> but tuya is zero Zoigbee standard and we is trying doing the best of it.
Yeah, that's what I have already learned while building my own system. I started with openhab and zigbee2mqtt first, but later changed to HA because this software really interacts with the setup process very well. Even though Linux is my daily business I sometime prefer not to write every config file by hand ;-)

Also, it's a pity what the European zigbee-market is offering right now. So many products having 'zigbee' written on their package but using proprietary implementations (like Bosch) or some EA-style subscription moneymaker licensing (like Tado). At least, Zigbee imho seems to be the best protocol right now aside from zwave where at least we have _some_ open standard...

The quirk seems to be working fine. Thanks for telling me the correct order of the switches. I have had some very steep learning curves in the past few days from both automation systems and sometimes I'm getting tired from reading doc after doc.

I'll close this issue as solved. Thank you for you help!

closed 2022-11-06 by jzielke84

MattWestb, 2022-11-06:
I was having one RaspBee I but is in the black box bad devices now then the company is not supporting there sold products you have paying for.

Its 2 versions of the Sonoff Zigbee 3 stick one with TI chip and one newer with Silabs EZSP both shall being OK but i think they still have some problems with it but i think it shall being fixed in upcoming firmware releases.

Some clue for not so bad Zigbee devices you can finding if they have being certificated at https://csa-iot.org/csa-iot_products/?p_keywords=&p_type%5B%5D=14&p_program_type%5B%5D=966&p_certificate=&p_family= but is no grantee that they is working 100% then can still using custom cluster and attributes that we is not knowing or bad hard and / or software (OSRAM).

tuya have 2 types of Zigbee devices most is normal Zigbee devices that can working OK or need some patching for working OK (like most LIDL devices) and some that is using one zuya MCU and is only using Zigbee for tunneling tuya DP commands to the tuya cloud. You device is the last type and its not documented and must being squeezed to one "normal Zigbee device" for working OK in ZHA.

If having problems keep asking its lot of brave users with different experience and some great devs that is helping then have time here and in the end most is being fixes then its possible.
And we is learn new things all the life :-)

jzielke84, 2022-11-06:
@MattWestb Just one last question in advance: would you recommend using tuya devices out of the box or use https://github.com/ct-Open-Source/tuya-convert ?

MattWestb, 2022-11-06:
The link is using ESP or other WiFI chips and tunneling to the cloud the same way your device is doing but your device cant being converted then its using one Zigbee module and the MCU is locked to the type of radio module (Its little different protocol steering / init the modules).

If using tuya WiFi device with ESP i have converting them to ESPHome but i dont have that.

And tuya is saying over 11 000 devices and many is using  Zigbee for tunneling and is only using different DP (tuya commands) for all devices you is understanding wi is lost until some is getting the right commands in place.

Lights i using mostly IKEA then they is updating the device firmware (most no name and also known name is not doing that) but good sensors that is not very expensive you only finding CE ones (CE = China Expert and **not** Certificate European).