zigpy/zha-device-handlers #1901: [Device Support Request] TS0601 _TZE200_h4cgnbzg

chrisius89, 2022-11-06:
**Is your feature request related to a problem? Please describe.**
I've bought multiple trvs "hama smart thermostat". Two of them do not appear as trv in ZHA and are only of type "device".

After modifiying ts0601_trv_sas.py (according to https://github.com/zigpy/zha-device-handlers/issues/819#issuecomment-800488953) the trv is recognized and does show me current and wanted temperature, but im not able to control ist e.g. turn it off or modifiy the temperature.

Here are the zigbee signatures:
<details>
<summary>Device signature working</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>,
	complex_descriptor_available=0,
	user_descriptor_available=0,
	reserved=0,
	aps_flags=0,
	frequency_band=<FrequencyBand.Freq2400MHz: 8>,
	mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>,
	manufacturer_code=4098,
	maximum_buffer_size=82,
	maximum_incoming_transfer_size=82,
	server_mask=11264,
	maximum_outgoing_transfer_size=82,
	descriptor_capability_field=<DescriptorCapability.NONE: 0>,
	*allocate_address=True,
	*is_alternate_pan_coordinator=False,
	*is_coordinator=False,
	*is_end_device=True,
	*is_full_function_device=False,
	*is_mains_powered=False,
	*is_receiver_on_when_idle=False,
	*is_router=False,
	*is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0201",
        "0xef00"
      ],
      "out_clusters": [
        "0x0001",
        "0x000a",
        "0x0019",
        "0x0201"
      ]
    }
  },
  "manufacturer": "_TZE200_yw7cahqs",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_trv_sas.Thermostat_TZE200_c88teujp"
}
```

</details>
<details>
<summary>Device signature NOT working</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>,
	complex_descriptor_available=0,
	user_descriptor_available=0,
	reserved=0,
	aps_flags=0,
[... 472 lines cut ...]
                "value": 1870
              },
              "0x0012": {
                "attribute_name": "occupied_heating_setpoint",
                "value": 1800
              },
              "0x001c": {
                "attribute_name": "system_mode",
                "value": 4
              },
              "0x001e": {
                "attribute_name": "running_mode",
                "value": 4
              }
            },
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

javicalle, 2022-11-12:
According to Z2M it seems that this is the quirk for your device.
Have you tried to remove and pair the device again? If not, remove the device, let pass one minute and pair the device again.

MattWestb, 2022-11-12:
The device shall being supported with basic functionality for the working ones.
Then adding the new device ID in the quirk the new shall also working OK if the firmware is not changed functionality (tuya MCU firmware).
ZHA is sometime not loading the quirk OK so try restarting HA and see if the device is having the same functionality.

If you like getting more functionality you can trying one extender quirk but its not merged in the main repro so need installing it as local quirk https://github.com/zigpy/zha-device-handlers/issues/1189.

BigBadPhysicist, 2022-12-06:
I have the same hardware with the same signature.

I added the custom quirk and can now get the correct measured temperatur, but only get Mode off and Heat in the ZHA integration.

If I try to change the Target Temperatur nothing changes.

I have the logs attached if i try to set the target temp and hvac mode via automation.

`
2022-12-06 20:26:31.786 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=automation, service=trigger, service_data=entity_id=automation.anti_freeze_arbeit, skip_condition=True>
2022-12-06 20:26:31.789 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event automation_triggered[L]: name=Anti-Freeze Arbeit, entity_id=automation.anti_freeze_arbeit>
2022-12-06 20:26:31.790 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=automation.anti_freeze_arbeit, old_state=<state automation.anti_freeze_arbeit=on; last_triggered=2022-12-06T20:01:56.920367+01:00, mode=single, current=0, id=1670333163089, friendly_name=Anti-Freeze Arbeit @ 2022-12-06T19:50:53.461041+01:00>, new_state=<state automation.anti_freeze_arbeit=on; last_triggered=2022-12-06T20:26:31.789988+01:00, mode=single, current=1, id=1670333163089, friendly_name=Anti-Freeze Arbeit @ 2022-12-06T19:50:53.461041+01:00>>
2022-12-06 20:26:31.795 INFO (MainThread) [homeassistant.components.automation.anti_freeze_arbeit] Anti-Freeze Arbeit: Running automation actions
2022-12-06 20:26:31.795 INFO (MainThread) [homeassistant.components.automation.anti_freeze_arbeit] Anti-Freeze Arbeit: Executing step call service
2022-12-06 20:26:31.796 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=climate, service=set_temperature, service_data=temperature=18, hvac_mode=heat, device_id=['7d61eb4c558255c1eb2b325e8b9cc24a']>
2022-12-06 20:26:31.801 DEBUG (MainThread) [zhaquirks.tuya] [0x8ad7:1:0x0201] Mapping standard system_mode (0x001c) with value <SystemMode.Heat: 4> to custom {357: 1, 364: 0}
2022-12-06 20:26:31.802 DEBUG (MainThread) [zigpy.zcl] [0x8AD7:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4417, tsn=161, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-12-06 20:26:31.803 DEBUG (MainThread) [zigpy.zcl] [0x8AD7:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=161, command_id=357, function=0, data=[1, 1]))
2022-12-06 20:26:31.804 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x8AD7), dst_ep=1, source_route=None, extended_timeout=False, tsn=161, profile_id=260, cluster_id=61184, data=Serialized[b'\x05A\x11\xa1\x00\x00\xa1e\x01\x00\x01\x01'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-12-06 20:26:31.805 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (27, 162, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x8ad7 endpoint=1>, 260, 61184, 1, b'\x05A\x11\xa1\x00\x00\xa1e\x01\x00\x01\x01', <DeconzTransmitOptions.USE_APS_ACKS|USE_NWK_KEY_SECURITY: 6>, 0)
2022-12-06 20:26:31.823 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_request[2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 162]
2022-12-06 20:26:31.823 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 162]
2022-12-06 20:26:35.474 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=weather.forecast_zuhause_bad_oey, old_state=<state weather.forecast_zuhause_bad_oey=cloudy; temperature=3.5, temperature_unit=°C, humidity=98, pressure=1017.0, pressure_unit=hPa, wind_bearing=237.4, wind_speed=9.4, wind_speed_unit=km/h, visibility_unit=km, precipitation_unit=mm, forecast=[{'condition': 'rainy', 'datetime': '2022-12-07T11:00:00+00:00', 'wind_bearing': 272.8, 'temperature': 4.3, 'templow': 0.2, 'wind_speed': 19.1, 'precipitation': 0.4}, {'condition': 'cloudy', 'datetime': '2022-12-08T11:00:00+00:00', 'wind_bearing': 234.2, 'temperature': 2.5, 'templow': 0.0, 'wind_speed': 16.6, 'precipitation': 1.4}, {'condition': 'cloudy', 'datetime': '2022-12-09T11:00:00+00:00', 'wind_bearing': 229.7, 'temperature': 1.2, 'templow': -0.7, 'wind_speed': 10.4, 'precipitation': 0.0}, {'condition': 'sunny', 'datetime': '2022-12-10T11:00:00+00:00', 'wind_bearing': 206.0, 'temperature': 1.7, 'templow': -2.4, 'wind_speed': 11.2, 'precipitation': 0.0}, {'condition': 'cloudy', 'datetime': '2022-12-11T11:00:00+00:00', 'wind_bearing': 99.5, 'temperature': 0.8, 'templow': -3.1, 'wind_speed': 9.0, 'precipitation': 0.0}], attribution=Weather forecast from met.no, delivered by the Norwegian Meteorological Institute., friendly_name=Forecast Zuhause Bad Oey @ 2022-12-06T19:23:35.957767+01:00>, new_state=<state weather.forecast_zuhause_bad_oey=cloudy; temperature=2.8, temperature_unit=°C, humidity=97, pressure=1016.8, pressure_unit=hPa, wind_bearing=231.0, wind_speed=10.1, wind_speed_unit=km/h, visibility_unit=km, precipitation_unit=mm, forecast=[{'condition': 'cloudy', 'datetime': '2022-12-07T11:00:00+00:00', 'wind [... cut ...]
2022-12-06 20:26:51.809 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 164: 00
2022-12-06 20:26:51.810 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xa4 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x8ad7 endpoint=1>, status: 0x00
2022-12-06 20:26:51.811 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x8AD7:1:0x0201]: wrote {'occupied_heating_setpoint': 1800} attrs, Status: [[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]`

[image]
look to the device in ZHA

526f73656e0a, 2022-12-13:
I followed the instructions, but am also unable to control the temperature from HA. It gets reset back to 16C after some time

StrangeD0s, 2022-12-18:
I also have the same issue with the same thermostats as @chrisius89 which have not yet been resolved. One is working as expected (_TZE200_yw7cahqs) and the other is not (_TZE200_h4cgnbzg). After adding the custom quirk like @BigBadPhysicist I can get it to show the measured temperatur but trying to set anything has no effect.

lukasmd1989, 2022-12-23:
Recently received the "_TZE200_h4cgnbzg" version of this TRV. It is not correctly working in HomeAssistant using ZHA. No custom quirks solve all issues. 

This is a similar issue about the same model: #1189 

ZHA diagnose file: https://justpaste.it/4o5gv

What can I do to help you create a quirk for this model?

soonic6, 2022-12-28:
same for me, Quirk from #1189 won't work for "_TZE200_h4cgnbzg"

timontsiolis, 2022-12-30:
For me, it is also not working for the "_TZE200_h4cgnbzg". I would really appreciate it if this would be added.

k-dani, 2022-12-31:
With debugging on and changing the temperature i get this communication. I see the target temperature is set via command_id 615 (SASWELL_TARGET_TEMP_ATTR = 0x0267  # uint32 - target temp 615) but it has no effect on the thermostat setting. I can manually dial in any temperature on the hardware knob (reporting current temperatures back to HA is ok) but changing it in home assistant has no effekt at all:

`
2022-12-31 12:07:26.621 DEBUG (MainThread) [zigpy.zcl] [0xF1CE:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4417, tsn=222, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
`

`
2022-12-31 12:07:26.622 DEBUG (MainThread) [zigpy.zcl] [0xF1CE:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=222, command_id=615, function=0, data=[4, 0, 0, 0, 70]))
2022-12-31 12:07:30.542 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xF1CE:1:0x0201]: wrote {'occupied_heating_setpoint': 700} attrs, Status: [[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]
`

Removing and pairing the thermostat again i got several **zigpy.appdb errors** for foreign key constraint violations like this:

`2022-12-31 12:47:13.604 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (a4:c1:38:8c:28:c2:02:fe, 1, 0, 4, '_TZE200_h4cgnbzg') params: FOREIGN KEY constraint failed`

Log section:
[has_pairing_TZE200_h4cgnbzg.log](https://github.com/zigpy/zha-device-handlers/files/10327351/has_pairing_TZE200_h4cgnbzg.log)

RolandoMagico, 2023-01-01:
Hi, as I have now more of the _TZE200_h4cgnbzg devices. I tried zigbee2mqtt together with a Raspbee 2. From the first impression, the work out of the box as expected. I can set the temperature, heat/off/auto mode and child protection/window detection.
Also they are quite "responsive", after adding them to zigbee2mqtt/Home Assistant, the can be used directly without any delay.

MattWestb, 2023-01-01:
Try configure local quirks and adding the device id / model info in this section and restarting HA and look if the system is loading the quirk for your device.
https://github.com/zigpy/zha-device-handlers/blob/071190e7ebaff6e3601c328726be585820f8adb0/zhaquirks/tuya/ts0601_trv_sas.py#L195-L206

timontsiolis, 2023-01-01:
> Try configure local quirks and adding the device id / model info in this section and restarting HA and look if the system is loading the quirk for your device.
> 
> https://github.com/zigpy/zha-device-handlers/blob/071190e7ebaff6e3601c328726be585820f8adb0/zhaquirks/tuya/ts0601_trv_sas.py#L195-L206
I tried to use this custom quirk for "_TZE200_h4cgnbzg" and the behavior is the same. I can see the data from the thermostat but I am not able to change it.

timontsiolis, 2023-01-02:
Can someone confirm that this is working now?

Mytronus, 2023-01-02:
Hey, didn´t work for me. i receive the Stats but can´t sond commands.

MattWestb, 2023-01-03:
I dont have the device but it shall being the same DPs used as the original Sas TRVs but it can being that they have doing one upgraded firmware and doing some changes.
Pleas post debug logs from the device then paring it new and also then changing parameters on the device and in HA so the devs can see what is going wrong.

timontsiolis, 2023-01-03:
[_TZE200_h4cgnbzg.log](https://github.com/zigpy/zha-device-handlers/files/10339051/_TZE200_h4cgnbzg.log)

This is a debug log with every line containing _TZE200_h4cgnbzg or the entity climate.thermostat_kuche_thermostat.  
I reconnected the device, tried to set a value using ha and changed the temperature on the device. 

hopefully this helps to solve this problem.

pticon91, 2023-01-07:
Hello gents,

after few days of debugging **I've managed to make the  _TZE200_h4cgnbzg work**.

## Debug
While sniffing on communication between **TUYA > TRV** and **HASS > TRV** (tested with **_TZE200_h4cgnbzg** and **_TZE200_yw7cahqs** for comparison), I did not find basically any difference in packets, except for **Cluster Library Frame**.

### TUYA > TRV
```
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
        .... ..01 = Frame Type: Cluster-specific (0x1)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 30
    Command: Unknown (0x00)
Data (7 bytes)
    Data: 00096501000101
    [Length: 7]
```

### HASS > TRV
```
ZigBee Cluster Library Frame, Mfr: Telink Micro (0x1141)
    Frame Control Field: Cluster-specific (0x05)
        .... ..01 = Frame Type: Cluster-specific (0x1)
        .... .1.. = Manufacturer Specific: True
        .... 0... = Direction: Client to Server
        ...0 .... = Disable Default Response: False
    Manufacturer Code: Telink Micro (0x1141)
    Sequence Number: 102
    Command: Unknown (0x00)
Data (7 bytes)
    Data: 00666501000101
    [Length: 7]
```
### _TZE200_h4cgnbzg response
The packets above show that TUYA sends the payload with field `Manufacturer Specific: False`, while ZHA is sending this field as `True` . This doesn't seem to have an impact on **_TZE200_yw7cahqs**, but **_TZE200_h4cgnbzg** responds with:

```
Frame 3: 50 bytes on wire (400 bits), 48 bytes captured (384 bits) on interface \\.\pipe\zboss_sniffer_COM3, id 0
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0xb7f2
ZigBee Network Layer Data, Dst: 0x0000, Src: 0xb7f2
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
ZigBee Cluster Library Frame, Command: Default Response, Seq: 102
    Frame Control Field: Profile-wide (0x18)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 1... = Direction: Server to Client
        ...1 .... = Disable Default Response: True
    Sequence Number: 102
    Command: Default Response (0x0b)
    Response to Command: 0x00
    Status: Unsupported Manufacturer Cluster Command (0x83)
```
## Solution (Workaround ?)
As am not very knowledgeable about quirks I was looking for some help on Home Assistant Discord and guy called @puddly point me on [TUYA init file](https://github.com/zigpy/zha-device-handlers/blob/22968954cacaff4fc2b00e02ca1bd5622f637393/zhaquirks/tuya/__init__.py#L329-L366) where this parameter is hardcoded with value `True`. When I changed this value to `False`, the **_TZE200_h4cgnbzg** starts working properly and as a bonus, behaviour of **_TZE200_yw7cahqs** was not affected.

I don't know if the discovery above can be fixed directly in the quirk (this will be probably up to you to find out), but for those who need to control this TRV now you can use it as temporary workaround.

To do this, just edit all `is_manufacturer_specific=True` to `False` (lines 329 - 366, probably editing of server_commands only would suffice) in this file in your instance: `usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py` (am not sure if you can do this in HASS OS)

## Wireshark data
trv1 = _TZE200_yw7cahqs
trv3 = _TZE200_h4cgnbzg

[wireshark_debug.zip](https://github.com/zigpy/zha-device-handlers/files/10366360/wireshark_debug.zip)

MattWestb, 2023-01-07:
@javicalle Is it possible implanting in the tuya INIT so can setting it to true or face from normal quirks ?
Then its possible using it for different device that need manufacture set and other that need it not being set for working OK.

### Great findings and workaround @pticon91 !!!

timontsiolis, 2023-01-08:
@pticon91 Thank you very much for your investigation! I can confirm the workaround is working on my system as well. I am currently running _TZE200_h4cgnbzg and _TZE200_yw7cahqs and get exactly the same behavior of both TRVs.

For anybody having problems editing the file. (As I had, as I am relatively new to homeassistant.)  
I used this [guide](https://developers.home-assistant.io/docs/operating-system/debugging/) to access the file under `usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py`.   
I also had to remove the custom quirk [zha-device-handlers/zhaquirks/tuya/ts0601_trv_sas.py](https://github.com/zigpy/zha-device-handlers/blob/071190e7ebaff6e3601c328726be585820f8adb0/zhaquirks/tuya/ts0601_trv_sas.py#L195-L206) for the TRVs to work correctly.

StrangeD0s, 2023-01-08:
This is great! Thank you @pticon91 for finding this!

I'm not familiar with the development process of HA/ZHA so how likely is it, that this solution will be implemented in the next official update?

Mytronus, 2023-01-08:
Unfortunately, i never had to dive this deep in HA. I managed to connect via SSH over the port 22222 but i can´t find this file to get this working. 
@timontsiolis can you give me a hint, please.

pticon91, 2023-01-09:
@Mytronus there's already a [fix in this custom quirk](https://github.com/zigpy/zha-device-handlers/issues/1189#issuecomment-1374874469). To make it work, you don't need to change the tuya init file. However if you are running the latest HASS, somebody has reported some sort of collision with latest release so bare this on mind.

Mytronus, 2023-01-09:
@pticon91 thanks a lot, didn't saw this.

timontsiolis, 2023-01-09:
@Mytronus is it working for you now?
if you followed every step of the guide you get ssh access without using the addon. after that you need to run `docker exec -it homeassistant /bin/bash` in order to get access to the file. But I think the updated custom quirk is also working and is probably the better solution. Need to test that in the coming days.

StrangeD0s, 2023-01-10:
The new quirk was working fine for me until the Home Assistant update 2023.1.3 today. After the update ZHA was not working at all and I had to disable the custom quirk in the configuration.yaml to make my zigbee devices work again. Of course now my Hama thermostat is not working any more :(. Any thoughts on that matter?

MattWestb, 2023-01-10:
@StrangeD0s If you was having one working quirk and cant finding one updated for your devices you can patching the x-working one your self.
Instruction witch command that need being changed is in this post https://github.com/zigpy/zha-device-handlers/issues/1061#issuecomment-1375212509.

StrangeD0s, 2023-01-10:
> @StrangeD0s If you was having one working quirk and cant finding one updated for your devices you can patching the x-working one your self. Instruction witch command that need being changed is in this post [#1061 (comment)](https://github.com/zigpy/zha-device-handlers/issues/1061#issuecomment-1375212509).

Thanks @MattWestb , I just found the link fixed quirk that @pticon91 mentioned :). This also fixed the issue for me.

k-dani, 2023-01-12:
> @Mytronus there's already a [fix in this custom quirk](https://github.com/zigpy/zha-device-handlers/issues/1189#issuecomment-1374874469). To make it work, you don't need to change the tuya init [...]

I can confirm that this custom quirk https://github.com/zigpy/zha-device-handlers/issues/1189#issuecomment-1376215137 (quirk corrected, jan 09 203) fixed it for me as well and the thermostat _TZE200_h4cgnbzg / TS0601 is running fine for two days now. Thanks for the awesome work to @jacekk015 👍

thprim, 2023-01-12:
I have the trv _TZE200_h4cgnbzg, but the quirk ist not working for setting the temperature. It shows in HA the manual changes at the TRV, but changig the heat from HASS ist still not working despite the several hints and modifications in the published quirk files.
I used also the hint at [https://github.com/zigpy/zha-device-handlers/issues/1189 but with the same result. Nothing helped so far.

javicalle, 2023-01-14:
There are a `NoManufacturerCluster` to fix this kind of issues:
https://github.com/zigpy/zha-device-handlers/blob/cec721eceb85fb7c680b6f468c53610b606355d7/zhaquirks/tuya/__init__.py#L266

It can be implemented like that:
```python
class ManufacturerThermostatCluster(NoManufacturerCluster, TuyaManufClusterAttributes):
    """Tuya manufacturer specific cluster."""
```

That class overwrite the `is_manufacturer_specific=True` to False as require that device.

jacekk015, 2023-01-15:
@javicalle Good to know.

soonic6, 2023-01-21:
will the fix be pushed into zha?

soonic6, 2023-01-23:
will the fix from https://github.com/zigpy/zha-device-handlers/issues/1901#issuecomment-1382965739 migrate into zha?

soonic6, 2023-01-28:
i got two of this, but one of them won't add correctly
[zha-84bcb125a6f47cc533527b469cf30367-_TZE200_h4cgnbzg TS0601-9be245fded1a5e220db897909bcf9a46.json.txt](https://github.com/zigpy/zha-device-handlers/files/10526958/zha-84bcb125a6f47cc533527b469cf30367-_TZE200_h4cgnbzg.TS0601-9be245fded1a5e220db897909bcf9a46.json.txt)

jacekk015, 2023-01-28:
@soonic6 use Saswell quirk file

https://github.com/jacekk015/zha_quirks/blob/main/trv_saswell.py

soonic6, 2023-01-28:
> 

i don't wanna use that quirk, it breakse all my other TZE200_yw7cahqs devices and automations :-/

MattWestb, 2023-01-28:
Then delete this line and the quirk is not being loaded for your  `TZE200_yw7cahqs` devices.

https://github.com/jacekk015/zha_quirks/blob/69b2b08e65999231d91d10afe40b102f89acfff6/trv_saswell.py#L508

soonic6, 2023-01-28:
> Then delete this line and the quirk is not being loaded for your `TZE200_yw7cahqs` devices.
> 
> https://github.com/jacekk015/zha_quirks/blob/69b2b08e65999231d91d10afe40b102f89acfff6/trv_saswell.py#L508

I don't want to sound ungrateful, but i would prefer the default zha quirk. it irritated me, that one of the TZE200_h4cgnbzg is showing me the temperatures, but the other one does't.

[image]

[image]

javicalle, 2023-01-28:
@soonic6 Your device is matching the `Saswell_Thermostat_TZE200` quirk:
https://github.com/zigpy/zha-device-handlers/blob/07f38fc78fe5656e72bc53e8270cce7ec836aff0/zhaquirks/tuya/ts0601_trv_sas.py#L265-L287

You can't force it to match another quirk because the device signature doesn't match (is just another device).
If your device is working wrong then the `Saswell_Thermostat_TZE200` quirk must be fixed or create another quirk for your device but it can't be added in another quirk.

MattWestb, 2023-01-28:
Z2M is having this device as one Saswell TRV.
https://github.com/Koenkk/zigbee-herdsman-converters/blob/84edb807202652de80071437fcefa33c6b205fee/devices/saswell.js#L24
It shall  being in HA 2023.1.

soonic6, 2023-01-29:
> Z2M is having this device as one Saswell TRV. https://github.com/Koenkk/zigbee-herdsman-converters/blob/84edb807202652de80071437fcefa33c6b205fee/devices/saswell.js#L24 It shall being in HA 2023.1.

But both of my TRV's doesn't work. The "Küche" TRV showes temperature, but it can't be change from HA. "Schlafzimmer" doesn't work either. It also missing the temperature as well.
So there must be somethink broken in the ZHA  quirk,

javicalle, 2023-01-29:
> Both Devices are the same, or not?

Yes, it seems to be the same device. Same signature, same quirk applied.

It can be a pairing problem? Have you tried to remove the not working device, wait a minute and pair it again? As a battery device I will suggest to you to keep the device 'awake' during all the pairing proccess.

jacekk015, 2023-01-30:
To correctly verify problem user needs to enable debug logging and restart HA first.
```
logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks: debug
```
Next both TRV's need to be removed from HA and paired like a brand new.
Then we need to wait still 10 minutes for TRV to complete initialization.
Last thing to do is to upload FULL HA log as a file here.
Logs will show if initialization process is the problem.

soonic6, 2023-01-31:
> Last thing to do is to upload FULL HA log as a file here.

okay thank you. didn't know that the complete initialization takes 10 minutes.
now both TRV's showes correctly in HA, thank you!
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/10546779/home-assistant.log)

now there is only the old problem, that you can't change the temperature from the HA UI.

jacekk015, 2023-02-04:
You have errors.
```
2023-01-31 14:07:54.628 DEBUG (MainThread) [zigpy.zcl] [0xD019:1:0xef00] Decoded ZCL frame: ManufacturerThermostatCluster:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
2023-01-31 14:07:54.629 DEBUG (MainThread) [zigpy.zcl] [0xD019:1:0xef00] Received command 0x0B (TSN 177): Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
```
TRV does not accept commands. That's because some Chinese TRV's gives on initialization process wrong Manufacturer ID.
Then the quirk try to send commands with this wrong ID and TRV responds back that they doesn't match.

Solution is to change the code to just doesn't send any Manufacturer ID with commands sent to TRV.
Since you don't want to use my custom quirk you need to wait for the official bugfix.
@javicalle should know the fix.
https://github.com/zigpy/zha-device-handlers/issues/1901#issuecomment-1382965739

Funny question - did you bought those TRV together?? Since one is working correct you say.

soonic6, 2023-02-04:
> Funny question - did you bought those TRV together?? Since one is working correct you say.

yes, both are new and official Hama products ( https://de.hama.com/00176592/hama-smartes-heizkoerperthermostat-fuer-hama-heizungssteuerung-wlan / https://zigbee.blakadder.com/Hama_00176592.html ) i buyed from computeruniverse.net .

soonic6, 2023-02-05:
> Solution is to change the code to just doesn't send any Manufacturer ID with commands sent to TRV.
> Since you don't want to use my custom quirk you need to wait for the official bugfix.
> @javicalle should know the fix.
> [#1901 (comment)](https://github.com/zigpy/zha-device-handlers/issues/1901#issuecomment-1382965739)

Okay, thanks for help and clarification. I am using HA as a docker container, so i can't use that fix. Hope for an offical fix.

MattWestb, 2023-02-05:
Saswell TRV is being around for around 3 years now and with different hardware config and more then likely is the last batch getting one new Zigbee module or firmware on it that is behaving different. 
The interesting thing is if all with the `_TZE200_h4cgnbzg` is needing the  `NoManufacturerCluster` or if some is not working with it then its one very bad situation then we cant see the difference of them. Perhaps they is having different software build but then we need all parameters read from the basic cluster from both versions and look if we can finding some differences that we can using for making 2 device classes that working OK for both models.
@soonic6 can you reading all attribute on the basic cluster of both new and old version and look if you is getting the same values from then and if what is different ?

javicalle, 2023-02-05:
> @soonic6 can you reading all attribute on the basic cluster of both new and old version and look if you is getting the same values from then and if what is different ?

I believe that you can get all this information from the device's diagnosis

MattWestb, 2023-02-05:
M2

soonic6, 2023-02-05:
> @soonic6 can you reading all attribute on the basic cluster of both new and old version and look if you is getting the same values from then and if what is different ?

> I believe that you can get all this information from the device's diagnosis

you man that ones:
working (Küche): 
[zha-84bcb125a6f47cc533527b469cf30367-_TZE200_h4cgnbzg TS0601-a61e884ccf40a10e0ea70e12e27a099f.json (1).txt](https://github.com/zigpy/zha-device-handlers/files/10610670/zha-84bcb125a6f47cc533527b469cf30367-_TZE200_h4cgnbzg.TS0601-a61e884ccf40a10e0ea70e12e27a099f.json.1.txt)

Problems: (Schlafzimmer): 
[zha-84bcb125a6f47cc533527b469cf30367-_TZE200_h4cgnbzg TS0601-9be245fded1a5e220db897909bcf9a46.json (1).txt](https://github.com/zigpy/zha-device-handlers/files/10610671/zha-84bcb125a6f47cc533527b469cf30367-_TZE200_h4cgnbzg.TS0601-9be245fded1a5e220db897909bcf9a46.json.1.txt)

javicalle, 2023-02-05:
The 'schlafzimmer' is unavailable and `last_seen` at 02/02.
Could you try to pair it again and download the diagnosis once repaired?. Try removing the device from HA and wait for a couple of minutes before pairing it againg.

Some diferences:
[image]
[image]

soonic6, 2023-02-05:
i don't know, what i am doing wrong.
i deleted both, waited 15minutes... then i paired both. "küche" got an empty Log, but "schlafzimmer" had still his whole history in the device Log.

Here are both Diagnostics:
Küche: 
[zha-84bcb125a6f47cc533527b469cf30367-_TZE200_h4cgnbzg TS0601-a61e884ccf40a10e0ea70e12e27a099f.json (1).txt](https://github.com/zigpy/zha-device-handlers/files/10610946/zha-84bcb125a6f47cc533527b469cf30367-_TZE200_h4cgnbzg.TS0601-a61e884ccf40a10e0ea70e12e27a099f.json.1.txt)

Schlafzimmer:
[zha-84bcb125a6f47cc533527b469cf30367-_TZE200_h4cgnbzg TS0601-9be245fded1a5e220db897909bcf9a46.json (1).txt](https://github.com/zigpy/zha-device-handlers/files/10610950/zha-84bcb125a6f47cc533527b469cf30367-_TZE200_h4cgnbzg.TS0601-9be245fded1a5e220db897909bcf9a46.json.1.txt)

javicalle, 2023-02-05:
Now both devices are similar in attributes:
[image]

But there are less than before... maybe it need to keep awake during the whole pairing proccess?

Anyway, if the problem is related to the `NoManufacturerCluster` it would be necessary to validate before create a PR and then cross fingers that don't break old devices behavior.

MattWestb, 2023-02-05:
@soonic6 Can you taking out the battery so the device is being dead and putting it back so the MCU is sending all commands its supporting and also populating the missing attributes ?
I dont know if its helping here but its looks having problem reading the device properties then paring. 
Can you also looking one the network map after forcing it updating and waiting 2 minutes witch parent they is having if the coordinator or some good or bad routers ?
Also the brand / model of the parent router can helping here.
Then tuya TRVs is sleeping end devices they is sleeping most of the time so can being little lazy on responding but if the parent is bad behaving its not working at all with them.

scolastico, 2023-02-08:
Im coming from the [forum](https://community.home-assistant.io/t/zigbee-smart-radiator-hama-tuya-ts0601-is-read-only/532183) and have the same issue with a new _TZE200_h4cgnbzg. I also have some old one which work as expected. Here my diagnostics if it helps: 

https://www.diffchecker.com/u9WB2olC/

The “original text” (on the left side) is the new one which is read only.
The “changed text” (on the right side) is the old one which works as expected.

javicalle, 2023-02-08:
Devices are showing diferent `manufacturer_code` so my guess would be a `NO_MANUFACTURER_ID` issue.

My proposal would be:
1. Enable the local quirk
2. Copy the current `ts0601_trv_sas.py` file:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_trv_sas.py
3. Update the `ManufacturerThermostatCluster` class like this (insert the new `write_attributes` method):
```python
    .../...
    TEMPERATURE_ATTRS = {
        SASWELL_HEATING_SETPOINT_ATTR: "occupied_heating_setpoint",
        SASWELL_LOCAL_TEMP_ATTR: "local_temperature",
    }

    async def write_attributes(self, attributes, manufacturer=None):
        """Overwrite to force manufacturer code."""

        return await super().write_attributes(
            attributes, manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID
        )

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid in self.TEMPERATURE_ATTRS:
            .../...
```

4. Save changes, delete devices and restart HA
5. Pair the devices again

Please test the quirk for both device 'versions' to be sure that doesn't break anything.

scolastico, 2023-02-09:
@javicalle I would love to try that but do you have any clue where the file is stored on [hassio os](https://www.home-assistant.io/installation/raspberrypi)?
I tryd to search for it (eg `sudo find / -name "ts0601_trv_sas.py"`) but got no results.

soonic6, 2023-02-10:
So, now i removed the batterie from both TRVs and waited 10 minutes. Then i restarted HA and waited till its booted completly.
Then i started the pairing process. First for "Küche", then "Schlafzimmer".
Now, after 10 minutes here are the Logs from HA and both TRVs:

[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/10706598/home-assistant.log)

Küche:
[image]
[zha-84bcb125a6f47cc533527b469cf30367-_TZE200_h4cgnbzg TS0601-a61e884ccf40a10e0ea70e12e27a099f.json (1).txt](https://github.com/zigpy/zha-device-handlers/files/10706603/zha-84bcb125a6f47cc533527b469cf30367-_TZE200_h4cgnbzg.TS0601-a61e884ccf40a10e0ea70e12e27a099f.json.1.txt)

Schlafzimmer: 
[image]
[zha-84bcb125a6f47cc533527b469cf30367-_TZE200_h4cgnbzg TS0601-9be245fded1a5e220db897909bcf9a46.json (1).txt](https://github.com/zigpy/zha-device-handlers/files/10706614/zha-84bcb125a6f47cc533527b469cf30367-_TZE200_h4cgnbzg.TS0601-9be245fded1a5e220db897909bcf9a46.json.1.txt)

hope this will help.

MattWestb, 2023-02-10:
@javicalle This TRV is using the new Telink chips is the Zigbee module that can behaving little different git the MCU shall working the same if its being on the same firmware as other ones.

[49 more replies]

PR #2052 (2023-01-02): Adding 5 "found" Sas TRVs
Fixes #1901
commit 5b7a3519 (2023-01-02): Adding 5 "found" Sas TRVs (#2052)

Fixes #1901
closed 2024-07-20 by github-actions[bot]