zigpy/zha-device-handlers #1904: [Device Support Request] TZ3000_xkap8wtb

KaherdinTristan, 2022-11-07:
**Is your feature request related to a problem? Please describe.**
The device; TZ3000_xkap8wtb, is reported under the domain "LIGHT". Also the device turns off if the signal to the associated mesh is too weak.  

**Describe the solution you'd like**
The correct device description has already been created
https://pullanswer.com/questions/tz3000_xkap8wtb-paired-as-ts0001#7107603

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_xkap8wtb",
  "model": "TS0001",
  "class": "zhaquirks.tuya.ts000x.Switch_1G_Metering"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Stockholm",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.10.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "fontawesome": {
      "version": "2.1.5",
[... 649 lines cut ...]
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.rmsVoltage(endpoint, {change: 5});
        await reporting.rmsCurrent(endpoint, {change: 50});
        await reporting.activePower(endpoint, {change: 10});
        await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE)],
};

module.exports = definition;

save this as file next to configuration.yaml as ext_converter.js
add it to configuration.yaml:
external_converters:
  - ext_converter.js
start z2m, check if issue is fixed```

</details>

**Additional context**

KaherdinTristan, 2022-11-07:
I am not in any way shape or form accustomed to debugging zigbee devices, but I'll do my best to help out with anything, just as long as You help me with describing what I should do... I'm currently running HA on RPi4, and using a Conbee 2 as coordinator. I have a bunch of zigbee devices, and this (and one similar) is really bugging me out.

KaherdinTristan, 2022-11-07:
This thread also talks about some sort of sollution
https://github.com/Koenkk/zigbee2mqtt/discussions/14214?notification_referrer_id=NT_kwDOADPbwrI0NDg3MDAyMjU4OjMzOTg1OTQ#discussioncomment-3829787

javicalle, 2022-11-08:
> This thread also talks about some sort of sollution

The thread explains how update the device firmware. This can't be done in ZHA (not yet) and will require a Tuya gateway to update the firmware (if there is some firmware update available, I don't know).

There is a full discussion about that issue:
* https://github.com/Koenkk/zigbee2mqtt/discussions/13343

Reading above it seems that there is a design problem that with a firmware update may or may not be solved (this part is not clear to me).

In any case, it doesn't seem to be a problem that can be fixed with a quirk.

KaherdinTristan, 2022-11-09:
The device defaults to domain LIGHT but it should default to domain SWITCH. 
I also think (haven't checked this entirely yet) that one or two "accumulated power consumption" 'entities'/reported values are wrong by a factor of 10. 

```
manufacturer: _TZ3000_xkap8wtb,
model: TS0001
name: _TZ3000_xkap8wtb TS0001
```

[image]

KaherdinTristan, 2022-11-10:
Has power metering. 
Is getting random "turned off" events. 
[2022-11-10_zha-e56ec98032982a689e88dc2163f8471d-_TZ3000_xkap8wtb TS0001-b1c9c9ad351532cfe672157b7eda0f36.json.txt](https://github.com/zigpy/zha-device-handlers/files/9981041/2022-11-10_zha-e56ec98032982a689e88dc2163f8471d-_TZ3000_xkap8wtb.TS0001-b1c9c9ad351532cfe672157b7eda0f36.json.txt)

MattWestb, 2022-11-10:
The same as the no power metering device plus if its getting to much voltage its shutting down it self for protecting the equipment. Also some is reporting zero in voltage in the voltage readings little now and then.
Pleas config debut logging in (Z)HA and post the log part then you is getting the unwanted turned offs.

Can you posting  the complete or the firs 6 numbers of the device IEEE (you finding it under device info on the device card) then im curious with chip tuya is using for this device.

KaherdinTristan, 2022-11-10:
**Device _TZ3000_xkap8wtb WITH power metering.** 
Constantly turning off. 
```
TS0001
by _TZ3000_rmjr4ufz
Connected via [Zigbee Coordinator](http://10.0.0.64:8123/config/devices/device/7b84426a50a9c164cf76c0b0f8c7f856)
Zigbee info
IEEE: a4:c1:38:dd:a5:6f:3d:07
Nwk: 0xd3c7
Device Type: Router
LQI: 223
RSSI: -66
Last Seen: 2022-11-10T15:58:12
Power Source: Mains
Quirk: zhaquirks.tuya.ts000x.Switch_1G_Metering
```

KaherdinTristan, 2022-11-10:
I have the latest update fw
[image]

MattWestb, 2022-11-10:
I think you is having one bad device that is having firmware problems.
One issue that  is converted to one discussion https://github.com/Koenkk/zigbee2mqtt/issues/11648 and all devices is having Telink chips.

You can testing if your device can being updated with the firmware in the issue but on own risk.