zigpy/zha-device-handlers #1905: [Device Support Request] TZ3000_rmjr4ufz

KaherdinTristan, 2022-11-07:
**Is your feature request related to a problem? Please describe.**
Device TZ3000_rmjr4ufz reported into domain "LIGHT". 
It is a wall switch (without energy metering) 

**Describe the solution you'd like**
The device to be identified as domain LIGHT. 

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_rmjr4ufz",
  "model": "TS0001",
  "class": "zhaquirks.tuya.ts000x.Switch_1G_GPP"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Stockholm",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.10.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "fontawesome": {
      "version": "2.1.5",
      "requirements": []
    },
[... 513 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

KaherdinTristan, 2022-11-07:
I am not in any way shape or form accustomed to debugging zigbee devices, but I'll do my best to help out with anything, just as long as You help me with describing what I should do... I'm currently running HA on RPi4, and using a Conbee 2 as coordinator. I have a bunch of zigbee devices, and this (and one similar) is really bugging me out.

KaherdinTristan, 2022-11-07:
This thread alo talks about some sort of sollution
https://github.com/Koenkk/zigbee2mqtt/discussions/14214?notification_referrer_id=NT_kwDOADPbwrI0NDg3MDAyMjU4OjMzOTg1OTQ#discussioncomment-3829787

KaherdinTristan, 2022-11-07:
Im sorry, but I donät understand you. 
Where is the file?

javicalle, 2022-11-08:
> **Is your feature request related to a problem? Please describe.** Device TZ3000_rmjr4ufz reported into domain "LIGHT". It is a wall switch (without energy metering)
> 
> **Describe the solution you'd like** The device to be identified as domain LIGHT.

Is already identified as `light` or isn't it?
Which is the problem here?

KaherdinTristan, 2022-11-09:
> Is already identified as `light` or isn't it? Which is the problem here?

The device defaults to domain light. 
But the device is a switch. 

"manufacturer": "_TZ3000_rmjr4ufz",
"model": "TS0001",
"name": "_TZ3000_rmjr4ufz TS0001"

[image]

MattWestb, 2022-11-09:
@javicalle _think i knowing what the problem is.
One switch in ZHA is it possible changing how the device is being shown like one of my IKEA outlets:
[image]
And if changing to outlet i getting one power cord plug icon.
The outlet is "device_type": "0x010a" =  `ON_OFF_PLUG_IN_UNIT`.
Lights is locked and cant being changed and i think (not tested) if changing it to one light it cant being changed (I getting  one warning in ZHA).

tuya is using `DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT` but i think all of them is switches (relays) for manpower so perhaps it wrong and shall being changed for all devices in the `ts000x.py`.
ts011f plugs and switches is using `ON_OFF_PLUG_IN_UNIT` or `MAIN_POWER_OUTLET` = OK.

I think we need some OK from maintainers for doing then can braking some automatons.

The device looks having loading the quirk OK so i think its working OK.

javicalle, 2022-11-10:
I see the point.
No dimmers in `ts000x`, just `TuyaZBExternalSwitchTypeCluster` and `TuyaZBOnOffAttributeCluster`  (also some metering and electricalmeasurement).

The custom attributes defined here are:
* external_switch_type
* child_lock
* backlight_mode
* power_on_state
* switch_mode

All of them pretty related to switches (more than to lights), so I also believe that would be fine to configure all the `ts000x` devices as switches.
@dmulcahey any thoughts? Which DeviceType would be appropiate?

KaherdinTristan, 2022-11-10:
Since I don't fully understand the coding or technical side of zigbee or ZHA, I am just going to ask;
- Is there a way of debugging what causes the device to turn off at "random times"? 
- Is there a way to prevent this from happening, by changing a quirk, or to edit some code? 

[image]

dmulcahey, 2022-11-10:
> I see the point. No dimmers in `ts000x`, just `TuyaZBExternalSwitchTypeCluster` and `TuyaZBOnOffAttributeCluster` (also some metering and electricalmeasurement).
>
> The custom attributes defined here [...]

Ugh… I just commented on another issue saying we wouldn’t change this… but I’m torn. Maybe we need a standard here.

@Adminiuga @puddly do you have any thoughts? 

i do agree that switch is more appropriate if there is no level control cluster but I don’t yet see a clean way to automatically do this across the board that won’t have the opposite effect on bulbs that don’t support dimming (potentially making correctly defined lights become switches)

MattWestb, 2022-11-10:
Those devices is quirked and can being fixed in the replace part only need one OK doing the "transformation".

One general with all light domain with no color and / or level cluster is better but then its more one ZHA issue.

MattWestb, 2022-11-10:
@KaherdinTristan  Hej Andres !
Can you posting one diagnostic from the device how its now (i like to knowing if its have loading the quirk) ?

And is one device with power reading and its have protection that can being triggered false in some tuya device but we is looking later for that.

dmulcahey, 2022-11-10:
> Those devices is quirked and can being fixed in the replace part only need one OK doing the "transformation".
> 
> One general with all light domain with no color and / or level cluster is better but then its more one ZHA issue.

I understand that. What I don’t want to do is have inconsistencies AND I don’t want to have a quirk for every device just to change its device type if functionality just works

KaherdinTristan, 2022-11-10:
Diagnostics;
[2022-11-10_zha-e56ec98032982a689e88dc2163f8471d-_TZ3000_rmjr4ufz TS0001-2f9faf016439f7e83b952b0d0af6b808.json.txt](https://github.com/zigpy/zha-device-handlers/files/9981013/2022-11-10_zha-e56ec98032982a689e88dc2163f8471d-_TZ3000_rmjr4ufz.TS0001-2f9faf016439f7e83b952b0d0af6b808.json.txt)

**This** device, the "_TZ3000_rmjr4ufz" has **NO** power metering, and right now I'm **not** getting any random "turned off" events for it. 

_Please also see my other Issue here https://github.com/Koenkk/zigbee2mqtt/discussions/14214 for a similar device I have and also facing issues with_

MattWestb, 2022-11-10:
@KaherdinTristan OK then is not one power shutdown problem.
The device is one full router so it shall normally not having problems like sleeping device is / can having.
If the mesh network is good configured and all router firmware is working OK the network shall being stable.
You is running your Sonoff stick on one very edge firmware (EZSP 7.X) and we have not testing it so well then its not supported by all hardware and have some problems that need being fixed.
That is very OK but if you is getting problem set up debug in (Z)HA and posting the logs for the devs to looking in if can finding any interesting.

It can being the the device firmware is getting problem (crashing) and restarting then you is getting one device accouterments in the log and then is the setting in the device if its starting in On, Off or last state.
Also possible can being radio interference that is blocking the network but all is easier if getting logs to digging in.

MattWestb, 2022-11-10:
Both devices is Telink and is one new chip tuya have start using and i have some that working very good but Z2M user have problems with power metering device.
Pleas post one log from HA then the power of is happening so we can see if the device is having problems in the network or its the device itself that is doing it.

Also check that CornBee IIs firmware is up to date https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog.

KaherdinTristan, 2022-11-10:
**Device _TZ3000_rmjr4ufz. In wall switch. No power metering**
- Actually I now checked all my TZ3000_rmjr4ufz devices, and the only issue that I can identify right now is that the device is by default set as domain LIGHT. 

The other issues are with the other https://github.com/zigpy/zha-device-handlers/issues/1904 
Device _TZ3000_xkap8wtb WITH power metering

KaherdinTristan, 2022-11-10:
To be clear; These devices "_TZ3000_rmjr4ufz" have had random turn offs in the past for me. It went so far, that I had to design an automation that detects the turn off, and turns the device back on. But right now, for the last three days, I have had no such issues. It might have been one of several "reconfiguration" runs that did the trick for them. 

My Conbee 2 is up to date; 
[image]

TheJulianJES, 2023-01-04:
Did you try https://www.home-assistant.io/integrations/zha/#modifying-the-device-type?

closed 2023-07-10 by github-actions[bot]