zigpy/zha-device-handlers #1906: [Device Support Request] TS0601 _TZE200_1n2kyphz Zemismart Tuya 4 Gangs Smart Wall Light Switch 4x4 TB26-4

nohans, 2022-11-07:
**Is your feature request related to a problem? Please describe.**
Device is connected but no entities are exposed, only lqi and rssi.

**Describe the solution you'd like**
Please add support for the device. This is the same device family as the TB26-6 (6-gang version, which was addressed by #1718). Here's the [product link](https://www.zemismart.com/products/tb26-4) for reference.

I have no experience debugging or testing patches for Zigbee devices, but I am willing to help and test in any way I can.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_1n2kyphz",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Unsupported Third Party Container",
    "version": "2022.11.0",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Sao_Paulo",
    "os_name": "Linux",
    "os_version": "5.15.62-051562-generic",
    "run_as_root": false
  },
  "custom_components": {
    "alarmo": {
      "version": "v1.9.6",
[... 407 lines cut ...]
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
No additional info.

javicalle, 2022-11-07:
You will need to enable the local quirk in your instalation. There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Copy the `ts0601_switch.py` file in your local quirk folder:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_switch.py

Edit the file content around line 425, adding your device model like:
```python
class TuyaQuadrupleSwitchTO(TuyaSwitch):
    """Tuya quadruple channel switch time on out cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_aqnazj70", "TS0601"),
            ("_TZE200_1n2kyphz", "TS0601"),
            ("_TZE200_1ozguk6x", "TS0601"),
            ("_TZE200_k6jhsr0q", "TS0601"),
        ],
```

Restart HA, remove and pair you device again. Check that the device is loading the quirk and that the device signature has changed.

nohans, 2022-11-08:
Thanks for the instructions! 

I was able to make it work, but I had to add the device on another class (on line 495, class TuyaQuadrupleSwitch_GP). Should I make a pull request with that change?

PR #1909 (2022-11-09): Add Zemismart Switch 4x4 TB26-4 support
Fixes #1906
closed 2022-11-09 by dmulcahey