zigpy/zha-device-handlers #1907: [Device Support Request] TZ3000_18ejxno0

orso7, 2022-11-07:
**Is your feature request related to a problem? Please describe.**
The device is a 2-Gang (2 button) zigbee switch to be installed in the wall. Pairing it showed both switches as entities but if I change one state (on/off) via the web interface both get turned on. When I push the buttons on the device only the pushed one changes state.

**Describe the solution you'd like**
Having the two switches to be controlled separately through the web interface. 

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_18ejxno0",
  "model": "TS0012",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
[image]

javicalle, 2022-11-07:
Duplicate of #1744

orso7, 2022-11-08:
> Duplicate of #1744 And already implemented in #1827 Which HA version do you have?

I found a link to this git where they told me to search for the device but I might have searched for the wrong ID sorry

Installed Home Assistant some time ago set up a few things and forgot about it, never updated so I am on "core-2021.10.6"

MattWestb, 2022-11-08:
You need one of the latest HA for getting the quirk in the package.
Also if you is running one so old HA is very likely not working installing the quirk as local quirk then some changes is being made that need being fixed for working with old versions of ZHA.

Try updating your HA and testing is its working OK.

If have CLI (its from my windows test so can being different for other OS / installs):
Updating:
`docker stop Home-Assistant`
`docker rm Home-Assistant`
`docker pull homeassistant/home-assistant`
Then install HA Core as you was doing the first time (the same parameters) and it shall working.

If being usecure or need help look in Home-assistant community forms for help with updating HA core.

javicalle, 2022-11-08:
Also you can check in the docs:
https://www.home-assistant.io/common-tasks/container/

orso7, 2022-11-09:
> Duplicate of #1744 And already implemented in #1827 Which HA version do you have?

Updating HA didn't fix it, and so far I am not really understanding the solution in #1744, I am quite inexperienced with HA

MattWestb, 2022-11-09:
@orso7 Can you posting the Device signature **after** you have updated (Z)HA to 2022.11.X ?
If the system is loading the quirk OK then the first step is done and you need deleting the device and wait one minute and adding it agen for getting it working OK (option restarting HA between deleting and adding the device).

orso7, 2022-11-09:
@MattWestb: I updated to the latest Version 

Here is the Device signature
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_18ejxno0",
  "model": "TS0012",
  "class": "zhaquirks.tuya.ts001x.TuyaDoubleNoNeutralSwitch_2"
}`

I could also download the diagnostic info now that I am updated
[zha-fa8c2c9719b636a567165bbde62ac0ae-_TZ3000_18ejxno0 TS0012-52585e7468ac40bad1ca1d225658b491.json.txt](https://github.com/zigpy/zha-device-handlers/files/9969626/zha-fa8c2c9719b636a567165bbde62ac0ae-_TZ3000_18ejxno0.TS0012-52585e7468ac40bad1ca1d225658b491.json.txt)

javicalle, 2022-11-27:
Have you checked to remove and pair the device again?
Is the issue fixed?

closed 2023-07-10 by github-actions[bot]