zigpy/zha-device-handlers #1912: [BUG] Relay switches should register as switch type, not light.

JoryHogeveen, 2022-11-10:
**Describe the bug**
There are many relays switches, few examples:

- Aqara T1
- Tuya TS0001, TS0002, TS0003, TS0004, TS0011, TS0012, TS0013, TS0014
- Sonoff ZBMINI
- etc.

There are currently registering as `light` entities.
I think that this is incorrect since it's a relay, it could switch anything. It's an on/off switch for any type, not necessarily a light.

Especially since HA now supports registering a switch as a different type of entity with the new helpers I think this should definitely be fixed in ZHA.

**Expected behavior**
The relays to be registered as switches.

<details>

<summary>Aqara T1</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x000a",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "31": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "41": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.n0agl1",
  "class": "zigpy.device.Device"
}
```
</details>

<details>

<summary>Tuya TS0002</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_bvrlqyj7",
  "model": "TS0002",
  "class": "zhaquirks.tuya.ts000x.Switch_2G_GPP"
}
```

</details>

dmulcahey, 2022-11-10:
https://github.com/zigpy/zigpy/blob/82b324d4c76b690b815c2c3fa8e6cd00af00fed5/zigpy/profiles/zha.py#L27

The devices identify as lights per their signatures. You can also override the type per entity in HA in the  ZHA configuration. This is probably something we will not change here

JoryHogeveen, 2022-11-10:
@dmulcahey 
Thanks for your quick reply, I didn't realize the device itself would register as a light type.
That explains a lot :)

> You can also override the type per entity in HA in the ZHA configuration.
Did this instead, thanks for the pointer here!

Closing issue.

closed 2022-11-10 by JoryHogeveen

dmulcahey, 2022-11-10:
https://www.home-assistant.io/integrations/zha/#modifying-the-device-type

Just for reference

MattWestb, 2022-11-10:
@dmulcahey That is true for no lights.
For IKEA outlet with type https://github.com/zigpy/zigpy/blob/82b324d4c76b690b815c2c3fa8e6cd00af00fed5/zigpy/profiles/zha.py#L39 its possible changing to many types in the GUI but for lights is blocked and can only changing the icon.

dmulcahey, 2022-11-10:
Not sure what you mean. In the zha config you can override the type which changes entity assignment and then normal HA configuration takes over?

TheJulianJES, 2022-11-10:
"Switch as X" (underlying integration of that function in HA) creates another entity and hides the original one.

Modifying the type via ZHA/YAML changes the original entity.

MattWestb, 2022-11-10:
@TheJulianJES Its depends of what type you is changing to / from.
IKEA plugs to Outlet is only cosmetic and not getting any warning but most you is getting warning changing type = hiding the original and making one new.

One proposal only changing in quirks that is needed for other fixes and not adding new quirks that  is only changing the type. 
Then most devises is right or can being changed in GUI.
If the user like doing more fin-tuning then doing in HA config.