zigpy/zha-device-handlers #1913: [BUG] SmartThings Button sends same command 3 times

alexruffell, 2022-11-10:
**Describe the bug**
I have several SmartThings Buttons M/N IM6001-BTP01 that have always worked flawlessly in ST and Home Assistant. Firmware version is 0x00000014. A while back they started repeating the same command 3 times. I checked the output of zha_event and confirmed it

Event 2 fired 7:36:
```
event_type: zha_event
data:
  device_ieee: 28:6d:97:00:01:04:1e:de
  unique_id: 28:6d:97:00:01:04:1e:de:1:0x0500
  device_id: f59f3b82102504ee70ab649e4300fde9
  endpoint_id: 1
  cluster_id: 1280
  command: button_single
  args:
    press_type: single
    command_id: 0
    args:
      - 1
      - 0
      - 0
      - 0
  params: {}
origin: LOCAL
time_fired: "2022-11-10T13:36:04.166296+00:00"
context:
  id: 01GHGW2EJ6R7HEWYYDDQPQW7B8
  parent_id: null
  user_id: null
```

Event 1 fired 7:36:
```
event_type: zha_event
data:
  device_ieee: 28:6d:97:00:01:04:1e:de
  unique_id: 28:6d:97:00:01:04:1e:de:1:0x0500
  device_id: f59f3b82102504ee70ab649e4300fde9
  endpoint_id: 1
  cluster_id: 1280
  command: button_single
  args:
    press_type: single
    command_id: 0
    args:
      - 1
      - 0
      - 0
      - 0
  params: {}
origin: LOCAL
time_fired: "2022-11-10T13:36:01.806008+00:00"
context:
  id: 01GHGW2C8ECN2VY6SNNKB53263
  parent_id: null
  user_id: null
```

Event 0 fired 7:36:
```
event_type: zha_event
data:
  device_ieee: 28:6d:97:00:01:04:1e:de
  unique_id: 28:6d:97:00:01:04:1e:de:1:0x0500
  device_id: f59f3b82102504ee70ab649e4300fde9
  endpoint_id: 1
  cluster_id: 1280
  command: button_single
  args:
    press_type: single
    command_id: 0
[... 482 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

MattWestb, 2022-11-10:
Witch Zigbee radio chip / model / brand is you Zigbee dongle ?
Also can you posting one log from (Z)HA then its happening with debug logging enable in HA ?

alexruffell, 2022-11-10:
@MattWestb I am using a Sonoff Zigbee 3.0 USB Dongle Plus with the TI CC2652P chip. I changed the logging for ZHA to debug, rebooted, pressed the button once and saw the 3 commands on zha_event, turned off debug and rebooted again. I then took the backed up log file (log.1) and pasted it here. Hope this is what you wanted.

[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/9982338/home-assistant.log)

MattWestb, 2022-11-10:
Im missing the communication with the device that need `zigpy: debug` also `zigpy_znp: debug` can being good but you dont need ZHA debug then its only spamming the log.
Also the NWK address of the device is good so not need trying finding it log.

alexruffell, 2022-11-10:
@MattWestb Below is the new log. I pressed the button more than once as I wasn't seeing the event... hopefully that doesn't confuse things. You should find presses in multiples of 3. ID is 0x6573

[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/9984701/home-assistant.log)

MattWestb, 2022-11-10:
### Great logging !!
```
2022-11-10 15:20:20.320 DEBUG (MainThread) [zigpy.zcl] [0x6573:1:0x0500] Received command 0x00 (TSN 9): status_change_notification(zone_status=<ZoneStatus.Alarm_1: 1>, extended_status=<bitmap8.0: 0>, zone_id=0, delay=0)
```
Also 15:20:22.091 and 15:20:24.260 the same frame with the same TSN and its one broadcasts to GroupId=0x0000.

@puddly Is not TI filtering group broadcast with the same TSN ?
I dont know if its one direct child or the device is having one no Zigbee 3 parent -> the parent is transmitting the frame 3 times as one ZHA router and not using  passive ack by listening that the neighbors have re sending the frame.

Is IAS Zone using broadcast in one different way as the light is doing ?

alexruffell, 2022-12-09:
@puddly Have you had a chance to look into this? Even with the latest release the issue is still present.

OliverDudgeon, 2023-01-09:
Started seeing this about a month back. Didn't happen before. I have a Aeotec Samjij Button with a TI chip. Other zigbee devices work normally.

LxonWWW, 2023-05-08:
I have the same problem with one of my Aeotec buttons (same as SmartThings Button). I even got it replaced and it started again 3 days later. The strange thing is, that one of the buttons still works flawlessly (this one was never replaced).

alexruffell, 2023-05-08:
Mine used to work flawlessly and then started having issues after some HA upgrade. Now I know that a single press to turn my LIFX light off will result in off, on, off. I just got used to it...

LxonWWW, 2023-05-08:
> Mine used to work flawlessly and then started having issues after some HA upgrade. Now I know that a single press to turn my LIFX light off will result in off, on, off. I just got used to it...

I have also looked at other zigbee buttons but none come close to this beautiful design while being this small. It's just frustrating to get a light show at night before going to bed. I might switch to the Aqara buttons, even though they don't look nearly as good as the Aeotec ones.

OliverDudgeon, 2023-05-08:
This has stopped being an issue for me now. Not sure what change but after some updates I noticed it wasn't happening anymore. Best of luck folks.

alexruffell, 2023-11-04:
I have several of these ST buttons that I was not using. Now that I am, I noticed that just one keeps doing this on, off, on business so I suspect that the issue is device or network specific. Maybe a bad router in between or a failing button (does poor signal cause it to try more than once?), etc.

loosrob, 2023-12-21:
I'm also experiencing this issue. I noticed that removing and replacing the battery on the Aeotec Smart Button (temporarily) solves the issue.

My HomeAssistant is on the latest version; I'm using the HomeAssistant Skyconnect zigbee stick.

alexruffell, 2023-12-21:
Must be my same issue as mine no longer does it. I likely replaced the battery since then. Thank you for letting me know the possible cause.

closed 2023-12-21 by alexruffell