zigpy/zha-device-handlers #1916: [BUG] Xiaomi Aqara Weather sensor - pressure reported rounded to 0 digits

hajo62, 2022-11-11:
Pressure has zero digits behind full value
﻿
I have moved from zigbee2mqtt to Zigbee Home Automation Integration.
Now my Xiaomi Aqara sensor show the pressure with zero digits behind the full value, but before it had one digit behind the full value.
See my forum [post](https://community.home-assistant.io/t/xiaomi-aquara-sensor-less-precise-after-move-from-zigbee2mqtt-to-zigbee-home-automation-integration/487427).

<details>

javicalle, 2022-11-20:
Let's see how it goes...
Zigbee standar define the mesured values in _kPa_ this way:
[image]

And HA shows the Pressure measurement in _hPa_ (and `_decimals = 0`):
https://github.com/home-assistant/core/blob/38b84620bdbf241a6b246b386dbc1cee21e95472/homeassistant/components/zha/sensor.py#L522-L531

That makes sense because the Zigbee `MeasuredValue` will have a precission of 0.1 _kPa_ or what is the same 1 _hPa_

That means that Z2M is doing something wrong?
Not necessarily. I haven't looked at the Z2M implementation, but let's say hello to the Zigbee `ScaledValue` and `Scale` attributes:
[image]

If with the `MeasuredValue` we can get a precision of $\frac{1}{10} kPa$ or $1 hPa$, the `ScaledValue` we should have a precision of $\frac{1}{10^{scale-1}} hPa$

The bad news is that ZHA don't manage other attributes than the `measured_value`:
https://github.com/home-assistant/core/blob/38b84620bdbf241a6b246b386dbc1cee21e95472/homeassistant/components/zha/core/channels/measurement.py#L72-L78

The good news is that we can ask for a review. 
But this will be just one part of the solution, if the change is done in ZHA, the Xiaomi quirk will require from some ajustement.

hajo62, 2022-11-20:
Thanks for caring.
If I understand right, the zigbee2mqtt integration had _offered_ the ScaledValue (as attribute pressure), but the ZHA integration exposes the MeasuredValue which is _rounded_ less precise?

That's what I did to _calibrate_ the values
```
  mqtt:
     - name: "XiaomiAqara-1 Pressure"
       state_topic: "zigbee2mqtt/XiaomiAqara-1"
       unit_of_measurement: 'hPa'
       value_template: "{{ (value_json.pressure + 10.3) | round(1) }}"
       availability_topic: "zigbee2mqtt/bridge/state"
       device_class: "pressure"
```

javicalle, 2022-11-20:
> the zigbee2mqtt integration had _offered_ the ScaledValue (as attribute pressure), but the ZHA integration exposes the MeasuredValue which is _rounded_ less precise?

I wouldn't say this way.
Z2M allows to expose any precision because allow the user to use a template. That has nothing to do with the Zigbee standard. 
It could be interpreted that the precision of the measurements is 1 hPa, but the standard really allows for greater precision. That part has not been accounted for in ZHA (yet).
My comment was along the lines of the fact that it _might seem_ that the precision offered in Z2M is greater than what the Zigbee standard would allow, and that in that case ZHA would not go beyond the standard.

For now, let's see what can be done in ZHA adhering to what defines the standard. 😉

hajo62, 2022-11-20:
Thanks.
One more question alongside this topic.
My two sensors both are around 10 hPa to low, but differ in mean by 0,5 hPa:
[image]

Wouldn't it be useful to have a calibrating option without creating a second template sensor?

MattWestb, 2022-11-20:
Issue is posted https://github.com/zigpy/zha-device-handlers/issues/622 with example of some Aquara sensor readings but nothing is made for implanting it.

Adminiuga, 2022-11-20:
What devices support scaled value and scale attribute?

javicalle, 2022-11-20:
> What devices support scaled value and scale attribute?

None that I know of, but the `zhaquirks.xiaomi.aqara.weather.Weather2` appears to be accurate to within 0.01kPa.
The current ZHA sensor implementation takes the `measured_value` attribute with a 0 precision:
* https://github.com/home-assistant/core/blob/38b84620bdbf241a6b246b386dbc1cee21e95472/homeassistant/components/zha/sensor.py#L522-L531

That's correct in the ZHA standar.

The `XiaomiCluster` quirk is doing a 100 factor correction, so it is dropping the ¿0.01? precision:
* https://github.com/zigpy/zha-device-handlers/blob/8e541a17a84139794738ca4698a212e967cf3a92/zhaquirks/xiaomi/__init__.py#L268-L271

IMHO the way to allow the higher precision without quirking ZHA would be reporting the values as a `scaled_value`. This way ZHA will just implement the Extended attribute set and the quirk would do all the dirty job.

javicalle, 2022-11-20:
> Wouldn't it be useful to have a calibrating option without creating an additional template sensor?

I'm not saying it can't be, but in this case it's something I haven't found defined in the Zigbee specification, so IMHO is something that shouldn't be implemented in ZHA (or in a quirk).

If the device already implements it, we could evaluate if it is exposed through a quirk, but adding functionality that does not exist in the devices or in the Zigbee standard is something that IMHO the library should not address.

MattWestb, 2022-11-20:
@javicalle If you is reading the link in my post before its explaining most of the problematic.
It shall being made in the integration so HA and also log service (influxDB) is using the corrected measurements or its no sin doing it at all so HA is showing corrected measurements in current and historic graphs.
Now its possible doing it very nice (before all attribute was not saved in Zigbee.BD / device) and can also storing all correction / device / sensor and being used by the integration (ZHA / Zigpy) and only adding the GUI for setting it in ZHA.

Adminiuga, 2022-11-21:
> IMHO the way to allow the higher precision without quirking ZHA would be reporting the values as a scaled_value. This way ZHA will just implement the Extended attribute set and the quirk would do all the dirty job.

Unless there're other devices making use of the scaled attributes, then it is easier to drop correction in the quirk and implement xiaomi specific pressure sensor in HA. 
Question is: How really accurate are those 0.01kPa measurements

hajo62, 2022-11-21:
> Question is: How really accurate are those 0.01kPa measurements

Absolute it makes no sense. But - from my point of view - relatively it does; see my post regarding calibrating where both sensors differ in mean by 0.4 to 0.6 hPa...

zauberflute, 2023-04-24:
> Unless there're other devices making use of the scaled attributes, then it is easier to drop correction in the quirk and implement xiaomi specific pressure sensor in HA. Question is: How really [...]

The device is advertised as having a precision of +-/ 0.12 hpa, and the sensor it appears to be using should support it. The data is there, the data appears to be valid - can we expose it?

zauberflute, 2023-05-02:
I cloned zha into a custom integration and changed one line in zha/sensor.py to _decimals = 2 and now I am getting the measurement essentially in the native pa in home assistant (hpa with 2 decimals).
The reason you are only getting whole-hpa measurements in home assistant is because ZHA is rounding to match the Zigbee standard, on this line: https://github.com/home-assistant/core/blob/a9d992c2def8c3da7e97a3100efc8e3b3f5e8531/homeassistant/components/zha/sensor.py#L617
As I read the quirks file, the device sends the value in pa but quirks converts it to hpa with 2 decimals precision with "/ 100"; but then zha rounds it down to an integer so you lose the pa-level precision.

closed 2023-11-05 by github-actions[bot]