zigpy/zha-device-handlers #1918: [Device Support Request] TZE200_9yapgbuv

FeignMan, 2022-11-11:
**Is your feature request related to a problem? Please describe.**
This is to request support for the Tuya Temperature Humidity and Luminance sensor. The device pairs with ZHA but none of the sensor values are exposed - temp, humidity, luminance or even battery percentage.

**Describe the solution you'd like**
It would be great to have the sensor values exposed in ZHA as one would expect.

<details>

<summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_9yapgbuv",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.11.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "5.15.61-v7+",
    "supervisor": "2022.10.2",
    "host_os": "Raspbian GNU/Linux 11 (bullseye)",
    "docker_version": "20.10.19",
    "chassis": "",
    "run_as_root": true
  },
  "custom_components": {
    "spotcast": {
      "version": "v3.6.30",
      "requirements": [
        "spotify_token==1.0.0"
      ]
    },
    "myEnedis": {
      "version": "1.5.1",
[... 269 lines cut ...]
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

Purchase link:
https://fr.aliexpress.com/item/1005004838346776.html

javicalle, 2022-11-11:
Maybe it can fit in the `TuyaTempHumiditySensor`.
Enable the custom quirk folder and copy the `ts0601_sensor.py` file inside:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_sensor.py

Then edit the content adding your device to the `MODELS_INFO` like:
```python
class TuyaTempHumiditySensor(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    # RelativeHumidity multiplier
    RH_MULTIPLIER = 10

    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=81
        # device_version=1
        # input_clusters=[4, 5, 61184, 0]
        # output_clusters=[25, 10]>
        MODELS_INFO: [
            ("_TZE200_bjawzodf", "TS0601")],
            ("_TZE200_9yapgbuv", "TS0601")],
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TemperatureHumidityManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }
    .../...
```

Save changes and restart HA.
Check if the device signature change after the restart. And double check if the values are correct.
If something is not fine, enable the debug logs and attach a couple of reports from the device.

FeignMan, 2022-11-14:
Hey there @javicalle, thanks for your reply!

So I tried doing as you said, and now when I pair the device I'm able to see 3 new sensors. One each for battery, temperature and humidity. Initially all 3 of them showed their status as "Unknown".

However, now as I'm writing this first the Temperature and then the Humidity value showed up. I guess the sensor doesn't report values too frequently. The battery percentage still shows as 'Unknown' but that's fine - I don't really care.

Thanks a lot for your help! Btw, should I or someone be submitting a MR for this change?

closed 2022-11-14 by FeignMan

challs, 2022-11-18:
There is a [forum post](https://community.home-assistant.io/t/sonoff-zigbee-usb-3-0-and-tuya-sensor-temperature-umidity/486874) asking about this device too.

javicalle, 2022-11-18:
@FeignMan if you can confirm that the battery entity is working too, would you to create a PR for the device. You can do it directly from the web page, just editing the file content and following the rules:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_sensor.py

Thanks in advanced.

sprokipchyn, 2024-03-25:
> Maybe it can fit in the `TuyaTempHumiditySensor`. Enable the custom quirk folder and copy the `ts0601_sensor.py` file inside:
>
> * [...]

Could you please explain how to do this with Home Assistant Yellow? As far as I understand I need to replace the standard version of Zigbee plugin with a forked one. Can this be done via Home Assistant UI? Ideally I would like to avoid moving files directly to my HA device.

javicalle, 2024-03-25:
You just need to enable the local quirks configuration.
There is a good guide:
https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

sprokipchyn, 2024-03-26:
Yeah, I was able to do this using VSCode plugin. Thanks